DIGImend/digimend-kernel-drivers #80: No pressure with Genius Mouse Pen M508

spbnick, 2017-10-14:
Originally reported by @mccouto in DIGImend/digimend.github.io#7:

I have a Genius Mouse Pen M508 tablet. I use Lubuntu 16.04.3 Kenel 4.10.0-37-generic
The table appears to be normally recognized by Linux but not fully Stylus. The Stylus works like a mouse. When I click on the windows everything works but if I open any file in the programs like Gimp, MyPaint, Krita, even configuring and activating the M508 the Stylus does not draw in open files.
When applying the two lsusb and xinput commands the terminal returned this:

     xinput
    ⎡ Virtual core pointer                    	id=2	[master pointer  (3)]
    ⎜   ↳ Virtual core XTEST pointer              	id=4	[slave  pointer  (2)]
    ⎜   ↳ Dell Dell Universal Receiver            	id=11	[slave  pointer  (2)]
    ⎜   ↳ AlpsPS/2 ALPS GlidePoint                	id=13	[slave  pointer  (2)]
    ⎜   ↳ UC-LOGI  5x8  Tablet                    	id=15	[slave  pointer  (2)]
    ⎜   ↳ UC-LOGI  5x8  Tablet                    	id=16	[slave  pointer  (2)]
    ⎣ Virtual core keyboard                   	id=3	[master keyboard (2)]
        ↳ Virtual core XTEST keyboard             	id=5	[slave  keyboard (3)]
        ↳ Video Bus                               	id=6	[slave  keyboard (3)]
        ↳ Power Button                            	id=7	[slave  keyboard (3)]
        ↳ Sleep Button                            	id=8	[slave  keyboard (3)]
        ↳ Integrated_Webcam_1.3M                  	id=9	[slave  keyboard (3)]
        ↳ Dell Dell Universal Receiver            	id=10	[slave  keyboard (3)]
        ↳ AT Translated Set 2 keyboard            	id=12	[slave  keyboard (3)]
        ↳ Dell WMI hotkeys                        	id=14	[slave  keyboard (3)]

    ~$ lsusb
    Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
    Bus 008 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
    Bus 007 Device 002: ID 5543:0772 UC-Logic Technology Corp. 
    Bus 007 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
    Bus 006 Device 002: ID 413c:301b Dell Computer Corp. 
    Bus 006 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
    Bus 001 Device 003: ID 0c45:63ee Microdia 
    Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
    Bus 005 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
    Bus 004 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
    Bus 003 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub

It seems that just the stylus needs to work properly under pressure and I can use this tablet day-to-day on Linux.

spbnick, 2017-10-14:
Hi Mozart,

First of all, let's check if you get the pressure in X.org at all. Run `xinput test` for your tablet and do a stroke with the pen.

mccouto, 2017-10-14:
Oi, Nick.
The terminal show me :
mozart@mz1545:~$ xinput test
usage: xinput test [-proximity] <device name>

Other graphic tablets I use (Wacom, Huion, and Genius) work correctly on
this same Lubuntu 16.04 and on this same notebook. I forgot to explain this
in the previous text. Of the many tables I have, only this M508 has this
problem with the stylus.

Outras mesas gráficas que utilizo ( Wacom, Huion, e Genius ) funcionam
corretamente neste mesmo Lubuntu 16.04 e neste mesmo notebook. Esqueci de
explicar isso no texto anterior. Das muitas mesas que tenho, só mesmo essa
M508 tem esse problema com a stylus.
-- 
Mozart Couto
www.mozartcouto.com.br
http://blogdodesenhador.blogspot.com
<http://mozartcoutoimagens.blogspot.com>

mccouto, 2017-10-14:
Sorry, nick
I did the wrong test. Now I'm sending the tests of ids 12 and 13 that are
referring to what appeared with xinput. As you can see, no log appears on
the terminal with
the id = 13 test.

Desculpe, nick
Eu fiz o teste errado. Agora envio os testes dos ids 12 e 13 que são os
referentes ao que apareceu com xinput. Como pode ver, não aparece nenhum
log no terminal com
o teste da id=13.

mozart@mz1545:~$ xinput
⎡ Virtual core pointer                    id=2 [master pointer  (3)]
⎜   ↳ Virtual core XTEST pointer              id=4 [slave  pointer  (2)]
⎜   ↳ Dell Dell Universal Receiver            id=11 [slave  pointer  (2)]
⎜   ↳ UC-LOGI  5x8  Tablet                    id=12 [slave  pointer  (2)]
⎜   ↳ UC-LOGI  5x8  Tablet                    id=13 [slave  pointer  (2)]
⎜   ↳ AlpsPS/2 ALPS GlidePoint                id=15 [slave  pointer  (2)]
⎣ Virtual core keyboard                    id=3 [master keyboard (2)]
    ↳ Virtual core XTEST keyboard              id=5 [slave  keyboard (3)]
    ↳ Video Bus                                id=6 [slave  keyboard (3)]
    ↳ Power Button                            id=7 [slave  keyboard (3)]
    ↳ Sleep Button                            id=8 [slave  keyboard (3)]
    ↳ Integrated_Webcam_1.3M                  id=9 [slave  keyboard (3)]
    ↳ Dell Dell Universal Receiver            id=10 [slave  keyboard (3)]
    ↳ AT Translated Set 2 keyboard            id=14 [slave  keyboard (3)]
    ↳ Dell WMI hotkeys                        id=16 [slave  keyboard (3)]
mozart@mz1545:~$ xinput test 12
motion a[0]=11910 a[1]=6780
motion a[0]=12029 a[1]=6707
motion a[0]=12142 a[1]=6626
motion a[0]=12247 a[1]=6536
motion a[0]=12351 a[1]=6432
motion a[0]=12455 a[1]=6328
motion a[0]=12559 a[1]=6220
motion a[0]=12657 a[1]=6111
motion a[0]=12764 a[1]=6000
motion a[0]=12876 a[1]=5889
motion a[0]=12985 a[1]=5772
motion a[0]=13101 a[1]=5671
motion a[0]=13217 a[1]=5566
motion a[0]=13331 a[1]=5475
motion a[0]=13449 a[1]=5395
motion a[0]=13562 a[1]=5313
motion a[0]=13679 a[1]=5243
motion a[0]=13799 a[1]=5178
motion a[0]=13908 a[1]=5112
motion a[0]=14023 a[1]=5051
motion a[0]=14139 a[1]=4990
motion a[0]=14253 a[1]=4928
motion a[0]=14362 a[1]=4865
motion a[0]=14472 a[1]=4797
motion a[0]=14575 a[1]=4727
motion a[0]=14674 a[1]=4650
motion a[0]=14775 a[1]=4571
motion a[0]=14874 a[1]=4490
motion a[0]=14963 a[1]=4412
motion a[0]=15052 a[1]=4338
motion a[0]=15133 a[1]=4266
motion a[0]=15219 a[1]=4200
motion a[0]=15294 a[1]=4139
motion a[0]=15367 a[1]=4081
motion a[0]=15427 a[1]=4029
motion a[0]=15485 a[1]=3984
motion a[0]=15528 a[1]=3943
motion a[0]=15562 a[1]=3907
motion a[0]=15590 a[1]=3869
motion a[0]=15609 a[1]=3836
motion a[0]=15632 a[1]=3810
[... 170 lines cut ...]
motion a[0]=17817 a[1]=3340
motion a[1]=3336

mozart@mz1545:~$ xinput
⎡ Virtual core pointer                    id=2 [master pointer  (3)]
⎜   ↳ Virtual core XTEST pointer              id=4 [slave  pointer  (2)]
⎜   ↳ Dell Dell Universal Receiver            id=11 [slave  pointer  (2)]
⎜   ↳ UC-LOGI  5x8  Tablet                    id=12 [slave  pointer  (2)]
⎜   ↳ UC-LOGI  5x8  Tablet                    id=13 [slave  pointer  (2)]
⎜   ↳ AlpsPS/2 ALPS GlidePoint                id=15 [slave  pointer  (2)]
⎣ Virtual core keyboard                    id=3 [master keyboard (2)]
    ↳ Virtual core XTEST keyboard              id=5 [slave  keyboard (3)]
    ↳ Video Bus                                id=6 [slave  keyboard (3)]
    ↳ Power Button                            id=7 [slave  keyboard (3)]
    ↳ Sleep Button                            id=8 [slave  keyboard (3)]
    ↳ Integrated_Webcam_1.3M                  id=9 [slave  keyboard (3)]
    ↳ Dell Dell Universal Receiver            id=10 [slave  keyboard (3)]
    ↳ AT Translated Set 2 keyboard            id=14 [slave  keyboard (3)]
    ↳ Dell WMI hotkeys                        id=16 [slave  keyboard (3)]
mozart@mz1545:~$ xinput test 13
-- 
Mozart Couto
www.mozartcouto.com.br
http://blogdodesenhador.blogspot.com
<http://mozartcoutoimagens.blogspot.com>

mccouto, 2017-10-14:
I also have this test done in MyPaint. Maybe it can be useful.

tenho também este teste feito no MyPaint. Talvez pode ser útil.

https://youtu.be/NoGyeBEMLyY
-- 
Mozart Couto
www.mozartcouto.com.br
http://blogdodesenhador.blogspot.com
<http://mozartcoutoimagens.blogspot.com>

spbnick, 2017-10-15:
There doesn't seem to be any pressure information reaching X.org applications. Could you please check if the kernel produces it? Run `sudo evtest` on your tablet, do a stroke with the pen and post the result here. Thank you.

mccouto, 2017-10-16:
Hi, Nick,
The result is:
mozart@mz1545:~$ sudo evtest
No device specified, trying to scan all of /dev/input/event*
Available devices:
/dev/input/event0: Lid Switch
/dev/input/event1: Power Button
/dev/input/event2: Sleep Button
/dev/input/event3: AT Translated Set 2 keyboard
/dev/input/event4: Video Bus
/dev/input/event5: Dell Dell Universal Receiver
/dev/input/event6: Dell Dell Universal Receiver
/dev/input/event7: AlpsPS/2 ALPS GlidePoint
/dev/input/event8: UC-LOGI  5x8  Tablet
/dev/input/event9: UC-LOGI  5x8  Tablet
/dev/input/event10: HDA Intel Mic
/dev/input/event11: HDA Intel Headphone
/dev/input/event12: Integrated_Webcam_1.3M
/dev/input/event13: Dell WMI hotkeys
Select the device event number [0-13]:

spbnick, 2017-10-16:
Could you please select your device after you start `evtest`, just as it prompts, so we could see the input events for your pen stroke? Also, this doesn't look like the proper driver is loaded. Could you also post the output of `dmesg` after you plug in your tablet?

mccouto, 2017-10-18:
Hi, Nick.
Sorry by delay.

dmesg:

[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000001000-0x000000000009efff]
[    0.000000]   node   0: [mem 0x0000000000100000-0x00000000dd8befff]
[    0.000000]   node   0: [mem 0x0000000100000000-0x000000011fffffff]
[    0.000000] Initmem setup node 0 [mem
0x0000000000001000-0x000000011fffffff]
[    0.000000] On node 0 totalpages: 1038429
[    0.000000]   DMA zone: 64 pages used for memmap
[    0.000000]   DMA zone: 21 pages reserved
[    0.000000]   DMA zone: 3998 pages, LIFO batch:0
[    0.000000]   DMA32 zone: 14115 pages used for memmap
[    0.000000]   DMA32 zone: 903359 pages, LIFO batch:31
[    0.000000]   Normal zone: 2048 pages used for memmap
[    0.000000]   Normal zone: 131072 pages, LIFO batch:31
[    0.000000] Reserving Intel graphics memory at
0x00000000de000000-0x00000000dfffffff
[    0.000000] ACPI: PM-Timer IO Port: 0x1008
[    0.000000] ACPI: Local APIC address 0xfee00000
[    0.000000] ACPI: LAPIC_NMI (acpi_id[0x00] high edge lint[0x1])
[    0.000000] ACPI: LAPIC_NMI (acpi_id[0x01] high edge lint[0x1])
[    0.000000] IOAPIC[0]: apic_id 2, version 32, address 0xfec00000, GSI
0-23
[    0.000000] ACPI: INT_SRC_OVR (bus 0 bus_irq 0 global_irq 2 dfl dfl)
[    0.000000] ACPI: INT_SRC_OVR (bus 0 bus_irq 9 global_irq 9 high level)
[    0.000000] ACPI: IRQ0 used by override.
[    0.000000] ACPI: IRQ9 used by override.
[    0.000000] Using ACPI (MADT) for SMP configuration information
[    0.000000] ACPI: HPET id: 0x8086a201 base: 0xfed00000
[    0.000000] smpboot: Allowing 2 CPUs, 0 hotplug CPUs
[    0.000000] PM: Registered nosave memory: [mem 0x00000000-0x00000fff]
[    0.000000] PM: Registered nosave memory: [mem 0x0009f000-0x0009ffff]
[    0.000000] PM: Registered nosave memory: [mem 0x000a0000-0x000fffff]
[    0.000000] PM: Registered nosave memory: [mem 0xdd8bf000-0xdd8bffff]
[    0.000000] PM: Registered nosave memory: [mem 0xdd8c0000-0xddac0fff]
[    0.000000] PM: Registered nosave memory: [mem 0xddac1000-0xddac1fff]
[    0.000000] PM: Registered nosave memory: [mem 0xddac2000-0xdfffffff]
[    0.000000] PM: Registered nosave memory: [mem 0xe0000000-0xf7ffffff]
[    0.000000] PM: Registered nosave memory: [mem 0xf8000000-0xfbffffff]
[    0.000000] PM: Registered nosave memory: [mem 0xfc000000-0xfebfffff]
[    0.000000] PM: Registered nosave memory: [mem 0xfec00000-0xfec0ffff]
[    0.000000] PM: Registered nosave memory: [mem 0xfec10000-0xfed17fff]
[    0.000000] PM: Registered nosave memory: [mem 0xfed18000-0xfed1bfff]
[    0.000000] PM: Registered nosave memory: [mem 0xfed1c000-0xfed1ffff]
[    0.000000] PM: Registered nosave memory: [mem 0xfed20000-0xfed8ffff]
[    0.000000] PM: Registered nosave memory: [mem 0xfed90000-0xfed9ffff]
[    0.000000] PM: Registered nosave memory: [mem 0xfeda0000-0xfeda5fff]
[    0.000000] PM: Registered nosave memory: [mem 0xfeda6000-0xfedfffff]
[    0.000000] PM: Registered nosave memory: [mem 0xfee00000-0xfee0ffff]
[    0.000000] PM: Registered nosave memory: [mem 0xfee10000-0xffdfffff]
[    0.000000] PM: Registered nosave memory: [mem 0xffe00000-0xffffffff]
[    0.000000] e820: [mem 0xe0000000-0xf7ffffff] available for PCI devices
[    0.000000] Booting paravirtualized kernel on bare hardware
[    0.000000] clocksource: refined-jiffies: mask: 0xffffffff max_cycles:
0xffffffff, max_idle_ns: 7645519600211568 ns
[    0.000000] setup_percpu: NR_CPUS:8192 nr_cpumask_bits:2 nr_cpu_ids:2
nr_node_ids:1
[    0.000000] percpu: Embedded 36 pages/cpu @ffff8ab91fc00000 s107992
r8192 d31272 u1048576
[    0.000000] pcpu-alloc: s107992 r8192 d31272 u1048576 alloc=1*2097152
[    0.000000] pcpu-alloc: [0] 0 1
[    0.000000] Built 1 zonelists in Node order, mobility grouping on.
Total pages: 1022181
[    0.000000] Policy zone: Normal
[    0.000000] Kernel command line:
BOOT_IMAGE=/boot/vmlinuz-4.10.0-37-generic
[... 1003 lines cut ...]
⎜   ↳ AlpsPS/2 ALPS GlidePoint                    id=13    [slave  pointer
(2)]
⎜   ↳ UC-LOGI  5x8  Tablet                        id=15    [slave  pointer
(2)]
⎜   ↳ UC-LOGI  5x8  Tablet                        id=16    [slave  pointer
(2)]
⎣ Virtual core keyboard                       id=3    [master keyboard (2)]
    ↳ Virtual core XTEST keyboard                 id=5    [slave  keyboard
(3)]
    ↳ Video Bus                                   id=6    [slave  keyboard
(3)]
    ↳ Power Button                                id=7    [slave  keyboard
(3)]
    ↳ Sleep Button                                id=8    [slave  keyboard
(3)]
    ↳ Integrated_Webcam_1.3M                      id=9    [slave  keyboard
(3)]
    ↳ Dell Dell Universal Receiver                id=10    [slave  keyboard
(3)]
    ↳ AT Translated Set 2 keyboard                id=12    [slave  keyboard
(3)]
    ↳ Dell WMI hotkeys                            id=14    [slave  keyboard
(3)]
mozart@mz1545:~$ evtest UC-LOGI 5x8 Tablet id=15
evtest: No such file or directory

spbnick, 2017-10-18:
Right, your tablet doesn't seem to be supported by any drivers yet, we'll need to update them. Please post the output of `sudo lsusb -d 5543:0772` for the start.

mccouto, 2017-10-26:
Hi, Nick. Sorry for the delay again.
The result of the command that asked me to do with the tablet plugged in
and also with the tablet unplugged, shows this: Nothing.

mozart@mz1545:~$ sudo lsusb -d 5543:07
[sudo] senha para mozart:
mozart@mz1545:~$

mccouto, 2017-10-26:
Sorry, Sorry, Sorry!
I got a new result on second time:

mozart@mz1545:~$ sudo lsusb -d 5543:0772
[sudo] senha para mozart:
Bus 006 Device 002: ID 5543:0772 UC-Logic Technology Corp.
mozart@mz1545:~$

spbnick, 2017-10-26:
Thanks, Mozart! Sorry, I forgot to add that you have to use "-v" option as well. Could you please run this instead and post the output: `sudo lsusb -v -d 5543:0772`?

mccouto, 2017-10-26:
Hi, Nick,
This is the new result.
Thanks.
mozart@mz1545:~$ sudo lsusb -v -d 5543:0772
[sudo] senha para mozart: 

Bus 006 Device 002: ID 5543:0772 UC-Logic Technology Corp. 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.10
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0         8
  idVendor           0x5543 UC-Logic Technology Corp.
  idProduct          0x0772 
  bcdDevice            1.10
  iManufacturer           1 UC-LOGI
  iProduct                2  5x8  Tablet 
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           59
    bNumInterfaces          2
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.10
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     237
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.10
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     276
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
Device Status:     0x0000
  (Bus Powered)
mozart@mz1545:~$