DIGImend/digimend-kernel-drivers #81: GT-220 (8192), not calibrated properly, Kernel 3.19

Yalyn, 2017-10-16:
Using Linux Mint 17.2 (MATE) with 3.19 kernel. Installed the latest DIGImend's drivers yesterday.

So far the pen is recognized (sometimes I have to re-plug the cord in order for the pen to work) and it actually has pressure sensitivity working too (tested on MyPaint/GIMP/Krita). Pretty good so far.

The only thing now is the calibration. When I have the pen/stylus around the top right, the mouse pointer would be around the left side or the center depending on how I move it.

The device is just called "Tablet Pen Monitor" while I access the Display settings, it is called HUI 21"

Please and thanks.

spbnick, 2017-10-16:
I don't know if this tablet was ever supported properly, but it seems this one is a newer version, one of those that started appearing and it wouldn't work properly with the existing drivers anyway. Indeed, it doesn't seem to. Could you please download, build and run [the latest development version of uclogic-probe](https://github.com/DIGImend/uclogic-tools/archive/master.zip) on your tablet and post results here, so we can check that?

Yalyn, 2017-10-17:
Okay here is what I got for when I used uclogic-probe :

P 54 00 61 00 62 00 6C 00 65 00 74 00 20 00 4D 00 6F 00 6E 00 69 00 74 00 6F 00 72 00
S 64 0E 03 A4 92 80 52 03 00 FF 1F D0 07 08 00
S 65 04 03 09 00
S 6E 04 03 31 00
S 79 02 03
S 7A 08 03 01 10 00 00 00 00
S 7B 0C 03 48 00 4B 00 20 00 4F 00 6E 00
S C8 13 03 78 74 01 8D D1 00 FF 1F D8 13 03 00 00 00 02 00 3E
S C9 24 03 48 00 55 00 49 00 4F 00 4E 00 5F 00 4D 00 31 00 36 00 35 00 5F 00 31 00 37 00 30 00 35 00 31 00 38 00
S CA 46 03 48 00 55 00 49 00 4F 00 4E 00 20 00 41 00 6E 00 69 00 6D 00 61 00 74 00 69 00 6F 00 6E 00 20 00 54 00 65 00 63 00 68 00 6E 00 6F 00 6C 00 6F 00 67 00 79 00 20 00 43 00 6F 00 2E 00 2C 00 6C 00 74 00 64 00

syntaxintact, 2017-11-28:
Any other data we can add to this report to assist? The calibration on the gt-220 v2 is concentrated in the upper right corner of the tablet. I'm running Ubuntu 17.04 Kernel 4.10.0-40-generic.

jcmoralespena, 2018-01-09:
Hello. 

I have the same problem on my Ubuntu 14.04. Recently (past november) I bought a Huion Kamvas GT221 pro; and despite the tablet recognize de pen, this is out of phase or shifted. I mean; the tip of the pen is in one place but the cursor is in another one. I have tested the pressure of the pen and its buttons, and both works well on Gimp and MyPaint, but I do not achieve to configure the displacement of the pen. The express-keys on the tablet doesn't work at all, but that's irrelevant for me at this moment. 

I ran lsusb to get the bus number and the device adress and the terminal dumped as follow:

Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 006: ID 04f2:b3fd Chicony Electronics Co., Ltd 
Bus 001 Device 004: ID 046d:c534 Logitech, Inc. Unifying Receiver
Bus 001 Device 009: ID 258a:0001  
Bus 001 Device 008: ID 256c:006e  
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

But when I run: sudo uclogic-probe 1 8 
the terminal dumped:

"Failed to get manufacturer string descriptor: Pipe error"

I used the command: sudo lsusb -v to trying to get the manufacter string descriptor and the terminal dumped:

Bus 001 Device 008: ID 256c:006e  
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.10
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0         8
  idVendor           0x256c 
  idProduct          0x006e 
  bcdDevice            0.00
  iManufacturer           5 (error)
  iProduct                6 Tablet Monitor
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           59
    bNumInterfaces          2
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      18
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               2
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      83
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               2
Device Status:     0x0000
  (Bus Powered)

I've uninstalled and installed again the wacom drivers but the problem continues. 

Could you help me please. Thank you.