DIGImend/digimend-kernel-drivers #82: HUION 1060 plus kernel issues

Nephilux, 2017-10-17:
Hello, 
I've just installed the digimend drivers in order to use my tablet Huion 1060 plus new. I can make and install the source because no errors show up (except some SSL errors). After installing the drivers I reboot the system but my tablet is just recognized as a usb drive. Here's my xinput --list result:
` ⎡ Virtual core pointer                     id=2    [master pointer  (3)]
⎜   ↳ Virtual core XTEST pointer                id=4    [slave  pointer  (2)]
⎜   ↳ ELAN0501:00 04F3:3019 Touchpad            id=15   [slave  pointer  (2)]
⎜   ↳ ASUS   ASUS Wireless Mouse                id=12   [slave  pointer  (2)]
⎜   ↳ HID 256c:006e Pad                         id=19   [slave  pointer  (2)]
⎣ Virtual core keyboard                     id=3    [master keyboard (2)]
    ↳ Virtual core XTEST keyboard               id=5    [slave  keyboard (3)]
    ↳ Power Button                              id=6    [slave  keyboard (3)]
    ↳ Video Bus                                 id=7    [slave  keyboard (3)]
    ↳ Video Bus                                 id=8    [slave  keyboard (3)]
    ↳ Power Button                              id=9    [slave  keyboard (3)]
    ↳ Sleep Button                              id=10   [slave  keyboard (3)]
    ↳ HD WebCam                                 id=14   [slave  keyboard (3)]
    ↳ AT Translated Set 2 keyboard              id=16   [slave  keyboard (3)]
    ↳ Acer WMI hotkeys                          id=17   [slave  keyboard (3)]
    ↳ Beats Electronics Beats Solo Wireless     id=13   [slave  keyboard (3)]
    ↳ ASUS   ASUS Wireless Mouse                id=11   [slave  keyboard (3)]
    ↳ ASUS   ASUS Wireless Mouse                id=18   [slave  keyboard (3)]
    ↳ HID 256c:006e Pen                         id=20   [slave  keyboard (3)]`

I tried using the xinput test to the pad and the pen but nothing seems to work.

When I try to capture events through the kernel nothing shows up 
`root@stome-Aspire-E5-575G:/sys/kernel/debug/hid# cd 0003:256C:006E.000A
root@stome-Aspire-E5-575G:/sys/kernel/debug/hid/0003:256C:006E.000A# 
root@stome-Aspire-E5-575G:/sys/kernel/debug/hid/0003:256C:006E.000A# 
root@stome-Aspire-E5-575G:/sys/kernel/debug/hid/0003:256C:006E.000A# 
root@stome-Aspire-E5-575G:/sys/kernel/debug/hid/0003:256C:006E.000A# 
root@stome-Aspire-E5-575G:/sys/kernel/debug/hid/0003:256C:006E.000A# ls -l
total 0
-r-------- 1 root root 0 Okt 17 09:15 events
-r-------- 1 root root 0 Okt 17 09:15 rdesc
root@stome-Aspire-E5-575G:/sys/kernel/debug/hid/0003:256C:006E.000A# cat events
^C
root@stome-Aspire-E5-575G:/sys/kernel/debug/hid/0003:256C:006E.000A# cd ..
root@stome-Aspire-E5-575G:/sys/kernel/debug/hid# cd 0003:256C:006E.000B
root@stome-Aspire-E5-575G:/sys/kernel/debug/hid/0003:256C:006E.000B# ls -l
total 0
-r-------- 1 root root 0 Okt 17 09:15 events
-r-------- 1 root root 0 Okt 17 09:15 rdesc
root@stome-Aspire-E5-575G:/sys/kernel/debug/hid/0003:256C:006E.000B# cat events
^C
`

The only thing that seems to work is the `sudo usbhid-dump -es -m 256c:006e`

which gave me a perfect output (considering pressure and position).
Example: 
`001:009:000:STREAM             1508224515.194654
 08 81 C8 8B 72 28 FF 1F 00 00 00 00

001:009:000:STREAM             1508224515.198741
 08 81 C8 8B 72 28 FF 1F 00 00 00 00

001:009:000:STREAM             1508224515.202742
 08 81 C8 8B 72 28 FF 1F 00 00 00 00

001:009:000:STREAM             1508224515.204732
 08 81 C8 8B 72 28 FF 1F 00 00 00 00

001:009:000:STREAM             1508224515.208771
 08 81 C8 8B 72 28 FF 1F 00 00 00 00

001:009:000:STREAM             1508224515.212659
 08 81 C8 8B 72 28 FF 1F 00 00 00 00

001:009:000:STREAM             1508224515.216647
 08 81 C8 8B 72 28 FF 1F 00 00 00 00

001:009:000:STREAM             1508224515.218642
 08 81 C8 8B 72 28 FF 1F 00 00 00 00

001:009:000:STREAM             1508224515.222748
 08 81 C8 8B 72 28 FF 1F 00 00 00 00

001:009:000:STREAM             1508224515.226753
 08 81 C8 8B 72 28 FF 1F 00 00 00 00

001:009:000:STREAM             1508224515.230750
 08 81 C8 8B 72 28 FF 1F 00 00 00 00

001:009:000:STREAM             1508224515.232744
 08 81 C8 8B 72 28 FF 1F 00 00 00 00

001:009:000:STREAM             1508224515.236754
 08 81 C8 8B 72 28 FF 1F 00 00 00 00

001:009:000:STREAM             1508224515.240757
 08 81 C8 8B 72 28 FF 1F 00 00 00 00

001:009:000:STREAM             1508224515.244620
 08 81 C8 8B 72 28 FF 1F 00 00 00 00

001:009:000:STREAM             1508224515.248654
 08 81 C8 8B 72 28 FF 1F 00 00 00 00

001:009:000:STREAM             1508224515.250645
 08 81 C8 8B 72 28 FF 1F 00 00 00 00

001:009:000:STREAM             1508224515.254703
 08 81 C8 8B 72 28 FF 1F 00 00 00 00

001:009:000:STREAM             1508224515.258762
 08 81 C8 8B 72 28 FF 1F 00 00 00 00

001:009:000:STREAM             1508224515.262759
 08 81 C8 8B 72 28 FF 1F 00 00 00 00

001:009:000:STREAM             1508224515.264775
 08 81 C8 8B 72 28 FF 1F 00 00 00 00

001:009:000:STREAM             1508224515.268667
 08 81 C8 8B 72 28 FF 1F 00 00 00 00

001:009:000:STREAM             1508224515.272767
 08 81 C8 8B 72 28 FF 1F 00 00 00 00
`

The tablet is just recognized as a USB drive at the moment. 
Please help me solve this issue.

spbnick, 2017-10-17:
These "new" tablets use a somewhat different protocol and the driver needs to be updated to handle it. See #77.

Nephilux, 2017-10-17:
@spbnick Thanks for the reply, so I have to wait until a new release? what will happen if i run a windows virtual machine through linux? It will detect the pad?

M-Klekowicki, 2017-11-02:
Hello guys. Just got the new 1060 plus and I'm naturally unable to use it. Any news about a fix/temporary driver or a new release?

jsprada, 2017-11-10:
Ditto.

dbpatankar, 2017-11-16:
Got into the same issue now. uclogic-probe is identical to given in #77 .

spbnick, 2017-11-16:
Sorry for the delay everyone. We moved house at the end of October and we're still settling in, so there will be more delays before I'm able to work on the driver. I didn't have much time for this as it were.

@Nephilux, yes, Windows in a VM will work with your tablet, if you pass the tablet USB device to it.

M-Klekowicki, 2017-11-17:
@spbnick That's understandable. Can we help in anyway at this point?

Also, currently on Wayland distributions libinput also supports Wacom tablets out of the box, maybe there is a chance to extend the support for at least some non-wacom tablets by helping out the people developing it?

Also did Huion made a contact with you ever again about supporting your work?

spbnick, 2017-11-17:
@M-Klekowicki Would be good if you dug around Wayland and figured out what's working and what's not with non-Wacom tablets. I personally am still to give Wayland a try with tablets.

No, Huion did not contact me again. However, I'm figuring out a way to get paid for my work and already have some work done and queued. Perhaps they'll be more interested in working with me once they see that could work.

M-Klekowicki, 2017-11-17:
@spbnick I'll soon going to try Ubuntu 17.10 on Wayland and I'll try to see how it will relate to my Huion NEW 1060 Plus. It didn't work when I did try the live version, but my second borrowed tablet (an old Medion model) worked out of the box. Check out: [https://wayland.freedesktop.org/libinput/doc/latest/tablet-support.html](url) ; [https://github.com/wayland-project/libinput](url)

I will probably write Huion in this matter to point them here and also suggest official support.

Huion NEW 1060 Plus however seems to be the only Huion tablet not working correctly with your drivers, I know this must be an irritating question with all your work on your back, but is there a chance you will resolve the problem this month or next month?

spbnick, 2017-11-17:
It wouldn't do much good to try Huion NEW 1060 Plus with Wayland, since the kernel doesn't support it. Something supported would be useful instead. If the Medion worked that's probably good news. It should work in principle, since Wayland uses libinput, which is also used by X.org, but it would be good to test it well.

Huion released a bunch of tablets recently, mostly refreshes of the existing models, but with newer hardware and firmware which is incompatible. So, the NEW 1060 Plus isn't the only one.

EvilMav, 2017-11-18:
@spbnick
Just in case it doesn't work out with Huion (which would be a shame), I wouldn't mind to "buy you a beer" on e.g. PayPal for supporting the new device. I'd hope other requesters think the same. Please let us know if you have a way for accepting donations.

ralleal, 2017-11-23:
@spbnick I would also gladly "buy you some beers" (craft beers, only :-) ) if you can find a way to support this new 1060 Plus, although I fully understand It's probably not easy to try to attend to all this requests and to have to say 'no' sometimes. And thank for all you work, even if I only heard about it today (never had a pen/tablet before).

asbaklm, 2017-12-02:
@sbpnick

Another NEW 1060 Plus owner here. I have two choices, buy another model that does (hopefully) work or fund driver dev for the 1060 Plus. So I would be happy to help fund driver dev for this model if you have a Patreon page or Paypal or similar.

spbnick, 2017-12-02:
Alright, I just finished setting up a Patreon page:

https://www.patreon.com/spbnick

If you'd like to support my work on DIGImend and help me find time to make new tablets work consider pledging there. I would also be glad to receive any comments about my Patreon setup.

I'll proceed to adding Patreon links to the website and drivers README, plus I'll also write an update on my blog about what was going on recently.

Thanks!

asbaklm, 2017-12-02:
I've signed up, thanks for your efforts and hopefully the Tablet manufacturers come to recognise your work. =)

spbnick, 2017-12-02:
Thanks, @asbaklm :) I sent you a request to join DIGImend's "Supporters" team, since you pledged enough for the 24-hour response guarantee, so I would recognize you on GitHub. It's up to you to accept it or not, of course.

asbaklm, 2017-12-02:
I joined, thanks (I didn't even know what this feature was, haha)

anpaza, 2017-12-17:
I think I've found the reason why hid-uclogic does not work with 1060 plus.

The problem is that drivers/hid/usbhid/hid-core.c computes the max report size in bytes from max report size in bits. The reports listed by device are max 56 bits in length, plus one byte header this makes 8 bytes per report.

However, the tablet reports 12 bytes, appending 4 zero bytes. That's the reason why hid_irq_in() gets urb->status being EOVERFLOW (which is 75), so all the reports are just ignored.

I'm still looking into how to make HID core enlarge the receiving buffer size. Unfortunately, in my kernel (fedora 25) the hid core driver is statically built into the kernel, so I can't change it easily. I'm looking if I can make it change behaviour by changing the fields in the USB HID structure somehow. If that'll work, the rest should start working automagically.

anpaza, 2017-12-17:
Whoops, just found #85. Seems to partially work, but only 8 buttons work okay.
Looks like report descriptors on this tablet are broken. Will look into that patch deeper.

spbnick, 2018-02-13:
Adding [financed](https://github.com/DIGImend/digimend-kernel-drivers/labels/financed) label, as Huion is sending me the [NEW 1060 PLUS(8192)](https://www.huiontablet.com/all-products/graphic-tablets/huion-new-1060-plus-159.html) and is willing to pay for making the driver.

ialebedev, 2018-02-13:
I am using Debian 9.3.0 (x86_64)
Today I've received my new 1060 plus (8192).

Tried to install deb driver, but it was unsuccessful. Installation finished with error.
`$ sudo dpkg -i ./digimend-dkms_6_all.deb 
[sudo] пароль для master: 
(Чтение базы данных … на данный момент установлено 172798 файлов и каталогов.)
Подготовка к распаковке ./digimend-dkms_6_all.deb …

------------------------------
Deleting module version: 6
completely from the DKMS tree.
------------------------------
Done.
Распаковывается digimend-dkms (6) на замену (6) …
Настраивается пакет digimend-dkms (6) …
Loading new digimend-6 DKMS files...
Building for 4.9.0-5-amd64
Building initial module for 4.9.0-5-amd64
Error! Bad return status for module build on kernel: 4.9.0-5-amd64 (x86_64)
Consult /var/lib/dkms/digimend/6/build/make.log for more information.
`
Then i've tried to compile driver from sources. Unsuccessful.
Here is an error.
`$ make
make -C /lib/modules/4.9.0-5-amd64/build SUBDIRS=/home/master/digimend-kernel-drivers-6 modules
make[1]: вход в каталог «/usr/src/linux-headers-4.9.0-5-amd64»
  CC [M]  /home/master/digimend-kernel-drivers-6/hid-kye.o
  CC [M]  /home/master/digimend-kernel-drivers-6/hid-uclogic.o
/home/master/digimend-kernel-drivers-6/hid-uclogic.c:1101:22: error: initialization from incompatible pointer type [-Werror=incompatible-pointer-types]
  .input_configured = uclogic_input_configured,
                      ^~~~~~~~~~~~~~~~~~~~~~~~
/home/master/digimend-kernel-drivers-6/hid-uclogic.c:1101:22: note: (near initialization for ‘uclogic_driver.input_configured’)
cc1: some warnings being treated as errors
/usr/src/linux-headers-4.9.0-5-common/scripts/Makefile.build:304: ошибка выполнения рецепта для цели «/home/master/digimend-kernel-drivers-6/hid-uclogic.o»
make[4]: *** [/home/master/digimend-kernel-drivers-6/hid-uclogic.o] Ошибка 1
/usr/src/linux-headers-4.9.0-5-common/Makefile:1510: ошибка выполнения рецепта для цели «_module_/home/master/digimend-kernel-drivers-6»
make[3]: *** [_module_/home/master/digimend-kernel-drivers-6] Ошибка 2
Makefile:150: ошибка выполнения рецепта для цели «sub-make»
make[2]: *** [sub-make] Ошибка 2
Makefile:8: ошибка выполнения рецепта для цели «all»
make[1]: *** [all] Ошибка 2
make[1]: выход из каталога «/usr/src/linux-headers-4.9.0-5-amd64»
Makefile:9: ошибка выполнения рецепта для цели «modules»
make: *** [modules] Ошибка 2
`
Then I've tried to compile the latest dev version of driver. Successfully!
But after reboot the tablet still doesn't work! ((

Any ideas?

1796426, 2018-02-28:
I've recently bought a Huion 1060 plus and am also trying to get it to work on Ubuntu 17.10. I haven't managed to get it to work at all - my laptop doesn't even recognize it when plugged in. Has anyone managed to get this to work at all?

jsprada, 2018-02-28:
I returned mine, and got a Wacom.  The Wacom works great!

ialebedev, 2018-03-28:
I see that this issue has high priority, and was financed. Will we see the proper driver in near future?

spbnick, 2018-03-28:
I should have functional code available for testing in a couple weeks.

ialebedev, 2018-03-28:
Sounds good! Thanks

dalesolucion, 2018-05-10:
Hello, I just bought a HUION 1060 plus tablet, but I did not know it was not yet compatible with Linux, I'm using the Linux Mint 18 distribution,
Do you know any way to make it work?

I see that they are working on the driver,
Is this still in progress?

Thank you very much

ThePreviousOne, 2018-05-14:
@dalesolucion the tablet needs the driver from master to work, it doesnt work with v7.

ThePreviousOne, 2018-05-14:
@Hyddra   what have you tried?
Are you using the latest driver or v7 release?

have you set up `/etc/X11/xorg.conf.d/50-tablet.conf` ?

`Section "InputClass"
        Identifier "UC-Logic on Wacom"
        MatchUSBID "256c:006e"
        MatchDevicePath "/dev/input/event*"
        Driver "wacom"
EndSection
`
What is version is your kernel? OS? `uname -a`

what are the outputs of
`xsetwacom  --list devices`  `xinput` `xinput --test "<Device Pad pad"`

asbaklm, 2018-05-14:
I have a similar problem and have been unable to get the NEW 1060 Plus to work no matter what I tried. I'm not experienced with tablet device configuration on Linux and unfortunately many of the instructions out there may or may not be completely relevant or up to date.

In the hope that this could help myself and others, I'll document what I've done so far. Would somebody with knowledge of getting this working PRETTY PLEASE fill in the missing gaps and correct mistakes or mistaken assumptions? It really shouldn't have to be so hard to get this working for the average user :(

**OS:** Linux Mint / Ubuntu 16.04
**Kernel** 14.4.6 (shouldn't matter too much I guess)

**Installed prereqs:**
apt-get -y install libwacom-bin libwacom-common xserver-xorg-input-wacom git-core xserver-xorg-input-libinput

**Followed David Revoy's instructions to make use of dkms to install under Ubuntu**
sudo apt install dkms git-core
sudo mkdir /usr/src/digimend-6
sudo git clone https://github.com/DIGImend/digimend-kernel-drivers.git /usr/src/digimend-6
sudo dkms build digimend/6
sudo dkms install digimend/6

This all works fine.
**Reboot System.**

xinput --list
⎡ Virtual core pointer                    	id=2	[master pointer  (3)]
⎜   ↳ Virtual core XTEST pointer              	id=4	[slave  pointer  (2)]
⎜   ↳ HID 256c:006e Pen stylus                	id=9	[slave  pointer  (2)]
⎜   ↳ HID 256c:006e Pad pad                   	id=10	[slave  pointer  (2)]
⎜   ↳ SynPS/2 Synaptics TouchPad              	id=12	[slave  pointer  (2)]
⎜   ↳ TPPS/2 IBM TrackPoint                   	id=13	[slave  pointer  (2)]
⎣ Virtual core keyboard                   	id=3	[master keyboard (2)]
    ↳ Virtual core XTEST keyboard             	id=5	[slave  keyboard (3)]
    ↳ Power Button                            	id=6	[slave  keyboard (3)]
    ↳ Video Bus                               	id=7	[slave  keyboard (3)]
    ↳ Sleep Button                            	id=8	[slave  keyboard (3)]
    ↳ AT Translated Set 2 keyboard            	id=11	[slave  keyboard (3)]
    ↳ ThinkPad Extra Buttons                  	id=14	[slave  keyboard (3)]

xsetwacom --list
HID 256c:006e Pen stylus        	id: 9	type: STYLUS    
HID 256c:006e Pad pad           	id: 10	type: PAD

**/usr/share/X11/xorg.conf.d/52-tablet.conf**
# Huion 1060 New
Section "InputClass" 
   Identifier "UC-Logic on Wacom" 
   MatchUSBID "256c:006e" 
   MatchIsTablet "on"
   MatchDevicePath "/dev/input/event*" 
   Driver "wacom" 
EndSection

Section "InputClass"
   Identifier "Huion buttons"
   MatchProduct "HID 256c:006e Pad"
   MatchIsKeyboard "on"
   MatchDevicePath "/dev/input/event*"
   Driver "evdev"
EndSection
 
Section "InputClass"
   Identifier "Huion scroll"
   MatchProduct "HID 256c:006e Pen"
   MatchIsPointer "off"
   MatchIsKeyboard "off"
   MatchIsTouchpad "off"
   MatchIsTablet "off"
   MatchIsTouchscreen "off"
   MatchDevicePath "/dev/input/event*"
   Driver "evdev"
EndSection

It doesn't work and it's not clear what behaviour to expect or even where to begin to find the fault/s.

1.) For example, should the device show up in the "graphics tablet" section in Preferences? (Perhaps not, since this is not a Wacom device???)
2.) Should the device function as a pointing device (like a mouse) when it is plugged in? (Presumably yes?)
3.) Should the LED's on the tablet light up when PEN input is applied to the surface? (Presumably yes?)
4.) What configuration files need to be set apart from /usr/share/X11/xorg.conf.d/52-tablet.conf?
5.) Are the configuration settings of the 52-tablet.conf file that I posted accurate or relevant? (Presumably not accurate?)
6.) What else is missing ie packages, configuration files etc?
7.) Will the device work in a USB3 port? (Presumably yes?)

ThePreviousOne, 2018-05-14:
@asbaklm see above post

asbaklm, 2018-05-15:
Hi there ThePreviousOne

I built the driver last night just before typing my previous post.

I tried the 50-tablet.conf file from your post at first but this didn't seem to work, ie the tablet didn't appear to generate events when the pen was applied. I think 2 buttons may have worked but with older drivers that has also worked. 

No matter what I do there doesn't appear to be any response to pen input and it isn't clear or apparent where things are going wrong or how to diagnose and narrow down the possible causes.

dalesolucion, 2018-05-15:
@ThePreviousOne Thank you very much, I have tried and it works well, the operation is not complete in terms of buttons, but it helps me. Thank you very much

ThePreviousOne, 2018-05-16:
@dalesolucion xsetwacom --set '<Tablet> Pad pad' Button 12 "key <Key>"

ThePreviousOne, 2018-05-16:
@asbaklm https://digimend.github.io/support/howto/trbl/locating_failure/

asbaklm, 2018-05-16:
Thanks for that link ThePreviousOne, something to add to the collection.
As ridiculous as this will sound, the problem turned out to be.... 

wait for it.... 

a flat stylus battery.

Kids, don't let this happen to you. =)