DIGImend/digimend-kernel-drivers #84: Debian 9: Error building for 4.9.0-3-amd64

JP-OPDevelop, 2017-11-06:
When I try to install the driver using `sudo dpkg -i digimend-dkms_6_all.deb` it display this:

```
Building initial module for 4.9.0-3-amd64
Error! Bad return status for module build on kernel: 4.9.0-3-amd64 (x86_64)
Consult /var/lib/dkms/digimend/6/build/make.log for more information.
```

Im new using GNU/Linux. Any advise? 
My tablet is a Ugee M708
BTW thank you for create this driver!

ThePreviousOne, 2018-02-10:
Have you tried installing building from source? Its actually pretty easy (I mention this because I remember the first time I had to and I thought it was going to be a nightmare)

spbnick, 2018-02-17:
This should be fixed with [v7 release](https://github.com/DIGImend/digimend-kernel-drivers/releases/tag/v7).

closed 2018-02-17 by spbnick