DIGImend/digimend-kernel-drivers #86: Installing .deb issue

monkeyfortstudios, 2017-11-08:
Hi, I've just tried installing (on xubuntu 17.10) and run into problems so am posting as requested in the install section.

The command I was using:
# sudo dpkg -i digimend-dkms_6_all.deb 

Output:

Selecting previously unselected package digimend-dkms.
(Reading database ... 154737 files and directories currently installed.)
Preparing to unpack digimend-dkms_6_all.deb ...
Unpacking digimend-dkms (6) ...
Setting up digimend-dkms (6) ...
Loading new digimend-6 DKMS files...
Building for 4.13.0-16-generic
Building initial module for 4.13.0-16-generic
ERROR (dkms apport): unable to determine source package for digimend-dkms
Error! Bad return status for module build on kernel: 4.13.0-16-generic (x86_64)
Consult /var/lib/dkms/digimend/6/build/make.log for more information.

And below is the contents of /var/lib/dkms/digimend/6/build/make.log:

DKMS make.log for digimend-6 for kernel 4.13.0-16-generic (x86_64)
Wed  8 Nov 14:18:09 AEDT 2017
make -C /lib/modules/4.13.0-16-generic/build SUBDIRS=/var/lib/dkms/digimend/6/build modules
make[1]: Entering directory '/usr/src/linux-headers-4.13.0-16-generic'
  CC [M]  /var/lib/dkms/digimend/6/build/hid-kye.o
  CC [M]  /var/lib/dkms/digimend/6/build/hid-uclogic.o
  CC [M]  /var/lib/dkms/digimend/6/build/hid-polostar.o
/var/lib/dkms/digimend/6/build/hid-uclogic.c:1101:22: error: initialization from incompatible pointer type [-Werror=incompatible-pointer-types]
  .input_configured = uclogic_input_configured,
                      ^~~~~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/digimend/6/build/hid-uclogic.c:1101:22: note: (near initialization for ‘uclogic_driver.input_configured’)
cc1: some warnings being treated as errors
scripts/Makefile.build:309: recipe for target '/var/lib/dkms/digimend/6/build/hid-uclogic.o' failed
make[2]: *** [/var/lib/dkms/digimend/6/build/hid-uclogic.o] Error 1
Makefile:1546: recipe for target '_module_/var/lib/dkms/digimend/6/build' failed
make[1]: *** [_module_/var/lib/dkms/digimend/6/build] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-4.13.0-16-generic'
Makefile:9: recipe for target 'modules' failed
make: *** [modules] Error 2

Obviously it looks like it's an issues with the uclogic files. Incidentally the tablet I'm trying to use (and have with partial success on 16.10) is a Ugee UG-2150. I believe it's almost identical to the UC-Logic XP-Pen Artist 22HD being discussed here https://github.com/DIGImend/digimend-kernel-drivers/pull/73
(My uclogic-probe results are identical to the results posted by miharenthas.)

Let me know if there's any more info that would be helpful, I'm happy to jump in and help out in that pull request if it's still being looked into.

Cheers,

ggolin, 2017-12-29:
+1
<pre>
j@ubuntu:~/Documents$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 17.10
Release:	17.10
Codename:	artful
j@ubuntu:~/Documents$ uname -a
Linux ubuntu 4.13.0-21-generic #24-Ubuntu SMP Mon Dec 18 17:29:16 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux
j@ubuntu:~/Documents$
j@ubuntu:~/Documents$ sudo dpkg -i digimend-dkms_6_all.deb 
Selecting previously unselected package digimend-dkms.
(Reading database ... 136473 files and directories currently installed.)
Preparing to unpack digimend-dkms_6_all.deb ...
Unpacking digimend-dkms (6) ...
Setting up digimend-dkms (6) ...
Loading new digimend-6 DKMS files...
Building for 4.13.0-21-generic
Building initial module for 4.13.0-21-generic
ERROR (dkms apport): unable to determine source package for digimend-dkms
Error! Bad return status for module build on kernel: 4.13.0-21-generic (x86_64)
Consult /var/lib/dkms/digimend/6/build/make.log for more information.
j@ubuntu:~/Documents$ cat /var/lib/dkms/digimend/6/build/make.log
DKMS make.log for digimend-6 for kernel 4.13.0-21-generic (x86_64)
Thu Dec 28 19:35:21 PST 2017
make -C /lib/modules/4.13.0-21-generic/build SUBDIRS=/var/lib/dkms/digimend/6/build modules
make[1]: Entering directory '/usr/src/linux-headers-4.13.0-21-generic'
  CC [M]  /var/lib/dkms/digimend/6/build/hid-kye.o
  CC [M]  /var/lib/dkms/digimend/6/build/hid-uclogic.o
  CC [M]  /var/lib/dkms/digimend/6/build/hid-polostar.o
/var/lib/dkms/digimend/6/build/hid-uclogic.c:1101:22: error: initialization from incompatible pointer type [-Werror=incompatible-pointer-types]
  .input_configured = uclogic_input_configured,
                      ^~~~~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/digimend/6/build/hid-uclogic.c:1101:22: note: (near initialization for 'uclogic_driver.input_configured')
cc1: some warnings being treated as errors
scripts/Makefile.build:309: recipe for target '/var/lib/dkms/digimend/6/build/hid-uclogic.o' failed
make[2]: *** [/var/lib/dkms/digimend/6/build/hid-uclogic.o] Error 1
Makefile:1546: recipe for target '_module_/var/lib/dkms/digimend/6/build' failed
make[1]: *** [_module_/var/lib/dkms/digimend/6/build] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-4.13.0-21-generic'
Makefile:9: recipe for target 'modules' failed
make: *** [modules] Error 2
</pre>

spbnick, 2018-02-17:
This should be fixed with [v7 release](https://github.com/DIGImend/digimend-kernel-drivers/releases/tag/v7).

closed 2018-02-17 by spbnick