DIGImend/digimend-kernel-drivers #87: Add Artisul D13 support

DennaGherlyn, 2017-11-10:
I have an Artisul D13 which is an UC-Logic tablet and I'm searching for tablet drivers for it. No driver known to me works. I can use the tablet as a monitor (out of the box) but its pen doesn't get recognized at all.

`lsusb` gives me this output for the tablet:
```
Bus 003 Device 005: ID 5543:0046 UC-Logic Technology Corp.
```

`uclogic-probe` gives me the following:
```
M 55 00 43 00 2D 00 4C 00 6F 00 67 00 69 00 63 00
P 41 00 52 00 54 00 49 00 53 00 55 00 4C 00 20 00 44 00 31 00 33 00
S 64 0C 03 99 72 7E 40 03 00 FF 07 EC 09
S 65 04 03 00 67
S 6E 04 03 31 00
S 79 0C 03 4F 00 31 00 31 00 30 00 35 00
S 7A 08 03 02 07 01 54 00 00
S 7B 0C 03 48 00 4B 00 20 00 4F 00 6E 00
```

`xinput --list` gives me this:
```
⎡ Virtual core pointer                          id=2    [master pointer  (3)]
⎜   ↳ UC-Logic ARTISUL D13                      id=8    [slave  pointer  (2)]
⎜   ↳ UC-Logic ARTISUL D13                      id=9    [slave  pointer  (2)]
⎣ Virtual core keyboard                         id=3    [master keyboard (2)]
    ↳ UC-Logic ARTISUL D13                      id=15   [slave  keyboard (3)]
```

[artisul-d13-dmesg.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/1461450/artisul-d13-dmesg.txt)
[artisul-d13-lsub-v.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/1461451/artisul-d13-lsub-v.txt)

It would be nice if you could add support for this device. Or point me to an other driver that works.

FexelFox, 2018-01-03:
Yeah, and I can confirm that Artisul themselves aren't planning to provide linux support AFAIK with my correspondence with them, so I would love to see support for it that way I don't need to be running Windows to use it :(

Attached my own DMESG and LSUSB dumps
[artisul-d13-dmesg.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/1601369/artisul-d13-dmesg.txt)
[artisul-d13-lsub-v.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/1601370/artisul-d13-lsub-v.txt)

Also, I couldn't get any activity out of `xinput test` of the two listings for the pointer, do get keypress activity out of the hotkeys for the device, they are using their default settings that came with the tablet when there is no driver installed. (the circle wheel acts as a scroll wheel in this mode)

DennaGherlyn, 2018-01-27:
So my friend and I were fiddling around with the code to try and make it work without knowing what we were actually doing. No good results so far. But we have some new data from the Artisul D13 that might help a person with more knowledge in writing drivers.

The data from the following dump are in the attachments:
`usbhid-dump -a3:20 | grep -v : | xxd -r -p | hidrd-convert -o code`
[d13.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/1669873/d13.txt)
[d13_i0.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/1669874/d13_i0.txt)
[d13_i1.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/1669875/d13_i1.txt)
[d13_i2.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/1669876/d13_i2.txt)

spbnick, 2019-05-05:
I threw together a quick patch trying to add support for Artisul D13, if anybody still needs it, try installing and using https://github.com/spbnick/digimend-kernel-drivers/archive/support_uc-logic_artisul_d13.zip

radeoflier, 2019-06-24:
Awesome - thanks for putting this together!
FEEDBACK:
this gave functionality to my D13 but no real pen-pressure
I'm still running Ubuntu from a flash drive and evaluating switching based on how effective I can manage the tablet portions of my pipeline -- I do animation / vfx and have become a tablet-addict... :) It save sooo much time in the concept phases.
(using with Krita, Blender, GIMP - will try other apps)

novice at code /data dump, but let me know if I can be helpful - many thanks!

radeoflier, 2022-06-20:
This worked great in Ubuntu 18.04. However, on 20.04 I'm having problems... should I have installed DIGImend before spbnick's .zip package? Thanks for any help you feel like giving!

Error:
```
make -C /lib/modules/5.15.0-39-generic/build SUBDIRS=/home/radeoflier/Desktop/DIGImend_Drivers/digimend-kernel-drivers-support_uc-logic_artisul_d13 modules
make[1]: Entering directory '/usr/src/linux-headers-5.15.0-39-generic'
  SYNC    include/config/auto.conf.cmd
  HOSTCC  scripts/basic/fixdep
  HOSTCC  scripts/kconfig/conf.o
  HOSTCC  scripts/kconfig/confdata.o
  HOSTCC  scripts/kconfig/expr.o
  LEX     scripts/kconfig/lexer.lex.c
/bin/sh: 1: flex: not found
make[3]: *** [scripts/Makefile.host:9: scripts/kconfig/lexer.lex.c] Error 127
make[2]: *** [Makefile:622: syncconfig] Error 2
make[1]: *** [Makefile:735: include/config/auto.conf.cmd] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.15.0-39-generic'
make: *** [Makefile:22: modules] Error 2
```