DIGImend/digimend-kernel-drivers #92: Add Priority List Or Progress Page

Garethp, 2017-12-04:
I think it would be useful or interesting if you had a list of future work planned. Maybe even a quick write up on where you were with adding new tablets? For example, I'd love to know how high or low on your priority list the New 1060 Pro+ is. Your Patreon says your working on a signing tablet, so I know the 1060+ isn't what you're working on right now. I figured rather than having people posting saying "XYZ isn't working!!!" or "When will {{tablet}} be ready?" now that you have the Patreon wouldn't be useful. Having just a short list in the README of your priority list might help to spread that information without clogging the issue list

spbnick, 2017-12-05:
Hi Gareth! First of all, thank you for your support! Please acknowledge my project member invitation, so I could recognize you as a "supported supporter" :)

Yes, you're right, I need to make such a list. In the very least, I owe it to supporters. I think I'll go with issue milestones and put issues I'm going to work on there, assigning due dates to milestones, so you could see the priority. At the moment I'm working on stuff financed by two companies, as Patreon pledges haven't amounted to much yet. Still, however little they might be, I'll spend the proportional time on community issues, including tablet support, if I can.

I think I'll also put "financed" and "community financed" tags on issues which are financed by companies and supporters respectively, to show which issues are ready to be worked on. And of course we could discuss which of them are higher priority. Once I get enough support from community, or perhaps Huion, I'll work on the new Huion tablets first thing, unless there's something else you would prefer.

@DIGImend/supporters, what do you think?

Regarding where to put such messages, I think "[team discussions](https://github.com/orgs/DIGImend/teams/supporters/discussions)" would be a good place, as @asbaklm has shown. I'll close this issue once everyone had a chance to voice their opinion.

spbnick, 2017-12-05:
Alright, here is my shot at this.

I added "[financed](https://github.com/DIGImend/digimend-kernel-drivers/labels/financed)" and "[community financed](https://github.com/DIGImend/digimend-kernel-drivers/labels/community%20financed)" labels and marked a few issues with them. I also added three priority labels: "pri: low", "pri: medium", and "pri:high", and marked a few issues which I think should be financed first as "pri:high": https://github.com/DIGImend/digimend-kernel-drivers/labels/pri%3A%20high

I added two release milestones "Release v7" and "Release v8", you can see them here: https://github.com/DIGImend/digimend-kernel-drivers/milestones I'm going to only add (company or community) financed issues to release milestones.

Let me know what you think. I'll make a post describing the system on Patreon once everyone had a chance to say something about this.

asbaklm, 2017-12-05:
I think this is a good way to do it. 

Prioritise tablets with the most requests for drivers and from a technical point of view, the best chances of successful driver implementation. It makes sense to concentrate on tablets with the best chances of attracting community & vendor funding and interest.

As you mentioned, create community & vendor funding targets. Show status of driver development & rough estimate of time to completion. It goes without saying that attracting the "full funding target" will prove challenging unless the vendors provide direct funding so it will probably have to be "best efforts for best case funding".

It may also be worth pitching to vendors (such as Huion, maybe others) that efforts will be made to include support for additional tablets in the mainline kernel and to add this as a "service" to them in exchange for funding. That way the vendor doesn't need to go through the hassle of getting their devices supported "out of the box". The vendor can then also market their products as being "Linux compatible".

It may be worth also to work with enthusiasts who can create customised versions of Ubuntu / Mint / Fedora with the latest tablet drivers baked in and a comprehensive suite of graphics related packages pre-installed. 

This can also be pitched to tablet vendors as an incentive, by letting them provide download links for such a "distro" which already supports their products. That way the vendors get to reach out to Linux markets and their (less technical) end users are saved the headaches of trying to get their software & hardware working. They can simply install the distro and be up and running.

Garethp, 2017-12-05:
I like this idea. You can also consolidate the current issues in to one and collect information in those instead (For example, there's two or three issues at least for 1060 Pro+, when there only needs to be one, and one of the tickets has links on how to help from a programming perspective while the others don't).

I'd love to help contribute, and I'll try to get the dev tools running. My knowledge of C isn't perfect, but as long as I can get the debugging in a nice state I should be able to do something