DIGImend/digimend-kernel-drivers #98: Merging In Driver for Monoprice Pen Display

odysseywestra, 2017-12-08:
Hi, I recently taken over maintenance of a kernel driver for my [Monoprice Pen Display](https://www.monoprice.com/product?p_id=14481) and was wondering if you would be interested in helping merge in the driver. So far the tablet can track, has pressure, has one button besides the pen tip working. The only thing left is to get the second button working and recognize the eraser as it's own input. I don't know much about kernel drivers or programming, but I'm willing to learn. So what do I need to do?

https://github.com/odysseywestra/Monoprice_22_linux_kernel_module

btw, thank you for bringing the project back to life. You have my financial support! I can also help you with issue tracker management,  organization, and website work too if you want.

spbnick, 2017-12-08:
Hi Albert! First of all, thanks a lot for your support!
We can work on this, and it can be done, no question.

However, unless you specifically want your driver be in digimend-kernel-drivers, I would recommend going straight into the Linux kernel, because since I have to sync with the upstream kernel from time to time, I have to apply similar criteria before letting the code in.

Since I'm not a Linux USB/HID/Input subsystem maintainer and digimend-kernel-drivers is somewhat different, I would still have different requirements. So you would have to go through my requirements, and then also through upstream requirements to have your driver properly finished (i.e. have it in the kernel).

Make your choice, and simply create a pull request if you'd still like to get into digimend-kernel-drivers :)

spbnick, 2017-12-08:
It's late here, and I missed the last paragraph - that would be absolutely great, we have lots of work :) We need more data on tablets, tidying up the issues, documentation, and update to now very old HOWTOs, everything. Take your pick :)! Write what you would prefer and I will fill you up on the details, or find some other stuff to do, if you'd like.

odysseywestra, 2017-12-12:
Since I have no kernel experience, it probably would be better to try to get the driver into the digimend-kernel-drivers. That way I can learn from you and also document the process. So I'll go ahead and create a pull request and we'll go from there.

Also if you want, you are more than welcome to the [mypaint community](https://community.mypaint.org) site if you need a place other than here or patreon for user support.. I can create a custom category for you under the development area category.

As far as website, I could update the old HOWTO. That way we can get more data for tablets. I'll be wiling to create a simple bash script we could use to gather the data needed as well. Just start creating issues for the website repository and I'll get start working.

spbnick, 2017-12-13:
Sounds great :) Let's work on the PR. I'm not sure how I can use the mypaint community site, as I have more issues than I can deal with here already. Can you explain how I can use that custom category? I'll file a few issues on the website repo and CC you.

spbnick, 2017-12-13:
Regarding a script for collecting diagnostics, I wanted to create one myself quite a while ago, but decided against it - debugging that script as a developer, and navigating it as a user in all possible situations didn't feel quite nice. I won't hold you from creating one, though. It should be fun, and might be useful.

spbnick, 2017-12-13:
Ah, I just found my attempt to start a diagnostics script: https://github.com/DIGImend/digimend-diag
Feel free to use it as you see fit :)

sibrydionmawr, 2018-07-25:
How is this project progressing? As a usually very happy Parblo Coast 22 user, I'd be over the moon if this project became part of the kernel, as it's a right royal pain in the proverbial to have to constantly re-install the drivers after every kernel update, and find that I have to usually use the older kernel as this project hasn't been updated to.