DIGImend/digimend-kernel-drivers #100: PPA Management and Testing

doctormo, 2017-12-13:
Here is the new PPA for older ubuntu releases:

https://launchpad.net/~digimend/+archive/ubuntu/releases

So far I've only pushed 16.04 (LTS) but will do more as testers become available.

spbnick, 2017-12-13:
Awesome, thank you! Is it the latest master, though, or the actual v6 release?

doctormo, 2017-12-13:
Ah yes, you're right, it's dev. You've got your debian files committed to your main repository which makes this a bit harder to do for the tagged release, but I'll sort it out going forwards.

spbnick, 2017-12-13:
Yeah, I know, having debian files in the repo makes it more difficult for packagers. However as far as I do my own packaging I'll need to keep them somewhere too. We can work on that, though.

One proper fix for this would be updating the debian files, and all other files mentioning the version number to v7. This should've been done immediately after v6 release, but I probably forgot. Can you do that? Then you'll have your PPA version correct, although you'll need to mark it as pre-release somehow (with something like git hash in the version?).

doctormo, 2017-12-13:
Yes I can do that.

The recommended build process is `git pull; make release; run fancy script to take tarball and put existing debian file in a fresh directory; debuild` but most upstreams find that tedious compared to debuild from git directory. I notice there's no make release for digimend anyway, so moot point.

spbnick, 2017-12-13:
We can add "make release" if that would help.

doctormo, 2017-12-13:
It would be something debian people would want to see I think. But I can work around the unclean tarball problem (if you do add it, make sure not to include the debian directory)

spbnick, 2017-12-13:
Yes, of course. Filed #101.

doctormo, 2017-12-13:
I've reset the v6 package to the v6 code (as tagged) and created a digimend-kernel-git package which has the latest version. We can't use the v7 version directly as launchpad will check the contents of the tarball for the same version and reject tarballs that change but claim to be the same version. So I'm using the date as the version.

spbnick, 2017-12-13:
OK, great, thanks :)

thwaller, 2018-05-25:
Will this PPA be updated for use with 18.04, the latest LTS Ubuntu? For such users, do you recommend using the PPA, or using a local install of the deb?

spbnick, 2018-05-25:
@doctormo, I guess @thwaller's question is for you to answer :)