DIGImend/digimend-kernel-drivers #104: Add support for Huion H640P

ericpareja, 2017-12-25:
I got a Huion H640P as a gift for my son, since it's a very affordable entry level tablet with attractive features.

Although Ubuntu 16.04 sees the USB device, it doesn't identify it properly even after I've compiled and installed the latest digimend kernel driver.

Output of lsusb for the device is:
Bus 001 Device 002: ID 256c:006e  

Here are the output files for the different diagnostics as listed in 
[huion-h640p-lsusb-v.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/1585322/huion-h640p-lsusb-v.txt)

[huion-h640p-frame_buttons.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/1585324/huion-h640p-frame_buttons.txt)
[huion-h640p-pen_buttons.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/1585325/huion-h640p-pen_buttons.txt)
[huion-h640p-pen_pressure.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/1585326/huion-h640p-pen_pressure.txt)
[huion-h640p-pen_coords.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/1585327/huion-h640p-pen_coords.txt)
[huion-h640p-hid_report_descriptors.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/1585328/huion-h640p-hid_report_descriptors.txt)
[huion-h640p-descriptors.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/1585329/huion-h640p-descriptors.txt)

Queatz, 2018-01-20:
Please!

jw-0, 2018-01-21:
I've got one of these tablets and C experience, so I'll have a shot at getting it working, but I'm not too sure exactly what I need to do. Will look over the source code for an idea, but does anyone have any tips/a general framework for what the implementation would look like?

spbnick, 2018-01-21:
@jw-0, this will likely boil down to either the initialization issue, if this is a newer tablet (a new initialization protocol is used by those), or a report descriptor issue. Take a look at http://digimend.github.io/support/howto/trbl/locating_failure/ and also my videos working on some other tabelts (but not the newer kind): https://www.youtube.com/channel/UCioG-vCm2WB6nSz0Y0JTXoA

Queatz, 2018-01-21:
It is one of their newest tablets

jw-0, 2018-01-21:
Great thanks, I've got some time in the evenings over the next few weeks and I'll spend it seeing what I can do.

ScislaC, 2018-02-08:
The diff from https://github.com/DIGImend/digimend-kernel-drivers/pull/114 has added support for the H640P from my testing.

jw-0, 2018-02-08:
Just tested and can also confirm that #114 has the tablet working on Fedora 27. Thanks very much to the developers!

spbnick, 2018-02-13:
Adding [financed](https://github.com/DIGImend/digimend-kernel-drivers/labels/financed) label, as Huion is sending me an [H640P](https://www.huiontablet.com/all-products/graphic-tablets/huion-h640p.html) and is willing to pay for making the driver.

AlyShmahell, 2018-02-16:
Hello,
has anyone tried to get the H640P to work with [#114](https://github.com/DIGImend/digimend-kernel-drivers/pull/114) on Ubuntu 16.04? if so, does it work? I see from the comments that it works on Fedora 27.

ericpareja, 2018-02-17:
I tested #114 on Ubuntu 16.04.3 and 17.04 and it works by all appearances. Although there are some minor / cosmetic issues, functionality seems to be in place. I'm waiting for an official patch or merge though. It's good to hear that Huion is willing to provide a device as well as pay for driver development. :)

AlyShmahell, 2018-02-17:
Thank you @ericpareja , I was weighing my buying options between the Huion640P and Wacom Intuos Art, but I'm unwilling to use windows so I had to make sure the 640P would work.
The patch coming up is exciting news as well, I will be using the [#114](https://github.com/DIGImend/digimend-kernel-drivers/pull/114) for the time being and I will comment any issues I find with it here.

simon-77, 2018-02-25:
Dear all,

I also got a H640P and managed to make it work with #114.
I could not find a way to download the modified version, I instead downloaded the original source code and copied and pasted the modified content into the modified files. It has worked for me, but Is there a way to download the modified version?

I have a second question. The upper three keys of my tablet are mapped to the three mouse keys. The lower three keys seem to not be mapped at all. Is there a way to modify the behaviour of the keys on the tablet?

Kind regards
Simon

simon-77, 2018-02-25:
close

AlyShmahell, 2018-02-25:
Hey Simon,

I assume you know your way around the terminal, so you just have to:

first clone the repo:
`git clone https://github.com/DIGImend/digimend-kernel-drivers.git`

navigate to the repo's directory:
`cd digimend-kernel-drivers`

then fetch the head of pull [114](https://github.com/DIGImend/digimend-kernel-drivers/pull/114):
`git pull origin pull/114/head`

then run make:
`make`

finally install everything:
`sudo make install`

Please note that I haven't tested this myself yet, but it should work.

simon-77, 2018-02-26:
Hey  AlyShmahell

thank you very much for your answer. This approach worked perfectly for me.

Nevertheless, is there a way to map the buttons one the H640P to some Keys?

Kind regards
Simon

AlyShmahell, 2018-02-26:
@simon-77 
I think your best bet is to try and modify the commands in this issue comment:
[#issuecomment-97044199](https://github.com/DIGImend/digimend-kernel-drivers/issues/10#issuecomment-97044199)
At least until someone posts a more definite answer.
And please if you get the shortcuts working, post the commands you used here as a reference for the rest of us :) 
Also, I've seen some posts about a non-linear jump in pressure sensitivity when pressure is set to +1, +2, if that is the case for you as well, make sure to make a comment so that the devs may try and address the issue.

simon-77, 2018-02-26:
Dear all

as mentioned I have a Huion H640P and I have working shortcuts now!

Some key data of my OS:
My Operating System is Debian Stretch and I use KDE.
"$ uname -a" output:
`Linux Debian-Simon 4.9.0-6-amd64 #1 SMP Debian 4.9.82-1+deb9u2 (2018-02-21) x86_64 GNU/Linux`

I have found the required information in this issue https://github.com/DIGImend/digimend-kernel-drivers/issues/26 and on the official DIGImend project page.

=======================================
**Instruction how to setup shortcuts for Huion H640P**

1. Install the modified DIGImend kernel driver https://github.com/DIGImend/digimend-kernel-drivers/pull/114
@AlyShmahell wrote an excellent guide how to do that yesterday

2. Create a file (and its directory if necessary) `/etc/X11/xorg.conf.d/52-tablet.conf` with the following content:

> Section "InputClass"
        Identifier "Huion on wacom"
        #MatchIsTablet "on"
        MatchProduct "Huion"
        MatchDevicePath "/dev/input/event*"
        Driver "wacom"
        # Apply custom Options below.
EndSection

Notice that all lines except the first and last one should be indented. Here you can download the config file. Note the additional .txt extension, I could not upload it up to github with the .conf extension 
[52-tablet.conf.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/1760491/52-tablet.conf.txt)

3. Reboot your Computer
Afterwards the utility "xsetwacom" should recognise the tablet and you can use it to configure shortcuts. Note that the buttons have the indices (up to down): 1,2,3, 8,9,10
For example you can map the 'a' to the top button, 'b' to the second one until 'f' for the lowest button with the following commands:
`$ xsetwacom --set "Huion  Tablet Pad pad" Button 1 "key a"`
`$ xsetwacom --set "Huion  Tablet Pad pad" Button 2 "key b"`
`$ xsetwacom --set "Huion  Tablet Pad pad" Button 3 "key c"`
`$ xsetwacom --set "Huion  Tablet Pad pad" Button 8 "key d"`
`$ xsetwacom --set "Huion  Tablet Pad pad" Button 9 "key e"`
`$ xsetwacom --set "Huion  Tablet Pad pad" Button 10 "key f"`

=======================================
Some more notes:
The command `$ xsetwacom --list` lists all tablets the utility can find. Before step 3. the output of this command was empty. After rebooting, the output was:
> Huion  Tablet Pen stylus                id: 19  type: STYLUS    
Huion  Tablet Pad pad                   id: 20  type: PAD

The command `$ xinput list` has the following relevant output, when my tablet is connected:
> ⎜   ↳ Huion  Tablet Pen stylus                  id=19   [slave  pointer  (2)]
⎜   ↳ Huion  Tablet Pad pad                     id=20   [slave  pointer  (2)]

As far as my understanding for the `52-tablet.conf` file goes, the 'Identifier' is just some made-up name and not functionally relevant. The 'MatchProduct' field on the other hand should hold a name to uniquely identify a device by the name of `$ xinput list`. In my case 'Huion' is unique enough to identify my tablet.

=======================================
Other:
I haven't noticed a problem with the pressure sensitivity, although I just gave it a very short try in 'xournal'. What I noticed is, that there is a lag (I estimate it to about 100ms) from moving the pen until my mouse will move. Moreover I need to press the pen relatively hard to the tablet, until a mouse click will be done (hard is the wrong word, but a slight touch is not sufficient). Does someone know how to address these issues?

I hope that I could help
Kind regards
Simon

jw-0, 2018-02-28:
@simon-77 Great instructional, thanks. I'll test this when I have my tablet handy, but I'm assuming these instructions would only work for X11 and not for Wayland environments?

simon-77, 2018-03-01:
@jw-0 
I am not a linux specialist for graphical environments and until now I didn't know that there are practically used alternatives to X11.
But since the .conf files are for configuring X11 I assume that these instructions won't work with Wayland. Though, let me know whether it works or not if you try it out.

spbnick, 2018-05-04:
Everyone concerned with this issue, you're welcome to join the new [#DIGImend channel on irc.freenode.net](https://webchat.freenode.net/?channels=DIGImend), discuss this and other issues, and help each other! Come and say "Hi!"

partialkakera, 2018-06-23:
I've been trying the instructions @AlyShmahell posted and keep running into this.
```
~/digimend-kernel-drivers $ git pull origin pull/114/head
remote: Counting objects: 23, done.
remote: Total 23 (delta 16), reused 16 (delta 16), pack-reused 7
Unpacking objects: 100% (23/23), done.
From https://github.com/DIGImend/digimend-kernel-drivers
 * branch            refs/pull/114/head -> FETCH_HEAD
Auto-merging hid-uclogic-rdesc.h
Auto-merging hid-uclogic-rdesc.c
CONFLICT (content): Merge conflict in hid-uclogic-rdesc.c
Auto-merging hid-uclogic-core.c
CONFLICT (content): Merge conflict in hid-uclogic-core.c
Auto-merging Makefile
CONFLICT (content): Merge conflict in Makefile
Automatic merge failed; fix conflicts and then commit the result.
```
I am fairly new to Linux and do not know how to proceed or get a different result.

partialkakera, 2018-06-23:
Once I write git commit, I get # Changes to be committed:
```
#       modified:   Makefile
#       modified:   hid-uclogic-core.c
#       new file:   hid-uclogic-proxemu.c
#       new file:   hid-uclogic-proxemu.h
#       modified:   hid-uclogic-rdesc.c
#       modified:   hid-uclogic-rdesc.h

^G Get Help  ^O Write Out ^W Where Is  ^K Cut Text  ^J Justify   ^C Cur Pos
^X Exit      ^R Read File ^\ Replace   ^U Uncut Text^T To Spell  ^_ Go To Line
```
Pressing G or O or anything from the keyboard options doesn't do anything at all. This is probably a result of me being new to linux. I forgot to mention I'm on Linux Mint CInnamon 18.3 64bit if that's needed information. Thank you for your help thus far!

AlyShmahell, 2018-06-23:
at this stage what you are seeing is inside nano editor, just press "CTRL + x" to exit.

partialkakera, 2018-06-24:
Thank you for the information! I was unsure.
I've pressed CTRL + X and now am faced with this dialog. I'm not sure which option to select.
```
File Name to Write: /home/partial/digimend-kernel-drivers/.git/COMMIT_EDITMSG   
^G Get Help     M-D DOS Format  M-A Append      M-B Backup File
^C Cancel       M-M Mac Format  M-P Prepend     ^T To Files
```

imyxh, 2018-06-24:
Uh, can you run a `cat -e -t -v Makefile` and post the output? That usually happens when you accidentally change a tab into spaces in the makefile.

Oh, and to explain nano shortcuts, "^" indicates control, and "M-" indicates meta (alt).

partialkakera, 2018-06-24:
I ran the command posted above. This is the output. I hope it's useful.
Thank you for your help thus far. I've recently switched to Linux and am frustrated at my lack of understanding in situations such as this.
```
partial@partial-portion ~/digimend-kernel-drivers $ cat -e -t -v Makefile
obj-m := hid-kye.o hid-uclogic.o hid-polostar.o hid-viewsonic.o$
<<<<<<< HEAD$
hid-uclogic-objs := \$
^Ihid-uclogic-core.o \$
^Ihid-uclogic-rdesc.o \$
^Ihid-uclogic-params.o$
=======$
hid-uclogic-objs := hid-uclogic-core.o hid-uclogic-rdesc.o hid-uclogic-proxemu.o$
>>>>>>> 244b96926cd48e80875bea7e7844d15070999880$
KVERSION := $(shell uname -r)$
KDIR := /lib/modules/$(KVERSION)/build$
PWD := $(shell pwd)$
UDEV_RULES := /lib/udev/rules.d/90-hid-rebind.rules$
DEPMOD_CONF := /etc/depmod.d/digimend.conf$
HID_REBIND := /sbin/hid-rebind$
PACKAGE_NAME = digimend-kernel-drivers$
PACKAGE_VERSION = 8$
PACKAGE = $(PACKAGE_NAME)-$(PACKAGE_VERSION)$
DKMS_MODULES_NAME = digimend$
DKMS_MODULES = $(DKMS_MODULES_NAME)/$(PACKAGE_VERSION)$
$
modules modules_install clean:$
^I$(MAKE) -C $(KDIR) SUBDIRS=$(PWD) $@$
$
files_install:$
^Iinstall -D -m 0644 digimend.conf $(DEPMOD_CONF)$
^Idepmod -a$
^Iinstall hid-rebind $(HID_REBIND)$
^Iinstall -m 0644 90-hid-rebind.rules $(UDEV_RULES)$
^Iudevadm control --reload$
$
files_uninstall:$
^Irm -vf $(UDEV_RULES) $(HID_REBIND) $(DEPMOD_CONF)$
^Iudevadm control --reload$
^Idepmod -a$
$
modules_uninstall:$
^Irm -vf /lib/modules/*/extra/hid-kye.ko \$
^I       /lib/modules/*/extra/hid-polostar.ko \$
^I       /lib/modules/*/extra/hid-uclogic.ko \$
^I       /lib/modules/*/extra/hid-viewsonic.ko$
$
install: modules_install files_install$
$
uninstall: modules_uninstall files_uninstall$
$
dkms_check:$
^I@if ! which dkms >/dev/null; then \$
^I    echo "DKMS not found, aborting." >&2; \$
^I    echo "Make sure DKMS is installed,"; \$
^I    echo "and \"make\" is running under sudo, or as root."; \$
^I    exit 1; \$
^Ifi$
$
dkms_modules_install: dkms_check$
^I@if dkms status $(DKMS_MODULES_NAME) | grep . >/dev/null; then \$
^I    echo "DKMS has DIGImend modules added already, aborting." >&2; \$
^I    echo "Run \"make dkms_uninstall\" first" >&2; \$
^I    exit 1; \$
^Ifi$
^Idkms add .$
^Idkms build $(DKMS_MODULES)$
^Idkms install --no-depmod $(DKMS_MODULES)$
$
dkms_modules_uninstall: dkms_check$
^Iset -e -x; \$
^Idkms status $(DKMS_MODULES_NAME) | \$
^I    while IFS=':' read -r modules status; do \$
^I        IFS=', ' read -r modules_name modules_version \$
^I                         kernel_version kernel_arch ignore \$
^I                     <<<"$$modules"; \$
^I        if [ -z "$$kernel_version" ]; then \$
^I            dkms remove --no-depmod \$
^I                        "$$modules_name/$$modules_version" \$
^I                        --all; \$
^I        else \$
^I            dkms remove --no-depmod \$
^I                        "$$modules_name/$$modules_version" \$
^I                        -k "$$kernel_version/$$kernel_arch"; \$
^I        fi; \$
^I    done$
$
dkms_install: dkms_modules_install files_install$
$
dkms_uninstall: dkms_modules_uninstall files_uninstall$
$
dist:$
^Igit archive --format=tar.gz --prefix=$(PACKAGE)/ HEAD > $(PACKAGE).tar.gz$
```

partialkakera, 2018-06-24:
Any ideas what to do with that information?

Sorry for the bump.

danghai, 2018-06-24:
I think you should remove this one, and git clone the new one

partialkakera, 2018-06-24:
I'm not entirely sure how to git clone a single file. (Which I'm assuming in this case is the makefile?)
Are you meaning I should git clone from the base branch? (sorry for my terminology; new to github/terminal)

danghai, 2018-06-24:
I mean remove the folder `digimend-kernel-drivers`, and then 
```
git clone https://github.com/DIGImend/digimend-kernel-drivers.git
```
And try to redo again at the beginning in `README`

imyxh, 2018-06-25:
@partialkakera  
Ah, yes, I see the problem. Git inserted a few things into the file while merging. The first part that trips `make` is `<<<<<<< HEAD$` on line 2, because obviously that isn't proper makefile syntax.

I recommend just copying the entirety of the [master makefile](https://raw.githubusercontent.com/DIGImend/digimend-kernel-drivers/master/Makefile) and pasting it into yours (overwriting what you currently have, of course).

I'm not really sure why that happened, but it should be a simple fix. :smile:

danghai, 2018-06-25:
It happens because it has conflict when you merge by git, and you do not solve the conflict. It has `<<<<< HEAD$` to determine which one you should choose for conflict. I see this, and I think it is better to remove the old one and re-install

partialkakera, 2018-06-25:
@imyxh I copied the master makefile and pasted it into my own. [The terminal now gave me all of this.](https://pastebin.com/raw/eVxLxWii)

@danghai I did try reinstalling digimend entirely. Is there more to re-installing it than just deleting the folder and [doing the steps listed in this post](https://github.com/DIGImend/digimend-kernel-drivers/issues/104#issuecomment-368347718) once more?

**I'm new to Linux. Sorry.**
**Thank you for the help thus far.**

danghai, 2018-06-25:
@partialkakera No, I mean delete the folder `digimend-kernel-driver`, and git clone by: `git clone https://github.com/DIGImend/digimend-kernel-drivers.git`. And then you can start here: https://github.com/DIGImend/digimend-kernel-drivers#installing-from-source-with-dkms

partialkakera, 2018-06-25:
I tried the steps listed above. [Here's my terminal result.](https://pastebin.com/raw/eUusX3Tm) Please tell me what I did wrong. Where do I go from here?

imyxh, 2018-06-25:
Well, it says to `run "make dkms_uninstall" first` since it's picking up the version you already have. I would try that, and perhaps follow it with a `make modules_uninstall` if you still have issues.

partialkakera, 2018-06-25:
```
partial@partial-portion ~/digimend-kernel-drivers $ make modules_uninstall
rm -vf /lib/modules/*/extra/hid-kye.ko \
       /lib/modules/*/extra/hid-polostar.ko \
       /lib/modules/*/extra/hid-uclogic.ko \
       /lib/modules/*/extra/hid-viewsonic.ko
```
Should I try to run `sudo make dkms_install` after this?

imyxh, 2018-06-25:
Yep!

partialkakera, 2018-06-25:
```
partial@partial-portion ~/digimend-kernel-drivers $ sudo make dkms_install
dkms add .

Creating symlink /var/lib/dkms/digimend/8/source ->
                 /usr/src/digimend-8

DKMS: add completed.
dkms build digimend/8

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
make -j4 KERNELRELEASE=4.10.0-38-generic KVERSION=4.10.0-38-generic...
cleaning build area...

DKMS: build completed.
dkms install --no-depmod digimend/8

hid-kye.ko:
Running module version sanity check.
Error! Module version 8 for hid-kye.ko
is not newer than what is already found in kernel 4.10.0-38-generic (FA4DE49655F2029A86617E9).
You may override by specifying --force.

hid-uclogic.ko:
Running module version sanity check.
Error! Module version 8 for hid-uclogic.ko
is not newer than what is already found in kernel 4.10.0-38-generic (57296091DE7034E8BD70CB5).
You may override by specifying --force.

hid-polostar.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/4.10.0-38-generic/extra/

hid-viewsonic.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/4.10.0-38-generic/extra/

depmod...

DKMS: install completed.
install -D -m 0644 digimend.conf /etc/depmod.d/digimend.conf
depmod -a
install hid-rebind /sbin/hid-rebind
install -m 0644 90-hid-rebind.rules /lib/udev/rules.d/90-hid-rebind.rules
udevadm control --reload
```
Should I be concerned with the lines saying 'no module exists within this kernel'?
Or should I move on and do this?
'sudo modprobe -r hid-kye hid-uclogic hid-polostar hid-viewsonic'

Thank you for your help! I feel like I'm getting closer to fixing my problems.

imyxh, 2018-06-25:
I'm pretty sure that just means that it's not installed yet—because you just uninstalled them.

partialkakera, 2018-06-25:
Right, right.
I ran `make modules_install` just now.
```
partial@partial-portion ~/digimend-kernel-drivers $ make modules_install
make -C /lib/modules/4.10.0-38-generic/build SUBDIRS=/home/partial/digimend-kernel-drivers modules_install
make[1]: Entering directory '/usr/src/linux-headers-4.10.0-38-generic'
  DEPMOD  4.10.0-38-generic
make[1]: Leaving directory '/usr/src/linux-headers-4.10.0-38-generic'
```
Which seems to have worked.
I'm guessing now the next step is `sudo modprobe -r hid-kye hid-uclogic hid-polostar hid-viewsonic`?
Sorry to ask so much. Last time I tried to do something on my own I made things worse. Lack of experience and all.

imyxh, 2018-06-25:
Yeah, `sudo modprobe -r hid-kye hid-uclogic hid-polostar hid-viewsonic` should be what you want, to update the modules from the last time they were loaded. Don't worry about asking.

partialkakera, 2018-06-25:
Thank you for your replies. I cannot tell you how grateful I am. I've been at such a loss. I'm definitely keeping this in mind for future reference.

I ran that command, entered my password.
There's no error or anything coming up.
Should I plug my tablet in and see now, or is there something else I should do?
Should I try `git pull origin pull/114/head` again? Or leave it as is?

imyxh, 2018-06-25:
It does look like the fix for your tablet was already merged into master, so you can probably just try it as-is. If it doesn't work, or you want to set some more things up, I would recommend asking on IRC as I hastily mentioned in an edit in my previous comment.

spbnick, 2018-06-25:
Hi @partialkakera, judging from the output of `sudo make dkms_install` you seem to be encountering a [bug in dkms](https://github.com/DIGImend/digimend-kernel-drivers#dkms-issue-preventing-correct-installation).

Try to run this command:

    sudo sed -i -e 's/\<unset res$/res=()/' /usr/sbin/dkms

And then do `sudo make dkms_uninstall` and `sudo make dkms_install` again.

partialkakera, 2018-06-25:
I've tried that. Upon doing so, I get this.
```
partial@partial-portion ~ $ sudo sed -i -e 's/\<unset res$/res=()/' /usr/sbin/dkms
[sudo] password for partial: 
partial@partial-portion ~ $ sudo make dkms_uninstall
make: *** No rule to make target 'dkms_uninstall'.  Stop.
```

imyxh, 2018-06-25:
You're in the wrong directory :) `cd` to the github repo.

partialkakera, 2018-06-25:
RIght, right. Thank you.
https://pastebin.com/raw/ciHEeYqa

It seems to have worked.
My next step now would just be trying the tablet, I assume?

spbnick, 2018-06-25:
@partialkakera, the output you linked doesn't seem complete, but as far as it can show what was happening, it happened successfully :) Please post the rest and/or reboot and try the tablet.

danghai, 2018-06-25:
@spbnick It is not related to above issue. I just want to ask you that you change something in makefile regarding to `dkms`. Should it change them in `travis` too? Or It is still fine to keep it

partialkakera, 2018-06-25:
Ah, sorry. I may not have copied it all! [Here. This should be better.](https://pastebin.com/raw/4UfZ4zgf)

spbnick, 2018-06-25:
@partialkakera, @danghai, right it doesn't seem to be related to the dkms bug. First of all, the first module fails to install already, and then the full terminal output @partialkakera posted shows it still fails after the fix.

spbnick, 2018-06-25:
@danghai, yes I think we need to test at least that we can install and uninstall with dkms_install/dkms_uninstall.

partialkakera, 2018-06-25:
Thank you for your help.
When I do things on my own I tend to make them worse, so I'll just wait until someone has an idea of the next step I should take.

spbnick, 2018-06-25:
@partialkakera, which distribution are you running? I'd like to try to reproduce your issue.

Meanwhile, try to install the driver manually as described in https://github.com/DIGImend/digimend-kernel-drivers/#installing-from-source-manually

Before that, run `sudo make dkms_uninstall`.

Do not mess with git, just download the latest development code as a .zip archive from https://github.com/DIGImend/digimend-kernel-drivers/archive/master.zip then unpack and follow the instructions linked above. Do not reuse the existing directory you have.

It's normal to make a mess when you start learning a new complex system. I made enormous messes multiple times when I was learning Linux  (eh, 15 years ago?), and had to reinstall many times. That happens with every new system I learn. Just recently I started learning FreeCAD and made a bunch of messes of it too, don't sweat it :)

partialkakera, 2018-06-25:
I am running Linux Mint 18.3 Cinnamon 64-bit.
Cinnamon version 3.6.7. Linux Kernel 4.10.0-38-generic.
I hope that's the information you were looking to know.

I will try the instructions you've listed and report back.
Thank you for your help!

danghai, 2018-06-25:
@spbnick I open this PR: #163 
@partialkakera : Could you elaborate `Still doesn't seem to work at all.`? Because, sometime, it has issue relating hardware. For example, are you sure your tablet already connect your laptop? you can check by `lsusb`, ...

partialkakera, 2018-06-25:
partial@partial-portion ~ $ lsusb
```
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 006: ID 04f2:b573 Chicony Electronics Co., Ltd 
Bus 001 Device 005: ID 04ca:3015 Lite-On Technology Corp. 
Bus 001 Device 008: ID 256c:006e  
Bus 001 Device 003: ID b58e:9e84 Blue Microphones Yeti Stereo Microphone
Bus 001 Device 002: ID 046d:c52b Logitech, Inc. Unifying Receiver
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```
Yes, my tablet is connected to my laptop.
What I mean when I say it isn't working is the pen does not move the cursor at all.

spbnick, 2018-06-25:
@partialkakera, your manual install attempt looks fine, try doing it again after you upgrade your kernel, and then try using your tablet. If it still doesn't work, post the output of your installation attempt, and the output of `sudo dmesg` after you reconnect your tablet.

Also, please post new comments if you want to share something new, because we don't get updates when you edit them. Thanks!

partialkakera, 2018-06-25:
No luck thus far.
https://pastebin.com/raw/fnMeh6UR
`sudo dmesg` output: https://pastebin.com/raw/MVXqNhDR

[11 more replies]