DIGImend/digimend-kernel-drivers #105: the tablet does not recognize the driver

f5melo, 2017-12-31:
Hello people
I'm new here and on linux too 
I would like to install my tablet Gaomon 8600 pro+  
I installed and uninstalled many different drivers of Digimend v6 and later v5
and my tablet does not work: '(

there are two parallel bands that the lines are shaken

Could someone help me clean everything up and start from scratch, please?
Thank you

Ubuntu Studio
DISTRIB_ID=Ubuntu
DISTRIB_RELEASE=16.04
DISTRIB_CODENAME=xenial
DISTRIB_DESCRIPTION="Ubuntu 16.04.1 LTS"

Linux MW-G41T-M7 4.4.0-104-lowlatency #127-Ubuntu SMP PREEMPT Mon Dec 11 13:04:21 UTC 2017 i686 i686 i686 GNU/Linux

His model visually is this
https://digimend.github.io/tablets/Huion_H58L/Huion_H58L.jpg

sorry I'm layman

spbnick, 2018-05-04:
Everyone concerned with this issue, you're welcome to join the new [#DIGImend channel on irc.freenode.net](https://webchat.freenode.net/?channels=DIGImend), discuss this and other issues, and help each other! Come and say "Hi!"

ThePreviousOne, 2018-05-14:
Incase you havent solved this yet:
`sudo apt remove digimend` or `digimend-dkms` if thats what you installed or `sudo make uninstall` if you built it yourself, I suggest you do at least until v8 is out.

I think `make` is part of the autotools package install that and either git or git-core (sorry, I dont run ubuntu)
git clone https://github.com/DIGImend/digimend-kernel-drivers.git
cd digimend-kernel-drivers
`make && sudo make install`