DIGImend/digimend-kernel-drivers #106: Huion H610PRO (8192) Undetected by Event Drivers

prijatelj, 2018-01-06:
OS: Arch Linux
Using [digimend drivers from AUR](https://aur.archlinux.org/packages/digimend-kernel-drivers-dkms-git/).

I previously used a Huion K58 with my previous set up and all was well. I used xorgs libinput, didn't need to use the digimend drivers in AUR, had full pressure sensitivity, and the whole works. However, my tablet recently broke so  bought a new one, the Huion H610 Pro (8192). I initially tested it and it works on Mac OsX. However it is not recognized as an input device for xorg libinput, evdev, or wacom in-kernel drivers/libs, nor with the Digimend AUR drivers.

I want my system to be able to recognize the pen and tablet to allow me to use it at least such that the pen and drawing space are fully functional (ie. tablet recognized and pen mouse works with pressure sensitivity on tablet.). The buttons would be nice too, but not main priority.

The device does light up when buttons are pressed, pen is within range or touching work space. There also is output from `sudo usbhid-dump -es -m 256c:006e` when using the pen to draw on the tablet, however no response from button presses on the tablet. Output is as seen [from troubleshoot webpage](https://digimend.github.io/support/howto/trbl/locating_failure/#tablet-hardware).

lsusb indicates the Tablet is connected and detected by the system as a usb device (256c:006e)
`
$ lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 004: ID 1b1c:1b36 Corsair 
Bus 001 Device 003: ID 256c:006e  
Bus 001 Device 002: ID 12c9:100f  
Bus 001 Device 005: ID 8087:07dc Intel Corp. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
`

evtest is unable to detect any Huion Pen Tablet input events
`
$ sudo evtest
No device specified, trying to scan all of /dev/input/event*
Available devices:
/dev/input/event0:	E-Signal USB Gaming Mouse
/dev/input/event1:	E-Signal USB Gaming Mouse
/dev/input/event2:	Sleep Button
/dev/input/event3:	Power Button
/dev/input/event4:	Power Button
/dev/input/event5:	PC Speaker
/dev/input/event6:	HDA Intel PCH Front Mic
/dev/input/event7:	HDA Intel PCH Rear Mic
/dev/input/event8:	HDA Intel PCH Line
/dev/input/event9:	HDA Intel PCH Line Out
/dev/input/event10:	HDA Intel PCH Front Headphone
/dev/input/event11:	HDA NVidia HDMI/DP,pcm=3
/dev/input/event12:	HDA NVidia HDMI/DP,pcm=7
/dev/input/event13:	HDA NVidia HDMI/DP,pcm=8
/dev/input/event14:	HDA NVidia HDMI/DP,pcm=9
/dev/input/event15:	Corsair Corsair Gaming K70 LUX Keyboard 
/dev/input/event16:	Corsair Corsair Gaming K70 LUX Keyboard 
Select the device event number [0-16]: ^C
`

xinput also fails to detect the Huion Pen and Tablet.
`
$ xinput
⎡ Virtual core pointer                    	id=2	[master pointer  (3)]
⎜   ↳ Virtual core XTEST pointer              	id=4	[slave  pointer  (2)]
⎜   ↳ E-Signal USB Gaming Mouse               	id=10	[slave  pointer  (2)]
⎜   ↳ Corsair Corsair Gaming K70 LUX Keyboard 	id=12	[slave  pointer  (2)]
⎜   ↳ E-Signal USB Gaming Mouse               	id=14	[slave  pointer  (2)]
⎣ Virtual core keyboard                   	id=3	[master keyboard (2)]
    ↳ Virtual core XTEST keyboard             	id=5	[slave  keyboard (3)]
    ↳ Power Button                            	id=6	[slave  keyboard (3)]
    ↳ Power Button                            	id=7	[slave  keyboard (3)]
    ↳ Sleep Button                            	id=8	[slave  keyboard (3)]
    ↳ Corsair Corsair Gaming K70 LUX Keyboard 	id=9	[slave  keyboard (3)]
    ↳ E-Signal USB Gaming Mouse               	id=11	[slave  keyboard (3)]
    ↳ Corsair Corsair Gaming K70 LUX Keyboard 	id=13	[slave  keyboard (3)]
`

lsmod | grep hid indicates that hid_uclogic module is loaded, but not used by anything
`
$ lsmod | grep hid
hid_uclogic            16384  0
mac_hid                16384  0
i2c_hid                24576  0
hid_generic            16384  0
usbhid                 57344  0
hid                   131072  4 i2c_hid,hid_generic,usbhid,hid_uclogic
usbcore               282624  5 usbhid,xhci_pci,btusb,hid_uclogic,xhci_hcd
`
Please help me solve this problem.
___
related links I have visited:
[Same problem, similar model posted on reddit](https://www.reddit.com/r/archlinux/comments/4eheqq/drawing_tablets_on_arch_huion_h610_not_registered/?st=jc2l9htg&sh=52717b9d) (uncertain if differences exist between newer H610 Pro models)
[Others can get it working out of the box, but not me.](https://www.reddit.com/r/archlinux/comments/3zctxo/configuring_huion_h610_pro_tablet/?st=jc2g7mt1&sh=0d8f6224) This may be due to newer model?

americapat, 2018-02-19:
Hi Nikolai, I am running the latest git code I think: 
`# modinfo hid_uclogic
filename:       /lib/modules/4.14.0-3-amd64/extra/hid-uclogic.ko
version:        7
license:        GPL
author:         Nikolai Kondrashov
author:         Martin Rusko
srcversion:     949312AF94BBD2A77A78F23`

H610PRO seems to be probed correctly:

`[ 5803.424493] usb 3-10.1: new full-speed USB device number 25 using xhci_hcd
[ 5803.571860] usb 3-10.1: New USB device found, idVendor=256c, idProduct=006e
[ 5803.571862] usb 3-10.1: New USB device strings: Mfr=5, Product=6, SerialNumber=0
[ 5803.581294] input: HID 256c:006e Pen as /devices/pci0000:00/0000:00:14.0/usb3/3-10/3-10.1/3-10.1:1.0/0003:256C:006E.0014/input/input32
[ 5803.581505] input: HID 256c:006e Pad as /devices/pci0000:00/0000:00:14.0/usb3/3-10/3-10.1/3-10.1:1.0/0003:256C:006E.0014/input/input33
[ 5803.581721] uclogic 0003:256C:006E.0014: input,hidraw3: USB HID v1.11 Keypad [HID 256c:006e] on usb-0000:00:14.0-10.1/input0
[ 5803.585468] uclogic 0003:256C:006E.0015: No inputs registered, leaving
[ 5803.585548] uclogic 0003:256C:006E.0015: hidraw4: USB HID v1.11 Device [HID 256c:006e] on usb-0000:00:14.0-10.1/input1
`

but there is an overflow somewhere when I try to use the pen:

`gtk3-demo --gdk-debug=input --run=event_axes
`
`[ 5845.041754] usb 3-10.1: input irq status -75 received
....
`
as long as I use the pen

and evtest, xinput test and so on fail at the user space level just like the gtk3-demo above.
Otherwise I think I have Xfce4 initialized ok (?):

`# xsetwacom --list
HID 256c:006e Pad pad           	id: 10	type: PAD       
HID 256c:006e Pen stylus        	id: 11	type: STYLUS`

`# xinput list-props 11
Device 'HID 256c:006e Pen stylus':
	Device Enabled (138):	1
	Coordinate Transformation Matrix (140):	1.000000, 0.000000, 0.000000, 0.000000, 1.000000, 0.000000, 0.000000, 0.000000, 1.000000
	Device Accel Profile (264):	0
	Device Accel Constant Deceleration (265):	1.000000
	Device Accel Adaptive Deceleration (266):	1.000000
	Device Accel Velocity Scaling (267):	10.000000
	Device Node (261):	"/dev/input/event7"
	Wacom Tablet Area (274):	0, 0, 40000, 25000
	Wacom Rotation (275):	0
	Wacom Pressurecurve (276):	0, 0, 100, 100
	Wacom Serial IDs (277):	110, 0, 2, 0, 0
	Wacom Serial ID binding (278):	0
	Wacom Pressure Threshold (279):	26
	Wacom Sample and Suppress (280):	2, 4
	Wacom Enable Touch (281):	0
	Wacom Hover Click (282):	1
	Wacom Enable Touch Gesture (283):	0
	Wacom Touch Gesture Parameters (284):	0, 0, 250
	Wacom Tool Type (285):	"STYLUS" (263)
	Wacom Button Actions (286):	"Wacom button action 0" (287), "Wacom button action 1" (288), "Wacom button action 2" (289)
	Wacom button action 0 (287):	1572865
	Wacom button action 1 (288):	1572866
	Wacom button action 2 (289):	1572867
	Wacom Pressure Recalibration (290):	1
	Device Product ID (262):	9580, 110
	Wacom Debug Levels (291):	0, 0`

`# usbhid-dump  -es -m 256c:006e` is alive:

`003:025:000:STREAM             1519011459.047448
 08 80 F7 90 9D 53 00 00 00 00 00 00

003:025:000:STREAM             1519011459.051451
 08 80 DD 90 98 53 00 00 00 00 00 00
`

`# uclogic-probe 03 25 | uclogic-decode
 64 |        Params block #1: ??????
    |                          Max X: 40000
    |                          Max Y: 25000
    |                   Max pressure: 8191
    |                     Resolution: 5080
    |
 79 |         Internal model: HA60-F400
 7b |         Buttons status: HK On
 c8 |        Params block #2: ??|?????
    |                          Max X: 50800
    |                          Max Y: 31750
    |                   Max pressure: 8191
    |                     Resolution: 5080
    |
 c9 |      Unknown string #1: HUION_T174_170707
 ca |  Internal manufacturer: HUION Animation Technology Co.,ltd`

what would you need to debug this? Thanks. Best regards. --Pat.

spbnick, 2018-02-19:
@americapat at the moment your only option seems to be using the code from PR #114. I'm working on integrating it into the drivers.

I fixed up formatting of your comment, take a look at the source to figure it out.

americapat, 2018-02-19:
Thanks so much, will have a look and report. Best regards. Pat.

americapat, 2018-03-09:
Hi spbnick, sorry for the very late response; 
`dkms status: digimend,` 114, 4.15.0-1-amd64, x86_64: `installed`
works for me,  tried gtk3-demo and MyPaint not in depth, but it's up and running. Anything I can help, let me know. Cheers. - Pat.

gplanon, 2018-03-20:
@americapat here's my info, we're close to the same kernel version:

digimend-kernel-drivers, 7.r4.g195dabc, 4.15.10-1-MANJARO, x86_64: installed

Running what seems to be the latest git version from the AUR and I still get no cursor movement. Are you saying everything seems to be working for you? Any files I should add/change/modify?

anhtuann, 2018-03-21:
@gplanon You should use the version from [this pull request](https://github.com/DIGImend/digimend-kernel-drivers/pull/114). For the time being, it hasn't been merged yet into the digimend-kernel-drivers. 
Here are the commands I used : 
```
git clone https://github.com/DIGImend/digimend-kernel-drivers.git /tmp/digimend
cd /tmp/digimend
git pull origin pull/114/head
make
sudo make install
```
It should work after a reboot

gplanon, 2018-03-21:
@anhtuann hmmm no dice. I modified the AUR packagebuild to pull ampaza's fork which is referenced in that pull request, no cursor movement. Just tried the good ol' make install on Ubuntu 16.10 and no cursor movement. Not sure if my tablet is cursed with a different firmware or what is going on. lsusb still isn't identifying the tablet by name or brand.

On Arch I also installed xf86-input-wacom and libinput+libwacom, but that didn't seem to change anything.

Happy to run commands or test things on any OS to the best of my ability if compatibility with H610Pro-8192 is pursued.

anhtuann, 2018-03-21:
@gplanon Try to create /etc/X11/xorg.conf.d/10-tablet.conf with :
``` 
Section "InputClass"
    Identifier "Huion on wacom"
    # MatchIsTablet "on"
    MatchProduct "HUION"
    MatchDevicePath "/dev/input/event*"
    Driver "wacom"
EndSection 
```
That conf file + the pull request should solve your problem

gplanon, 2018-03-22:
Do not want to ping anyone with extra comments, but still no luck. I'm posting both in my own interests and in hopes of supporting the project. I will post output of any command I'm asked, here's some info from the device when connected to Windows 10.

Hardware IDs
USB\VID_256C&PID_006E&REV_0000&MI_00
USB\VID_256C&PID_006E&MI_00
Compatible IDs
USB\Class_03&SubClass_01&Prot_02
USB\Class_03&SubClass_01
USB\Class_03
Physical Device Object Name
\Device\00000078
Power Data
Current power state:
D0

Power capabilities:
0000007F
PDCAP_D0_SUPPORTED
PDCAP_D1_SUPPORTED
PDCAP_D2_SUPPORTED
PDCAP_D3_SUPPORTED
PDCAP_WAKE_FROM_D0_SUPPORTED
PDCAP_WAKE_FROM_D1_SUPPORTED
PDCAP_WAKE_FROM_D2_SUPPORTED

Power state mappings:
S0 -> D0
S1 -> D2
S2 -> D2
S3 -> D2
S4 -> D2
S5 -> D3

Reported Device IDs Hash

spbnick, 2018-05-04:
Everyone concerned with this issue, you're welcome to join the new [#DIGImend channel on irc.freenode.net](https://webchat.freenode.net/?channels=DIGImend), discuss this and other issues, and help each other! Come and say "Hi!"

closed 2018-05-04 by spbnick