DIGImend/digimend-kernel-drivers #107: Installing digimend .deb package fails (linux mint, 18.3)

BlackXanthus, 2018-01-12:
The error 
(Reading database ... 917571 files and directories currently installed.)
Preparing to unpack digimend-dkms_6_all.deb ...

------------------------------
Deleting module version: 6
completely from the DKMS tree.
------------------------------
Done.
Unpacking digimend-dkms (6) over (6) ...
Setting up digimend-dkms (6) ...
Loading new digimend-6 DKMS files...
Building only for 4.13.0-26-generic
Building initial module for 4.13.0-26-generic
Error! Bad return status for module build on kernel: 4.13.0-26-generic (x86_64)
Consult /var/lib/dkms/digimend/6/build/make.log for more information.

The result of /var/lib/dkms/digimend/6/build/make.log:

DKMS make.log for digimend-6 for kernel 4.13.0-26-generic (x86_64)
Fri 12 Jan 21:32:10 GMT 2018
make -C /lib/modules/4.13.0-26-generic/build SUBDIRS=/var/lib/dkms/digimend/6/build modules
make[1]: Entering directory '/usr/src/linux-headers-4.13.0-26-generic'
  CC [M]  /var/lib/dkms/digimend/6/build/hid-kye.o
  CC [M]  /var/lib/dkms/digimend/6/build/hid-uclogic.o
/var/lib/dkms/digimend/6/build/hid-uclogic.c:1101:22: error: initialization from incompatible pointer type [-Werror=incompatible-pointer-types]
  .input_configured = uclogic_input_configured,
                      ^
/var/lib/dkms/digimend/6/build/hid-uclogic.c:1101:22: note: (near initialization for ‘uclogic_driver.input_configured’)
cc1: some warnings being treated as errors
scripts/Makefile.build:315: recipe for target '/var/lib/dkms/digimend/6/build/hid-uclogic.o' failed
make[2]: *** [/var/lib/dkms/digimend/6/build/hid-uclogic.o] Error 1
Makefile:1550: recipe for target '_module_/var/lib/dkms/digimend/6/build' failed
make[1]: *** [_module_/var/lib/dkms/digimend/6/build] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-4.13.0-26-generic'
Makefile:9: recipe for target 'modules' failed
make: *** [modules] Error 2

spbnick, 2018-02-17:
This should be fixed with [v7 release](https://github.com/DIGImend/digimend-kernel-drivers/releases/tag/v7).

closed 2018-02-17 by spbnick

ThePreviousOne, 2018-02-17:
Is this a reinstall or upgrade? Like from the non-dkms driver. I'm finding a temp fix is to just reboot and try again.