DIGImend/digimend-kernel-drivers #109: XP-PEN STAR 06 Issues

bahaazahika, 2018-01-17:
> lsusb
`Bus 001 Device 002: ID 28bd:0078  
`

> cat /proc/bus/input/devices

```
I: Bus=0003 Vendor=28bd Product=0078 Version=0100
N: Name="XP-PEN STAR 06"
P: Phys=usb-0000:00:14.0-3/input0
S: Sysfs=/devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3:1.0/0003:28BD:0078.0001/input/input2
U: Uniq=
H: Handlers=event1 mouse0 
B: PROP=0
B: EV=1f
B: KEY=400 3f0000 0 0 0 0
B: REL=303
B: ABS=1000003
B: MSC=10

I: Bus=0003 Vendor=28bd Product=0078 Version=0100
N: Name="XP-PEN STAR 06"
P: Phys=usb-0000:00:14.0-3/input1
S: Sysfs=/devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3:1.1/0003:28BD:0078.0002/input/input3
U: Uniq=
H: Handlers=sysrq kbd event2 mouse1 
B: PROP=0
B: EV=10001b
B: KEY=c01 1ffffffff 1000000000007 ff9f207ac14057ff febeffdfffefffff fffffffffffffffe
B: ABS=1000003
B: MSC=10
```

> xinput
```
Virtual core pointer                    	id=2	[master pointer  (3)]
⎜   ↳ Virtual core XTEST pointer              	id=4	[slave  pointer  (2)]
⎜   ↳ ELAN1010:00 04F3:3012 Touchpad          	id=15	[slave  pointer  (2)]
**⎜   ↳ XP-PEN STAR 06                          	id=12	[slave  pointer  (2)]
⎜   ↳ XP-PEN STAR 06                          	id=11	[slave  pointer  (2)]**
⎜   ↳ Atmel Atmel maXTouch Digitizer          	id=14	[slave  pointer  (2)]
⎜   ↳ ETPS/2 Elantech Touchpad                	id=19	[slave  pointer  (2)]
⎜   ↳ Logitech USB Receiver
```

> xinput list-props 11
```
Device 'XP-PEN STAR 06':
	Device Enabled (141):	1
	Coordinate Transformation Matrix (143):	1.000000, 0.000000, 0.000000, 0.000000, 1.000000, 0.000000, 0.000000, 0.000000, 1.000000
	Device Accel Profile (277):	0
	Device Accel Constant Deceleration (278):	1.000000
	Device Accel Adaptive Deceleration (279):	1.000000
	Device Accel Velocity Scaling (280):	10.000000
	Device Product ID (268):	10429, 120
	Device Node (267):	"/dev/input/event1"
	Evdev Axis Inversion (281):	0, 0
	Evdev Axis Calibration (282):	<no items>
	Evdev Axes Swap (283):	0
	Axis Labels (284):	"Abs X" (274), "Abs Y" (275), "Abs Pressure" (276), "Rel Vert Wheel" (273)
	Button Labels (285):	"Button Left" (144), "Button Middle" (145), "Button Right" (146), "Button Wheel Up" (147), "Button Wheel Down" (148), "Button Horiz Wheel Left" (149), "Button Horiz Wheel Right" (150), "Button Side" (270), "Button Extra" (271), "Button Forward" (272), "Button Unknown" (269), "Button Unknown" (269), "Button Unknown" (269), "Button Unknown" (269)
	Evdev Scrolling Distance (286):	1, 1, 1
	Evdev Middle Button Emulation (287):	0
	Evdev Middle Button Timeout (288):	50
	Evdev Middle Button Button (289):	2
	Evdev Third Button Emulation (290):	0
	Evdev Third Button Emulation Timeout (291):	1000
	Evdev Third Button Emulation Button (292):	3
	Evdev Third Button Emulation Threshold (293):	20
	Evdev Wheel Emulation (294):	0
	Evdev Wheel Emulation Axes (295):	0, 0, 4, 5
	Evdev Wheel Emulation Inertia (296):	10
	Evdev Wheel Emulation Timeout (297):	200
	Evdev Wheel Emulation Button (298):	4
	Evdev Drag Lock Buttons (299):	0
```

> xinput list-props 12
```
Device 'XP-PEN STAR 06':
	Device Enabled (141):	1
	Coordinate Transformation Matrix (143):	1.000000, 0.000000, 0.000000, 0.000000, 1.000000, 0.000000, 0.000000, 0.000000, 1.000000
	Device Accel Profile (277):	0
	Device Accel Constant Deceleration (278):	1.000000
	Device Accel Adaptive Deceleration (279):	1.000000
	Device Accel Velocity Scaling (280):	10.000000
	Device Product ID (268):	10429, 120
	Device Node (267):	"/dev/input/event2"
	Evdev Axis Inversion (281):	0, 0
	Evdev Axis Calibration (282):	<no items>
	Evdev Axes Swap (283):	0
	Axis Labels (284):	"Abs X" (274), "Abs Y" (275), "Abs Pressure" (276)
	Button Labels (285):	"Button Left" (144), "Button Middle" (145), "Button Right" (146), "Button Wheel Up" (147), "Button Wheel Down" (148), "Button Horiz Wheel Left" (149), "Button Horiz Wheel Right" (150), "Button Side" (270), "Button Extra" (271), "Button Forward" (272), "Button Back" (310), "Button Task" (311), "Button 8" (308), "Button 9" (309), "Button Unknown" (269), "Button Unknown" (269), "Button Unknown" (269), "Button Unknown" (269), "Button Unknown" (269), "Button Unknown" (269)
	Evdev Scrolling Distance (286):	0, 0, 0
	Evdev Middle Button Emulation (287):	0
	Evdev Middle Button Timeout (288):	50
	Evdev Middle Button Button (289):	2
	Evdev Third Button Emulation (290):	0
	Evdev Third Button Emulation Timeout (291):	1000
	Evdev Third Button Emulation Button (292):	3
	Evdev Third Button Emulation Threshold (293):	20
	Evdev Wheel Emulation (294):	0
	Evdev Wheel Emulation Axes (295):	0, 0, 4, 5
	Evdev Wheel Emulation Inertia (296):	10
	Evdev Wheel Emulation Timeout (297):	200
	Evdev Wheel Emulation Button (298):	4
	Evdev Drag Lock Buttons (299):	0
```

> evtest
```
Input driver version is 1.0.1
Input device ID: bus 0x3 vendor 0x28bd product 0x78 version 0x100
Input device name: "XP-PEN STAR 06"
Supported events:
  Event type 0 (EV_SYN)
  Event type 1 (EV_KEY)
    Event code 272 (BTN_LEFT)
    Event code 273 (BTN_RIGHT)
    Event code 274 (BTN_MIDDLE)
    Event code 275 (BTN_SIDE)
    Event code 276 (BTN_EXTRA)
    Event code 277 (BTN_FORWARD)
    Event code 330 (BTN_TOUCH)
  Event type 2 (EV_REL)
    Event code 0 (REL_X)
    Event code 1 (REL_Y)
    Event code 8 (REL_WHEEL)
    Event code 9 (REL_MISC)
  Event type 3 (EV_ABS)
    Event code 0 (ABS_X)
      Value      0
      Min        0
      Max    32767
    Event code 1 (ABS_Y)
      Value      0
      Min        0
      Max    32767
    Event code 24 (ABS_PRESSURE)
      Value      0
      Min        0
      Max     8191
  Event type 4 (EV_MSC)
    Event code 4 (MSC_SCAN)
Properties:
Testing ... (interrupt to exit)
```
everything with the pen seems to be working perfectly except the following problem
https://github.com/DIGImend/digimend-kernel-drivers/issues/64#issuecomment-302025857

> but my big problem is that the distance movement works before the pen touches the table.
> When i touch the table, the mouse pointer stucks on the screen when i take a short distance (distance movement is lost). evtest works (i mean the distance stream) but the pointer stucks.
> it's like some buttons stay in one click status even if i release the pen from table.
> of course the pointer starts moving when i move the pen on the table as it should be.

The tablet hard buttons work but they act like keyboard buttons, but this is not that important for me at the moment

bahaazahika, 2018-01-20:
So I have edited the drivers similar to the following reply and the stylus works perfectly with its buttons
https://github.com/DIGImend/digimend-kernel-drivers/issues/64#issuecomment-302513223

added this:

> 90-hid-rehind.rules

```
SUBSYSTEM=="hid", ACTION=="add", ENV{HID_ID}=="0003:000028BD:00000078", \
    RUN+="/sbin/hid-rebind"
```

> hid-ids.h
```
#define USB_VENDOR_ID_XPPEN	0x28bd
#define USB_DEVICE_ID_XPPEN_STAR_06	0x0078
```

> hid-uclogic.c
after line 1172
`case USB_DEVICE_ID_UGEE_TABLET_EX07S:`
I added
`case USB_DEVICE_ID_XPPEN_STAR_06:`
after line 1317
`{ HID_USB_DEVICE(USB_VENDOR_ID_UGEE, USB_DEVICE_ID_UGEE_XPPEN_TABLET_G540) },`
I added
`{ HID_USB_DEVICE(USB_VENDOR_ID_XPPEN, USB_DEVICE_ID_XPPEN_STAR_06) },`

I compiled the drivers and installed them.

and with in xorg.conf.d I added this

> 52-tablet.conf
```
Section "InputClass"
       Identifier "Graphics tablet mouse"
       MatchIsKeyboard "on"
       MatchProduct "XP-PEN STAR 06"
       MatchDevicePath "/dev/input/event*"
       Driver "evdev"
       # Apply custom Options below.
EndSection

Section "InputClass"
       Identifier "Graphics tablet mouse"
       MatchIsPointer "on"
       MatchProduct "XP-PEN STAR 06"
       MatchDevicePath "/dev/input/event*"
       Driver "evdev"
       # Apply custom Options below.
EndSection

Section "InputClass"
       Identifier "Graphics tablet pen"
       MatchIsTablet "on"
       MatchProduct "XP-PEN STAR 06"
       MatchDevicePath "/dev/input/event*"
       Driver "evdev"
       # Apply custom Options below.
EndSection
```

xinput now shows this:
```
⎜   ↳ XP-PEN STAR 06 Pad                      	id=13	[slave  pointer  (2)]
⎜   ↳ XP-PEN STAR 06 Pen                      	id=14	[slave  pointer  (2)]
```

The only problem now left is the buttons , I used

> sudo usbhid-dump -es -m 28bd:0078
```
to collect the following:
row 1:

button 1:

001:003:001:STREAM             1516466794.008388
 03 00 05 00 00 00 00 00

001:003:001:STREAM             1516466794.254327
 03 00 00 00 00 00 00 00

button 2:

001:003:001:STREAM             1516466840.436753
 03 00 2C 00 00 00 00 00

001:003:001:STREAM             1516466840.562724
 03 00 00 00 00 00 00 00

row 2:
button 1:

001:003:001:STREAM             1516466864.426910
 03 00 08 00 00 00 00 00

001:003:001:STREAM             1516466864.552926
 03 00 00 00 00 00 00 00

button 2:
001:003:001:STREAM             1516466886.937078
 03 00 0C 00 00 00 00 00

001:003:001:STREAM             1516466887.033072
 03 00 00 00 00 00 00 00
 
 row 3:

button 1:
001:003:001:STREAM             1516466915.647496
 03 05 1D 00 00 00 00 00

001:003:001:STREAM             1516466915.733323
 03 00 00 00 00 00 00 00

button 2:
001:003:001:STREAM             1516466939.415655
 03 01 16 00 00 00 00 00

001:003:001:STREAM             1516466939.561593
 03 00 00 00 00 00 00 00
 
wheel:

rotate right:
001:003:001:STREAM             1516466992.755907
 03 01 2E 00 00 00 00 00

001:003:001:STREAM             1516466992.757941
 03 00 00 00 00 00 00 00

rotate left:
001:003:001:STREAM             1516467017.148133
 03 01 2D 00 00 00 00 00

001:003:001:STREAM             1516467017.150133
03 00 00 00 00 00 00 00
```

iommu, 2018-03-13:
Can confirm your patches work with me on solus so they seem to be stable (for the drawing side at least). The buttons on mine seem to have the same usbhid-dump as well

iommu, 2018-03-15:
Update : It seems to be an issue with the /* If this is the pen interface */ section 

checking "intf->cur_altsetting->desc.bInterfaceNumber == 0" allows button usage but the 3rd button on the stylus becomes disabled.
checking "intf->cur_altsetting->desc.bInterfaceNumber == 1" (or 2) re-enables the third button on the stylus but the buttons no longer work

reya276, 2018-04-16:
@bahaazahika I also have an XP-Pen Star 06 but I can't get it to work. Which drivers can I install? I download and installed the all.deb package.

reya276, 2018-04-16:
@bahaazahika I ran that cat command and I got this:

I: Bus=0003 Vendor=28bd Product=0078 Version=0100
N: Name="XP-PEN STAR 06"
P: Phys=usb-0000:00:14.0-3/input1
S: Sysfs=/devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3:1.1/0003:28BD:0078.0008/input/input25
U: Uniq=
H: Handlers=sysrq kbd mouse2 event10 
B: PROP=0
B: EV=10001b
B: KEY=c01 1ffffffff 1000000000007 ff9f207ac14057ff febeffdfffefffff fffffffffffffffe
B: ABS=1000003
B: MSC=10

I: Bus=0003 Vendor=28bd Product=0078 Version=0100
N: Name="XP-PEN STAR 06"
P: Phys=usb-0000:00:14.0-3/input0
S: Sysfs=/devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3:1.0/0003:28BD:0078.0009/input/input26
U: Uniq=
H: Handlers=mouse3 event11 
B: PROP=0
B: EV=1f
B: KEY=400 3f0000 0 0 0 0
B: REL=303
B: ABS=1000003
B: MSC=10

the xinput command gives me:

⎡ Virtual core pointer                    	id=2	[master pointer  (3)]
⎜   ↳ Virtual core XTEST pointer              	id=4	[slave  pointer  (2)]
⎜   ↳ Logitech USB Receiver                   	id=13	[slave  pointer  (2)]
⎜   ↳ DLL0798:00 06CB:7E92 Touchpad           	id=16	[slave  pointer  (2)]
⎜   ↳ SynPS/2 Synaptics TouchPad              	id=21	[slave  pointer  (2)]
⎜   ↳ XP-PEN STAR 06                          	id=15	[slave  pointer  (2)]
⎣ Virtual core keyboard                   	id=3	[master keyboard (2)]
    ↳ Virtual core XTEST keyboard             	id=5	[slave  keyboard (3)]
    ↳ Power Button                            	id=6	[slave  keyboard (3)]
    ↳ Video Bus                               	id=7	[slave  keyboard (3)]
    ↳ Video Bus                               	id=8	[slave  keyboard (3)]
    ↳ Power Button                            	id=9	[slave  keyboard (3)]
    ↳ Sleep Button                            	id=10	[slave  keyboard (3)]
    ↳ Integrated_Webcam_HD: Integrate         	id=11	[slave  keyboard (3)]
    ↳ Logitech USB Receiver                   	id=12	[slave  keyboard (3)]
    ↳ Intel HID events                        	id=17	[slave  keyboard (3)]
    ↳ Intel HID 5 button array                	id=18	[slave  keyboard (3)]
    ↳ Dell WMI hotkeys                        	id=19	[slave  keyboard (3)]
    ↳ AT Translated Set 2 keyboard            	id=20	[slave  keyboard (3)]
    ↳ Logitech USB Receiver                   	id=22	[slave  keyboard (3)]
    ↳ XP-PEN STAR 06                          	id=14	[slave  keyboard (3)]

reya276, 2018-04-17:
@bahaazahika So It looks like I'm getting almost what you had. However it is only recognized when I set the driver for wacom instead of evdev.

xinput list
⎡ Virtual core pointer                    	id=2	[master pointer  (3)]
⎜   ↳ Virtual core XTEST pointer              	id=4	[slave  pointer  (2)]
⎜   ↳ Logitech USB Receiver                   	id=13	[slave  pointer  (2)]
⎜   ↳ XP-PEN STAR 06 touch                    	id=14	[slave  pointer  (2)]
⎜   ↳ XP-PEN STAR 06 stylus                   	id=15	[slave  pointer  (2)]
⎜   ↳ DLL0798:00 06CB:7E92 Touchpad           	id=16	[slave  pointer  (2)]
⎜   ↳ SynPS/2 Synaptics TouchPad              	id=21	[slave  pointer  (2)]
⎜   ↳ XP-PEN STAR 06 pad                      	id=23	[slave  pointer  (2)]
⎜   ↳ XP-PEN STAR 06 pad                      	id=24	[slave  pointer  (2)]
⎣ Virtual core keyboard                   	id=3	[master keyboard (2)]
    ↳ Virtual core XTEST keyboard             	id=5	[slave  keyboard (3)]
    ↳ Power Button                            	id=6	[slave  keyboard (3)]
    ↳ Video Bus                               	id=7	[slave  keyboard (3)]
    ↳ Video Bus                               	id=8	[slave  keyboard (3)]
    ↳ Power Button                            	id=9	[slave  keyboard (3)]
    ↳ Sleep Button                            	id=10	[slave  keyboard (3)]
    ↳ Integrated_Webcam_HD: Integrate         	id=11	[slave  keyboard (3)]
    ↳ Logitech USB Receiver                   	id=12	[slave  keyboard (3)]
    ↳ Intel HID events                        	id=17	[slave  keyboard (3)]
    ↳ Intel HID 5 button array                	id=18	[slave  keyboard (3)]
    ↳ Dell WMI hotkeys                        	id=19	[slave  keyboard (3)]
    ↳ AT Translated Set 2 keyboard            	id=20	[slave  keyboard (3)]
    ↳ Logitech USB Receiver                   	id=22	[slave  keyboard (3)]

reya276, 2018-04-17:
@bahaazahika One more thing. When I run this command xsetwacom --list

I get this:

xsetwacom --list
XP-PEN STAR 06 stylus           	id: 15	type: STYLUS    
XP-PEN STAR 06 pad              	id: 23	type: PAD

spbnick, 2018-05-04:
Everyone concerned with this issue, you're welcome to join the new [#DIGImend channel on irc.freenode.net](https://webchat.freenode.net/?channels=DIGImend), discuss this and other issues, and help each other! Come and say "Hi!"

BEpresent, 2018-07-22:
Tried to implement the patch from ghost above, but there does not seem to be a hid-uclogic.c file.

is this patch already implemented in the latest drivers or what method is recommended to recognize the STAR 06 ?

BEpresent, 2018-07-24:
Thanks! Do you know when this will be implemented into the kernel drivers? I don't feel confident to make the changes myself.

There is also a beta driver for Linux on the xp-pen site https://www.xp-pen.com/download/index/cid/36.html , but it does not work for me.

alkumaish, 2018-09-24:
this works in recent pulls (at the time of writing this):

add this to "hid-ids.h" after line 53
#define USB_DEVICE_ID_UGEE_XPPEN_TABLET_STAR06  0x0078

add this to hid-uclogic-core.c after line 389
       { HID_USB_DEVICE(USB_VENDOR_ID_UGEE,
                               USB_DEVICE_ID_UGEE_XPPEN_TABLET_STAR06) },

and this to "hid-uclogic-params.c" after line 1035
       case VID_PID(USB_VENDOR_ID_UGEE,
                    USB_DEVICE_ID_UGEE_XPPEN_TABLET_STAR06):

That should compile and work. choppy and buggy but it works. 

One thing I noticed: There is a clear difference in behavior between using the wireless USB dongle vs using the regular usb cable.

EliterScripts, 2018-12-06:
> and this to "hid-uclogic-params.c" after line 1035
> case VID_PID(USB_VENDOR_ID_UGEE,
> USB_DEVICE_ID_UGEE_XPPEN_TABLET_STAR06):
> 
> That should compile and work. choppy and buggy but it works.

Sorry, but I don't do too much C in particular, but this just looks like bad code, especially where it's been placed. I've tried this before (I tried something else, so trying it again), and it didn't work for my tablet.

You are suggesting that we we add a "case" statement without a "break", which I would assume is code that is just bound to break. I instantly got an error after creating a .deb file of this.

jeffreyantony, 2019-03-10:
Hello All,

I am very much interested in getting the support of this tablet. I can make changes and test if needed. I also own this hardware. Can anyone tell me what is the difficulty in getting this tablet supported in this driver ? I want to get the support iof scroll wheel and buttons.

Meanwhile, I successfully installed this driver(no button support or scroll wheel - only the basic stlylus) on Ubuntu 18.04. I will list down all the changes done.

1) The major change from the above mentioned comments I had to do was to change the drivers to "wacom" instead of "evdev" in the file 52-tablet.conf

This file needs to be kept in the folder _/etc/X11/xorg.conf.d/_

> 52-tablet.conf

```
Section "InputClass"
       Identifier "Graphics tablet mouse"
       MatchIsKeyboard "on"
       MatchProduct "XP-PEN STAR 06"
       MatchDevicePath "/dev/input/event*"
       Driver "wacom"
       # Apply custom Options below.
EndSection

Section "InputClass"
       Identifier "Graphics tablet mouse"
       MatchIsPointer "on"
       MatchProduct "XP-PEN STAR 06"
       MatchDevicePath "/dev/input/event*"
       Driver "wacom"
       # Apply custom Options below.
EndSection

Section "InputClass"
       Identifier "Graphics tablet pen"
       MatchIsTablet "on"
       MatchProduct "XP-PEN STAR 06"
       MatchDevicePath "/dev/input/event*"
       Driver "wacom"
       # Apply custom Options below.
EndSection
```

2) Fixes on the drivers

add below line to "hid-ids.h" after line "#define USB_DEVICE_ID_UGEE_TABLET_EX07S		0x0071"
`#define USB_DEVICE_ID_UGEE_XPPEN_TABLET_STAR06 0x0078`

add below line to hid-uclogic-core.c after line 	"{ HID_USB_DEVICE(USB_VENDOR_ID_UGEE,
				USB_DEVICE_ID_UGEE_XPPEN_TABLET_DECO01) },"
```
{ HID_USB_DEVICE(USB_VENDOR_ID_UGEE,
USB_DEVICE_ID_UGEE_XPPEN_TABLET_STAR06) },
```

and below line to "hid-uclogic-params.c" after line 	"case VID_PID(USB_VENDOR_ID_UGEE,
		     USB_DEVICE_ID_UGEE_TABLET_EX07S):"
```
case VID_PID(USB_VENDOR_ID_UGEE,
USB_DEVICE_ID_UGEE_XPPEN_TABLET_STAR06):
```

EliterScripts, 2019-03-10:
So I contacted XP-pen, and got the engineers to fix the issue with their “driver”. They sent me back a file of a working version of their driver, and I reminded them a while later that they didn’t update their driver on the website. It now works on Ubuntu 18.04, but I don’t like how it works.

You have to run an application and keep it running to get the tablet to work. It would be nice if it could work without seeing the “driver” on the dock (whatever the left bar is called), it ran automatically, and was open source.

I would definitely recommend reaching out to XP-pen and asking if they would be so kind as to release the source code and make it open source. But I’m not good at convincing people to do things. I imagine they wouldn’t mind opening up their driver code to the public, as they didn’t even bother to make sure it worked before having it sit on the website, nor update the publicly available driver for Linux once they put in the work to fix it. To me, they don’t care about the driver, so it shouldn’t be difficult for a smooth talker to convince them to open the driver’s source code and open license it. If they did care about the driver, they would keep it maintained much much better.

Once you can get your hands on the source code, you could reverse engineer it and make it a part of DIGImend.

cme-linux, 2019-10-10:
Like @jeffreyantony I have access to this model of tablet, I'd like better support for it, and I may be able to help with simple testing to bring about that better support.  

For those who don't understand enough C to interpret the patches in step 2 of jeffreyantony's comment, it looks like the patches say, "Treat this tablet exactly the same as an EX07S."  That would explain why the `case` has no `break`; the `break` would be supplied by the existing code for the EX07S.  

My test machine has Ubuntu 16.04.  I've made the following software changes to that machine:  
* Upgraded the kernel from 4.4.x to 4.6.x to troubleshoot an unrelated issue.  
* Installed (from the distro repo) `xserver-xorg-input-libinput` 0.18.0 
* Installed the `52-tablet.conf` from jeffreyantony's comment (it says `Driver "wacom"`) 

I did not modify any source code.  I didn't install `digimend`.  
I did not modify any udev `.rules` file.  
I did not add to `52-tablet.conf` the line `InputDevice "tablet" "AlwaysCore"` that I found somewhere on the Web.  

Results:  
* The tablet ( *with reservations noted below* ) can be used interchangeably with a mouse, even if both are connected to the PC at the same time, even though I didn't add the `AlwaysCore` statement.  When I lift the stylus off of the tablet, the mouse cursor disappears (until the mouse or stylus is used again), which is nice.  :-)
* Reservation 1:  When the stylus's tip is on the tablet, the PC acts like I'm holding down one of the stylus buttons, even though I'm not pushing any button.  If I press and release the primary stylus button, then the PC acts like I released the button and then started holding the button again.  Maybe this is the same problem as in the comment quoted from #64 in the opening comment of this issue.  
* Reservation 2:  I haven't tested the tablet frame's 6 buttons nor its rotary dial, because I suppose the mapping of those to application functions would need to be done by a configuration utility (either standalone or in each application), which I haven't heard of yet.  I suppose I could use `xev` to find out whether the 8 buttons (6 buttons as-such, plus dial-left & dial-right) generate X11 keysyms.  
* The command `ls /dev` does NOT show `/dev/tablet` 
* The tablet is detected by the command `xsetwacom --list devices` 

=> Is there a better driver than `xserver-xorg-input-libinput` that I can install now in `.deb` format?  

=> Are there source-code modifications I can make to `digimend` that will fix the stylus-button issue (Reservation 1) without the result being "choppy and buggy" as @alkumaish noted?  

Other than what I stated above, I haven't noticed my current setup as being "choppy or buggy."  But the stylus-button problem is keeping me from using the tablet enough to tell for sure.  :)  

=> Is there test data I can provide that will help the digimend programmers find out what source-code changes to make, so the stylus buttons will work right & the driver won't be "choppy and buggy" with this tablet?

cme-linux, 2019-10-18:
Attached is USB data as outlined in http://digimend.github.io/support/howto/trbl/diagnostics/

The XP-Pen website indicates in some text (as opposed to a specifications table) that the P01 stylus used with this tablet has two buttons.  (The buttons share an actuator whose shape makes it ambiguous as to how many buttons there are.)  I assumed that this tablet doesn't collect tilt information.  Some of the tests I did only twice, instead of the 3-5 times suggested in the diagnostic instructions.  

[GitHub_109_XP-Pen_Star06_USBHID-dump.zip](https://github.com/DIGImend/digimend-kernel-drivers/files/3745557/GitHub_109_XP-Pen_Star06_USBHID-dump.zip)

Tentatively I will work on setting up a development environment so I can test the code snippets that were given earlier in this issue thread.  (I have a development environment, but not on a machine that has access to this tablet.)  

Earlier contributors to this thread have noted that support for this tablet is almost working already.  I would think that someone familiar with the DIGImend source code could fix right away (at least some of) the issues noted by those contributors, just by looking at the USB data.

gen1321, 2020-05-01:
Do some one have working fork ? is there any progress on issue?  I saw that xpen realised driver, but I cant get it to work properly on Lininputux because of 21:9 display.  I need ability to limit tablet  to some part of the screen. Is it possible?

gen1321, 2020-05-02:
Ok, got it working. The issue was that my tablet had different device id, 0062 so that's why nothing in this thread worked for me, so for anyone else with same problem, pay attention to your deviceid