DIGImend/digimend-kernel-drivers #110: GAOMON PD1560 button mapping

loentar, 2018-01-17:
This pen display is detected as HUION tablet and everything works fine in exception of 10th display button. It's detected as `ABS_X`, not as `BUTTON_9` like other buttons. Because of that it cannot be mapped as hotkey with `xsetwacom` (I didn't find a way to do this).

Here is the log from `evtest` (I press every button in order):

```
Event: time 1516177835.334813, type 4 (EV_MSC), code 4 (MSC_SCAN), value 90001
Event: time 1516177835.334813, type 1 (EV_KEY), code 256 (BTN_0), value 1
Event: time 1516177835.334813, -------------- SYN_REPORT ------------
Event: time 1516177835.456808, type 4 (EV_MSC), code 4 (MSC_SCAN), value 90001
Event: time 1516177835.456808, type 1 (EV_KEY), code 256 (BTN_0), value 0
Event: time 1516177835.456808, -------------- SYN_REPORT ------------
Event: time 1516177835.794793, type 4 (EV_MSC), code 4 (MSC_SCAN), value 90002
Event: time 1516177835.794793, type 1 (EV_KEY), code 257 (BTN_1), value 1
Event: time 1516177835.794793, -------------- SYN_REPORT ------------
Event: time 1516177835.948795, type 4 (EV_MSC), code 4 (MSC_SCAN), value 90002
Event: time 1516177835.948795, type 1 (EV_KEY), code 257 (BTN_1), value 0
Event: time 1516177835.948795, -------------- SYN_REPORT ------------
Event: time 1516177836.226794, type 4 (EV_MSC), code 4 (MSC_SCAN), value 90003
Event: time 1516177836.226794, type 1 (EV_KEY), code 258 (BTN_2), value 1
Event: time 1516177836.226794, -------------- SYN_REPORT ------------
Event: time 1516177836.380791, type 4 (EV_MSC), code 4 (MSC_SCAN), value 90003
Event: time 1516177836.380791, type 1 (EV_KEY), code 258 (BTN_2), value 0
Event: time 1516177836.380791, -------------- SYN_REPORT ------------
Event: time 1516177836.770793, type 4 (EV_MSC), code 4 (MSC_SCAN), value 90004
Event: time 1516177836.770793, type 1 (EV_KEY), code 259 (BTN_3), value 1
Event: time 1516177836.770793, -------------- SYN_REPORT ------------
Event: time 1516177836.862793, type 4 (EV_MSC), code 4 (MSC_SCAN), value 90004
Event: time 1516177836.862793, type 1 (EV_KEY), code 259 (BTN_3), value 0
Event: time 1516177836.862793, -------------- SYN_REPORT ------------
Event: time 1516177837.212793, type 4 (EV_MSC), code 4 (MSC_SCAN), value 90005
Event: time 1516177837.212793, type 1 (EV_KEY), code 260 (BTN_4), value 1
Event: time 1516177837.212793, -------------- SYN_REPORT ------------
Event: time 1516177837.318793, type 4 (EV_MSC), code 4 (MSC_SCAN), value 90005
Event: time 1516177837.318793, type 1 (EV_KEY), code 260 (BTN_4), value 0
Event: time 1516177837.318793, -------------- SYN_REPORT ------------
Event: time 1516177837.618792, type 4 (EV_MSC), code 4 (MSC_SCAN), value 90006
Event: time 1516177837.618792, type 1 (EV_KEY), code 261 (BTN_5), value 1
Event: time 1516177837.618792, -------------- SYN_REPORT ------------
Event: time 1516177837.734791, type 4 (EV_MSC), code 4 (MSC_SCAN), value 90006
Event: time 1516177837.734791, type 1 (EV_KEY), code 261 (BTN_5), value 0
Event: time 1516177837.734791, -------------- SYN_REPORT ------------
Event: time 1516177838.034790, type 4 (EV_MSC), code 4 (MSC_SCAN), value 90007
Event: time 1516177838.034790, type 1 (EV_KEY), code 262 (BTN_6), value 1
Event: time 1516177838.034790, -------------- SYN_REPORT ------------
Event: time 1516177838.146794, type 4 (EV_MSC), code 4 (MSC_SCAN), value 90007
Event: time 1516177838.146794, type 1 (EV_KEY), code 262 (BTN_6), value 0
Event: time 1516177838.146794, -------------- SYN_REPORT ------------
Event: time 1516177838.446793, type 4 (EV_MSC), code 4 (MSC_SCAN), value 90008
Event: time 1516177838.446793, type 1 (EV_KEY), code 263 (BTN_7), value 1
Event: time 1516177838.446793, -------------- SYN_REPORT ------------
Event: time 1516177838.562792, type 4 (EV_MSC), code 4 (MSC_SCAN), value 90008
Event: time 1516177838.562792, type 1 (EV_KEY), code 263 (BTN_7), value 0
Event: time 1516177838.562792, -------------- SYN_REPORT ------------
Event: time 1516177838.842790, type 4 (EV_MSC), code 4 (MSC_SCAN), value d0020
Event: time 1516177838.842790, type 1 (EV_KEY), code 264 (BTN_8), value 1
Event: time 1516177838.842790, -------------- SYN_REPORT ------------
Event: time 1516177838.962790, type 4 (EV_MSC), code 4 (MSC_SCAN), value d0020
Event: time 1516177838.962790, type 1 (EV_KEY), code 264 (BTN_8), value 0
Event: time 1516177838.962790, -------------- SYN_REPORT ------------
Event: time 1516177839.238791, type 3 (EV_ABS), code 0 (ABS_X), value 1
Event: time 1516177839.238791, -------------- SYN_REPORT ------------
Event: time 1516177839.368790, type 3 (EV_ABS), code 0 (ABS_X), value 0
Event: time 1516177839.368790, -------------- SYN_REPORT ------------
```

loentar, 2018-02-11:
Output generated by uclogin-tools:

[uclogic-decode-gaomon-pd1560.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/1714108/uclogic-decode-gaomon-pd1560.txt)
[uclogic-probe-gaomon-pd1560.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/1714109/uclogic-probe-gaomon-pd1560.txt)

Eqlipse, 2018-03-03:
Out of curiosity, how did you get your system to even see it as a HUION product? Mine doesn't even show up using "xsetwacom list."

Here's what shows up under some simple investigations:
Using xinput...

> eqlipse@eqlipse-Secondary-Linux:$ xinput list
> ⎡ Virtual core pointer                          id=2    [master pointer  (3)]
> ⎜   ↳ Virtual core XTEST pointer                id=4    [slave  pointer  (2)]
> ⎜   ↳ Tablet Monitor Pad                        id=8    [slave  pointer  (2)]
> ⎜   ↳ Logitech M570                             id=10   [slave  pointer  (2)]
> ⎜   ↳ Logitech USB-PS/2 Optical Mouse           id=11   [slave  pointer  (2)]
> ⎜   ↳ ckb1: Corsair STRAFE Gaming Keyboard      id=12   [slave  pointer  (2)]
> ⎜   ↳ ckb1: Corsair STRAFE Gaming Keyboard      id=13   [slave  pointer  (2)]
> ⎜   ↳ Tablet Monitor Pen Pen (0)                id=16   [slave  pointer  (2)]
> ⎣ Virtual core keyboard                         id=3    [master keyboard (2)]
>     ↳ Virtual core XTEST keyboard               id=5    [slave  keyboard (3)]
>     ↳ Power Button                              id=6    [slave  keyboard (3)]
>     ↳ Power Button                              id=7    [slave  keyboard (3)]
>     ↳ Tablet Monitor Pen                        id=9    [slave  keyboard (3)]
>     ↳ ckb1: Corsair STRAFE Gaming Keyboard      id=14   [slave  keyboard (3)]
>     ↳ ckb1: Corsair STRAFE Gaming Keyboard      id=15   [slave  keyboard (3)]
> eqlipse@eqlipse-Secondary-Linux:$ xinput query-state 8
> 2 classes :
> ButtonClass
>         button[1]=up
>         button[2]=up
>         button[3]=up
>         button[4]=up
>         button[5]=up
>         button[6]=up
>         button[7]=up
>         button[8]=up
>         button[9]=up
>         button[10]=up
>         button[11]=up
>         button[12]=up
>         button[13]=up
> ValuatorClass Mode=Absolute Proximity=In
>         valuator[0]=0
>         valuator[1]=0
>         valuator[2]=0
>         valuator[3]=0
>         valuator[4]=0
>         valuator[5]=0
>         valuator[6]=0
> eqlipse@eqlipse-Secondary-Linux:~$ xinput list-props 8
> Device 'Tablet Monitor Pad':
>         Device Enabled (141):   1
>         Coordinate Transformation Matrix (143): 1.000000, 0.000000, 0.000000, 0.000000, 1.000000, 0.000000, 0.000000, 0.000000, 1.000000
>         libinput Send Events Modes Available (261):     1, 0
>         libinput Send Events Mode Enabled (262):        0, 0
>         libinput Send Events Mode Enabled Default (263):        0, 0
>         Device Node (264):      "/dev/input/event7"
>         Device Product ID (265):        9580, 110

Using evtest...
> eqlipse@eqlipse-Secondary-Linux:~$ sudo evtest
> No device specified, trying to scan all of /dev/input/event*
> Available devices:
> /dev/input/event0:      Power Button
> /dev/input/event1:      Power Button
> /dev/input/event2:      Logitech M570
> /dev/input/event3:      ckb1: Corsair STRAFE Gaming Keyboard
> /dev/input/event4:      ckb1: Corsair STRAFE Gaming Keyboard
> /dev/input/event5:      Logitech USB-PS/2 Optical Mouse
> /dev/input/event6:      Tablet Monitor Pen
> /dev/input/event7:      Tablet Monitor Pad
> /dev/input/event8:      HDA ATI HDMI HDMI/DP,pcm=3
> /dev/input/event9:      HDA ATI HDMI HDMI/DP,pcm=7
> /dev/input/event10:     HDA ATI HDMI HDMI/DP,pcm=8
[... 72 lines cut ...]
> Event: time 1520051407.933282, type 4 (EV_MSC), code 4 (MSC_SCAN), value 90005
> Event: time 1520051407.933282, type 1 (EV_KEY), code 260 (BTN_4), value 0
> Event: time 1520051407.933282, -------------- SYN_REPORT ------------
> Event: time 1520051409.527292, type 4 (EV_MSC), code 4 (MSC_SCAN), value 90007
> Event: time 1520051409.527292, type 1 (EV_KEY), code 262 (BTN_6), value 1
> Event: time 1520051409.527292, -------------- SYN_REPORT ------------
> Event: time 1520051409.741292, type 4 (EV_MSC), code 4 (MSC_SCAN), value 90007
> Event: time 1520051409.741292, type 1 (EV_KEY), code 262 (BTN_6), value 0
> Event: time 1520051409.741292, -------------- SYN_REPORT ------------
> Event: time 1520051410.087296, type 4 (EV_MSC), code 4 (MSC_SCAN), value 90008
> Event: time 1520051410.087296, type 1 (EV_KEY), code 263 (BTN_7), value 1
> Event: time 1520051410.087296, -------------- SYN_REPORT ------------
> Event: time 1520051410.301295, type 4 (EV_MSC), code 4 (MSC_SCAN), value 90008
> Event: time 1520051410.301295, type 1 (EV_KEY), code 263 (BTN_7), value 0
> Event: time 1520051410.301295, -------------- SYN_REPORT ------------
> Event: time 1520051410.663298, type 4 (EV_MSC), code 4 (MSC_SCAN), value d0020
> Event: time 1520051410.663298, type 1 (EV_KEY), code 264 (BTN_8), value 1
> Event: time 1520051410.663298, -------------- SYN_REPORT ------------
> Event: time 1520051410.845298, type 4 (EV_MSC), code 4 (MSC_SCAN), value d0020
> Event: time 1520051410.845298, type 1 (EV_KEY), code 264 (BTN_8), value 0
> Event: time 1520051410.845298, -------------- SYN_REPORT ------------
> Event: time 1520051411.331302, type 3 (EV_ABS), code 0 (ABS_X), value 1
> Event: time 1520051411.331302, -------------- SYN_REPORT ------------
> Event: time 1520051411.525302, type 3 (EV_ABS), code 0 (ABS_X), value 0
> Event: time 1520051411.525302, -------------- SYN_REPORT ------------

loentar, 2018-03-03:
Probably it was captured by evdev driver.

Try to edit your `/usr/share/X11/xorg.conf.d/10-evdev.conf` commenting out the tablet section like this:

```
#Section "InputClass"
#        Identifier "evdev tablet catchall"
#        MatchIsTablet "on"
#        MatchDevicePath "/dev/input/event*"
#        Driver "evdev"
#EndSection
```

Also I've added a new conf like this: 
`/usr/share/X11/xorg.conf.d/52-tablet.conf`:

```
Section "InputClass"
  Identifier "GAOMON"
  MatchUSBID "256c:006e"
  MatchDevicePath "/dev/input/event*"
  Driver "wacom"
EndSection
```

Eqlipse, 2018-03-03:
I do not have a `/usr/share/X11/xorg.conf.d/52-tablet.conf`

However, I do seem to have an equivalent `/usr/share/X11/xorg.conf.d/70-wacom.conf`and placing the suggested 52-tablet.conf change in there, as well as disabling the evdev capture seemed to work. 

xsetwacom can now see and interact with the tablet ^_^ Thanks!

However, directly tinkering with the drivers on LInux is something I've never done before. Where exactly would I be inserting that if-then line of coding?

loentar, 2018-03-04:
Well.. I applied it over anpaza's repository. Here is short instruction (didn't checked it):

```
mkdir -p ~/install
cd ~/install
git clone https://github.com/anpaza/digimend-kernel-drivers.git digimend-kernel-drivers-gaomon
cd digimend-kernel-drivers-gaomon
mcedit +736 hid-uclogic-core.c
```

Copy and paste lines in my previous comment before this line:
```
if (drvdata->has_virtual_pad_interface && (data[1] & 0x20))
```
save and exit.

Then compile and install driver like this:

```
sudo rmmod hid-uclogic
make clean
make
sudo make install
sudo modprobe hid-uclogic
```

Also I have this script to initialize device parameters. The pen display is configured to be second display to the right of main display:

```
#!/bin/bash

keyboard="Huion  Tablet Pad pad"
tablet="Huion  Tablet Pen stylus"

bindings=(
"+shift"
"+ctrl"
"+z" # undo
"+x" # redo

"/"  # last brush
";"  # last layer

"+e"
"+a"
"+home"
"+m"
)

buttons=(1 2 3 8 9 10 11 12 13 14)

i=0
for binding in "${bindings[@]}"
do
  xsetwacom --set "$keyboard" Button ${buttons[$((i++))]} key $binding
done

xinput set-prop "$tablet" --type=float "Coordinate Transformation Matrix" 0.5 0 0.5 0 1 0 0 0 1
xsetwacom --set "$tablet" Area 200 0 65100 38500
xsetwacom --set "$tablet" Button 2 button +2
xsetwacom --set "$tablet" Button 3 button +3
```

Eqlipse, 2018-03-18:
Sorry for the late reply. Thanks, that did the trick! Now it's seeing the full sensitivity range, too, or at least much better than it had before (something I previously forced myself to "get used to" instead). Of course I switched up the keybindings to suite my preferences. 

Some of those variables I'll have to look into, such as the "Coordinate Transformation Matrix." I like to know what's actually being done inside the code, after all. :)

spbnick, 2018-05-04:
@danghai, do you think you could add this tablet to the tablets repo?

spbnick, 2018-05-04:
Everyone concerned with this issue, you're welcome to join the new [#DIGImend channel on irc.freenode.net](https://webchat.freenode.net/?channels=DIGImend), discuss this and other issues, and help each other! Come and say "Hi!"

veryprofessionaldodo, 2018-05-05:
I just recently bought this device, and it works great! Though there are two things that are really bugging me, and one of them is really strange... If I go to Windows, and then switch back to Linux, the Tablet will not recognize anything the stylus does. 
Evtest shows absolutely nothing. I need to power down the pc, wait for it to drain all power, and then if I restart back into Linux, everything is working fine again...

The other thing is that Windows show significantly less lag on input. I already tweaked the Sample and Supress values of the xinput, and that fixed a great deal (why was it so smoothed out in the first place, I find it really strange). 

Does anyone know how to improve latency of this device? Sorry for necro-ing this old thread, but I'm really at a loss, I much rather work on Linux over Windows.

veryprofessionaldodo, 2018-05-05:
Okay, so quick answer to my previous comment in regards to that strange Windows thing. To solve it you need to have the Windows Gaomon driver UNINSTALLED. Uninstalled it now, powered down pc, restarted, switched OSes without a hitch. I have a feeling though that the lag just might be something out of my reach... Will report later!

maxim-sheronov, 2018-05-05:
Hi. Maybe it can help. I have similar problem, that tablet does not recognize the pen. To fix it, I have to do:
sudo rmmod hid-uclogic
sudo modprobe hid-uclogic

And then it works.

veryprofessionaldodo, 2018-05-05:
Thanks! Do you know of a way of reducing pen latency though? That is my biggest concern right now... If I can't get this at least imperceptibly on par with Windows, then I'm going to have to work there, which I really didn't want...

danghai, 2018-05-05:
@spbnick Yeah sure. 
@loentar Hello, could you help me provide information about descriptor, frame_buttons, pen_buttons, pen_coords, pen_pressure, pen_tilt, hid_report_descriptors? Thanks a lot

veryprofessionaldodo, 2018-05-05:
@danghai I can't right now, but I'll do that tomorrow in the afternoon. Thanks for the help!

veryprofessionaldodo, 2018-05-05:
But just to make sure, you would like the output of xinput, and maybe lsusb? I can also present xsetwacom --list devices, and a little bit of evtest

loentar, 2018-05-05:
@danghai sorry, I do not own this device anymore.

veryprofessionaldodo, 2018-05-05:
It's ok, thanks anyway!

veryprofessionaldodo, 2018-05-06:
What a brainfart, I didn't read the tags and thought you were talking to me, @danghai... Sorry guys!

danghai, 2018-05-08:
@veryprofessionaldodo Thanks, would you mind to provide your `probe.txt`? You can review the command from there: https://github.com/DIGImend/uclogic-tools to get information your probe. Thanks

spbnick, 2018-05-08:
@veryprofessionaldodo, could you please use the latest development version of uclogic-tools from here: https://github.com/DIGImend/uclogic-tools/archive/master.zip (I need to make a release of that).

veryprofessionaldodo, 2018-05-08:
I had installed it from the AUR, but it looks like it isn't that up to date. I uninstalled my previous version, installed directly from the repository, and now I get these results.

[probe.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/1983402/probe.txt)

veryprofessionaldodo, 2018-05-16:
And I think I got it. The difference in latency in Windows is barely perceptible, and something I'll easily get used to. I have made this script in case anyone wants to use it, with button bindings made for Krita.

This is my particular setup in terms of display, if you want to know more about the coordinate transformation matrix https://github.com/linuxwacom/xf86-input-wacom/wiki/Dual-and-Multi-Monitor-Set-Up

[gaomon.zip](https://github.com/DIGImend/digimend-kernel-drivers/files/2010780/gaomon.zip)

teleportingtortoise, 2018-07-14:
Tried @loentar 's fix but got issue #160 instead when trying to compile.

spbnick, 2018-07-14:
@teleportingtortoise, please try the new release: https://github.com/DIGImend/digimend-kernel-drivers/releases/v8

teleportingtortoise, 2018-07-14:
@spbnick Thank you, the new release seems to work with this pen display without any noticeable issues.

jonnor, 2020-04-02:
So it seems the PD1560 works OK for a while. Does anyone know if the smaller version, PD1161 can work also?

dexter30, 2020-07-16:
@jonnor I just tried it out with my PD1161 on manjaro. Seems to work all fine. Just getting the transformation set up now.

BloodyMess, 2021-01-21:
What about the PD2200? Thinking of getting it, or one of the XP-Pen Artist 22 Pro models, but I want to be sure it will work on linux first.

loentar, 2021-01-24:
@BloodyMess As I remember PD1650 had noticable lags and pen offset with digimend drivers when comparing to windows version. 

XP-PEN 22E HD Pro still works well for me, see #168 for details. I use it almost every day. There is a newer version of this display on market, can't say anything about it.

BloodyMess, 2021-01-25:
> @BloodyMess As I remember PD1650 had noticable lags and pen offset with digimend drivers when comparing to windows version.
> 
> XP-PEN 22E HD Pro still works well for me, see #168 for details. I use it almost every day. There is a newer version of this display on market, can't say anything about it.

@loentar Have you tried the methods that @veryprofessionaldodo and @teleportingtortoise say they've had success with?

I went ahead and ordered the PD2200. I would have gone with the XP-Pen 22E, but it doesn't have support for detecting/reporting stylus tilt. It also isn't fully laminated, and I don't want to deal with parallax.

loentar, 2021-01-25:
@BloodyMess Unfortunately no, I sold it before theirs solution. I wanted a larger pen display with better characteristics at that moment.

Well, if you haven't had a pen display before, the parallax effect is unavoidable. The difference between PD1650 and 22E was very noticable. PD1650 had a parallax gap about 2.5mm in worst case on the right bottom edge.

I'm not sure if the "laminate" technology can solve the parallax effect, because it's more connected to the display itself not the pen sensor. But I hope they greatly improved their sensor, 3 years is a pretty long period to make it work better.

Once you have a chance to try PD2200, please say how it works for you.

NamesNetworkNic, 2021-11-03:
This issue hasn't been updated in a long time.  Has their been any developments?

veryprofessionaldodo, 2021-11-05:
@NamesNetworkNic I haven't had any problems with my Gaomon PD1560 on Linux, using standard drivers and digimend-drivers-git-dkms from AUR.