DIGImend/digimend-kernel-drivers #117: Find a way to provide default button configuration

spbnick, 2018-02-15:
Huion requested to have default hotkey configuration for buttons on tablet frames. We need to figure out how to do that.

spbnick, 2018-02-16:
Yes, libwacom is probably where we'll have to go, and also submitting descriptions of Huion tablets there.

reya276, 2018-04-17:
I'm actually trying to develop a config tool for each tablet listed within this project.

[image]

ozeszty, 2018-04-20:
Nice, are you going to support multiple (e.g. per application or task) hotkeys' configs? Even in one app, multiple configs could be useful (painting or animating in Krita).

Some ideas for development:

- Some pens have rubber on the other end serving as another button.
- It should be possible to emulate another button when both buttons are pressed together (or in minimal interval, e.g. < 300 ms), like with touchpads and getting middle-click by pressing left and right buttons together.

megamuteki, 2018-05-16:
1.I would like to translate this code into japanese.
2.I can provide photoes and default key-code.
3.I have many XP-PEN tablets, if you need and I have one, I can dump data.

Lvaskz, 2018-06-14:
Hi, this is very usefull. Do you have any advance about it?

reya276, 2018-06-14:
@Lvaskz Not yet, trying to see if I can get some developers to help me out.

reya276, 2018-06-14:
@megamuteki I sent you some emails about the project, send me anything you have.

reya276, 2018-07-02:
@Lvaskz no not really I've been swamped at work so I have not had time to get back to it. I'm waiting on some other folks to help me with the scripts side of it.

Lvaskz, 2018-07-02:
I see, thanks to reply. Well I will wait.

megamuteki, 2018-07-04:
I have been tried to work on Ubntu 16.04 LTS.
The scripts does not work on ubuntu 16.04 LTS easily  becuase of glib issues.
I reccomend to use higher than Ubuntu 18.04.

megamuteki, 2018-07-04:
I have retrieved XP-PEN tablet datas.
If  you need more info, please let me know.
I have many xp-pen tablet.
[XpPenTabletData.zip](https://github.com/DIGImend/digimend-kernel-drivers/files/2161580/XpPenTabletData.zip)

ozeszty, 2019-03-02:
https://github.com/KDE/wacomtablet rocks! On top of what GNOME's tablet settings provide, it supports,  among other features, config profiles. It played well with Huion H610 Pro.

Lvaskz, 2020-02-22:
hi, a long time has passed.!!! and I wonder if it has some progress?

Lvaskz, 2021-07-19:
Hi, any progress? Or definitively this willl not happen?