DIGImend/digimend-kernel-drivers #118: Mint 18 on Lenovo Y50 with UC-Logic Tablet PF1209 Not working

Zaldor, 2018-02-16:
I tried it all, help would be greatly appreciated.
Thanks

spbnick, 2018-02-16:
It would help people help you, if you defined "not working". Also please see https://digimend.github.io/support/howto/trbl/locating_failure/

spbnick, 2018-02-16:
Start here and go down: https://digimend.github.io/support/howto/trbl/locating_failure/#userspace-drivers
Check application settings.

Zaldor, 2018-02-16:
thanks, running

> sudo usbhid-dump -es -m 5543:0042
i can see all working fine... movement and pressure

but when i run

> xinput  test 13

it seams all fucked up...

Zaldor, 2018-02-16:
I have done another test, I installed MyPaint and it recognized the pen the movement and everything but only untill I clicked the Button on the pen, from that moment it seams like it started to consider the pen touching the surface as a button click like xinput test seams to do aswell

Zaldor, 2018-02-16:
Another out that may help figure out:

T:  Bus=03 Lev=01 Prnt=01 Port=00 Cnt=01 Dev#=  8 Spd=12  MxCh= 0
D:  Ver= 1.10 Cls=00(>ifc ) Sub=00 Prot=00 MxPS= 8 #Cfgs=  1
P:  Vendor=5543 ProdID=0042 Rev=00.00
S:  Manufacturer=               
S:  Product=TabletPF1209
C:  #Ifs= 1 Cfg#= 1 Atr=a0 MxPwr=100mA
I:  If#= 0 Alt= 0 #EPs= 1 Cls=03(HID  ) Sub=01 Prot=02 Driver=usbhid