DIGImend/digimend-kernel-drivers #120: Gaomon M106K tablet

BrunoVernay, 2018-02-17:
Looks like it has been done: https://github.com/DIGImend/digimend-kernel-drivers/pull/85 ... BUT not integrated yet.
Indeed, it does not appear in the release notes v7, nor in the Tablet list on the web site.

BUT: You can get the repo from https://github.com/anpaza/digimend-kernel-drivers and install from source.

I get some errors (but it works):
```
make -C /lib/modules/4.15.3-300.fc27.x86_64/build SUBDIRS=/srv/data/dev/zdev/digimend-kernel-drivers-master modules_install
make[1]: Entering directory '/usr/src/kernels/4.15.3-300.fc27.x86_64'
  INSTALL /srv/data/dev/zdev/digimend-kernel-drivers-master/hid-kye.ko
At main.c:160:
- SSL error:02001002:system library:fopen:No such file or directory: crypto/bio/bss_file.c:74
- SSL error:2006D080:BIO routines:BIO_new_file:no such file: crypto/bio/bss_file.c:81
sign-file: certs/signing_key.pem: No such file or directory
  INSTALL /srv/data/dev/zdev/digimend-kernel-drivers-master/hid-polostar.ko
At main.c:160:
- SSL error:02001002:system library:fopen:No such file or directory: crypto/bio/bss_file.c:74
- SSL error:2006D080:BIO routines:BIO_new_file:no such file: crypto/bio/bss_file.c:81
sign-file: certs/signing_key.pem: No such file or directory
  INSTALL /srv/data/dev/zdev/digimend-kernel-drivers-master/hid-uclogic.ko
At main.c:160:
- SSL error:02001002:system library:fopen:No such file or directory: crypto/bio/bss_file.c:74
- SSL error:2006D080:BIO routines:BIO_new_file:no such file: crypto/bio/bss_file.c:81
sign-file: certs/signing_key.pem: No such file or directory
  INSTALL /srv/data/dev/zdev/digimend-kernel-drivers-master/hid-viewsonic.ko
At main.c:160:
- SSL error:02001002:system library:fopen:No such file or directory: crypto/bio/bss_file.c:74
- SSL error:2006D080:BIO routines:BIO_new_file:no such file: crypto/bio/bss_file.c:81
sign-file: certs/signing_key.pem: No such file or directory
  DEPMOD  4.15.3-300.fc27.x86_64
make[1]: Leaving directory '/usr/src/kernels/4.15.3-300.fc27.x86_64'
install -D -m 0644 digimend.conf /etc/depmod.d/digimend.conf
depmod -a
install hid-rebind /sbin/hid-rebind
install -m 0644 90-hid-rebind.rules /lib/udev/rules.d/90-hid-rebind.rules
udevadm control --reload
```
I use a Fedora 27, I have to sign the drivers:
```
# https://gorka.eguileor.com/vbox-vmware-in-secureboot-linux-2016-update/
# https://access.redhat.com/documentation/en-us/red_hat_enterprise_linux/7/html/system_administrators_guide/sect-signing-kernel-modules-for-secure-boot

for f in $(dirname $(modinfo -n hid-kye))/*.ko; do 
    echo "Signing $f"; 
    sudo /usr/src/kernels/$(uname -r)/scripts/sign-file sha256 your.priv your.der $f;
done
```

Don't even need to reboot.

It might not be an "issue", but I spent 2 hours before realizing that the pull request was closed, but not yet merged ...

kolijoker, 2018-02-25:
I'm sorry, I'm using 17.10 and this is not work for my M106K, can you help me with it somehow, please?

kolijoker, 2018-02-26:
I found that my M106K is not listed in the evtest
`kolijoker@kolijoker-Lenovo-Y520-15IKBN:~$ lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 003: ID 5986:06b0 Acer, Inc 
Bus 001 Device 018: ID 256c:006e  
Bus 001 Device 004: ID 0bda:0821 Realtek Semiconductor Corp. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
kolijoker@kolijoker-Lenovo-Y520-15IKBN:~$ sudo usbhid-dump -es -m 256c:006e
[sudo] пароль для kolijoker: 
Starting dumping interrupt transfer stream
with 1 minute timeout.

001:018:001:STREAM             1519605285.826240
 0A C0 A3 5E D9 22 00 00

001:018:001:STREAM             1519605285.830427
 0A C0 A8 3A B4 36 00 00
`
I checked every of this events. I tried to connect tablet to every port.
`root@kolijoker-Lenovo-Y520-15IKBN:/sys/kernel/debug/hid/0003:256C:006E.001B# evtest
No device specified, trying to scan all of /dev/input/event*
Available devices:
/dev/input/event0:	Lid Switch
/dev/input/event1:	Power Button
/dev/input/event2:	Power Button
/dev/input/event3:	AT Translated Set 2 keyboard
/dev/input/event4:	Video Bus
/dev/input/event5:	Video Bus
/dev/input/event6:	SynPS/2 Synaptics TouchPad
/dev/input/event7:	Ideapad extra buttons
/dev/input/event8:	EasyCamera: EasyCamera
/dev/input/event9:	HDA Intel PCH Mic
/dev/input/event10:	HDA Intel PCH Headphone
/dev/input/event11:	HDA Intel PCH HDMI/DP,pcm=3
/dev/input/event12:	HDA Intel PCH HDMI/DP,pcm=7
/dev/input/event13:	HDA Intel PCH HDMI/DP,pcm=8
/dev/input/event14:	HDA Intel PCH HDMI/DP,pcm=9
/dev/input/event15:	HDA Intel PCH HDMI/DP,pcm=10
Select the device event number [0-15]: ^C`

BrunoVernay, 2018-02-26:
You mean Ubuntu 17.10?  In any case, until the pull request is merged and a new release is created v8 or more,  you will have to build it from sources using this repo:  https://github.com/anpaza/digimend-kernel-drivers 
Note that it stopped working on my side, I did not have time to investigate yet. (But I used Patreon to contribute.)

danghai, 2018-05-05:
Hello @BrunoVernay, @kolijoker  , could you help me provide the information about your tablet in terms of descriptor, frame_buttons, pen_buttons, pen_coords, pen_pressure, pen_tilt, hid_report_descriptors for better support?

BrunoVernay, 2018-05-06:
Also there are tons of outputs in issue #134 . and to me also, it is not clear which commend would give the outputs you expect.

danghai, 2018-05-06:
Hi @kolijoker @BrunoVernay 
I need some information about these command:
http://digimend.github.io/support/howto/trbl/diagnostics/
http://digimend.github.io/support/howto/trbl/locating_failure/
Thanks

kolijoker, 2018-05-06:
Hi @danghai , there is no pen_tilt in this tablet. And frame_buttons gives no output at all.
lsusb:
`Bus 001 Device 006: ID 256c:006e
`
[discriptors.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/1978067/discriptors.txt)
[hid_report_descriptor.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/1978068/hid_report_descriptor.txt)
[pen_buttons.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/1978069/pen_buttons.txt)
[pen_cords.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/1978070/pen_cords.txt)
[pen_pressure.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/1978071/pen_pressure.txt)

danghai, 2018-05-08:
@kolijoker Thanks, would you mind to provide your probe.txt? You can review the command from there: https://github.com/DIGImend/uclogic-tools to get information your probe. Thanks!

MadMouvies, 2018-11-27:
Hi @danghai  l think i may have the same issue as @kolijoker, is there any news about this problem ? Thanks!

Flashwalker, 2018-12-04:
Is there a case of drawing works fine on this tablet under Linux?

BrunoVernay, 2018-12-05:
I have one, it used to work about a year ago, I will test it again this week.

BrunoVernay, 2018-12-05:
lsusb first gives
`Bus 001 Device 020: ID 0416:a316 Winbond Electronics Corp. `
a few seconds after it is replaced by
`Bus 001 Device 021: ID 256c:006e`

[descriptors.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/2650460/descriptors.txt)
[hid_report_descriptors.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/2650464/hid_report_descriptors.txt)
[frame_wheel_srolling.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/2650466/frame_wheel_srolling.txt)

-  `evtest`  does not show any Pen, Pad or anything that looks related to tablets.
```
xinput list
⎡ Virtual core pointer                    	id=2	[master pointer  (3)]
⎜   ↳ Virtual core XTEST pointer              	id=4	[slave  pointer  (2)]
⎜   ↳ xwayland-pointer:15                     	id=6	[slave  pointer  (2)]
⎜   ↳ xwayland-relative-pointer:15            	id=7	[slave  pointer  (2)]
⎣ Virtual core keyboard                   	id=3	[master keyboard (2)]
    ↳ Virtual core XTEST keyboard             	id=5	[slave  keyboard (3)]
    ↳ xwayland-keyboard:15                    	id=8	[slave  keyboard (3)]
```

BrunoVernay, 2018-12-05:
Not sure if it is legit, but I have a 
```
cat /lib/udev/rules.d/90-hid-rebind.rules
SUBSYSTEM=="hid", ACTION=="add", ENV{HID_ID}=="0003:0000099A:00002620", \
	RUN+="/sbin/hid-rebind"
SUBSYSTEM=="hid", ACTION=="add", ENV{HID_ID}=="0003:0000256C:*", \
	RUN+="/sbin/hid-rebind"
SUBSYSTEM=="hid", ACTION=="add", ENV{HID_ID}=="0003:00005543:*", \
	RUN+="/sbin/hid-rebind"
SUBSYSTEM=="hid", ACTION=="add", ENV{HID_ID}=="0003:00000458:*", \
	RUN+="/sbin/hid-rebind"
SUBSYSTEM=="hid", ACTION=="add", ENV{HID_ID}=="0003:00002179:*", \
	RUN+="/sbin/hid-rebind"
SUBSYSTEM=="hid", ACTION=="add", ENV{HID_ID}=="0003:000028BD:*", \
	RUN+="/sbin/hid-rebind"
SUBSYSTEM=="hid", ACTION=="add", ENV{HID_ID}=="0003:00000543:0000E621", \
	RUN+="/sbin/hid-rebind"
SUBSYSTEM=="hid", ACTION=="add", ENV{HID_ID}=="0003:00002133:00000018", \
	RUN+="/sbin/hid-rebind"
```

Flashwalker, 2018-12-06:
Ok. Just bought ** gaomon m106k** , installed **digimend-dkms_8_all.deb** on Linux Mint 17  (Ubuntu 14.04), and looks like drawing and pen buttons works fine. :grinning: Tested on MyPaint, Gimp, Inkscape.

The only issue is low perfomance, because tested on veeery low netbook.

Flashwalker, 2018-12-06:
@BrunoVernay if you replied to me, i have tested on 3.16 kernel. Although the README specifies a minimum requirement of 3.5, however, it works on 3.16.
32 bit.

Flashwalker, 2018-12-06:
Oh, And actually working space a little bit smaller on right and bottom sides, somewhere about 5mm on each one. If i believe the labels on the drawing surface. But it's not critical.

Flashwalker, 2018-12-07:
Tested on windows. It's funy, but official gaomon windows drivers doesn’t works. Ha-ha... chinese crap.

BrunoVernay, 2018-12-07:
I will try to compile and install the drivers from sources.
But I hoped that they were integrated upstream in the Linux Kernel ...

Flashwalker, 2018-12-08:
A bit of fail. Not all of pad’s buttons works. First three, from right to left, top to bottom are behaves as mouse buttons. Last two (at far bottom) doesn't response at all.

spbnick, 2018-12-08:
@Flashwalker, the first three buttons are behaving as intended and are arranged that way to support the Wacom driver. The other buttons should report higher-numbered buttons as well, which should have no effect by default. Test if they're reporting anything by running `xinput test` for your tablet's "pad" device (see the device list with `xinput list`). Press every button and check that you get the output with a button number. If that works, you can use `xsetwacom` to assign keyboard shortcuts to those buttons. See README.md and `man xsetwacom` for details.

BrunoVernay, 2018-12-10:
OK It worked once I installed the drivers from this repo.   I really thought that it was integrated upstream, wasn't it?

DeenOConnor, 2019-08-20:
[deen@parrot] : [~]
[$]> uname -a
Linux parrot 5.2.0-2parrot1-amd64 #1 SMP Parrot 5.2.7-2parrot1 (2019-08-12) x86_64 GNU/Linux

Digimend release 9, M106K kinda works.
It works as a mouse pointer, but when i try drawing something in gimp 2.10.8 nothing happens. Device IS enabled in gimp exactly the way it is described in https://digimend.github.io/support/howto/apps/gimp/
BUT it says that device is not connected.

If i try lsusb i get the following:
[deen@parrot] : [~]
[$]> lsusb
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 003: ID 045e:07f8 Microsoft Corp. Wired Keyboard 600 (model 1576)
Bus 003 Device 005: ID 256c:006e  

Where 256c:0006e is actually gaomon m106k. 

I removed digimend 9 and installed digimend 8 instead, and guess what? I can draw now!
It works in gimp, now it displays "Hid 256c:006e Pad Pad" and "Hid 256c:006e Pen Stylus" in devices list.
However, i still can't draw in sai 2. Maybe wine is the problem here.

So maybe this is the solution? For M106K users i suggest installing digimend 8. It worked for me, hope it works for you too.