DIGImend/digimend-kernel-drivers #121: HUION H640P not supported

simon-77, 2018-02-25:
Dear all,

I have got a HUION H640P Graphic Tablet and want to use it on my Laptop. Though I saw that HUION Tablets are widely supported by DIGImend, I have hoped that my tablet would also work.

My Operating System is Debian Stretch and I use KDE.
"$ uname -a" output:
Linux Debian-Simon 4.9.0-6-amd64 #1 SMP Debian 4.9.82-1+deb9u2 (2018-02-21) x86_64 GNU/Linux

I found the how-to "Collecting tablet diagnostics" and followed it and attached the output files
https://digimend.github.io/support/howto/trbl/diagnostics/

[descriptor.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/1756552/descriptor.txt)
[frame_buttons.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/1756553/frame_buttons.txt)
[frame_wheel_srolling.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/1756554/frame_wheel_srolling.txt)
[hid_report_descriptors.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/1756555/hid_report_descriptors.txt)
[pen_buttons.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/1756556/pen_buttons.txt)
[pen_coords.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/1756557/pen_coords.txt)
[pen_pressure.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/1756558/pen_pressure.txt)

Could you please help me to make my tablet work.

Kind regards
Simon

Queatz, 2018-02-25:
Related https://github.com/DIGImend/digimend-kernel-drivers/issues/104

simon-77, 2018-02-25:
Thank you for the hint, I didn't notice it before.

I've noticed that there will be support soon, because HUION has financed it and that there is a "diff" that supports it with a link to this page:
https://github.com/DIGImend/digimend-kernel-drivers/pull/114
Where can I download the modified version? I just noticed that on this side there are 6 files that are different. Do I have to download the modified files differently or is there a way to download the modified work with original files?

Kind regards
Simon

simon-77, 2018-02-25:
I did not notice it. Thank you very much! I managed to make it work.
Is there a way to close this Issue?

simon-77, 2018-02-25:
close #121

simon-77, 2018-02-25:
close

closed 2018-02-25 by simon-77