DIGImend/digimend-kernel-drivers #122: Building Debian package

danghai, 2018-03-01:
It has error when I try to use the command:  `dpkg-buildpackage -b --no-sign`

```
dpkg-buildpackage: unknown option or argument --no-sign
```

I think it should be : `dpkg-buildpackage -us -uc` ?

spbnick, 2018-03-08:
Thank you for the report, Hai. Which distribution are you running this on? What is the output of `dpkg-buildpackage --version` and `dpkg-buildpackage --help`?

danghai, 2018-03-08:
I think my distribution may be old.  My distribution: 
```
DISTRIB_ID=Ubuntu
DISTRIB_RELEASE=16.04
DISTRIB_CODENAME=xenial
DISTRIB_DESCRIPTION="Ubuntu 16.04.4 LTS"
```
`version` of `buildpackage`: 
```
Debian dpkg-buildpackage version 1.18.4.
```

command `help`: 

```
Usage: dpkg-buildpackage [<option>...]

Options:
  -F             normal full build (binaries and sources, default).
  -g             source and arch-indep build.
  -G             source and arch-specific build.
  -b             binary-only, no source files.
  -B             binary-only, only arch-specific files.
  -A             binary-only, only arch-indep files.
  -S             source-only, no binary files.
  -nc            do not clean source tree (implies -b).
  -tc            clean source tree when finished.
  -D             check build dependencies and conflicts (default).
  -d             do not check build dependencies and conflicts.
  -P<profiles>   assume given build profiles as active (comma-separated list).
  -R<rules>      rules file to execute (default is debian/rules).
  -T<target>     call debian/rules <target> with the proper environment.
      --as-root  ensure -T calls the target with root rights.
  -j[<number>]   jobs to run simultaneously (passed to <rules>), forced mode.
  -J[<number>]   jobs to run simultaneously (passed to <rules>), opt-in mode.
  -r<gain-root-command>
                 command to gain root privileges (default is fakeroot).
  --check-command=<check-command>
                 command to check the .changes file (no default).
  --check-option=<opt>
                 pass <opt> to <check-command>.
  --hook-<hook-name>=<hook-command>
                 set <hook-command> as the hook <hook-name>, known hooks:
                   init preclean source build binary changes postclean
                   check sign done
  -p<sign-command>
                 command to sign .dsc and/or .changes files
                   (default is gpg2 or gpg).
  -k<keyid>      the key to use for signing.
  -ap            add pause before starting signature process.
  -us            unsigned source package.
  -uc            unsigned .changes file.
      --force-sign
                 force signing the resulting files.
      --admindir=<directory>
                 change the administrative directory.
  -?, --help     show this help message.
      --version  show the version.

Options passed to dpkg-architecture:
  -a, --host-arch <arch>    set the host Debian architecture.
  -t, --host-type <type>    set the host GNU system type.
      --target-arch <arch>  set the target Debian architecture.
      --target-type <type>  set the target GNU system type.

Options passed to dpkg-genchanges:
  -si            source includes orig, if new upstream (default).
  -sa            source includes orig, always.
  -sd            source is diff and .dsc only.
  -v<version>    changes since version <version>.
  -m<maint>      maintainer for package is <maint>.
  -e<maint>      maintainer for release is <maint>.
  -C<descfile>   changes are described in <descfile>.
      --changes-option=<opt>
                 pass option <opt> to dpkg-genchanges.

Options passed to dpkg-source:
  -sn            force Debian native source format.
  -s[sAkurKUR]   see dpkg-source for explanation.
  -z<level>      compression level to use for source.
  -Z<compressor> compression to use for source (gz|xz|bzip2|lzma).
  -i[<regex>]    ignore diffs of files matching regex.
  -I[<pattern>]  filter out files when building tarballs.
      --source-option=<opt>
                 pass option <opt> to dpkg-source.

```

tolga-balci, 2018-03-09:
@spbnick - Nikolai, maybe there is a need to rethink how the debian package is structured. dpkg -i should install without any hiccups, but it throws out a generic error, then it needs digging to understand what is happening

```
dpkg: error processing archive digimend-dkms_6_all.deb (--install):
cannot access archive: No such file or directory
Errors were encountered while processing:
digimend-dkms_6_all.deb
```

dpkg-buildpackage --version
`Debian dpkg-buildpackage version 1.18.4.`

dpkg-buildpackage --help

```
`Usage: dpkg-buildpackage [<option>...]

Options:
  -F             normal full build (binaries and sources, default).
  -g             source and arch-indep build.
  -G             source and arch-specific build.
  -b             binary-only, no source files.
  -B             binary-only, only arch-specific files.
  -A             binary-only, only arch-indep files.
  -S             source-only, no binary files.
  -nc            do not clean source tree (implies -b).
  -tc            clean source tree when finished.
  -D             check build dependencies and conflicts (default).
  -d             do not check build dependencies and conflicts.
  -P<profiles>   assume given build profiles as active (comma-separated list).
  -R<rules>      rules file to execute (default is debian/rules).
  -T<target>     call debian/rules <target> with the proper environment.
      --as-root  ensure -T calls the target with root rights.
  -j[<number>]   jobs to run simultaneously (passed to <rules>), forced mode.
  -J[<number>]   jobs to run simultaneously (passed to <rules>), opt-in mode.
  -r<gain-root-command>
                 command to gain root privileges (default is fakeroot).
  --check-command=<check-command>
                 command to check the .changes file (no default).
  --check-option=<opt>
                 pass <opt> to <check-command>.
  --hook-<hook-name>=<hook-command>
                 set <hook-command> as the hook <hook-name>, known hooks:
                   init preclean source build binary changes postclean
                   check sign done
  -p<sign-command>
                 command to sign .dsc and/or .changes files
                   (default is gpg2 or gpg).
  -k<keyid>      the key to use for signing.
  -ap            add pause before starting signature process.
  -us            unsigned source package.
  -uc            unsigned .changes file.
      --force-sign
                 force signing the resulting files.
      --admindir=<directory>
                 change the administrative directory.
  -?, --help     show this help message.
      --version  show the version.

Options passed to dpkg-architecture:
  -a, --host-arch <arch>    set the host Debian architecture.
  -t, --host-type <type>    set the host GNU system type.
      --target-arch <arch>  set the target Debian architecture.
      --target-type <type>  set the target GNU system type.

Options passed to dpkg-genchanges:
  -si            source includes orig, if new upstream (default).
  -sa            source includes orig, always.
  -sd            source is diff and .dsc only.
  -v<version>    changes since version <version>.
  -m<maint>      maintainer for package is <maint>.
  -e<maint>      maintainer for release is <maint>.
  -C<descfile>   changes are described in <descfile>.
      --changes-option=<opt>
                 pass option <opt> to dpkg-genchanges.

Options passed to dpkg-source:
  -sn            force Debian native source format.
  -s[sAkurKUR]   see dpkg-source for explanation.
  -z<level>      compression level to use for source.
  -Z<compressor> compression to use for source (gz|xz|bzip2|lzma).
  -i[<regex>]    ignore diffs of files matching regex.
  -I[<pattern>]  filter out files when building tarballs.
      --source-option=<opt>
                 pass option <opt> to dpkg-source.
```

spbnick, 2018-03-10:
@danghai, alright. Then we need to remove `--no-sign` option from the instructions in README.md.

spbnick, 2018-03-10:
@tolga-balci, could you please post the output of you trying to install the package, and the name of the distribution you're using? Thank you.

danghai, 2018-03-10:
@spbnick , it has error when I try to build `dpkg-buildpackage -b`:

```
dpkg-deb: building package 'digimend-dkms' in '../digimend-dkms_7_all.deb'.
 dpkg-genchanges -b >../digimend-dkms_7_amd64.changes
dpkg-genchanges: binary-only upload (no source code included)
 dpkg-source --after-build digimend-kernel-drivers
dpkg-buildpackage: binary-only upload (no source included)
 signfile digimend-dkms_7_amd64.changes
gpg: skipped "Nikolai Kondrashov <spbnick@gmail.com>": No secret key
gpg: dpkg-sign.L72NjcPn/digimend-dkms_7_amd64.changes: clearsign failed: No secret key

dpkg-buildpackage: error: failed to sign .changes file

```

It works under command `dpkg-buildpackage -us -uc` .

spbnick, 2018-03-11:
Riight, I forgot about those, thank you, Hai. Only we don't need to build the source, so `dpkg-buildpackage -b -uc` should work.

spbnick, 2018-03-18:
This is fixed with a note in README.md in f517f68d.

closed 2018-03-18 by spbnick