DIGImend/digimend-kernel-drivers #124: Make install leads to ssl error (Ubuntu 16.04)

danghai, 2018-03-10:
It has error when I use command `sudo make install`:
```
make -C /lib/modules/4.13.0-36-generic/build SUBDIRS=/home/danghai/digimend-kernel-drivers modules_install
make[1]: Entering directory '/usr/src/linux-headers-4.13.0-36-generic'
  INSTALL /home/danghai/digimend-kernel-drivers/hid-kye.ko
At main.c:160:
- SSL error:02001002:system library:fopen:No such file or directory: bss_file.c:175
- SSL error:2006D080:BIO routines:BIO_new_file:no such file: bss_file.c:178
sign-file: certs/signing_key.pem: No such file or directory
  INSTALL /home/danghai/digimend-kernel-drivers/hid-polostar.ko
At main.c:160:
- SSL error:02001002:system library:fopen:No such file or directory: bss_file.c:175
- SSL error:2006D080:BIO routines:BIO_new_file:no such file: bss_file.c:178
sign-file: certs/signing_key.pem: No such file or directory
  INSTALL /home/danghai/digimend-kernel-drivers/hid-uclogic.ko
At main.c:160:
- SSL error:02001002:system library:fopen:No such file or directory: bss_file.c:175
- SSL error:2006D080:BIO routines:BIO_new_file:no such file: bss_file.c:178
sign-file: certs/signing_key.pem: No such file or directory
  INSTALL /home/danghai/digimend-kernel-drivers/hid-viewsonic.ko
At main.c:160:
- SSL error:02001002:system library:fopen:No such file or directory: bss_file.c:175
- SSL error:2006D080:BIO routines:BIO_new_file:no such file: bss_file.c:178
sign-file: certs/signing_key.pem: No such file or directory
  DEPMOD  4.13.0-36-generic
make[1]: Leaving directory '/usr/src/linux-headers-4.13.0-36-generic'
install -D -m 0644 digimend.conf /etc/depmod.d/digimend.conf
depmod -a
install hid-rebind /sbin/hid-rebind
install -m 0644 90-hid-rebind.rules /lib/udev/rules.d/90-hid-rebind.rules
udevadm control --reload
```

danghai, 2018-03-10:
I try to use `dkms`, but it does not work: 

```
sudo cp -R . /usr/src/digimend-7
sudo dkms add -m digimend -v 7
sudo dkms build -m digimend -v 7
sudo dkms install -m digimend -v 7

```

spbnick, 2018-03-11:
The SSL errors are due to Ubuntu trying to sign the module and failing. I think we can't do much about it, except note in README.md that this is not an issue.

Regarding DKMS, could you please create a separate issue and post the output, and how exactly it "doesn not work"? Thank you.

danghai, 2018-03-11:
Ah I mention `dkms` because I see they discuss about this issue. `dkms` is considered as a tool to handle this issue: 
https://github.com/umlaeute/v4l2loopback/issues/139

ericpareja, 2018-03-17:
If you build the package using **dpkg-buildpackage -uc -us**, this will build an installable .deb file without signing the source or changes files.

spbnick, 2018-03-18:
I added a note about this issue to README.md in 654ddfb.

closed 2018-03-18 by spbnick

jessicaqz, 2019-12-17:
Could you please tell me how you solved this problem finally? I also met this problem when i install .