DIGImend/digimend-kernel-drivers #131: XP-Pen Deco 01 support

apcomptec, 2018-04-07:
I checked the list of supported devices and this one is not. Is there any chances to get it work?

iommu, 2018-04-12:
On the note of XP-pen it's looking like they're putting a lot of effort recently into becoming a bigger name in the graphics tablet industry. Might be worth looking into trying to reach out to them to make a deal

dominikh, 2018-04-14:
The above link mentions that the Deco 01 has 5080 LPI according to the spec sheet, but that the tablet only reports values corresponding to 2540 LPI. It hypothesizes that a special command is required to enable 5080 LPI. I have checked on Windows, and even there, only 2540 LPI are ever used. If a special command is required, the Windows driver isn't sending it.

kodkuce, 2018-04-25:
today I just received my, out of box it doesn't work at all but I am willing to try to make myself a driver but have some newbi issues.
1.How can know in what format tablet is sending msg to host?
2.How does linux handle input from those msg , do i emulate it like a mouse or does Linux have already some ploter/table input system?
3.Am i doomed?

kodkuce, 2018-05-07:
Ok so spbnick is working on a proper fix but atm here is a temp fix for those who cant w8, i didn't test all buttons and stuff but pen works for sure, will continue testing,  drawing and report/fix any issue i get if i can ... Ty @spbnick for a lot of spoonfeeding :)

LINK
https://my.pcloud.com/publink/show?code=XZ3clb7Zybe9Qi3X5vmwKUYtlOY45yo9YIcX
cd to dir then  "make && sudo make install"

Just to know this DECO01 fix is atm in conflict whit "UCLOGIC_TABLET_PF1209" tablet so if you install this that PF1209 tablet probably won't work.. Probably irrelevant but dunno if someone draws whit 2 tablets in same time :)

Deep-Six, 2018-05-20:
I've managed to get this working using this patched driver but my question is should the tablet buttons work? As they do not and show nothing in xev.

euoeuo

spbnick, 2018-05-20:
@kodkuce, I just merged support for checking for both VID and PID in uclogic_params_probe. Do you think you can rebase your code and submit a pull request?

kodkuce, 2018-05-20:
@Deep-Six  buttons did not work for me ,on that driver i posted, i think maybe in future they will ,  @spbnick  ye ofc will try ATM am working on something but in next day or 2 will be totally free :)

brallanAguilar, 2018-05-24:
Thank you so much for your efforts! I am able to use it on Linux (commit 842427df146e28935e786147791f6da07a7bbf15). Right now the buttons does nothing but I guess I need to perform some settings.

Deep-Six, 2018-05-24:
@kodkuce will build, test and let you know

Deep-Six, 2018-05-24:
@kodkuce no joy, no events showing via xev at least. Is there a better way to test?

razvanc-r, 2018-05-24:
I'm gonna receive an xp pen deco 01 later today, will let you guys know what I did and if it worked.

kodkuce, 2018-05-24:
@Deep-Six , @razvanc-r 
Check here , i writhed down what you need to do for buttons  , just copy paste and change buttons , 
 or do manualy just take care buttons are marked  1 2 3 then 8 9 10 11 12 , there is no 4 :)
  
https://github.com/DIGImend/digimend-kernel-drivers/pull/152#issuecomment-391398547

razvanc-r, 2018-05-24:
looks like the stylus isn't detected, I'm using ubuntu 18.04. Installed digimend with dkms from master, just this morning.

dmesg
```
    1094 [36841.628881] usb 3-3.2: new full-speed USB device number 13 using xhci_hcd
    1095 [36841.730410] usb 3-3.2: New USB device found, idVendor=28bd, idProduct=0042
    1096 [36841.730415] usb 3-3.2: New USB device strings: Mfr=5, Product=6, SerialNumber=0
    1097 [36841.730418] usb 3-3.2: Product: DECO 01
*   1098 [36841.730421] usb 3-3.2: Manufacturer: UGEE
*   1099 [36841.732944] input: UGEE DECO 01 as /devices/pci0000:00/0000:00:14.0/usb3/3-3/3-3.2/3-3.2:1.0/0003:28BD:0042.0012/input/input42
*   1100 [36841.793151] hid-generic 0003:28BD:0042.0012: input,hidraw4: USB HID v1.00 Mouse [UGEE DECO 01] on usb-0000:00:14.0-3.2/input0
*   1101 [36841.794861] hid-generic 0003:28BD:0042.0013: hiddev3,hidraw5: USB HID v1.00 Device [UGEE DECO 01] on usb-0000:00:14.0-3.2/input1
*   1102 [36841.796023] hid-generic 0003:28BD:0042.0014: hiddev4,hidraw6: USB HID v1.00 Device [UGEE DECO 01] on usb-0000:00:14.0-3.2/input2
```

```
 $ xinput list | grep UGEE
(standard input):5:⎜   ↳ UGEE DECO 01 touch                             id=15   [slave  pointer  (2)]
(standard input):6:⎜   ↳ UGEE DECO 01 pad                               id=21   [slave  pointer  (2)]
```

xorg
```
    1019 [ 37238.399] (II) config/udev: Adding input device UGEE DECO 01 (/dev/input/mouse0)
    1020 [ 37238.399] (II) No input driver specified, ignoring this device.
    1021 [ 37238.399] (II) This device may have been added with another device file.
    1022 [ 37238.475] (II) config/udev: Adding input device UGEE DECO 01 (/dev/input/event7)
    1023 [ 37238.475] (**) UGEE DECO 01: Applying InputClass "libinput pointer catchall"
    1024 [ 37238.475] (**) UGEE DECO 01: Applying InputClass "Tablet"
    1025 [ 37238.475] (II) Using input driver 'wacom' for 'UGEE DECO 01'
    1026 [ 37238.564] (II) systemd-logind: got fd for /dev/input/event7 13:71 fd 28 paused 1
    1027 [ 37238.564] (II) systemd-logind: releasing fd for 13:71
```

```
 $ modinfo hid-uclogic
filename:       /lib/modules/4.15.0-22-generic/updates/dkms/hid-uclogic.ko
version:        8
license:        GPL
author:         Nikolai Kondrashov
author:         Martin Rusko
srcversion:     03AB8F7DBACB5EAEAFD3258
alias:          hid:b0003g*v000028BDp00000042
alias:          hid:b0003g*v000028BDp00000075
alias:          hid:b0003g*v000028BDp00000071
alias:          hid:b0003g*v000028BDp00000074
alias:          hid:b0003g*v00002179p00000053
alias:          hid:b0003g*v00005543p00003031
alias:          hid:b0003g*v00005543p00000047
alias:          hid:b0003g*v00005543p00000045
alias:          hid:b0003g*v00005543p00000081
alias:          hid:b0003g*v00005543p0000004D
alias:          hid:b0003g*v00005543p0000006E
alias:          hid:b0003g*v0000256Cp0000006E
alias:          hid:b0003g*v00005543p00000781
alias:          hid:b0003g*v00005543p00000522
alias:          hid:b0003g*v00005543p00000064
alias:          hid:b0003g*v00005543p00000005
alias:          hid:b0003g*v00005543p00000004
alias:          hid:b0003g*v00005543p00000003
alias:          hid:b0003g*v00005543p00000042
depends:        hid
retpoline:      Y
name:           hid_uclogic
vermagic:       4.15.0-22-generic SMP mod_unload
```

[descriptors.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/2036330/descriptors.txt)
[hid_report_descriptors.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/2036333/hid_report_descriptors.txt)
[pen_coords.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/2036343/pen_coords.txt)
[pen_pressure.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/2036345/pen_pressure.txt)
[pen_buttons.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/2036349/pen_buttons.txt)
[frame_buttons.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/2036353/frame_buttons.txt)

does look like `pen_*.txt` has output so the stylus produces input that is reported by this `usbhid-dump` but the pointer on the screen doesn't move and the nothing really works. The tablet flashes the LED when the stylus is tapped but I guess that's a hardware thing. So anything else I can do let me know. Thanks!

spbnick, 2018-05-24:
@razvanc-r, could you please try running "sudo modprobe -r hid-uclogic" and then reconnect the tablet?

kodkuce, 2018-05-24:
for me it all shows normal 
xinput list | grep UGEE
⎜   ↳ UGEE DECO 01 Pad pad                    	id=12	[slave  pointer  (2)]
⎜   ↳ UGEE DECO 01 Pen stylus                 	id=13	[slave  pointer  (2)]
⎜   ↳ UGEE DECO 01 Mouse touch                	id=14	[slave  pointer  (2)]

am atm working on something can come later try help

razvanc-r, 2018-05-24:
@spbnick I did but nothing (visible) changed, xsetwacom returns still
```
 $ xsetwacom --list
UGEE DECO 01 touch                      id: 19  type: TOUCH     
UGEE DECO 01 pad                        id: 20  type: PAD 
```

razvanc-r, 2018-05-24:
@spbnick 
```
(standard input):5712:/lib/modules/4.15.0-22-generic/updates/dkms/hid-uclogic.ko needs "hid_unregister_driver": /lib/modules/4.15.0-22-generic/kernel/drivers/hid/hid.ko
(standard input):10883:updates/dkms/hid-uclogic.ko: kernel/drivers/hid/hid.ko
(standard input):36280:alias hid:b0003g*v000028BDp00000042 hid_uclogic
(standard input):36281:alias hid:b0003g*v000028BDp00000075 hid_uclogic
(standard input):36282:alias hid:b0003g*v000028BDp00000071 hid_uclogic
(standard input):36283:alias hid:b0003g*v000028BDp00000074 hid_uclogic
(standard input):36284:alias hid:b0003g*v00002179p00000053 hid_uclogic
(standard input):36285:alias hid:b0003g*v00005543p00003031 hid_uclogic
(standard input):36286:alias hid:b0003g*v00005543p00000047 hid_uclogic
(standard input):36287:alias hid:b0003g*v00005543p00000045 hid_uclogic
(standard input):36288:alias hid:b0003g*v00005543p00000081 hid_uclogic
(standard input):36289:alias hid:b0003g*v00005543p0000004D hid_uclogic
(standard input):36290:alias hid:b0003g*v00005543p0000006E hid_uclogic
(standard input):36291:alias hid:b0003g*v0000256Cp0000006E hid_uclogic
(standard input):36292:alias hid:b0003g*v00005543p00000781 hid_uclogic
(standard input):36293:alias hid:b0003g*v00005543p00000522 hid_uclogic
(standard input):36294:alias hid:b0003g*v00005543p00000064 hid_uclogic
(standard input):36295:alias hid:b0003g*v00005543p00000005 hid_uclogic
(standard input):36296:alias hid:b0003g*v00005543p00000004 hid_uclogic
(standard input):36297:alias hid:b0003g*v00005543p00000003 hid_uclogic
(standard input):36298:alias hid:b0003g*v00005543p00000042 hid_uclogic
```

razvanc-r, 2018-05-24:
@spbnick nothing changed with the "brute force method"
dmesg
```
    1217 [ 2523.376240] usb 3-3.2: new full-speed USB device number 22 using xhci_hcd
    1218 [ 2523.478037] usb 3-3.2: New USB device found, idVendor=28bd, idProduct=0042
    1219 [ 2523.478043] usb 3-3.2: New USB device strings: Mfr=5, Product=6, SerialNumber=0
    1220 [ 2523.478047] usb 3-3.2: Product: DECO 01
    1221 [ 2523.478050] usb 3-3.2: Manufacturer: UGEE
    1222 [ 2523.481107] input: UGEE DECO 01 as /devices/pci0000:00/0000:00:14.0/usb3/3-3/3-3.2/3-3.2:1.0/0003:28BD:0042.002C/input/input38
    1223 [ 2523.481499] hid-generic 0003:28BD:0042.002C: input,hidraw0: USB HID v1.00 Mouse [UGEE DECO 01] on usb-0000:00:14.0-3.2/input0
    1224 [ 2523.483612] hid-generic 0003:28BD:0042.002D: hiddev0,hidraw2: USB HID v1.00 Device [UGEE DECO 01] on usb-0000:00:14.0-3.2/input1
    1225 [ 2523.484964] hid-generic 0003:28BD:0042.002E: hiddev2,hidraw3: USB HID v1.00 Device [UGEE DECO 01] on usb-0000:00:14.0-3.2/input2
```

razvanc-r, 2018-05-24:
I completely forgot to mention that in the above I used `dkms` (following the instructions from travis.yml file) to build the driver so I don't have to worry about kernel upgrades, but just before you finished the session with me @spbnick on IRC I noticed that `dkms` doesn't take care of this `hid-rebind` script in a last effort I installed with `make`and with `make` the tablet freaking works! :)), but I don't think the above was wasted... I think it could have been shorter, but at least you know that with `dkms` there's this trouble :P...

Now, you guys know how to make a script that runs `xsetwacom` every time you plug the tablet in? :)...

kodkuce, 2018-05-25:
@razvanc-r   hmm you can make a script that runs in background and checks every 10 second if tablet is connected but that kinda sux ,  option 2 is  adding new  rule to udev

sudo nano /etc/udev/rules.d/10-local.rules
ATTRS{name}=="UGEE DECO 01 Mouse" , RUN+="/path/to/your/script.sh" 
but it will run script 6 times  , not really professional :(

razvanc-r, 2018-05-25:
Ya, I already tried udev and sometimes it works sometimes it doesn't, you also have to export DISPLAY and XAUTHORITY and I don't like the solution. You also have to delay the script with sleep.

I was looking into configuring through xorg.conf, the 50-tablet.conf file, but I can't find the relevat options explained well enough. I made a new inputclass section with matchdriver Wacom & matchproduct pad, but it doesn't seem to be picked up. Also with xsetwacom you can specify if a key should be repeated instead of pressed only once, I couldn't find anything explaining if this is possible with xorg.conf. Anyway, so far best solution seems to be to run a script manually after plugging the tablet

spbnick, 2018-05-26:
Thanks, @razvanc-r! Right, we need to fix that. Filed #154.

closed 2018-05-26 by spbnick

aveemashfaq, 2018-06-26:
Hey. I contacted xp-pen support. They gave me a linux driver. The pen is working. Not sure how to assign keys. I am not a developer. Just sharing the resource to help others
[Linux_PentabletDriver1.0.tar.gz](https://github.com/DIGImend/digimend-kernel-drivers/files/2137595/Linux_PentabletDriver1.0.tar.gz)

tydlwav, 2018-07-11:
I found this link on the XP Pen official website: [Linux_PentabletDriver1.0.tar.gz](https://www.xp-pen.com/upload/download/20180706/Linux_PentabletDriver1.0.tar.gz)
It is supposed to support "Star 03, 04, 05, 06, Deco01, 02, 03, G430S, G540, G640 for Linux beta driver", as quoted from the official website. I have not tested it yet, but I hope we would be able to mark tablets like the deco 03 and star 06 as supported (which we aren't now).

ShayBox, 2020-09-12:
I found that following [this](https://github.com/DIGImend/digimend-kernel-drivers#enabling-wacom-xorg-driver) will work perfectly (even without the xp-pen linux program or digimend), the it works perfectly with xf86-input-wacom and all wacom frontend programs