DIGImend/digimend-kernel-drivers #136: Install Huion

serikame, 2018-04-25:
sudo rmod hid-kye
rm mod: ERROR: Module hid_kye is not currently loaded

spbnick, 2018-04-25:
Those are not a problem. These commands are just trying to make sure the old modules are unloaded, and if they weren't loaded in the first place, that's fine.

closed 2018-04-25 by spbnick

serikame, 2018-04-25:
Well, it does not recognize the tablet, nor does it move the cursor.

spbnick, 2018-04-25:
@serikame You gotta give us more detail, like what tablet you use, what's the output of "sudo dmesg" is after plugging the tablet in, what's the output of "sudo evtest", what's the output of "xinput list", etc. Otherwise we can't help you.

serikame, 2018-04-25:
Huion New 1060 Plus (8192)

serikame, 2018-04-25:
[    0.014906] Spectre V2 mitigation: Mitigation: Full AMD retpoline
[    0.014907] Spectre V2 mitigation: Speculation control IBPB not-supported IBRS not-supported
[    0.015256] Freeing SMP alternatives memory: 32K
[    0.017730] smpboot: Max logical packages: 2
[    0.021003] Switched APIC routing to physical flat.
[    0.026514] ..TIMER: vector=0x30 apic1=0 pin1=2 apic2=-1 pin2=-1
[    0.037000] smpboot: CPU0: AMD Ryzen 7 1700 Eight-Core Processor (family: 0x17, model: 0x1, stepping: 0x1)
[    0.040008] Performance Events: Fam17h core perfctr, AMD PMU driver.
[    0.040013] ... version:                0
[    0.040013] ... bit width:              48
[    0.040014] ... generic registers:      6
[    0.040014] ... value mask:             0000ffffffffffff
[    0.040015] ... max period:             00007fffffffffff
[    0.040015] ... fixed-purpose events:   0
[    0.040016] ... event mask:             000000000000003f
[    0.042005] Hierarchical SRCU implementation.
[    0.046103] NMI watchdog: enabled on all CPUs, permanently consumes one hw-PMU counter.
[    0.048006] smp: Bringing up secondary CPUs ...
[    0.056015] x86: Booting SMP configuration:
[    0.056016] .... node  #0, CPUs:        #1  #2  #3  #4  #5  #6  #7  #8  #9 #10 #11 #12 #13 #14 #15
[    0.183026] smp: Brought up 1 node, 16 CPUs
[    0.183026] smpboot: Total of 16 processors activated (95822.27 BogoMIPS)
[    0.185841] devtmpfs: initialized
[    0.185841] x86/mm: Memory block size: 128MB
[    0.187734] evm: security.selinux
[    0.187735] evm: security.SMACK64
[    0.187735] evm: security.SMACK64EXEC
[    0.187736] evm: security.SMACK64TRANSMUTE
[    0.187736] evm: security.SMACK64MMAP
[    0.187737] evm: security.ima
[    0.187737] evm: security.capability
[    0.187753] PM: Registering ACPI NVS region [mem 0xdd57f000-0xdd956fff] (4030464 bytes)
[    0.187753] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 1911260446275000 ns
[    0.187753] futex hash table entries: 4096 (order: 6, 262144 bytes)
[    0.188030] pinctrl core: initialized pinctrl subsystem
[    0.188102] RTC time:  1:32:41, date: 01/01/16
[    0.189288] NET: Registered protocol family 16
[    0.189426] cpuidle: using governor ladder
[    0.189426] cpuidle: using governor menu
[    0.189426] PCCT header not found.
[    0.189426] ACPI: bus type PCI registered
[    0.189426] acpiphp: ACPI Hot Plug PCI Controller Driver version: 0.5
[    0.189426] PCI: MMCONFIG for domain 0000 [bus 00-3f] at [mem 0xf8000000-0xfbffffff] (base 0xf8000000)
[    0.189426] PCI: MMCONFIG at [mem 0xf8000000-0xfbffffff] reserved in E820
[    0.189426] pmd_set_huge: Cannot satisfy [mem 0xf8000000-0xf8200000] with a huge-page mapping due to MTRR override.
[    0.189426] PCI: Using configuration type 1 for base access
[    0.191047] HugeTLB registered 1.00 GiB page size, pre-allocated 0 pages
[    0.191047] HugeTLB registered 2.00 MiB page size, pre-allocated 0 pages
[    0.191069] ACPI: Added _OSI(Module Device)
[    0.191070] ACPI: Added _OSI(Processor Device)
[    0.191070] ACPI: Added _OSI(3.0 _SCP Extensions)
[    0.191071] ACPI: Added _OSI(Processor Aggregator Device)
[    0.191127] ACPI: [Firmware Bug]: BIOS _OSI(Linux) query ignored
[    0.191127] ACPI: Executed 2 blocks of module-level executable AML code
[    0.197933] ACPI: Interpreter enabled
[    0.197946] ACPI: (supports S0 S3 S4 S5)
[    0.197946] ACPI: Using IOAPIC for interrupt routing
[    0.198035] PCI: Using host bridge windows from ACPI; if necessary, use "pci=nocrs" and report a bug
[    0.198243] ACPI: Enabled 2 GPEs in block 00 to 1F
[    0.204034] ACPI: PCI Root Bridge [PCI0] (domain 0000 [bus 00-ff])
[    0.204038] acpi PNP0A08:00: _OSC: OS supports [ExtendedConfig ASPM ClockPM Segments MSI]
[    0.204193] acpi PNP0A08:00: _OSC: OS now controls [PCIeHotplug PME AER PCIeCapability]
[    0.204202] acpi PNP0A08:00: [Firmware Info]: MMCONFIG for domain 0000 [bus 00-3f] only partially covers this bridge
[    0.204447] PCI host bridge to bus 0000:00
[    0.204449] pci_bus 0000:00: root bus resource [io  0x0000-0x03af window]
[    0.204450] pci_bus 0000:00: root bus resource [io  0x03e0-0x0cf7 window]
[    0.204451] pci_bus 0000:00: root bus resource [io  0x03b0-0x03df window]
[    0.204452] pci_bus 0000:00: root bus resource [io  0x0d00-0xffff window]
[    0.204453] pci_bus 0000:00: root bus resource [mem 0x000a0000-0x000bffff window]
[    0.204454] pci_bus 0000:00: root bus resource [mem 0x000c0000-0x000dffff window]
[... 873 lines cut ...]
[  967.338158] usb 1-4.4: new high-speed USB device number 6 using xhci_hcd
[  967.459161] usb 1-4.4: New USB device found, idVendor=058f, idProduct=6366
[  967.459165] usb 1-4.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  967.459167] usb 1-4.4: Product: Mass Storage Device
[  967.459169] usb 1-4.4: Manufacturer: Generic
[  967.459171] usb 1-4.4: SerialNumber: 058F0O1111B1
[  967.480936] usb-storage 1-4.4:1.0: USB Mass Storage device detected
[  967.481184] scsi host9: usb-storage 1-4.4:1.0
[  967.481352] usbcore: registered new interface driver usb-storage
[  967.484197] usbcore: registered new interface driver uas
[  967.491147] input: HID 256c:006e Pen as /devices/pci0000:00/0000:00:01.3/0000:03:00.0/usb1/1-4/1-4.1/1-4.1:1.0/0003:256C:006E.0005/input/input20
[  967.543344] input: HID 256c:006e Pad as /devices/pci0000:00/0000:00:01.3/0000:03:00.0/usb1/1-4/1-4.1/1-4.1:1.0/0003:256C:006E.0005/input/input21
[... cut ...]
[  969.239732] sd 9:0:0:0: [sde] 15493120 512-byte logical blocks: (7.93 GB/7.39 GiB)
[  969.241263] sd 9:0:0:0: [sde] Write Protect is off
[  969.241266] sd 9:0:0:0: [sde] Mode Sense: 03 00 00 00
[  969.243264] sd 9:0:0:0: [sde] No Caching mode page found
[  969.243270] sd 9:0:0:0: [sde] Assuming drive cache: write through
[  969.249299]  sde: sde1
[  969.254299] sd 9:0:0:0: [sde] Attached SCSI removable disk
[  974.942803] FAT-fs (sde1): Volume was not properly unmounted. Some data may be corrupt. Please run fsck.

serikame, 2018-04-25:
sudo evtest, it does not work

serikame, 2018-04-25:
sudo xinput list
⎡ Virtual core pointer                    	id=2	[master pointer  (3)]
⎜   ↳ Virtual core XTEST pointer              	id=4	[slave  pointer  (2)]
⎜   ↳ USB USB Keyboard                        	id=9	[slave  pointer  (2)]
⎜   ↳ MOSART Semi. 2.4G Keyboard Mouse        	id=12	[slave  pointer  (2)]
⎣ Virtual core keyboard                   	id=3	[master keyboard (2)]
    ↳ Virtual core XTEST keyboard             	id=5	[slave  keyboard (3)]
    ↳ Power Button                            	id=6	[slave  keyboard (3)]
    ↳ Power Button                            	id=7	[slave  keyboard (3)]
    ↳ USB USB Keyboard                        	id=8	[slave  keyboard (3)]
    ↳ UVC Camera (046d:0825)                  	id=10	[slave  keyboard (3)]
    ↳ MOSART Semi. 2.4G Keyboard Mouse        	id=11	[slave  keyboard (3)]
    ↳ USB USB Keyboard                        	id=13	[slave  keyboard (3)]
    ↳ MOSART Semi. 2.4G Keyboard Mouse        	id=14	[slave  keyboard (3)]

serikame, 2018-04-25:
Sorry, I installed vtest

serikame, 2018-04-25:
sudo xinput list
⎡ Virtual core pointer                    	id=2	[master pointer  (3)]
⎜   ↳ Virtual core XTEST pointer              	id=4	[slave  pointer  (2)]
⎜   ↳ USB USB Keyboard                        	id=9	[slave  pointer  (2)]
⎜   ↳ MOSART Semi. 2.4G Keyboard Mouse        	id=12	[slave  pointer  (2)]
⎜   ↳ HID 256c:006e Pad                       	id=15	[slave  pointer  (2)]
⎣ Virtual core keyboard                   	id=3	[master keyboard (2)]
    ↳ Virtual core XTEST keyboard             	id=5	[slave  keyboard (3)]
    ↳ Power Button                            	id=6	[slave  keyboard (3)]
    ↳ Power Button                            	id=7	[slave  keyboard (3)]
    ↳ USB USB Keyboard                        	id=8	[slave  keyboard (3)]
    ↳ UVC Camera (046d:0825)                  	id=10	[slave  keyboard (3)]
    ↳ MOSART Semi. 2.4G Keyboard Mouse        	id=11	[slave  keyboard (3)]
    ↳ USB USB Keyboard                        	id=13	[slave  keyboard (3)]
    ↳ MOSART Semi. 2.4G Keyboard Mouse        	id=14	[slave  keyboard (3)]
    ↳ HID 256c:006e Pen                       	id=16	[slave  keyboard (3)]

serikame, 2018-04-25:
sudo evtest
[sudo] password for serikame: 
No device specified, trying to scan all of /dev/input/event*
Available devices:
/dev/input/event0:	Power Button
/dev/input/event1:	Power Button
/dev/input/event2:	MOSART Semi. 2.4G Keyboard Mouse
/dev/input/event3:	MOSART Semi. 2.4G Keyboard Mouse
/dev/input/event4:	USB USB Keyboard
/dev/input/event5:	USB USB Keyboard
/dev/input/event6:	HDA ATI HDMI HDMI/DP,pcm=3
/dev/input/event7:	HDA ATI HDMI HDMI/DP,pcm=7
/dev/input/event8:	HDA ATI HDMI HDMI/DP,pcm=8
/dev/input/event9:	HDA ATI HDMI HDMI/DP,pcm=9
/dev/input/event10:	HDA ATI HDMI HDMI/DP,pcm=10
/dev/input/event11:	HDA ATI HDMI HDMI/DP,pcm=11
/dev/input/event12:	HD-Audio Generic Front Mic
/dev/input/event13:	HD-Audio Generic Rear Mic
/dev/input/event14:	HD-Audio Generic Line
/dev/input/event15:	HD-Audio Generic Line Out Front
/dev/input/event16:	HD-Audio Generic Line Out Surround
/dev/input/event17:	HD-Audio Generic Line Out CLFE
/dev/input/event18:	HD-Audio Generic Front Headphone
/dev/input/event19:	UVC Camera (046d:0825)
/dev/input/event20:	HID 256c:006e Pen
/dev/input/event21:	HID 256c:006e Pad
Select the device event number [0-21]: 20
Input driver version is 1.0.1
Input device ID: bus 0x3 vendor 0x256c product 0x6e version 0x111
Input device name: "HID 256c:006e Pen"
Supported events:
  Event type 0 (EV_SYN)
  Event type 1 (EV_KEY)
    Event code 320 (BTN_TOOL_PEN)
    Event code 330 (BTN_TOUCH)
    Event code 331 (BTN_STYLUS)
    Event code 332 (BTN_STYLUS2)
  Event type 3 (EV_ABS)
    Event code 0 (ABS_X)
      Value      0
      Min        0
      Max    40000
      Resolution     200
    Event code 1 (ABS_Y)
      Value      0
      Min        0
      Max    25000
      Resolution     200
    Event code 24 (ABS_PRESSURE)
      Value      0
      Min        0
      Max     8191
  Event type 4 (EV_MSC)
    Event code 4 (MSC_SCAN)
Properties:
Testing ... (interrupt to exit)

serikame, 2018-04-25:
sudo evtest
[sudo] password for serikame: 
No device specified, trying to scan all of /dev/input/event*
Available devices:
/dev/input/event0:	Power Button
/dev/input/event1:	Power Button
/dev/input/event2:	MOSART Semi. 2.4G Keyboard Mouse
/dev/input/event3:	MOSART Semi. 2.4G Keyboard Mouse
/dev/input/event4:	USB USB Keyboard
/dev/input/event5:	USB USB Keyboard
/dev/input/event6:	HDA ATI HDMI HDMI/DP,pcm=3
/dev/input/event7:	HDA ATI HDMI HDMI/DP,pcm=7
/dev/input/event8:	HDA ATI HDMI HDMI/DP,pcm=8
/dev/input/event9:	HDA ATI HDMI HDMI/DP,pcm=9
/dev/input/event10:	HDA ATI HDMI HDMI/DP,pcm=10
/dev/input/event11:	HDA ATI HDMI HDMI/DP,pcm=11
/dev/input/event12:	HD-Audio Generic Front Mic
/dev/input/event13:	HD-Audio Generic Rear Mic
/dev/input/event14:	HD-Audio Generic Line
/dev/input/event15:	HD-Audio Generic Line Out Front
/dev/input/event16:	HD-Audio Generic Line Out Surround
/dev/input/event17:	HD-Audio Generic Line Out CLFE
/dev/input/event18:	HD-Audio Generic Front Headphone
/dev/input/event19:	UVC Camera (046d:0825)
/dev/input/event20:	HID 256c:006e Pen
/dev/input/event21:	HID 256c:006e Pad
Select the device event number [0-21]: 21
Input driver version is 1.0.1
Input device ID: bus 0x3 vendor 0x256c product 0x6e version 0x111
Input device name: "HID 256c:006e Pad"
Supported events:
  Event type 0 (EV_SYN)
  Event type 1 (EV_KEY)
    Event code 256 (BTN_0)
    Event code 257 (BTN_1)
    Event code 258 (BTN_2)
    Event code 259 (BTN_3)
    Event code 260 (BTN_4)
    Event code 261 (BTN_5)
    Event code 262 (BTN_6)
    Event code 263 (BTN_7)
  Event type 4 (EV_MSC)
    Event code 4 (MSC_SCAN)
Properties:
Testing ... (interrupt to exit)

spbnick, 2018-04-25:
Thank you. This doesn't look like the driver has installed. To confirm, please post the output of `sudo modinfo hid-uclogic`.

serikame, 2018-04-25:
filename:       /lib/modules/4.13.0-39-lowlatency/kernel/drivers/hid/hid-uclogic.ko
license:        GPL
author:         Nikolai Kondrashov
author:         Martin Rusko
srcversion:     1BEDCC552EDCC95D3DE2132
alias:          hid:b0003g*v000028BDp00000071
alias:          hid:b0003g*v00002179p00000053
alias:          hid:b0003g*v00005543p00003031
alias:          hid:b0003g*v00005543p00000045
alias:          hid:b0003g*v00005543p00000081
alias:          hid:b0003g*v00005543p0000004D
alias:          hid:b0003g*v00005543p0000006E
alias:          hid:b0003g*v0000256Cp0000006E
alias:          hid:b0003g*v00005543p00000781
alias:          hid:b0003g*v00005543p00000522
alias:          hid:b0003g*v00005543p00000064
alias:          hid:b0003g*v00005543p00000005
alias:          hid:b0003g*v00005543p00000004
alias:          hid:b0003g*v00005543p00000003
alias:          hid:b0003g*v00005543p00000042
depends:        hid
intree:         Y
name:           hid_uclogic
vermagic:       4.13.0-39-lowlatency SMP preempt mod_unload 
signat:         PKCS#7
signer:         
sig_key:        
sig_hashalgo:   md4

spbnick, 2018-04-25:
Right, this shows that only stock driver is available. Please try to install the driver again, but this time please post everything that you had on the screen while doing that - commands you entered, their output, etc.

serikame, 2018-04-25:
serikame@ZYX:~/Programas/Huion/digimend-kernel-drivers-hires_refactor$ 
sudo rmmod hid-kye
rmmod: ERROR: Module hid_kye is not currently loaded
serikame@ZYX:~/Programas/Huion/digimend-kernel-drivers-hires_refactor$ 
sudo rmmod hid-uclogic
serikame@ZYX:~/Programas/Huion/digimend-kernel-drivers-hires_refactor$ 
sudo rmmod hid-huion
rmmod: ERROR: Module hid_huion is not currently loaded
serikame@ZYX:~/Programas/Huion/digimend-kernel-drivers-hires_refactor$

spbnick, 2018-04-25:
Is that all you do to install the driver?

serikame, 2018-04-25:
Yes, I followed the instructions in the READme.md file and that is all it says to install. I'm not very good at installing things on Linux and I do not understand much English either. I am sorry. Thank you for being so understanding and helping me with this.

spbnick, 2018-04-25:
Please follow the "[Installing source package manually](https://github.com/DIGImend/digimend-kernel-drivers#installing-source-package-manually)" section.

spbnick, 2018-04-25:
You need to run `make` and then `sudo make install` in the driver source directory.

serikame, 2018-04-25:
serikame@ZYX:~/Programas/Huion/digimend-kernel-drivers-hires_refactor$ sudo make install
[sudo] password for serikame: 
make -C /lib/modules/4.13.0-39-lowlatency/build SUBDIRS=/home/serikame/Programas/Huion/digimend-kernel-drivers-hires_refactor modules_install
make[1]: se entra en el directorio '/usr/src/linux-headers-4.13.0-39-lowlatency'
  DEPMOD  4.13.0-39-lowlatency
make[1]: se sale del directorio '/usr/src/linux-headers-4.13.0-39-lowlatency'
install -D -m 0644 digimend.conf /etc/depmod.d/digimend.conf
depmod -a
install hid-rebind /sbin/hid-rebind
install -m 0644 90-hid-rebind.rules /lib/udev/rules.d/90-hid-rebind.rules
udevadm control --reload

spbnick, 2018-04-25:
This looks good. Run `sudo rmmod hid-uclogic`, reconnect the tablet and see if it works.

serikame, 2018-04-25:
I already did everything, but I can not get the digitizer tablet to move the pointer in any program

RJQuiralta, 2018-04-25:
Not sure if this applies here, there may be a need to rebuild the kernel modules with the mkinitcpio command after installing the driver, and then a full reboot may bring the tablet to recognition? (although re-plugging usually makes the job) Again, not sure if this step is required on all distros but that used to work for me with the older drivers.

spbnick, 2018-04-26:
@serikame, please post the output of `sudo modinfo hid-uclogic`, `sudo evtest`, and `xinput list` again.

serikame, 2018-04-26:
Hello Nikolai, I have committed the imprudence of installing Ubuntu Studio 18.04 on my computer, thinking that my configuration problems might be solved with the Huion digitizer card. The result of my management is that now I tango new problems, one of my monitors is unconfigured and I have to reinstall the drives that you sent me, so I will be delayed with what you ask me to monitor. I hope to have everything ready tomorrow afternoon. Thank you. A greeting. Omar

serikame, 2018-04-26:
serikame@ZYX:~/digimend-kernel-drivers-hires_refactor$ sudo modinfo hid-uclogic
filename:       /lib/modules/4.15.0-20-lowlatency/kernel/drivers/hid/hid-uclogic.ko
license:        GPL
author:         Nikolai Kondrashov
author:         Martin Rusko
srcversion:     BEA927C95852EC70343F18B
alias:          hid:b0003g*v000028BDp00000071
alias:          hid:b0003g*v00002179p00000053
alias:          hid:b0003g*v00005543p00003031
alias:          hid:b0003g*v00005543p00000045
alias:          hid:b0003g*v00005543p00000081
alias:          hid:b0003g*v00005543p0000004D
alias:          hid:b0003g*v00005543p0000006E
alias:          hid:b0003g*v0000256Cp0000006E
alias:          hid:b0003g*v00005543p00000781
alias:          hid:b0003g*v00005543p00000522
alias:          hid:b0003g*v00005543p00000064
alias:          hid:b0003g*v00005543p00000005
alias:          hid:b0003g*v00005543p00000004
alias:          hid:b0003g*v00005543p00000003
alias:          hid:b0003g*v00005543p00000042
depends:        hid
retpoline:      Y
intree:         Y
name:           hid_uclogic
vermagic:       4.15.0-20-lowlatency SMP preempt mod_unload 
signat:         PKCS#7
signer:         
sig_key:        
sig_hashalgo:   md4

serikame, 2018-04-26:
serikame@ZYX:~/digimend-kernel-drivers-hires_refactor$ sudo evtest
[sudo] contraseña para serikame: 
No device specified, trying to scan all of /dev/input/event*
Available devices:
/dev/input/event0:	Power Button
/dev/input/event1:	Power Button
/dev/input/event2:	MOSART Semi. 2.4G Keyboard Mouse
/dev/input/event3:	MOSART Semi. 2.4G Keyboard Mouse
/dev/input/event4:	USB USB Keyboard
/dev/input/event5:	USB USB Keyboard
/dev/input/event6:	HID 256c:006e Pen
/dev/input/event7:	HID 256c:006e Pad
/dev/input/event8:	HDA ATI HDMI HDMI/DP,pcm=3
/dev/input/event9:	HDA ATI HDMI HDMI/DP,pcm=7
/dev/input/event10:	HDA ATI HDMI HDMI/DP,pcm=8
/dev/input/event11:	HDA ATI HDMI HDMI/DP,pcm=9
/dev/input/event12:	HDA ATI HDMI HDMI/DP,pcm=10
/dev/input/event13:	HDA ATI HDMI HDMI/DP,pcm=11
/dev/input/event14:	HD-Audio Generic Front Mic
/dev/input/event15:	HD-Audio Generic Rear Mic
/dev/input/event16:	HD-Audio Generic Line
/dev/input/event17:	HD-Audio Generic Line Out Front
/dev/input/event18:	HD-Audio Generic Line Out Surround
/dev/input/event19:	HD-Audio Generic Line Out CLFE
/dev/input/event20:	HD-Audio Generic Front Headphone
/dev/input/event21:	UVC Camera (046d:0825)
Select the device event number [0-21]:

serikame, 2018-04-26:
serikame@ZYX:~/digimend-kernel-drivers-hires_refactor$ xinput list
⎡ Virtual core pointer                    	id=2	[master pointer  (3)]
⎜   ↳ Virtual core XTEST pointer              	id=4	[slave  pointer  (2)]
⎜   ↳ USB USB Keyboard                        	id=9	[slave  pointer  (2)]
⎜   ↳ MOSART Semi. 2.4G Keyboard Mouse        	id=14	[slave  pointer  (2)]
⎜   ↳ HID 256c:006e Pad                       	id=10	[slave  pointer  (2)]
⎣ Virtual core keyboard                   	id=3	[master keyboard (2)]
    ↳ Virtual core XTEST keyboard             	id=5	[slave  keyboard (3)]
    ↳ Power Button                            	id=6	[slave  keyboard (3)]
    ↳ Power Button                            	id=7	[slave  keyboard (3)]
    ↳ USB USB Keyboard                        	id=8	[slave  keyboard (3)]
    ↳ MOSART Semi. 2.4G Keyboard Mouse        	id=13	[slave  keyboard (3)]
    ↳ USB USB Keyboard                        	id=15	[slave  keyboard (3)]
    ↳ MOSART Semi. 2.4G Keyboard Mouse        	id=16	[slave  keyboard (3)]
    ↳ UVC Camera (046d:0825)                  	id=12	[slave  keyboard (3)]
    ↳ HID 256c:006e Pen                       	id=11	[slave  keyboard (3)]

spbnick, 2018-04-27:
I see that the new driver is still not being found. Could you please install it again on this new system and post your terminal contents, while doing it, again? Thank you.

serikame, 2018-04-27:
serikame@ZYX:~/digimend-kernel-drivers-hires_refactor$ sudo make install
[sudo] contraseña para serikame: 
make -C /lib/modules/4.15.0-20-lowlatency/build SUBDIRS=/home/serikame/digimend-kernel-drivers-hires_refactor modules_install
make[1]: se entra en el directorio '/usr/src/linux-headers-4.15.0-20-lowlatency'
Makefile:976: "Cannot use CONFIG_STACK_VALIDATION=y, please install libelf-dev, libelf-devel or elfutils-libelf-devel"
  DEPMOD  4.15.0-20-lowlatency
make[1]: se sale del directorio '/usr/src/linux-headers-4.15.0-20-lowlatency'
install -D -m 0644 digimend.conf /etc/depmod.d/digimend.conf
depmod -a
install hid-rebind /sbin/hid-rebind
install -m 0644 90-hid-rebind.rules /lib/udev/rules.d/90-hid-rebind.rules
udevadm control --reload
serikame@ZYX:~/digimend-kernel-drivers-hires_refactor$ sudo rmmod hid-kye
rmmod: ERROR: Module hid_kye is not currently loaded
serikame@ZYX:~/digimend-kernel-drivers-hires_refactor$ sudo rmmod hid-uclogic
rmmod: ERROR: Module hid_uclogic is not currently loaded
serikame@ZYX:~/digimend-kernel-drivers-hires_refactor$ sudo rmmod hid-huion

spbnick, 2018-04-27:
Right, I see now that the driver modules are not actually installed. Note how it says:

    Makefile:976: "Cannot use CONFIG_STACK_VALIDATION=y, please install libelf-dev, libelf-devel or elfutils-libelf-devel"

So install the required package with:

    sudo apt-get install libelf-dev

Then re-install the drivers again and post the output here.

serikame, 2018-04-27:
serikame@ZYX:~/digimend-kernel-drivers-hires_refactor$ sudo apt-get install libelf-dev
[sudo] contraseña para serikame: 
Leyendo lista de paquetes... Hecho
Creando árbol de dependencias       
Leyendo la información de estado... Hecho
Los paquetes indicados a continuación se instalaron de forma automática y ya no son necesarios.
  dkms libbsd0:i386 libdrm-amdgpu1:i386 libdrm-nouveau2:i386 libdrm-radeon1:i386 libdrm2:i386 libedit2:i386 libelf1:i386 libexpat1:i386 libffi6:i386 libllvm6.0:i386 libomxil-bellagio-bin libomxil-bellagio0
  libsensors4:i386 libstdc++6:i386 libtxc-dxtn0:i386 libtxc-dxtn0 libvdpau1:i386 libx11-6:i386 libx11-xcb1:i386 libxau6:i386 libxcb-dri2-0:i386 libxcb-dri3-0:i386 libxcb-glx0:i386 libxcb-present0:i386
  libxcb-sync1:i386 libxcb-xfixes0:i386 libxcb1:i386 libxdamage1:i386 libxdmcp6:i386 libxext6:i386 libxfixes3:i386 libxshmfence1:i386 libxxf86vm1:i386 mesa-vdpau-drivers:i386 vdpau-driver-all:i386
Utilice «sudo apt autoremove» para eliminarlos.
Se instalarán los siguientes paquetes NUEVOS:
  libelf-dev
0 actualizados, 1 nuevos se instalarán, 0 para eliminar y 0 no actualizados.
Se necesita descargar 57,2 kB de archivos.
Se utilizarán 373 kB de espacio de disco adicional después de esta operación.
Des:1 http://es.archive.ubuntu.com/ubuntu bionic/main amd64 libelf-dev amd64 0.170-0.4 [57,2 kB]
Descargados 57,2 kB en 0s (266 kB/s) 
Seleccionando el paquete libelf-dev:amd64 previamente no seleccionado.
(Leyendo la base de datos ... 277611 ficheros o directorios instalados actualmente.)
Preparando para desempaquetar .../libelf-dev_0.170-0.4_amd64.deb ...
Desempaquetando libelf-dev:amd64 (0.170-0.4) ...
Configurando libelf-dev:amd64 (0.170-0.4) ...
serikame@ZYX:~/digimend-kernel-drivers-hires_refactor$ sudo make install
make -C /lib/modules/4.15.0-20-lowlatency/build SUBDIRS=/home/serikame/digimend-kernel-drivers-hires_refactor modules_install
make[1]: se entra en el directorio '/usr/src/linux-headers-4.15.0-20-lowlatency'
  DEPMOD  4.15.0-20-lowlatency
make[1]: se sale del directorio '/usr/src/linux-headers-4.15.0-20-lowlatency'
install -D -m 0644 digimend.conf /etc/depmod.d/digimend.conf
depmod -a
install hid-rebind /sbin/hid-rebind
install -m 0644 90-hid-rebind.rules /lib/udev/rules.d/90-hid-rebind.rules
udevadm control --reload
serikame@ZYX:~/digimend-kernel-drivers-hires_refactor$ sudo rmmod hid-kye
rmmod: ERROR: Module hid_kye is not currently loaded
serikame@ZYX:~/digimend-kernel-drivers-hires_refactor$ sudo rmmod hid-uclogic
serikame@ZYX:~/digimend-kernel-drivers-hires_refactor$ sudo rmmod hid-huion
rmmod: ERROR: Module hid_huion is not currently loaded
serikame@ZYX:~/digimend-kernel-drivers-hires_refactor$

spbnick, 2018-04-27:
Alright, this is better, but the modules are still not installed for some reason. Could you please go to the source directory and run these commands there and post results:

    make clean
    make
    sudo make install

Thank you.

serikame, 2018-04-27:
serikame@ZYX:~/digimend-kernel-drivers-hires_refactor$ make clean
make -C /lib/modules/4.15.0-20-lowlatency/build SUBDIRS=/home/serikame/digimend-kernel-drivers-hires_refactor clean
make[1]: se entra en el directorio '/usr/src/linux-headers-4.15.0-20-lowlatency'
/bin/sh: 1: cannot create /home/serikame/digimend-kernel-drivers-hires_refactor/.cache.mk: Permission denied
/bin/sh: 1: cannot create /home/serikame/digimend-kernel-drivers-hires_refactor/.cache.mk: Permission denied
/bin/sh: 1: cannot create /home/serikame/digimend-kernel-drivers-hires_refactor/.cache.mk: Permission denied
/bin/sh: 1: cannot create /home/serikame/digimend-kernel-drivers-hires_refactor/.cache.mk: Permission denied
/bin/sh: 1: cannot create /home/serikame/digimend-kernel-drivers-hires_refactor/.cache.mk: Permission denied
/bin/sh: 1: cannot create /home/serikame/digimend-kernel-drivers-hires_refactor/.cache.mk: Permission denied
/bin/sh: 1: cannot create /home/serikame/digimend-kernel-drivers-hires_refactor/.cache.mk: Permission denied
/bin/sh: 1: cannot create /home/serikame/digimend-kernel-drivers-hires_refactor/.cache.mk: Permission denied
/bin/sh: 1: cannot create /home/serikame/digimend-kernel-drivers-hires_refactor/.cache.mk: Permission denied
/bin/sh: 1: cannot create /home/serikame/digimend-kernel-drivers-hires_refactor/.cache.mk: Permission denied
/bin/sh: 1: cannot create /home/serikame/digimend-kernel-drivers-hires_refactor/.cache.mk: Permission denied
/bin/sh: 1: cannot create /home/serikame/digimend-kernel-drivers-hires_refactor/.cache.mk: Permission denied
/bin/sh: 1: cannot create /home/serikame/digimend-kernel-drivers-hires_refactor/.cache.mk: Permission denied
/bin/sh: 1: cannot create /home/serikame/digimend-kernel-drivers-hires_refactor/.cache.mk: Permission denied
/bin/sh: 1: cannot create /home/serikame/digimend-kernel-drivers-hires_refactor/.cache.mk: Permission denied
/bin/sh: 1: cannot create /home/serikame/digimend-kernel-drivers-hires_refactor/.cache.mk: Permission denied
/bin/sh: 1: cannot create /home/serikame/digimend-kernel-drivers-hires_refactor/.cache.mk: Permission denied
/bin/sh: 1: cannot create /home/serikame/digimend-kernel-drivers-hires_refactor/.cache.mk: Permission denied
/bin/sh: 1: cannot create /home/serikame/digimend-kernel-drivers-hires_refactor/.cache.mk: Permission denied
/bin/sh: 1: cannot create /home/serikame/digimend-kernel-drivers-hires_refactor/.cache.mk: Permission denied
make[1]: se sale del directorio '/usr/src/linux-headers-4.15.0-20-lowlatency'
serikame@ZYX:~/digimend-kernel-drivers-hires_refactor$ make
make -C /lib/modules/4.15.0-20-lowlatency/build SUBDIRS=/home/serikame/digimend-kernel-drivers-hires_refactor modules
make[1]: se entra en el directorio '/usr/src/linux-headers-4.15.0-20-lowlatency'
  CC [M]  /home/serikame/digimend-kernel-drivers-hires_refactor/hid-kye.o
  CC [M]  /home/serikame/digimend-kernel-drivers-hires_refactor/hid-uclogic-core.o
  CC [M]  /home/serikame/digimend-kernel-drivers-hires_refactor/hid-uclogic-rdesc.o
  CC [M]  /home/serikame/digimend-kernel-drivers-hires_refactor/hid-uclogic-params.o
  LD [M]  /home/serikame/digimend-kernel-drivers-hires_refactor/hid-uclogic.o
  CC [M]  /home/serikame/digimend-kernel-drivers-hires_refactor/hid-polostar.o
  CC [M]  /home/serikame/digimend-kernel-drivers-hires_refactor/hid-viewsonic.o
  Building modules, stage 2.
  MODPOST 4 modules
  CC      /home/serikame/digimend-kernel-drivers-hires_refactor/hid-kye.mod.o
  LD [M]  /home/serikame/digimend-kernel-drivers-hires_refactor/hid-kye.ko
  CC      /home/serikame/digimend-kernel-drivers-hires_refactor/hid-polostar.mod.o
  LD [M]  /home/serikame/digimend-kernel-drivers-hires_refactor/hid-polostar.ko
  CC      /home/serikame/digimend-kernel-drivers-hires_refactor/hid-uclogic.mod.o
  LD [M]  /home/serikame/digimend-kernel-drivers-hires_refactor/hid-uclogic.ko
  CC      /home/serikame/digimend-kernel-drivers-hires_refactor/hid-viewsonic.mod.o
  LD [M]  /home/serikame/digimend-kernel-drivers-hires_refactor/hid-viewsonic.ko
make[1]: se sale del directorio '/usr/src/linux-headers-4.15.0-20-lowlatency'
serikame@ZYX:~/digimend-kernel-drivers-hires_refactor$ sudo make install
[sudo] contraseña para serikame: 
make -C /lib/modules/4.15.0-20-lowlatency/build SUBDIRS=/home/serikame/digimend-kernel-drivers-hires_refactor modules_install
make[1]: se entra en el directorio '/usr/src/linux-headers-4.15.0-20-lowlatency'
  INSTALL /home/serikame/digimend-kernel-drivers-hires_refactor/hid-kye.ko
At main.c:160:
- SSL error:02001002:system library:fopen:No such file or directory: ../crypto/bio/bss_file.c:74
- SSL error:2006D080:BIO routines:BIO_new_file:no such file: ../crypto/bio/bss_file.c:81
sign-file: certs/signing_key.pem: No such file or directory
  INSTALL /home/serikame/digimend-kernel-drivers-hires_refactor/hid-polostar.ko
At main.c:160:
- SSL error:02001002:system library:fopen:No such file or directory: ../crypto/bio/bss_file.c:74
- SSL error:2006D080:BIO routines:BIO_new_file:no such file: ../crypto/bio/bss_file.c:81
sign-file: certs/signing_key.pem: No such file or directory
  INSTALL /home/serikame/digimend-kernel-drivers-hires_refactor/hid-uclogic.ko
At main.c:160:
- SSL error:02001002:system library:fopen:No such file or directory: ../crypto/bio/bss_file.c:74
- SSL error:2006D080:BIO routines:BIO_new_file:no such file: ../crypto/bio/bss_file.c:81
sign-file: certs/signing_key.pem: No such file or directory
[... cut ...]
- SSL error:2006D080:BIO routines:BIO_new_file:no such file: ../crypto/bio/bss_file.c:81
sign-file: certs/signing_key.pem: No such file or directory
  DEPMOD  4.15.0-20-lowlatency
make[1]: se sale del directorio '/usr/src/linux-headers-4.15.0-20-lowlatency'
install -D -m 0644 digimend.conf /etc/depmod.d/digimend.conf
depmod -a
install hid-rebind /sbin/hid-rebind
install -m 0644 90-hid-rebind.rules /lib/udev/rules.d/90-hid-rebind.rules
udevadm control --reload
serikame@ZYX:~/digimend-kernel-drivers-hires_refactor$

spbnick, 2018-04-27:
This looks OK, although the "permission denied" errors should not be there. Now reconnect your tablet, post the output of `sudo modinfo hid-uclogic`, `sudo evtest`, `xinput list`, and then try using it.

spbnick, 2018-04-27:
Oh, and don't forget to execute `sudo rmmod hid-uclogic` before reconnecting the tablet.

serikame, 2018-04-27:
serikame@ZYX:~/digimend-kernel-drivers-hires_refactor$ sudo modinfo hid-uclogic
[sudo] contraseña para serikame: 
filename:       /lib/modules/4.15.0-20-lowlatency/extra/hid-uclogic.ko
version:        7
license:        GPL
author:         Nikolai Kondrashov
author:         Martin Rusko
srcversion:     C976222EA257C6A2728C449
alias:          hid:b0003g*v000028BDp00000075
alias:          hid:b0003g*v000028BDp00000071
alias:          hid:b0003g*v00002179p00000053
alias:          hid:b0003g*v00005543p00003031
alias:          hid:b0003g*v00005543p00000047
alias:          hid:b0003g*v00005543p00000045
alias:          hid:b0003g*v00005543p00000081
alias:          hid:b0003g*v00005543p0000004D
alias:          hid:b0003g*v00005543p0000006E
alias:          hid:b0003g*v0000256Cp0000006E
alias:          hid:b0003g*v00005543p00000781
alias:          hid:b0003g*v00005543p00000522
alias:          hid:b0003g*v00005543p00000064
alias:          hid:b0003g*v00005543p00000005
alias:          hid:b0003g*v00005543p00000004
alias:          hid:b0003g*v00005543p00000003
alias:          hid:b0003g*v00005543p00000042
depends:        hid
retpoline:      Y
name:           hid_uclogic
vermagic:       4.15.0-20-lowlatency SMP preempt mod_unload 
serikame@ZYX:~/digimend-kernel-drivers-hires_refactor$ sudo evtest
No device specified, trying to scan all of /dev/input/event*
Available devices:
/dev/input/event0:	Power Button
/dev/input/event1:	Power Button
/dev/input/event2:	MOSART Semi. 2.4G Keyboard Mouse
/dev/input/event3:	MOSART Semi. 2.4G Keyboard Mouse
/dev/input/event4:	USB USB Keyboard
/dev/input/event5:	USB USB Keyboard
/dev/input/event8:	HDA ATI HDMI HDMI/DP,pcm=3
/dev/input/event9:	HDA ATI HDMI HDMI/DP,pcm=7
/dev/input/event10:	HDA ATI HDMI HDMI/DP,pcm=8
/dev/input/event11:	HDA ATI HDMI HDMI/DP,pcm=9
/dev/input/event12:	HDA ATI HDMI HDMI/DP,pcm=10
/dev/input/event13:	HDA ATI HDMI HDMI/DP,pcm=11
/dev/input/event14:	HD-Audio Generic Front Mic
/dev/input/event15:	HD-Audio Generic Rear Mic
/dev/input/event16:	HD-Audio Generic Line
/dev/input/event17:	HD-Audio Generic Line Out Front
/dev/input/event18:	HD-Audio Generic Line Out Surround
/dev/input/event19:	HD-Audio Generic Line Out CLFE
/dev/input/event20:	HD-Audio Generic Front Headphone
/dev/input/event21:	UVC Camera (046d:0825)
Select the device event number [0-21]:
serikame@ZYX:~/digimend-kernel-drivers-hires_refactor$ xinput list
⎡ Virtual core pointer                    	id=2	[master pointer  (3)]
⎜   ↳ Virtual core XTEST pointer              	id=4	[slave  pointer  (2)]
⎜   ↳ USB USB Keyboard                        	id=9	[slave  pointer  (2)]
⎜   ↳ MOSART Semi. 2.4G Keyboard Mouse        	id=13	[slave  pointer  (2)]
⎣ Virtual core keyboard                   	id=3	[master keyboard (2)]
    ↳ Virtual core XTEST keyboard             	id=5	[slave  keyboard (3)]
    ↳ Power Button                            	id=6	[slave  keyboard (3)]
    ↳ Power Button                            	id=7	[slave  keyboard (3)]
    ↳ UVC Camera (046d:0825)                  	id=10	[slave  keyboard (3)]
    ↳ USB USB Keyboard                        	id=8	[slave  keyboard (3)]
    ↳ USB USB Keyboard                        	id=11	[slave  keyboard (3)]
    ↳ MOSART Semi. 2.4G Keyboard Mouse        	id=12	[slave  keyboard (3)]
    ↳ MOSART Semi. 2.4G Keyboard Mouse        	id=14	[slave  keyboard (3)]
serikame@ZYX:~/digimend-kernel-drivers-hires_refactor$ sudo rmmod hid-uclogic
[sudo] contraseña para serikame: 
rmmod: ERROR: Module hid_uclogic is not currently loaded
serikame@ZYX:~/digimend-kernel-drivers-hires_refactor$

serikame, 2018-04-28:
> Glad to hear it works, have fun using your tablet!

Nikolai thank you very much

serikame, 2018-05-04:
Hello Nikolai I had a conflict with Ubuntu 18.04 and I had to reinstall the operating system, but when I wanted to reinstall the digitizer card following the same steps that you had indicated to me now it throws an error in make: make[1]: *** There is no rule to build the goal 'y'.  Alto.
make[1]: se sale del directorio '/usr/src/linux-headers-4.15.0-20-lowlatency'
Makefile:13: recipe for target 'modules_install' failed

serikame, 2018-05-04:
I regret a lot of needing help again, but I had a problem with the installation of the AMD drives, the AMDGPUPRO, because I could not upload the resolution of my monitors and it turned out that once I installed the driver they blocked the Ubuntu Studio and I had to do a reinstallation and of course the Digimend drive was deleted, sorry.
The problem is that now $ sudo make install, fails and gives these messages: Directory does not exist, and Makefile: 13: recipe for target 'modules_install' failed
make: *** [modules_install] Error 2

spbnick, 2018-05-04:
@serikame, please post the full session of you trying to install the drivers. Also, when posting, surround the text with triple backticks, like this:

    ```
    FULL SESSION TEXT
    ```

serikame, 2018-05-04:
...
~/Controladores y Extras/Huion/digimend-kernel-drivers-hires_refactor$ sudo make install
[sudo] contraseña para serikame: 
make -C /lib/modules/4.15.0-20-lowlatency/build SUBDIRS=/home/serikame/Controladores y Extras/Huion/digimend-kernel-drivers-hires_refactor modules_install
make[1]: se entra en el directorio '/usr/src/linux-headers-4.15.0-20-lowlatency'
/bin/sh: 1: cannot create /home/serikame/Controladores/.cache.mk: Directory nonexistent
/bin/sh: 1: cannot create /home/serikame/Controladores/.cache.mk: Directory nonexistent
/bin/sh: 1: cannot create /home/serikame/Controladores/.cache.mk: Directory nonexistent
/bin/sh: 1: cannot create /home/serikame/Controladores/.cache.mk: Directory nonexistent
/bin/sh: 1: cannot create /home/serikame/Controladores/.cache.mk: Directory nonexistent
/bin/sh: 1: cannot create /home/serikame/Controladores/.cache.mk: Directory nonexistent
/bin/sh: 1: cannot create /home/serikame/Controladores/.cache.mk: Directory nonexistent
/bin/sh: 1: cannot create /home/serikame/Controladores/.cache.mk: Directory nonexistent
/bin/sh: 1: cannot create /home/serikame/Controladores/.cache.mk: Directory nonexistent
/bin/sh: 1: cannot create /home/serikame/Controladores/.cache.mk: Directory nonexistent
/bin/sh: 1: cannot create /home/serikame/Controladores/.cache.mk: Directory nonexistent
/bin/sh: 1: cannot create /home/serikame/Controladores/.cache.mk: Directory nonexistent
/bin/sh: 1: cannot create /home/serikame/Controladores/.cache.mk: Directory nonexistent
/bin/sh: 1: cannot create /home/serikame/Controladores/.cache.mk: Directory nonexistent
/bin/sh: 1: cannot create /home/serikame/Controladores/.cache.mk: Directory nonexistent
/bin/sh: 1: cannot create /home/serikame/Controladores/.cache.mk: Directory nonexistent
/bin/sh: 1: cannot create /home/serikame/Controladores/.cache.mk: Directory nonexistent
/bin/sh: 1: cannot create /home/serikame/Controladores/.cache.mk: Directory nonexistent
/bin/sh: 1: cannot create /home/serikame/Controladores/.cache.mk: Directory nonexistent
/bin/sh: 1: cannot create /home/serikame/Controladores/.cache.mk: Directory nonexistent
arch/x86/Makefile:156: CONFIG_X86_X32 enabled but no binutils support
/bin/sh: 1: cannot create /home/serikame/Controladores/.cache.mk: Directory nonexistent
/bin/sh: 1: cannot create /home/serikame/Controladores/.cache.mk: Directory nonexistent
/bin/sh: 1: cannot create /home/serikame/Controladores/.cache.mk: Directory nonexistent
/bin/sh: 1: cannot create /home/serikame/Controladores/.cache.mk: Directory nonexistent
/bin/sh: 1: cannot create /home/serikame/Controladores/.cache.mk: Directory nonexistent
/bin/sh: 1: cannot create /home/serikame/Controladores/.cache.mk: Directory nonexistent
/bin/sh: 1: cannot create /home/serikame/Controladores/.cache.mk: Directory nonexistent
/bin/sh: 1: cannot create /home/serikame/Controladores/.cache.mk: Directory nonexistent
/bin/sh: 1: cannot create /home/serikame/Controladores/.cache.mk: Directory nonexistent
/bin/sh: 1: cannot create /home/serikame/Controladores/.cache.mk: Directory nonexistent
/bin/sh: 1: cannot create /home/serikame/Controladores/.cache.mk: Directory nonexistent
/bin/sh: 1: cannot create /home/serikame/Controladores/.cache.mk: Directory nonexistent
/bin/sh: 1: cannot create /home/serikame/Controladores/.cache.mk: Directory nonexistent
/bin/sh: 1: cannot create /home/serikame/Controladores/.cache.mk: Directory nonexistent
/bin/sh: 1: cannot create /home/serikame/Controladores/.cache.mk: Directory nonexistent
/bin/sh: 1: cannot create /home/serikame/Controladores/.cache.mk: Directory nonexistent
/bin/sh: 1: cannot create /home/serikame/Controladores/.cache.mk: Directory nonexistent
/bin/sh: 1: cannot create /home/serikame/Controladores/.cache.mk: Directory nonexistent
/bin/sh: 1: cannot create /home/serikame/Controladores/.cache.mk: Directory nonexistent
/bin/sh: 1: cannot create /home/serikame/Controladores/.cache.mk: Directory nonexistent
/bin/sh: 1: cannot create /home/serikame/Controladores/.cache.mk: Directory nonexistent
/bin/sh: 1: cannot create /home/serikame/Controladores/.cache.mk: Directory nonexistent
/bin/sh: 1: cannot create /home/serikame/Controladores/.cache.mk: Directory nonexistent
/bin/sh: 1: cannot create /home/serikame/Controladores/.cache.mk: Directory nonexistent
/bin/sh: 1: cannot create /home/serikame/Controladores/.cache.mk: Directory nonexistent
/bin/sh: 1: cannot create /home/serikame/Controladores/.cache.mk: Directory nonexistent
/bin/sh: 1: cannot create /home/serikame/Controladores/.cache.mk: Directory nonexistent
/bin/sh: 1: cannot create /home/serikame/Controladores/.cache.mk: Directory nonexistent
/bin/sh: 1: cannot create /home/serikame/Controladores/.cache.mk: Directory nonexistent
/bin/sh: 1: cannot create /home/serikame/Controladores/.cache.mk: Directory nonexistent
/bin/sh: 1: cannot create /home/serikame/Controladores/.cache.mk: Directory nonexistent
/bin/sh: 1: cannot create /home/serikame/Controladores/.cache.mk: Directory nonexistent
/bin/sh: 1: cannot create /home/serikame/Controladores/.cache.mk: Directory nonexistent
/bin/sh: 1: cannot create /home/serikame/Controladores/.cache.mk: Directory nonexistent
/bin/sh: 1: cannot create /home/serikame/Controladores/.cache.mk: Directory nonexistent
/bin/sh: 1: cannot create /home/serikame/Controladores/.cache.mk: Directory nonexistent
/bin/sh: 1: cannot create /home/serikame/Controladores/.cache.mk: Directory nonexistent
[... cut ...]
/bin/sh: 1: cannot create /home/serikame/Controladores/.cache.mk: Directory nonexistent
/bin/sh: 1: cannot create /home/serikame/Controladores/.cache.mk: Directory nonexistent
/bin/sh: 1: cannot create /home/serikame/Controladores/.cache.mk: Directory nonexistent
/bin/sh: 1: cannot create /home/serikame/Controladores/.cache.mk: Directory nonexistent
make[1]: *** No hay ninguna regla para construir el objetivo 'y'.  Alto.
make[1]: se sale del directorio '/usr/src/linux-headers-4.15.0-20-lowlatency'
Makefile:13: recipe for target 'modules_install' failed
...

spbnick, 2018-05-04:
To get rid of the "Directory nonexistent" error please rename your `Controladores y Extras` directory to something without spaces. For example to `Controladores_y_Extras`. To let me read your output more easily (since I don't know Spanish), please execute `export LANG=C` before installing drivers. After that, please try again and post the session here. Please use the "backtick" character which I mention above, instead of dots to surround the output. Otherwise it doesn't work. Thank you.

serikame, 2018-05-04:
You were right, the problem was the name of the directory and the spaces between words, I solved it by replacing the spaces with scripts and I managed to install the digitizer tablet with your drives and it works well. Thank you very much and apologize for my computer ignorance.
regards

spbnick, 2018-05-04:
No need to apologize. We're all ignorant one way or another. Glad to hear it worked for you!