DIGImend/digimend-kernel-drivers #137: Add support for XP-Pen Artist 10s

ign1s-fatuus, 2018-04-26:
Here are the various files for the XP-PEN Artist 10s.

As it is the tablet, being a digital graphics tablet works awesome via xrandr for displayed output. With the rest of it being operational it would be a true wonder. 

[frame_buttons.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/1949710/frame_buttons.txt)
[pen_buttons.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/1949711/pen_buttons.txt)
[pen_pressure.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/1949712/pen_pressure.txt)
[pen_tilt.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/1949713/pen_tilt.txt)
[pen_coords.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/1949714/pen_coords.txt)
[hid_report_descriptors.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/1949716/hid_report_descriptors.txt)
[descriptors.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/1949717/descriptors.txt)

spbnick, 2018-05-04:
Thank you, @ign1s-fatuus! @danghai, would you be able to integrate these into the tablets repo?

ign1s-fatuus, 2018-05-04:
I can see what I can do. What would you need from me?

danghai, 2018-05-04:
Hello @ign1s-fatuus , 
You can review an example my commit in repo `tablet`: to add information about your tablet: https://github.com/DIGImend/tablets/commit/35457a0741b56f9841e9fe41f44fc7ad1e762029

And follow the `README` from this repo: https://github.com/DIGImend/digimend.github.io in order to run your Jekyll site locally. It helps us review what the digimend website looks like.
Thanks. Ping me if you need any help

danghai, 2018-05-08:
@ign1s-fatuus  would you mind to provide your probe.txt? You can review the command from there: https://github.com/DIGImend/uclogic-tools to get information your probe. Thanks

ign1s-fatuus, 2018-05-08:
Here you go. Lemme know if there is anything else I can do. I am in the middle of writing papers for  finals as well as writing my Master's thesis, so my time is limited, but I will address what I can when I get a second of free time. 
[probe.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/1984857/probe.txt)

jimmy00784, 2018-10-03:
Any timeline on this? XP-PEN's beta driver for Linux doesn't support Artist 10S and Wizardpen project is all but dead with no development. Digimend is the only hope for Artist 10S.

ign1s-fatuus, 2018-10-23:
Hi, just checking in after a while away. I too would be interested in knowing if there has been any movement on the 10S. While I will be busy university coursework for the foreseeable future, I will take any time needed to help in whatever way that I can.

ign1s-fatuus, 2018-10-28:
> Could somebody please try this version of the driver, where I tried to add
> support for XP-Pen Artist 10s:
> https://github.com/spbnick/digimend-kernel-drivers/archive/add_xp_pen_artist_10s.zip

I'll look into it ASAP and get back to you later this week at the
latest--grad school deadlines coming up but I will gladly show my support
by testing out what you've been laboring over.

kayn23, 2019-05-27:
How to progress with drivers for Artist 10s?

hooger, 2020-04-23:
I know it's an old post, however I try to use a device like this. A friend borrowed me this and a Wacom Intous S, so I've seen how a properly configured tablet should work.
I tried the zipped version from @spbnick and I also tried the .deb version (downloaded 2 days back). They produced the same behaviour. System:
```
Linux *** 4.19.0-8-amd64 #1 SMP Debian 4.19.98-1 (2020-01-26) x86_64 GNU/Linux
```
I'm on KDE, I also installed kde-config-tablet, which doesn't recognize the XP-Pen (with the Wacom tablet like 5 tabs appear, and printed it's name). The kde wizard offered me to setup a custom tablet, but it crashed with various options and I also wasn't sure what should I input.

Experience with the tablet:
- As a screen it was recognized at once. Set the system to use dual monitor use, merge them, set to the same resolution. This was needed, as the pen moved the mouse, as the table has been showing the whole screen.
- The usb device also appears and moves the mouse.
- There are various zones around the tablet where (I think) there is no recognition, like there would be  a push magnet, so the mouse runs around it. I think it's a HW problem, maybe it is dirty? Can you recommend some way to test? Is it possible to clean it with some window cleaner?
- Pen pressure works but I need to press really hard until it starts to appear in Krita. In the pen tester it works as it should, with normal touch, and it prints 0% pressure.

How can I help to make it work?
Thanks in advance!

hooger, 2020-04-25:
Some more info that may help:

- `lsusb` recognizes the device: `Bus 001 Device 002: ID 5543:004a UC-Logic Technology Corp. `
- `dmesg` output:
```
[  723.034588] usb 1-4: new full-speed USB device number 5 using xhci_hcd
[  723.191934] usb 1-4: New USB device found, idVendor=5543, idProduct=004a, bcdDevice= 0.00
[  723.191942] usb 1-4: New USB device strings: Mfr=5, Product=6, SerialNumber=0
[  723.191946] usb 1-4: Product: 10.1 Tablet Monitor
[  723.191950] usb 1-4: Manufacturer: UGEE
[  723.200590] input: UGEE 10.1 Tablet Monitor Pen as /devices/pci0000:00/0000:00:14.0/usb1/1-4/1-4:1.0/0003:5543:004A.0004/input/input31
[  723.200976] input: UGEE 10.1 Tablet Monitor Pad as /devices/pci0000:00/0000:00:14.0/usb1/1-4/1-4:1.0/0003:5543:004A.0004/input/input32
[  723.201373] uclogic 0003:5543:004A.0004: input,hidraw0: USB HID v1.11 Keypad [UGEE 10.1 Tablet Monitor] on usb-0000:00:14.0-4/input0
[  723.203343] input: UGEE 10.1 Tablet Monitor Mouse as /devices/pci0000:00/0000:00:14.0/usb1/1-4/1-4:1.1/0003:5543:004A.0005/input/input33
[  723.203655] uclogic 0003:5543:004A.0005: input,hiddev0,hidraw1: USB HID v1.11 Mouse [UGEE 10.1 Tablet Monitor] on usb-0000:00:14.0-4/input1
[  723.205171] input: UGEE 10.1 Tablet Monitor Keyboard as /devices/pci0000:00/0000:00:14.0/usb1/1-4/1-4:1.2/0003:5543:004A.0006/input/input34
[  723.267117] input: UGEE 10.1 Tablet Monitor Consumer Control as /devices/pci0000:00/0000:00:14.0/usb1/1-4/1-4:1.2/0003:5543:004A.0006/input/input35
[  723.267439] input: UGEE 10.1 Tablet Monitor System Control as /devices/pci0000:00/0000:00:14.0/usb1/1-4/1-4:1.2/0003:5543:004A.0006/input/input36
[  723.267749] uclogic 0003:5543:004A.0006: input,hidraw2: USB HID v1.11 Keyboard [UGEE 10.1 Tablet Monitor] on usb-0000:00:14.0-4/input2
```
- `xsetwacom`does not recognizes the device.
- According to `lsmod |grep hid`, the dkms module is not used. If I understand the output right:
```
hid_uclogic            28672  0
hid_generic            16384  0
usbhid                 57344  0
hid                   135168  3 usbhid,hid_uclogic,hid_generic
usbcore               294912  6 xhci_hcd,usbhid,hid_uclogic,uvcvideo,btusb,xhci_pci
```

How may I help?

hooger, 2020-04-25:
So I cloned the current stage of the repo, and made some changes based on merged PRs, that add a single device. The results is in this repo: https://github.com/hooger/digimend-kernel-drivers/tree/xppen-artist-10s
Currently the table is recognized, `xsetwacom` recognizes the tablet, pressure curve can be customized and also MapToOutput works properly. Is there anything else I should check or should I make the PR?

Thanks!

hooger, 2024-05-26:
I have an open pull request for this device, that I regularly use. Unfortunately, on the PR page there is no activity. Can anyone point me how can I speed up the merge process?
Thanks in advance!