DIGImend/digimend-kernel-drivers #140: Improve installation

spbnick, 2018-05-11:
People often write with tablet issues, where they don't actually have the out-of-tree driver installed. Think of a way to improve that: verify that driver installed after make, provide diagnostic utility checking which driver version is installed, provide a HOWTO, or something else.

ThePreviousOne, 2018-05-15:
For the HOWTO were you thinking just a simple thing in the readme? If you like I could start a Wiki here.

ThePreviousOne, 2018-05-15:
Forgot to mention, if you search "Digimend" or "Huion Linux Driver" most search engines this page is listed before the io.github page. So starting a wiki here or adding links to the other on could be of great help.

spbnick, 2018-06-02:
@ThePreviousOne Riight. It would be good to have the HOWTO in the README.md or somewhere close. Only we have to balance the size of it somehow. People are already not noticing things there. Perhaps we need to split it into basic installation and have troubleshooting somewhere else, clearly linked.