DIGImend/digimend-kernel-drivers #141: Huion KAMVAS GT-221 Pro dianostics

kelargo, 2018-05-12:
using Ubuntu 18-04 and installed kernel drivers from hires_refactor

the pen coordinates are 100% empty  no output appearing.

[descriptors.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/1997986/descriptors.txt)

[frame_buttons.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/1997987/frame_buttons.txt)

[hid_report_descriptors.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/1997988/hid_report_descriptors.txt)

[xorg_log.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/1997993/xorg_log.txt)

kelargo, 2018-05-12:
mm.. maybe the pen wasnt charged before? or something.  
from top, bottom, left, right.

[pen_coords.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/1997994/pen_coords.txt)

danghai, 2018-05-30:
Hi @kelargo , could you provide the `probe.txt`? Thanks

kelargo, 2018-05-30:
yes,  how do I generate "probe.txt"  info ?

kelargo, 2018-05-30:
never mind,  I think this is the info ?   The most recent drivers seem to be working for everything except buttons on the pen.  Im not seeing any output from pushing pen buttons. 

`
sudo uclogic-probe 1 6 > probe.txt 
sudo uclogic-decode < probe.txt 
    |                Product: Tablet Monitor
 64 |        Params block #1: ??????
    |                          Max X: 37540
    |                          Max Y: 21120
    |                   Max pressure: 8191
    |                     Resolution: 2000
    |
 79 |         Internal model: 
 7b |         Buttons status: HK On
 c8 |        Params block #2: ????????
    |                          Max X: 95352
    |                          Max Y: 53645
    |                   Max pressure: 8191
    |                     Resolution: 5080
    |
 c9 |      Unknown string #1: HUION_M167_170623
 ca |  Internal manufacturer: HUION Animation Technology Co.,ltd

 
cat probe.txt 
P 54 00 61 00 62 00 6C 00 65 00 74 00 20 00 4D 00 6F 00 6E 00 69 00 74 00 6F 00 72 00
S 64 0E 03 A4 92 80 52 03 00 FF 1F D0 07 08 00
S 65 04 03 09 00
S 6E 04 03 31 00
S 79 02 03
S 7A 08 03 01 10 00 00 00 00
S 7B 0C 03 48 00 4B 00 20 00 4F 00 6E 00
S C8 13 03 78 74 01 8D D1 00 FF 1F D8 13 03 0A 00 00 02 00 3E
S C9 24 03 48 00 55 00 49 00 4F 00 4E 00 5F 00 4D 00 31 00 36 00 37 00 5F 00 31 00 37 00 30 00 36 00 32 00 33 00
S CA 46 03 48 00 55 00 49 00 4F 00 4E 00 20 00 41 00 6E 00 69 00 6D 00 61 00 74 00 69 00 6F 00 6E 00 20 00 54 00 65 00 63 00 68 00 6E 00 6F 00 6C 00 6F 00 67 00 79 00 20 00 43 00 6F 00 2E 00 2C 00 6C 00 74 00 64 00
`

danghai, 2018-06-12:
Hi @kelargo , could you help me provide detail dumps descriptor? For particular, could you make a new dump with a specific button press order, and another dump of the touch bar with clearly specified sequence of full-range movement, repeated several times so that we are sure we get the full range and the direction correctly? It is very great because Nick will make better support for these tablet in the future. Thanks