DIGImend/digimend-kernel-drivers #145: Define soft buttons on Huion H1060P

a-l-e-x-d-s-9, 2018-05-15:
I'm using the latest drivers from the git. 
The tablet has 16 soft buttons at the upper part of the tablet pad. I can't find any information how to make them usable as fully functional buttons, any suggestions?
Also I tried to limit the usable area with:
"xsetwacom --set 'Tablet Monitor Pad pad' Area 90 2500 50800 31750"
But received the error: 
"Property 'Wacom Tablet Area' does not exist on device." 
Am I doing it wrong, or it just unsupported?
BTW, only the first 8 hard buttons (out of 12) are working with "xsetwacom --set 'Tablet Monitor Pad pad' Button ...", and the other return: "Unsupported offset into 'Wacom Button Actions' property.", any insights?

```console
$ xsetwacom --list
Tablet Monitor Pen stylus       	id: 16	type: STYLUS    
Tablet Monitor Pad pad          	id: 17	type: PAD

$ lsusb -v
Bus 003 Device 004: ID 256c:006e  
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.10
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0         8
  idVendor           0x256c 
  idProduct          0x006e 
  bcdDevice            0.00
  iManufacturer           5 (error)
  iProduct                6 Tablet Monitor
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           59
    bNumInterfaces          2
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      18
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               2
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      83
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               2
Device Status:     0x0000
  (Bus Powered)
```

spbnick, 2018-05-15:
Do not bother with the "soft buttons", they don't get reported by the hardware, must be implemented in the driver, and they're mostly useless. If you have to use something like that, just put the buttons on the screen (e.g. in a window, or in a taskbar) and you won't need to look at the tablet to hit them.

Regarding button numbers, run `xinput test` for your pad device, and look at which buttons are reported with which number, then use them in a mapping. Generally, button numbers go 1, 2, 3, 8, 9, and so on. Button numbers 4-7 are reserved for vertical and horizontal scrolling events.

spbnick, 2018-05-15:
All buttons should work on H1060P. If they don't please submit the exact `xsetwacom` command you're executing, for the start. Thank you.

a-l-e-x-d-s-9, 2018-05-15:
The "missing" 4-7 buttons confused me, but with 13-16 button numbers all 12 are working fine.
Thanks!

closed 2018-05-15 by a-l-e-x-d-s-9