DIGImend/digimend-kernel-drivers #146: Parblo A610s issues

avsamsonovjr, 2018-05-15:
Tablet Parblo A610s
System openSUSE 42.3
Linux linux-zf0g 4.4.126-48-default #1 SMP Sat Apr 7 05:22:50 UTC 2018 (f24992c) x86_64 x86_64 x86_64 GNU/Linux
Driver digimend-kernel-drivers-7
Download, make and install with success. 
The tablet does get recognized, but don't work in applications (e.g. GIMP, Krita)
[descriptors.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/2004819/descriptors.txt)
[frame_buttons.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/2004821/frame_buttons.txt)
[pen_buttons.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/2004822/pen_buttons.txt)
[pen_coords.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/2004823/pen_coords.txt)
[pen_pressure.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/2004824/pen_pressure.txt)

Identifiers is 5543:0081

>  linux-zf0g:/home/avsnh7 # lsusb
Bus 002 Device 002: ID 8087:8000 Intel Corp. 
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 002: ID 8087:8008 Intel Corp. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 010: ID 5543:0081 UC-Logic Technology Corp. 
Bus 003 Device 005: ID 058f:6364 Alcor Micro Corp. AU6477 Card Reader Controller
Bus 003 Device 004: ID 03f0:0c17 Hewlett-Packard LaserJet 1010
Bus 003 Device 003: ID 2341:0010 Arduino SA Mega 2560 (CDC ACM)
Bus 003 Device 002: ID 1a2c:0042 China Resource Semico Co., Ltd 
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

Output of dmesg

> [ 4139.856707] usb 3-6: new full-speed USB device number 11 using xhci_hcd
[ 4139.985756] usb 3-6: New USB device found, idVendor=5543, idProduct=0081
[ 4139.985758] usb 3-6: New USB device strings: Mfr=5, Product=6, SerialNumber=0
[ 4139.985759] usb 3-6: Product: A610s
[ 4139.985760] usb 3-6: Manufacturer: UC-Logic
[ 4139.987354] input: UC-Logic A610s as /devices/pci0000:00/0000:00:14.0/usb3/3-6/3-6:1.0/0003:5543:0081.0027/input/input132
[ 4139.987477] hid-generic 0003:5543:0081.0027: input,hiddev0,hidraw2: USB HID v1.11 Mouse [UC-Logic A610s] on usb-0000:00:14.0-6/input0
[ 4139.988751] input: UC-Logic A610s as /devices/pci0000:00/0000:00:14.0/usb3/3-6/3-6:1.1/0003:5543:0081.0028/input/input133
[ 4139.989544] hid-generic 0003:5543:0081.0028: input,hiddev0,hidraw3: USB HID v1.11 Mouse [UC-Logic A610s] on usb-0000:00:14.0-6/input1
[ 4139.991211] input: UC-Logic A610s as /devices/pci0000:00/0000:00:14.0/usb3/3-6/3-6:1.2/0003:5543:0081.0029/input/input134
[ 4140.044805] hid-generic 0003:5543:0081.0029: input,hidraw4: USB HID v1.11 Keyboard [UC-Logic A610s] on usb-0000:00:14.0-6/input2
[ 4140.130764] input: UC-Logic A610s Pen as /devices/pci0000:00/0000:00:14.0/usb3/3-6/3-6:1.0/0003:5543:0081.0027/input/input135
[ 4140.186258] input: UC-Logic A610s Mouse as /devices/pci0000:00/0000:00:14.0/usb3/3-6/3-6:1.1/0003:5543:0081.0028/input/input136
[ 4140.249179] input: UC-Logic A610s Pad as /devices/pci0000:00/0000:00:14.0/usb3/3-6/3-6:1.0/0003:5543:0081.0027/input/input137
[ 4140.249824] uclogic 0003:5543:0081.0027: input,hidraw2: USB HID v1.11 Keypad [UC-Logic A610s] on usb-0000:00:14.0-6/input0
[ 4140.263306] uclogic 0003:5543:0081.0028: input,hiddev0,hidraw3: USB HID v1.11 Mouse [UC-Logic A610s] on usb-0000:00:14.0-6/input1
[ 4140.273494] input: UC-Logic A610s Keyboard as /devices/pci0000:00/0000:00:14.0/usb3/3-6/3-6:1.2/0003:5543:0081.0029/input/input139
[ 4140.364915] input: UC-Logic A610s Consumer Control as /devices/pci0000:00/0000:00:14.0/usb3/3-6/3-6:1.2/0003:5543:0081.0029/input/input140
[ 4140.365149] input: UC-Logic A610s System Control as /devices/pci0000:00/0000:00:14.0/usb3/3-6/3-6:1.2/0003:5543:0081.0029/input/input141
[ 4140.365352] uclogic 0003:5543:0081.0029: input,hidraw4: USB HID v1.11 Keyboard [UC-Logic A610s] on usb-0000:00:14.0-6/input2

Xorg.0.log very large 
[Xorg.0.log.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/2004884/Xorg.0.log.txt)

And (at end of all) xinput list

> linux-zf0g:/home/avsnh7 # xinput list
⎡ Virtual core pointer                          id=2    [master pointer  (3)]
⎜   ↳ Virtual core XTEST pointer                id=4    [slave  pointer  (2)]
⎜   ↳ SEMICCHIP Usb Mouse                       id=9    [slave  pointer  (2)]
⎜   ↳ UC-Logic A610s Consumer Control           id=12   [slave  pointer  (2)]
⎜   ↳ UC-Logic A610s Mouse                      id=17   [slave  pointer  (2)]
⎣ Virtual core keyboard                         id=3    [master keyboard (2)]
    ↳ Virtual core XTEST keyboard               id=5    [slave  keyboard (3)]
    ↳ Power Button                              id=6    [slave  keyboard (3)]
    ↳ Video Bus                                 id=7    [slave  keyboard (3)]
    ↳ Power Button                              id=8    [slave  keyboard (3)]
    ↳ Eee PC WMI hotkeys                        id=15   [slave  keyboard (3)]
    ↳ AT Translated Set 2 keyboard              id=16   [slave  keyboard (3)]
    ↳ UC-Logic A610s System Control             id=10   [slave  keyboard (3)]
    ↳ UC-Logic A610s Keyboard                   id=11   [slave  keyboard (3)]
    ↳ UC-Logic A610s Consumer Control           id=13   [slave  keyboard (3)]
    ↳ UC-Logic A610s Pen                        id=14   [slave  keyboard (3)]

As you see, Pen in keyboard section....
Anyone help?

ThePreviousOne, 2018-05-16:
https://digimend.github.io/support/howto/trbl/locating_failure/
also try the git version

xunterr, 2022-02-18:
Same issue
Kernel: x86_64 Linux 5.15.21-1-MANJARO
Driver: digimend-kernel-drivers-10
Id: 5543:0081

[descriptors.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/8098483/descriptors.txt)
[frame_buttons.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/8098485/frame_buttons.txt)
[hid_report_descriptors.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/8098486/hid_report_descriptors.txt)
[pen_buttons.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/8098487/pen_buttons.txt)
[pen_coords.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/8098488/pen_coords.txt)
[pen_pressure.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/8098489/pen_pressure.txt)
[probe.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/8098490/probe.txt)