DIGImend/digimend-kernel-drivers #147: Configuring pad buttons on Monoprice 10594: xinput/xwacom not finding button presses

TheDaviot, 2018-05-16:
After referencing other threads, such as:
[https://github.com/DIGImend/digimend-kernel-drivers/issues/42](https://github.com/DIGImend/digimend-kernel-drivers/issues/42)
[https://github.com/DIGImend/digimend-kernel-drivers/issues/26](https://github.com/DIGImend/digimend-kernel-drivers/issues/26)

I was able to edit /etc/X11/xorg.conf.d/52-tablet.conf such that xsetwacom --list returns:
```
10594 Pen stylus                	id: 10	type: STYLUS    
10594 Pad pad                   	id: 11	type: PAD    
```
lsusb shows a "Bus 001 Device 008: ID 256c:006e" that is the tablet and evtest returns events for the button presses.  However, xinput test isn't receiving anything, and even after setting keypresses to the wacom drivers, neither they or the xinput test work.
I'm running Linux Mint 18.3 Cinnamon with kernel 4.13.0-41

Help/suggestions?

orbitaldigital, 2019-07-15:
Help some bug for me...

orbitaldigital, 2019-07-15:
```
#!/bin/sh
xsetwacom --set 'HID 256c:006e Pad pad' Button 1 "key +ctrl +z -z -ctrl"
xsetwacom --set 'HID 256c:006e Pad pad' Button 2 "key e"
xsetwacom --set 'HID 256c:006e Pad pad' Button 3 "key b"
xsetwacom --set 'HID 256c:006e Pad pad' Button 8 "key +"
xsetwacom --set 'HID 256c:006e Pad pad' Button 9 "key -"
xsetwacom --set 'HID 256c:006e Pad pad' Button 10 "key ]"
xsetwacom --set 'HID 256c:006e Pad pad' Button 11 "key ["
xsetwacom --set 'HID 256c:006e Pad pad' Button 12 "key p"
```

orbitaldigital, 2019-07-15:
work's for me