DIGImend/digimend-kernel-drivers #148: Huion H950P

Sylphien, 2018-05-17:
Hi, currently trying to install Huion H950P with no luck. Tried to follow the manual installation instructions but no dice:

[sylphien@Manjaro_Workstation digimend-kernel-drivers-master]$ make
make -C /lib/modules/4.16.7-1-MANJARO/build SUBDIRS=/home/sylphien/Downloads/digimend-kernel-drivers-master modules
make[1]: Entering directory '/usr/lib/modules/4.16.7-1-MANJARO/build'
make[1]: *** No rule to make target 'modules'.  Stop.
make[1]: Leaving directory '/usr/lib/modules/4.16.7-1-MANJARO/build'
make: *** [Makefile:13: modules] Error 2
[sylphien@Manjaro_Workstation digimend-kernel-drivers-master]$ 

Would you please help me? I have no idea where to do now.

danghai, 2018-05-18:
@Sylphien , Do you have the kernel headers installed?

Sylphien, 2018-05-18:
@a904guy , thanks for your response. I was just reading about that, googled "No rule to make target 'modules' and the first result was about headers". I'm not really skilled when it comes to Linux's kernels and command line. How do i know if the headers are installed?

danghai, 2018-05-18:
Try use the above command, and then using `make` again

Sylphien, 2018-05-18:
What command? This: "make -C /lib/modules/4.16.7-1-MANJARO/build SUBDIRS=/home/sylphien/Downloads/digimend-kernel-drivers-master modules"?

danghai, 2018-05-18:
`sudo apt-get install linux-headers-`uname -r``

Sylphien, 2018-05-18:
[sylphien@Manjaro_Workstation ~]$ sudo apt-get install linux-headers-`uname -r`
sudo: apt-get: command not found
[sylphien@Manjaro_Workstation ~]$

danghai, 2018-05-18:
What is your `linux distribution`? ( Ex: Fedora, Ubuntu, ...)

Sylphien, 2018-05-18:
Running Manjaro. Running Kernel 4.16.7-1

danghai, 2018-05-18:
Try this command then: 
```
sudo pacman -S linux414-headers
```

Sylphien, 2018-05-18:
The headers were successfully installed. What do i do now?

danghai, 2018-05-18:
What is your output `uname -r`

danghai, 2018-05-18:
Ah I do not notice your version, try it then

Sylphien, 2018-05-18:
Trying 416 and it worked. Already used "make" and then "sudo make install". Do I need to restart the computer?

Sylphien, 2018-05-18:
Rebooted and It's working already. Thank you very much! Just one thing. Buttons are not working. Any comment on that?

danghai, 2018-05-18:
Ok great! Sorry, I cannot help you debug the button issue now.

Sylphien, 2018-05-18:
Thanks, I'll look into that later.

danghai, 2018-05-19:
@Sylphien For the button issue. You can see `Userspace drivers` at: http://digimend.github.io/support/howto/trbl/locating_failure/ to see your output from button.

Sylphien, 2018-05-19:
@danghai . Shouldn't I try to configure the buttons using the wacom drivers as it's explained in the installation guide? Have you done it the way you're recommending?

danghai, 2018-05-19:
@Sylphien I mean when install driver and try to test something. You can use command: `sudo xinput list`
My output:
```
danghai@ubuntu:~$ xinput list
⎡ Virtual core pointer                    	id=2	[master pointer  (3)]
⎜   ↳ Virtual core XTEST pointer              	id=4	[slave  pointer  (2)]
⎜   ↳ VMware VMware Virtual USB Mouse         	id=7	[slave  pointer  (2)]
⎜   ↳ VirtualPS/2 VMware VMMouse              	id=9	[slave  pointer  (2)]
⎜   ↳ VirtualPS/2 VMware VMMouse              	id=10	[slave  pointer  (2)]
⎜   ↳ Tablet Monitor Mouse                    	id=11	[slave  pointer  (2)]
⎜   ↳ Tablet Monitor Pen                      	id=12	[slave  pointer  (2)]
⎜   ↳ Tablet Monitor Pad                      	id=13	[slave  pointer  (2)]
⎣ Virtual core keyboard                   	id=3	[master keyboard (2)]
    ↳ Virtual core XTEST keyboard             	id=5	[slave  keyboard (3)]
    ↳ Power Button                            	id=6	[slave  keyboard (3)]
    ↳ AT Translated Set 2 keyboard            	id=8	[slave  keyboard (3)]
```
If I want to test button of `pad`, I will use command: `xinput test 13`. And I try to use some buttons on my pad:
```
button press   1 
button release 1 
button press   2 
button press   8 
button release 8 
button press   9 
button release 9 
button press   10 
button release 10 
button press   2 
```
Hopefully, it is clear