DIGImend/digimend-kernel-drivers #149: Tablet Lost functionality after upgrading from Ubuntu 16.04 to 18.04, and the .deb package didn't work

bryanwillis7, 2018-05-20:
My tablet is the Ugee 2150, which is supported by these drivers. Any help at all would be greatly appreciated. A bit panicky, as this is an expensive tablet and this could cause some serious issues if I can't get this working. :/

spbnick, 2018-05-20:
Hi Bryan! Please describe how the .deb package "didn't work". Try installing the [latest release](https://github.com/DIGImend/digimend-kernel-drivers/releases/tag/v7), and if *that* doesn't work, directly from the [latest development source](https://github.com/DIGImend/digimend-kernel-drivers/archive/master.zip). Report results here.

spbnick, 2018-06-17:
Closing, since not getting required information about the issue.

closed 2018-06-17 by spbnick