DIGImend/digimend-kernel-drivers #156: XP Pen Deco 01 pressure sensitivity reset

razvanc-r, 2018-05-26:
Tried on Ubuntu 18.04 with commit 38585dfb78d23a24297433a381839c5bd762c96c.

I tried both xsetwacom (`xsetwacom --set "UGEE DECO 01 Pen stylus" PressureCurve "0 0 10 10") and xorg.conf (`Option "PressCurve" "0,0,10,10"`). It doesn't seem that the pressure curve is affected at all and sometimes (more often) after using the pen, the pressure curve resets to default as witnessed by running `xsetwacom --get "UGEE DECO 01 Pen stylus" PressureCurve` which reports "0 0 100 100". Sometimes this doesn't happen, it returns still "0 0 10 10" (in this example) but there's no visible change to the actual pressure sensitivity of the pen.

If there's any log or anything I could throw in here let me know, I'm not sure where this stuff is logged or even if it is logged in the first place.

mbellitti, 2020-08-03:
I don't know if you have solved your problem, but the pressure curve "0 0 10 10" and "0 0 100 100" are fully equivalent. See the `PressureCurve` section in [xsetwacom documentation](https://linux.die.net/man/1/xsetwacom).

The four parameters after `PressureCurve` are two anchors for a bezier curve in the x-y plane, where x=(force applied to the tip) y=(output intensity). Both these quantities range from 0 to 100, where x=100 means the maximum force supported by the tip (I don't know the exact number, let's say it's the equivalent of 100g). If you press harder than 100g it still registers as x=100.

The 0,0 point is fixed (if you don't press on the tip there will be no output) and the point 100,100 is too (if you press harder than the maximum force you always get full intensity output). What `PressureCurve` lets you fix is two extra intermediate values: if you feel you have to press harder than you would like, try (0  0  50  100), which means that the maximum output is reached with half of the maximum force supported by the tablet.

razcore-art, 2020-08-04:
I see what you mean. And no, I guess this issues isn't relevant any more, I ended up using software curves in the apps directly wherever possible. Thanks for clarifying the setting.

closed 2020-08-04 by razcore-art