DIGImend/digimend-kernel-drivers #157: Huion new H1060+ not detected

YPares, 2018-06-01:
Hi! I'm using Ubuntu 17.10
I tried to install the driver, I did `modprobe hid-uclogic` but the tablet doesn't even work as mouse cursor.

Here the output of lsusb :
```
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 003: ID 05ac:0273 Apple, Inc. 
Bus 001 Device 002: ID 05ac:8290 Apple, Inc. 
Bus 001 Device 009: ID 058f:6366 Alcor Micro Corp. Multi Flash Reader
Bus 001 Device 008: ID 256c:006e  
Bus 001 Device 007: ID 058f:6254 Alcor Micro Corp. USB Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub```

So I think tablet is device 008.
When I run `uclogic-probe 1 8` I get:
```
S 64 12 03 70 C6 00 06 7C 00 FF 1F D8 13 03 0C 10 00 01 00
S 65 04 03 20 A0
S 6E 04 03 00 30
S 79 14 03 48 00 41 00 36 00 30 00 2D 00 46 00 34 00 30 00 30 00
S 7A 08 03 01 0C 00 00 00 00
S 7B 0C 03 48 00 4B 00 20 00 4F 00 6E 00
S C8 12 03 70 C6 00 06 7C 00 FF 1F D8 13 03 0C 10 00 01 00
S C9 24 03 48 00 55 00 49 00 4F 00 4E 00 5F 00 54 00 31 00 37 00 34 00 5F 00 31 00 37 00 31 00 32 00 32 00 31 00
S CA 46 03 48 00 55 00 49 00 4F 00 4E 00 20 00 41 00 6E 00 69 00 6D 00 61 00 74 00 69 00 6F 00 6E 00 20 00 54 00 65 00 63 00 68 00 6E 00 6F 00 6C 00 6F 00 67 00 79 00 20 00 43 00 6F 00 2E 00 2C 00 6C 00 74 00 64 00
```

spbnick, 2018-06-01:
Hi Yves, which version of the driver did you download?

YPares, 2018-06-01:
The very last commit : 38585dfb78d23a24297433a381839c5bd762c96c

spbnick, 2018-06-01:
@YPares, alright, what's the output of `cat /sys/module/hid_uclogic/version`?

ThePreviousOne, 2018-06-01:
I am having this problem too, but with the GT-185. Previously in cccaed58d999a2f589774e7428387420b0420ac7 everything worked perfectly.
Also I have the older H1060Pro and interestingly both pens now work on it, though not perfectly or as intended.

spbnick, 2018-06-01:
Alright, let's figure it out! @ThePreviousOne, please post `cat /sys/module/hid_uclogic/version` too.

ThePreviousOne, 2018-06-01:
≻≻≻  cat /sys/module/hid_uclogic/version 
8

ThePreviousOne, 2018-06-01:
something is wrong. Ive tryied reverting to cccaed58d999a2f589774e7428387420b0420ac7 then restarting as well and `make uninstall` then restart and still nothing.

YPares, 2018-06-02:
`version` doesn't exist.

However I have `/sys/module/hid_uclogic/srcversion` that contains:

1BEDCC552EDCC95D3DE2132

YPares, 2018-06-02:
Note that I'm getting some certificates errors when installing. However, that didn't prevent `sudo modprobe hid-uclogic` to work and exit with 0:

```
$ sudo make install                                                                              
make -C /lib/modules/4.13.0-43-generic/build SUBDIRS=/home/yves/digimend-kernel-drivers modules_install                                    
make[1]: Entering directory '/usr/src/linux-headers-4.13.0-43-generic'                                                                     
  INSTALL /home/yves/digimend-kernel-drivers/hid-kye.ko                                                                                    
At main.c:160:                                                                                                                             
- SSL error:02001002:system library:fopen:No such file or directory: bss_file.c:175                                                        
- SSL error:2006D080:BIO routines:BIO_new_file:no such file: bss_file.c:178                                                                
sign-file: certs/signing_key.pem: No such file or directory                                                                                
  INSTALL /home/yves/digimend-kernel-drivers/hid-polostar.ko                                                                               
At main.c:160:                                                                                                                             
- SSL error:02001002:system library:fopen:No such file or directory: bss_file.c:175                                                        
- SSL error:2006D080:BIO routines:BIO_new_file:no such file: bss_file.c:178                                                                
sign-file: certs/signing_key.pem: No such file or directory                                                                                
  INSTALL /home/yves/digimend-kernel-drivers/hid-uclogic.ko                                                                                
At main.c:160:                                                                                                                             
- SSL error:02001002:system library:fopen:No such file or directory: bss_file.c:175                                                        
- SSL error:2006D080:BIO routines:BIO_new_file:no such file: bss_file.c:178                                                                
sign-file: certs/signing_key.pem: No such file or directory                                                                                
  INSTALL /home/yves/digimend-kernel-drivers/hid-viewsonic.ko                                                                              
At main.c:160:                                                                                                                             
- SSL error:02001002:system library:fopen:No such file or directory: bss_file.c:175                                                        
- SSL error:2006D080:BIO routines:BIO_new_file:no such file: bss_file.c:178                                                                
sign-file: certs/signing_key.pem: No such file or directory                                                                                
  DEPMOD  4.13.0-43-generic                                                                                                                
make[1]: Leaving directory '/usr/src/linux-headers-4.13.0-43-generic'                                                                      
install -D -m 0644 digimend.conf /etc/depmod.d/digimend.conf                                                                               
depmod -a                                                                                                                                  
install hid-rebind /sbin/hid-rebind                                                                                                        
install -m 0644 90-hid-rebind.rules /lib/udev/rules.d/90-hid-rebind.rules                                                                  
udevadm control --reload
```

YPares, 2018-06-02:
@danghai Both commands work with status 0. But dmesg show nothing related.

danghai, 2018-06-02:
My output after modprobe on dmesg: 
```
[Jun 2 06:59] input: Tablet Monitor Pen as /devices/pci0000:00/0000:00:15.0/0000:03:00.0/usb3/3-2/3-2:1.0/0003:256C:006E.0003/input/input11
[  +0.000238] input: Tablet Monitor Pad as /devices/pci0000:00/0000:00:15.0/0000:03:00.0/usb3/3-2/3-2:1.0/0003:256C:006E.0003/input/input12
[  +0.000112] uclogic 0003:256C:006E.0003: input,hidraw1: USB HID v1.11 Keypad [Tablet Monitor] on usb-0000:03:00.0-2/input0
[  +0.000214] input: Tablet Monitor Mouse as /devices/pci0000:00/0000:00:15.0/0000:03:00.0/usb3/3-2/3-2:1.1/0003:256C:006E.0004/input/input13
[  +0.000126] uclogic 0003:256C:006E.0004: input,hiddev0,hidraw2: USB HID v1.11 Mouse [Tablet Monitor] on usb-0000:03:00.0-2/input1
```

Your version is still 7 ?

spbnick, 2018-06-02:
If `/sys/modules/hid_uclogic/version` doesn't exist, then you're using the driver that comes with the kernel, which at the moment doesn't have support for the new Huion tablets.

You won't get support for your tablet by installing the .deb package. It's too old. You need to use the development version. At least until we do a new release.

spbnick, 2018-06-02:
@danghai could you please try to narrow the problem down? Those tablets should be supported. There must be something with the setup.

Also, the SSL errors shouldn't affect the driver. See also README.md about them.

YPares, 2018-06-02:
Ok, I removed the .deb and rebuilt/installed everything from the git.
Now `/sys/module/hid_uclogic/version` file is there and says 8

YPares, 2018-06-02:
Ok. So now the tablet seems to work, however it doesn't play well with the mouse.
Here is the dmesg output after activating debug output:

```
[  376.518747] FAT-fs (sdc1): Volume was not properly unmounted. Some data may be corrupt. Please run fsck.
[  414.668803] usb 1-1: USB disconnect, device number 4
[  414.668806] usb 1-1.1: USB disconnect, device number 5
[  414.705682] usb 1-1.4: USB disconnect, device number 6
[  477.803249] usb 1-1: new high-speed USB device number 7 using xhci_hcd
[  477.943826] usb 1-1: New USB device found, idVendor=058f, idProduct=6254
[  477.943833] usb 1-1: New USB device strings: Mfr=0, Product=1, SerialNumber=0
[  477.943837] usb 1-1: Product: USB2.0Hub
[  477.944707] hub 1-1:1.0: USB hub found
[  477.944763] hub 1-1:1.0: 4 ports detected
[  478.231330] usb 1-1.1: new full-speed USB device number 8 using xhci_hcd
[  478.380951] usb 1-1.1: New USB device found, idVendor=256c, idProduct=006e
[  478.380959] usb 1-1.1: New USB device strings: Mfr=5, Product=6, SerialNumber=0
[  478.389607] uclogic_params_probe:979: uclogic 0003:256C:006E.0009: pen v2 parameters found
[  478.389624] uclogic_probe:201: uclogic 0003:256C:006E.0009: parameters:
               .desc_ptr = ffff8c60db25a0c0
               .desc_size = 166
               .pen_unused = false
               .pen_id = 8
               .pen_inrange = none
               .pen_fragmented_hires = true
               .pen_frame_flag = 0x20
               .pen_frame_id = 247
               .frame_id = 0
               .frame_re_lsb = 0
               .frame_dev_id_byte = 0
[  478.390477] input: HID 256c:006e Pen as /devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1.1/1-1.1:1.0/0003:256C:006E.0009/input/input20
[  478.447990] input: HID 256c:006e Pad as /devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1.1/1-1.1:1.0/0003:256C:006E.0009/input/input21
[  478.448600] uclogic 0003:256C:006E.0009: input,hidraw6: USB HID v1.11 Keypad [HID 256c:006e] on usb-0000:00:14.0-1.1/input0
[  478.452648] uclogic_probe:201: uclogic 0003:256C:006E.000A: parameters:
               .desc_ptr =           (null)
               .desc_size = 0
               .pen_unused = true
               .pen_id = 0
               .pen_inrange = normal
               .pen_fragmented_hires = false
               .pen_frame_flag = 0x00
               .pen_frame_id = 0
               .frame_id = 0
               .frame_re_lsb = 0
               .frame_dev_id_byte = 0
[  478.452927] uclogic 0003:256C:006E.000A: No inputs registered, leaving
[  478.453195] uclogic 0003:256C:006E.000A: hidraw7: USB HID v1.11 Device [HID 256c:006e] on usb-0000:00:14.0-1.1/input1
[  478.551218] usb 1-1.4: new high-speed USB device number 9 using xhci_hcd
[  478.654223] usb 1-1.4: New USB device found, idVendor=058f, idProduct=6366
[  478.654230] usb 1-1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  478.654234] usb 1-1.4: Product: Mass Storage Device
[  478.654238] usb 1-1.4: Manufacturer: Generic
[  478.654241] usb 1-1.4: SerialNumber: 058F0O1111B1
[  478.655351] usb-storage 1-1.4:1.0: USB Mass Storage device detected
[  478.658542] scsi host2: usb-storage 1-1.4:1.0
```

ThePreviousOne, 2018-06-02:
@YPares xsetwacom --set 'Tablet Monitor Pad pad' Button <1-3, 8-?> "key +ctrl +z -z -ctrl"
@spbnick I went back to v7 using the .deb and it worked. So then I updated to 38585dfb78d23a24297433a381839c5bd762c96c and tablet no longer works, same as before.

YPares, 2018-06-02:
@ThePreviousOne How do I know the name of my device? `xsetwacom --list devices` yields nothing and the output of `dmesg` just shows something like "HID 256c:006e Pad" which doesn't work.

ThePreviousOne, 2018-06-02:
try `xinput` it is strange that `xsetwacom --list devices` doesnt work though

spbnick, 2018-06-03:
@YPares, you need to make the Wacom driver handle the tablet. See README.md.

YPares, 2018-06-03:
@spbnick Yes, I did try that but recent Ubuntu versions have removed xorg config files. Apparently they've been moved : https://askubuntu.com/questions/4662/where-is-the-x-org-config-file-how-do-i-configure-x-there?utm_medium=organic&utm_source=google_rich_qa&utm_campaign=google_rich_qa

ThePreviousOne, 2018-06-03:
@YPares Xorg is still in the same place, it is just no longer needed for a standard system to function.
I noticed the same thing when I set mine up on Debian, but `/etc/X11/xorg.conf` or `/etc/X11/xorg.conf.d/`
is still correct.

YPares, 2018-06-03:
@ThePreviousOne But I tried `/etc/X11/xorg.conf.d/` _and_ `/usr/share/X11/xorg.conf.d/70-wacom.conf` and neither worked. Do you need some special command to tell xorg to take into account these files

ThePreviousOne, 2018-06-03:
@YPares have you tried to re-log (kill X or reboot) for it to take effect. 
Personally I went all out and wrote my own Xorg.conf though I don't know if that's necessary.
I can show it to you if want to try that and need an example

YPares, 2018-06-03:
I did yeah, several times. Yes, if you can share it, yes please!

ThePreviousOne, 2018-06-03:
@YPares @spbnick I also found that I needed to build the debian package and install it, the  `sudo make install didnt work

ThePreviousOne, 2018-06-04:
@spbnick everything was in place. it even functioned after install, albiet incorrectly. It's more like it just compiled incorrectly or something.