DIGImend/digimend-kernel-drivers #160: Error when installing the drivers and don't know what to do

Ramengod44, 2018-06-16:
I'm trying to make the huion h610 pro work but I can't install the drivers

I downloaded them and placed them in /home/user/Documents/driver
Then I opened the terminal and typed
cd Documents
cd driver
make

and then I got this error

make -C /lib/modules/4.17.1-041701-generic/build SUBDIRS=/home/user/Documents/driver modules
make[1]: Entering directory '/usr/src/linux-headers-4.17.1-041701-generic'
  CC [M]  /home/user/Documents/driver/hid-kye.o
  CC [M]  /home/user/Documents/driver/hid-uclogic-core.o
/home/user/Documents/driver/hid-uclogic-core.c: In function ‘uclogic_probe’:
/home/user/Documents/driver/hid-uclogic-core.c:176:18: error: ‘HID_QUIRK_NO_EMPTY_INPUT’ undeclared (first use in this function); did you mean ‘HID_QUIRK_MULTI_INPUT’?
  hdev->quirks |= HID_QUIRK_NO_EMPTY_INPUT;
                  ^~~~~~~~~~~~~~~~~~~~~~~~
                  HID_QUIRK_MULTI_INPUT
/home/user/Documents/driver/hid-uclogic-core.c:176:18: note: each undeclared identifier is reported only once for each function it appears in
scripts/Makefile.build:312: recipe for target '/home/user/Documents/driver/hid-uclogic-core.o' failed
make[2]: *** [/home/user/Documents/driver/hid-uclogic-core.o] Error 1
Makefile:1577: recipe for target '_module_/home/user/Documents/driver' failed
make[1]: *** [_module_/home/user/Documents/driver] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-4.17.1-041701-generic'
Makefile:19: recipe for target 'modules' failed
make: *** [modules] Error 2

spbnick, 2018-06-17:
Hi @Ramengod44, this is due to an update in the linux kernel and requires an update in the DIGImend driver. Researching...

spbnick, 2018-06-17:
Alright, fix ready, gonna be merged today, likely. I'll close this when it's done. Then please download and build the latest master from https://github.com/DIGImend/digimend-kernel-drivers/archive/master.zip