DIGImend/digimend-kernel-drivers #162: dkms_uninstall fails

Pixdigit, 2018-06-20:
running `sudo make dkms_uninstall` returns the following error:
```
set -e -x; \
dkms status digimend | \
    while IFS=':' read -r modules status; do \
        IFS=', ' read -r modules_name modules_version \
                         kernel_version kernel_arch ignore \
                     <<<"$modules"; \
        if [ -z "$kernel_version" ]; then \
            dkms remove --no-depmod \
                        "$modules_name/$modules_version" \
                        --all; \
        else \
            dkms remove --no-depmod \
                        "$modules_name/$modules_version" \
                        -k "$kernel_version/$kernel_arch"; \
        fi; \
    done
/bin/sh: 6: Syntax error: redirection unexpected
Makefile:62: recipe for target 'dkms_modules_uninstall' failed
make: *** [dkms_modules_uninstall] Error 2
```

spbnick, 2018-06-24:
Thank you for the report, @Pixdigit. I'm going to see what can be done about this right now.

Pixdigit, 2018-06-24:
As far as I can tell it works now

spbnick, 2018-06-24:
Thank you for testing, @Pixdigit!