DIGImend/digimend-kernel-drivers #168: Add support for Artist 16 Pro

fire, 2018-07-03:
Was interested in Linux drivers for this pen.

Wanted to post a placeholder issue.

fire, 2018-07-23:
I wanted to poll the issues forum before I bought the non-cheap device for Linux.

loentar, 2018-09-11:
@fire, It's possible that it's already supported by the Linux kernel including the one which comes with latest updates of Ubuntu 16.04, because of my XP-PEN Artist 22E PRO works just well without any 3rd-party drivers (after restoring original Xorg configuration and removing this gitlab driver, I was using for UGEE tablet).

loentar, 2018-10-11:
@DanielJoyce did you try the official linux beta driver?

DanielJoyce, 2018-10-11:
It seems to mostly work, I'm still working on the Linux Wacom configuration, I've got the tablet buttons working. The only thing left is one of the stylus buttons doesn't work, but I may be using the wrong stylus model.

Once I've exhausted all options there, I will ask the linux wacom project, and then worry about filing a bug here. It may just entirely be tweaking a config file. And then you guys can say "XP-PEN 13.3 works"

:)

DanielJoyce, 2018-10-13:
> @DanielJoyce did you try the official linux beta driver?

Their 'driver' is some weird GUI utility that appears to read low-level usb messages and send xe events. It doesn't even run in 18.04 due to lib differences.

loentar, 2018-10-14:
Yes, but who cares if it provides a convenient way to change button mapping and performs 5-point calibration like similar utility in Win/Mac? As I know no normal driver can do this :). Unfortunately when I tried it, it just exits silently and nothing useful appear in strace/ltrace, anyway my display (22e pro) isn't marked as supported. And now I just use an ugly way to remap 14 of 16 buttons and standard calibration (luckily the difference between physical and screen coordinates is tiny and acceptable).

Barracuda77, 2018-11-22:
> It's possible that it's already supported by the Linux kernel including the one which comes with latest updates of Ubuntu 16.04, because of my XP-PEN Artist 22E PRO works just well without any [...]
Sorry to be Hijacking this thread but I have a quick question to @loentar 
I just ordered my XP-PEN Artist 22E PRO and I am running ubuntu 18.04 LTS on Gnome. can you elaborate on which driver are you running, please?

loentar, 2018-11-23:
@Barracuda77  I suspect it is HID-compliant, because it's handled by evdev. But I'm still on Ubuntu 16.04 and didn't checked if it works under 18.04.

If someone is interested I can provide diagnostic info for 22E PRO.

Barracuda77, 2018-11-23:
@loentar thank you for the info. it will arrive in a few hours will test it and see. I hope it just works out of the box.

Barracuda77, 2018-11-23:
@loentar So it arrived and I connected it and nothing happened? 
what would you recommend I do? go for Digimend driver or Xp pen driver? or something else?

loentar, 2018-11-24:
@Barracuda77 it just works for me. 

First check if your tablet works at all. Try windows or mac os for this.

When I start `usb-devices` it shows info such as:

```
T:  Bus=03 Lev=01 Prnt=01 Port=08 Cnt=02 Dev#=  5 Spd=12  MxCh= 0
D:  Ver= 1.10 Cls=00(>ifc ) Sub=00 Prot=00 MxPS=64 #Cfgs=  1
P:  Vendor=28bd ProdID=900b Rev=00.00
S:  Manufacturer=XP-PEN
S:  Product=Artist22E Pro_V3
C:  #Ifs= 3 Cfg#= 1 Atr=a0 MxPwr=70mA
I:  If#= 0 Alt= 0 #EPs= 1 Cls=03(HID  ) Sub=01 Prot=02 Driver=usbhid
I:  If#= 1 Alt= 0 #EPs= 1 Cls=03(HID  ) Sub=01 Prot=02 Driver=usbhid
I:  If#= 2 Alt= 0 #EPs= 2 Cls=03(HID  ) Sub=00 Prot=00 Driver=usbhid
```

Check if usb device is recognized - start `lsusb` - you should see device `28bd:900b`:

```
$ lsusb -d 28bd:900b
Bus 003 Device 007: ID 28bd:900b  
```

Try to check which driver is loaded for your tablet - run `usb-devices` and find XP-PEN. Check if `Driver=usbhid` appears like above. If there is some other driver claiming the device, remove/disable it. 

Check your `/usr/share/X11/xorg.conf.d` directory and find and disable any configs that claiming tablet devices. Only `/usr/share/X11/xorg.conf.d/10-evdev.conf` should be enabled and should contain the lines as follows:

```
Section "InputClass"
        Identifier "evdev tablet catchall"
        MatchIsTablet "on"
        MatchDevicePath "/dev/input/event*"
        Driver "evdev"
EndSection
```

or more specific:

```
Section "InputClass"
        Identifier "evdev tablet"
        MatchIsTablet "on"
        MatchDevicePath "/dev/input/event*"
        MatchUSBID "28bd:900b"
        Driver "evdev"
EndSection
```

Then reboot for sure.

loentar, 2018-11-24:
I just tried Ubuntu 18.04 under virtual box and can confirm, the tablet doesn't work there as under Ubuntu 16.04. It gets detected, appeared in `xinput`, recognized as pointer(!!), but doesn't work.

So... try Ubuntu 16.04 virtual machine and see if it works.

Barracuda77, 2018-11-24:
@loentar 
Thanks for your time and effort. I tested it on a mac laptop to see if it works and it worked. 
will test a virtual machine or a live usb stick of Ubuntu 16.04 and see.
but the thing is I can't downgrade now. Just did a fresh install a few weeks ago as I was on Cetos and switched to Ubuntu 18.04. and it was a pain getting everything to work again.

Can I ask you one more favor can you test the Digimend driver in that virtual box you did?
Thanks again your help is very appreciated :)

loentar, 2018-11-24:
@Barracuda77  I tested the digimend driver and it doesn't work with xp-pen 22e pro.

Barracuda77, 2018-11-24:
@loentar the problem is i dont have the required skill :(

Barracuda77, 2018-11-24:
@loentar 

So I get this when run usb-devices
T:  Bus=01 Lev=01 Prnt=01 Port=09 Cnt=01 Dev#=  5 Spd=12  MxCh= 0
D:  Ver= 1.10 Cls=00(>ifc ) Sub=00 Prot=00 MxPS=64 #Cfgs=  1
P:  Vendor=28bd ProdID=090b Rev=00.00
S:  Manufacturer=XP-PEN
S:  Product=Artist22E Pro
C:  #Ifs= 3 Cfg#= 1 Atr=a0 MxPwr=70mA
I:  If#= 0 Alt= 0 #EPs= 1 Cls=03(HID  ) Sub=01 Prot=02 Driver=usbhid
I:  If#= 1 Alt= 0 #EPs= 1 Cls=03(HID  ) Sub=01 Prot=02 Driver=usbhid
I:  If#= 2 Alt= 0 #EPs= 2 Cls=03(HID  ) Sub=00 Prot=00 Driver=usbhid

in my /usr/share/X11/xorg.conf.d I had a 70-wacom.conf file I moved it from there and I added an 10-evdev.conf with the lines you mentioned as I didn't have that file.

Should I disable them anywhere else or just moving that wacom file was enough?

loentar, 2018-11-25:
Figured out the problem, it's a known bug in libinput:

```
[    72.901] (II) config/udev: Adding input device XP-PEN Artist22E Pro_V3 (/dev/input/event7)
[    72.901] (**) XP-PEN Artist22E Pro_V3: Applying InputClass "libinput tablet"
[    72.901] (**) XP-PEN Artist22E Pro_V3: Applying InputClass "libinput tablet catchall"
[    72.902] (II) Using input driver 'libinput' for 'XP-PEN Artist22E Pro_V3'
[    72.960] (II) systemd-logind: got fd for /dev/input/event7 13:71 fd 28 paused 0
[    72.960] (**) XP-PEN Artist22E Pro_V3: always reports core events
[    72.960] (**) Option "Device" "/dev/input/event7"
[    72.960] (**) Option "_source" "server/udev"
[    72.960] (II) event7  - XP-PEN Artist22E Pro_V3: is tagged by udev as: Tablet
[    72.960] (EE) event7  - XP-PEN Artist22E Pro_V3: libinput bug: missing tablet capabilities: resolution.Ignoring this device.  <<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<
[    72.960] (II) event7  - XP-PEN Artist22E Pro_V3: device is a tablet
[    72.960] (II) event7  - failed to create input device '/dev/input/event7'.
[    72.960] (EE) libinput: XP-PEN Artist22E Pro_V3: Failed to create a device for /dev/input/event7
[    72.960] (EE) PreInit returned 2 for "XP-PEN Artist22E Pro_V3"
[    72.960] (II) UnloadModule: "libinput"
```

To resolve the issue you must configure Xorg to use evdev instead of libinput for the tablet device:

1) install `xserver-xorg-input-evdev`:

```
sudo apt install xserver-xorg-input-evdev
```

2) create a new config file (if you didn't created it yet) `/usr/share/X11/xorg.conf.d/10-evdev.conf`

```
Section "InputClass"
        Identifier "evdev tablet catchall"
        MatchIsTablet "on"
        MatchDevicePath "/dev/input/event*"
        Driver "evdev"
EndSection
```

3) edit `/usr/share/X11/xorg.conf.d/40-libibput.conf` and comment out the tablet section, else it will catch and disable the device:

```
#Section "InputClass"
#        Identifier "libinput tablet catchall"
#        MatchIsTablet "on"
#        MatchDevicePath "/dev/input/event*"
#        Driver "libinput"
#EndSection
```

4) restart Xorg or reboot

Barracuda77, 2018-11-25:
@loentar 
Thank you thank you. it works.
Yesterday I figured out that I had no evdev installed and went to install it. I suspected the conflict with libinput but didn't know how to disable that and I didn't want to just uninstall libinput cause I didn't know the consequences.
Thanks again for your help. I will bug you with another question is there is a way to set it up and change keys and active screen? and does this driver recognize all the pressure levels? 

I really can't thank you enough :)

Barracuda77, 2018-11-25:
@loentar again thank you.
my xinput looks like this
 Virtual core pointer                    	id=2	[master pointer  (3)]
⎜   ↳ Virtual core XTEST pointer              	id=4	[slave  pointer  (2)]
⎜   ↳ MADCATZ Mad Catz R.A.T.1                	id=11	[slave  pointer  (2)]
⎜   ↳ Microsoft Natural® Ergonomic Keyboard 4000	id=13	[slave  pointer  (2)]
⎜   ↳ XP-PEN Artist22E Pro                    	id=10	[slave  pointer  (2)]
⎜   ↳ XP-PEN Artist22E Pro                    	id=9	[slave  pointer  (2)]
⎣ Virtual core keyboard                   	id=3	[master keyboard (2)]
    ↳ Virtual core XTEST keyboard             	id=5	[slave  keyboard (3)]
    ↳ Power Button                            	id=6	[slave  keyboard (3)]
    ↳ Power Button                            	id=7	[slave  keyboard (3)]
    ↳ Sleep Button                            	id=8	[slave  keyboard (3)]
    ↳ Microsoft Natural® Ergonomic Keyboard 4000	id=12	[slave  keyboard (3)]
    ↳ XP-PEN Artist22E Pro                    	id=14	[slave  keyboard (3)]
    ↳ Microsoft Natural® Ergonomic Keyboard 4000	id=15	[slave  keyboard (3)]
the xp-pen appears twice do I need to change transformation matrix for both IDs
also, my other screen is 4k screen and it reads it as screen 2
for now, I just mirrored them but it set my 4k screen to HD and the res is not that great on it.

Barracuda77, 2018-11-26:
@loentar Sorry for nagging but I think I managed to set up the transformation matrix for the tablet but each time I restart I have to reapply it and can't seem to find which config file to use to store this?

loentar, 2018-11-26:
A simple solution would be to create a bash script with setup commands and add this script to "Startup applications" (`gnome-session-properties`).

Barracuda77, 2018-11-26:
@loentar Thank you. I just did that but I was wondering if that was the best solution?

Barracuda77, 2018-11-27:
@loentar I managed to store the coordinates and also set a different DPi for the tablet. but I cant set the stylus top bottom to right click. I tried to run  xinput set-button-map "device name" 1 2 3 
but with no luck. 
I am so sorry to keep nagging you like this but there is so little help for this.

loentar, 2018-11-27:
Looks like it's mapped to the keyboard key, so cannot be remapped as mouse key. I just noticed that I don't use that button :)

loentar, 2018-11-29:
I was wrong, for me it's mapped as left mouse button (BTN_0).. I suspect you can remap it with udev by creating specific `.hwdb`.

Barracuda77, 2018-11-29:
@loentar Can it be added to the same .hwdb file you did for the express keys?
and If you would remap that click to right click where can I find a reference for that command?

Barracuda77, 2018-12-02:
@loentar So I ran evtest, and yes scan codes on our tablets are a bit different. so I managed to fix tablet buttons mapping. but I still can't figure the stylus key. any help on that will be appreciated. 
Thanks again.

Barracuda77, 2018-12-08:
Ok I managed to fix my stylus top button right click issue added a line to the .hwdb file
KEYBOARD_KEY_d0045=273

Barracuda77, 2018-12-11:
@loentar Thanks a lot for your help I couldn't have done this without your help :).

fire, 2019-02-01:
Seems like things are solved.

closed 2019-02-01 by fire

kunzproductions, 2019-05-05:
i have problem for multiple displays, can you help me?
https://photos.app.goo.gl/UirbMWs7uci2nozT8
https://photos.app.goo.gl/GQwsgTbroJoMyY327

Barracuda77, 2019-05-05:
> i have problem for multiple displays, can you help me?
> https://photos.app.goo.gl/UirbMWs7uci2nozT8
> https://photos.app.goo.gl/GQwsgTbroJoMyY327

you will be better off correcting it from the command line
here is an article explaining it 
https://wiki.archlinux.org/index.php/Calibrating_Touchscreen

then you can add the command as a bash script to "Startup applications" (gnome-session-properties).

allo-, 2021-02-04:
@loentar Can one somehow map the hardware keys that send a sequence to a single key?

I have a similar tablet (Artist 12 pro) and button 3 sends LEFTALT, but button 8, for example, sends LEFTCTRL+LEFTALT+N and I cannot map button 3 without affecting other keys that send LEFTALT.

loentar, 2021-02-04:
@allo- I think it's not possible then, In my case I just muted both Shift and Ctrl, luckily I lost only 1 button.

BTW, Have you tried the original XP-PEN drivers for you tablet?

allo-, 2021-02-04:
@loentar: I tried and on some distributions it works fine (e.g. current debian stable), on others the pen stops working when the driver is loaded (e.g. newer Ubuntu versions. Currently I am testing with recent xorg from Gentoo as I am free to change between evdev and libinput and similar things).

Since January, they have a new beta driver online (the same as someone posted as a gdrive link earlier). It seems to work the same as the older driver, but has a nicer UI and more configuration options (e.g. not only choosing the monitor, but defining any rectangle on the screen and on the tablet). It also includes a debugging utility (in the menu), which shows X,Y and pressure. The debug utility shows that everything is working, but the driver does not forward the pen events properly on the mentioned systems.

Maybe someone here knows how to get it working? I do not know what the "xe events" mentioned by @DanielJoyce are, but with a few hints I would be happy to try to debug the driver and test patches when someone can help me where to start.
As long as the driver does not work, I would also be happy using hwdb or other approaches to map the keys and define an own transformation matrix for the pen. Pressure sensitivity seems to work out of the box even without their propritary driver.

Another question is, if there is some method to map the input events in a similar way as their driver does. udev hwdb seems only to allow to map the keycodes, but their driver seems to be able to identify the actual key instead of the emulated keycodes. As the hardware buttons do not need a low latency I could imagine writing some helper with python (and maybe the `evdev` module?) and `xdotool` or a python module with similar features.

I asked their support if they can provide hints about how to debug the driver, but I did not manage to get past the first-level support, who asked for videos of what is not working and described more or less what's in the documentation and the demo videos.

loentar, 2021-02-04:
@allo- I'm not sure what `xe events` for but I was debugging the events using `xev` and `evtest`.

The driver must have reading from hid device and emulating a device which Xorg can't find. Try listing the `/dev` before and after the driver start.

allo-, 2021-02-04:
Yes, I was getting the keycodes with `evtest` and device names with `xinput`.

This hwdb config works to remap the keys: https://gist.github.com/allo-/e7c2e97400f4c7c8bf878fd726aeff5e

But the 8th key produces `F15` and then `F20` instead of only `F20`, as key 3 (mapped to `F15`) produces `leftalt` by default. evtest only shows keycodes, i.e., the `leftalt`.

For remapping the hardware keys one needs another method to read the keys than what evdev and the udev mapping are able to provide. The proprietary driver has some method to read what key is pressed and avoid sending two mappings.
I wonder how they distinguish between key 8 producing multiple keycodes and the user pressing multiple keys. 

Either the hardware must be sending more than just the sequence, or their drive uses some ugly hack like measuring how fast the keys are "pressed" after each other?