DIGImend/digimend-kernel-drivers #169: Huion Kamvas GT191

SreckoM, 2018-07-04:
Have issue with Huion Kamvas. Drivers are installed and it is version 8. Tablet is recognized and using Wacom drivers to setup buttons. But when using Wacom drivers I have cursor offset few mm to the right from where I have pen. If I do not add entry in xorg than there are no offset but do not have pressure sensitvity in some of the apps for sculpting, like Blender and 3dCoat. 
Also interesting is that in login window offset is ok, when I log in, offset is changed. 
I am using Linux Deepin 15.6 (Debian). 
Any help?

spbnick, 2018-07-10:
@SreckoM this seems like a misbehaving Wacom driver setting. The kernel driver cannot have such an effect. Check what you're configuring and if anything else changes its settings. Ask Wacom driver developers.

SreckoM, 2018-07-10:
@TheJackiMonster Thanks for comment, I figured out that if I unload and load drivers, offset is gone. Only thing is that I miss pressure in some of the apps I use, but I guess that is up to apps not drivers. Probably not supporting evdev. Will try this for sure, I already tried few python drivers and those work nice too, but also without pressure being recognized in some of the apps.

TheJackiMonster, 2018-07-10:
@SreckoM Wait.. do you have zero offset? Because I tested with the kernel drivers and I just mapped the projection to the right screen like the example said and I had this offset which increases more close to the borders. I had the same problem at the beginning of my custom driver but fixed it mostly using a projection matrix which does not just translate and scale the values to map it to the screen. Is there a solution using the wacom drivers?

TheJackiMonster, 2018-07-11:
@SreckoM Some apps require the device to have the substring “pen“ or “stylus“ in its name to recognize them as pen with pressure and not as simple cursor/mouse. Krita is one of that apps for example.. had that bug with my driver to before I fixed it.

SreckoM, 2018-07-12:
@TheJackiMonster Krita works for me. What does not work is scultping app 3dcoat, which was mentioned in issues here I think. Also Foundry app MODO does not recognize pressure.  Will try ur driver when I grab some time, probably next week.