DIGImend/digimend-kernel-drivers #170: Huion H430P

caseneuve, 2018-07-05:
### Sys info:
OS:    Antergos Linux x86_64
dkms: digimend-kernel-drivers, 7.r80.gcda5bef, 4.17.3-1-ARCH, x86_64: installed

### Issue:
_After installation of drivers the tablet is not working._

### Troubleshooting:
#### Tablet hardware: 
* works on Windows10

* lsusb
```
$ lsusb
Bus 003 Device 006: ID 256c:006e
```

* usbhid-dump
```
$ sudo usbhid-dump -es -m 256c:006e
...
003:006:001:STREAM             1530738808.757385
 0A C0 C6 28 C5 66 00 00

003:006:001:STREAM             1530738808.761364
 0A C0 85 28 77 67 00 00

003:006:001:STREAM             1530738808.765382
 0A C0 41 28 27 68 00 00

003:006:001:STREAM             1530738808.769366
 0A 80 41 28 27 68 00 00
...
```

#### Kernel drivers:
* `evtest` sees the tablet but doesn't give any output for tests

#### Userspace drivers:
* xiput
```
$ xinput list
⎡ Virtual core pointer                    	id=2	[master pointer  (3)]
⎜   ↳ Virtual core XTEST pointer              	id=4	[slave  pointer  (2)]
⎜   ↳ Logitech USB Optical Mouse              	id=11	[slave  pointer  (2)]
⎜   ↳ SynPS/2 Synaptics TouchPad              	id=13	[slave  pointer  (2)]
⎣ Virtual core keyboard                   	id=3	[master keyboard (2)]
    ↳ Virtual core XTEST keyboard             	id=5	[slave  keyboard (3)]
    ↳ Power Button                            	id=6	[slave  keyboard (3)]
    ↳ Video Bus                               	id=7	[slave  keyboard (3)]
    ↳ Power Button                            	id=8	[slave  keyboard (3)]
    ↳ TOSHIBA Web Camera: TOSHIBA Web         	id=9	[slave  keyboard (3)]
    ↳ AT Translated Set 2 keyboard            	id=12	[slave  keyboard (3)]
    ↳ Toshiba input device                    	id=14	[slave  keyboard (3)]
    ↳ HUION Huion Tablet Pen                  	id=10	[slave  keyboard (3)]
```
```
$ xinput test 10
no event registered...
```

#### Applications:
* in MyPaint > Help > Debug > Test Input Devices gives no output.

### Request:
I would appreciate any help to make the tablet working. Thanks in advance.

spbnick, 2018-07-05:
Thank you for a detailed report, @caseneuve! Could you please download and install the latest development version from https://github.com/DIGImend/digimend-kernel-drivers/archive/master.zip ?

caseneuve, 2018-07-05:
Thanks for a quick response!

I downloaded the file, uninstalled `digimend-kernel-drivers-dkms-git`, run `make; sudo make install` in drivers' folder and rebooted: it seems to WORK! :+1: 

**Two more questions:**
1) how should I maintain the drivers (my OS is a rolling release)?
2) is it possible to configure the stylus and pad buttons?

spbnick, 2018-07-05:
To keep the driver installed through kernel upgrades, uninstall it with `sudo make uninstall` and install it again with `sudo make dkms_install`. See README.md for details. There are multiple ways to configure stylus buttons, depending on which driver you use, but if you want to configure the pad buttons, you will need to use the Wacom X.org driver. See README.md again for a brief introduction to that, and otherwise refer to the Wacom driver configuration.