DIGImend/digimend-kernel-drivers #175: Huion H580 with generic pen does not recognize pen touch (write)

pauloch8, 2018-07-11:
My original Huion pen is broken so I'm using a generic pen from a generic tablet.
It recognize pen position but does not recognize when I write.
Can anyone help me, please?

spbnick, 2018-07-11:
For the start, does it work with Windows? Could you please explain what "does not recognize when I write" means? Does the cursor jump to the position where the pen approached the tablet, but not move when you move the pen over it?

pauloch8, 2018-07-11:
Hello Nikolay!
It moves with the pen when I move it over the tablet. It doesn't click when
I touch and doesn't recognize the pens buttons.

spbnick, 2018-07-11:
OK, please test it with Windows, if you can. Otherwise please post the output of `usbhid-dump -es` for your tablet, when you do a stroke with the pen.

pauloch8, 2018-07-12:
Here: https://pastebin.com/9gQAJBnj

At Windows click doesn't work as well..

pauloch8, 2018-07-12:
Sorry. I did a mess. This last link https://pastebin.com/9gQAJBnj was the
output of a Huion drawing table with a C3 Tech pen.
It move the pen but does not click. ( I would like to use those two
together if possible)

Now, this link: https://pastebin.com/T8XqHbKw is of a C3 Tech table and
pen. It was a generic brand here in Brazil. It is working on Windows and
now I realize it is working on Linux. Working well in Blender 2.8(Grease
Pencil). It only is hiding mouse pointer over Ubuntu's interface
(Unity)....

Well, it is working for my purpose. If it not worth any enhance, think we
can close this Issue.
When I fix my Huion pen I'll try it again.
Thank you.

2018-07-11 22:15 GMT-03:00 Paulo_CH8 . <paulo.ch8@gmail.com>:

closed 2018-07-12 by spbnick