DIGImend/digimend-kernel-drivers #178: The pen button of XP-PEN deco 01 does not work

thinkedgatefoxe, 2018-07-19:
Well, we installed digimend 8 on kde-kubuntu using dkms
But deco 01 stylus pen button top does not work
It is not recognized even by xev
Did we miss something?
In huionpro 1060 it does not seem to recognize moving sh!
we describe our created xorg-conf here
conf on usr/x11/xorg.conf/
Help we
sorry, use google translater.

> 10-tablet.conf
Section "InputClass"
    Identifier "UGTABLET on wacom"
    # MatchIsTablet "on"
    MatchProduct "UGTABLET DECO 01"
    MatchDevicePath "/dev/input/event*"
    Driver "wacom"
EndSection 

> 50-deco.conf
Section "InputClass"
    Identifier "UGTABLET on wacom"
    MatchIsTablet "on"
    MatchProduct "UGTABLET DECO 01"
    MatchDevicePath "/dev/input/event*"
    Driver "wacom"
EndSection

The result of using this conf will be here
> xinput --list | grep UG
⎜   ↳ UGTABLET DECO 01 touch                    id=10   [slave  pointer  (2)]
⎜   ↳ UGTABLET DECO 01 pad                      id=11   [slave  pointer  (2)]
⎜   ↳ UGTABLET DECO 01 stylus                   id=15   [slave  pointer  (2)]
⎜   ↳ UGTABLET DECO 01 eraser                   id=16   [slave  pointer  (2)]

thinkedgatefoxe, 2018-07-19:
We think that it will be helpful if digsend user can publish sh of xsetwacom respectively

It does not move
> xsetwacom - set "UGTABLET DECO 01 stylus" Button 2 "+2"
xsetwacom - set "UGTABLET DECO 01 stylus" Button 3 "+3"

or It does not move
> xsetwacom - set "UGTABLET DECO 01 stylus" Button 2 2
xsetwacom - set "UGTABLET DECO 01 stylus" Button 3 3

thinkedgatefoxe, 2018-07-19:
> xinput list-props 15
Device 'UGTABLET DECO 01 stylus':
        Device Enabled (147):   1
        Coordinate Transformation Matrix (149): 1.000000, 0.000000, 0.000000, 0.000000, 1.000000, 0.000000, 0.000000, 0.000000, 1.000000
        Device Accel Profile (278):     0
        Device Accel Constant Deceleration (279):       1.000000
        Device Accel Adaptive Deceleration (280):       1.000000
        Device Accel Velocity Scaling (281):    10.000000
        Device Node (270):      "/dev/input/event6"
        Wacom Tablet Area (290):        0, 0, 25400, 12150
        Wacom Rotation (291):   3
        Wacom Pressurecurve (292):      0, 0, 100, 100
        Wacom Serial IDs (293): 2306, 1, 0, 0, 0
        Wacom Serial ID binding (294):  0
        Wacom Pressure Threshold (295): 26
        Wacom Sample and Suppress (296):        2, 20
        Wacom Enable Touch (297):       0
        Wacom Hover Click (312):        1
        Wacom Enable Touch Gesture (298):       0
        Wacom Touch Gesture Parameters (299):   0, 0, 250
        Wacom Tool Type (300):  "STYLUS" (308)
        Wacom Button Actions (301):     "Wacom button action 0" (302), "Wacom button action 1" (303), "Wacom button action 2" (304), "None" (0), "None" (0), "None" (0), "None" (0), "Wacom button action 3" (305)
        Wacom button action 0 (302):    1572865
        Wacom button action 1 (303):    1572866
        Wacom button action 2 (304):    1572867
        Wacom button action 3 (305):    1572872
        Wacom Pressure Recalibration (313):     1
        Wacom Panscroll Threshold (306):        1300
        Device Product ID (271):        10429, 2306
        Wacom Debug Levels (307):       0, 0

thinkedgatefoxe, 2018-07-19:
Please tell me if you need any other information,
We will present it.

thinkedgatefoxe, 2018-07-19:
> button press   2 a[0]=19995 a[1]=8872 a[2]=0 a[3]=0 a[4]=0 a[5]=-900 
motion a[0]=19996 a[1]=8875 a[2]=0 a[3]=0 a[4]=0 a[5]=-900 
xinput test "UGTABLET DECO 01 stylusmotion a[0]=19988 a[1]=8882 a[2]=0 a[3]=0 a[4]=0 a[5]=-900 
motion a[0]=19980 a[1]=8893 a[2]=0 a[3]=0 a[4]=0 a[5]=-900 
button release 2 a[0]=19949 a[1]=8934 a[2]=0 a[3]=0 a[4]=0 a[5]=-900 

thanks. we test.
pen botton top = botton 3. dose not work. no result.

spbnick, 2018-07-19:
This doesn't look like a complete output. Could you please post the complete output? You can capture it in a file called `xinput.log` by executing `xinput test "UGTABLET DECO 01 stylus" |& tee xinput.log`. Then attach the resulting `xinput.log` to your comment here.

thinkedgatefoxe, 2018-07-19:
I'm sorry
We pushed the top button many times, but nothing was outputted
It seems that click = 1 and under button = 2 are recognized
[xinput.log](https://github.com/DIGImend/digimend-kernel-drivers/files/2208726/xinput.log)

> button press   2 a[0]=9771 a[1]=7924 a[2]=0 a[3]=0 a[4]=0 a[5]=-900 
button release 2 a[0]=10177 a[1]=7827 a[2]=0 a[3]=0 a[4]=0 a[5]=-900 
button press   1 a[0]=9999 a[1]=7910 a[2]=2016 a[3]=0 a[4]=0 a[5]=-900 
button release 1 a[0]=10077 a[1]=7861 a[2]=0 a[3]=0 a[4]=0 a[5]=-900

spbnick, 2018-07-19:
Thank you. I see. OK, let's go lower in the driver stack. Run `sudo evtest`, select the tablet's stylus device, and see if you get any reaction to the top button there.

thinkedgatefoxe, 2018-07-19:
I tried, there was a reaction!
/ dev / input / event 2: UGTABLET DECO 01
/ dev / input / event 3: UGTABLET DECO 01
Of these, there was no response in event 2
There is a reaction in event 3, which is the result

Top button = not recognize it
> Event: time 1531986604.995987, -------------- SYN_REPORT ------------
Event: time 1531986605.101987, type 4 (EV_MSC), code 4 (MSC_SCAN), value d0045
Event: time 1531986605.101987, type 1 (EV_KEY), code 330 (BTN_TOUCH), value 0
Event: time 1531986605.101987, type 1 (EV_KEY), code 320 (BTN_TOOL_PEN), value 0
Event: time 1531986605.101987, -------------- SYN_REPORT ------------
Event: time 1531986605.141986, type 1 (EV_KEY), code 320 (BTN_TOOL_PEN), value 1
Event: time 1531986605.141986, type 3 (EV_ABS), code 0 (ABS_X), value 8303
Event: time 1531986605.141986, type 3 (EV_ABS), code 1 (ABS_Y), value 7908

Under button
> Event: time 1531986636.343242, -------------- SYN_REPORT ------------
Event: time 1531986636.347245, type 4 (EV_MSC), code 4 (MSC_SCAN), value d0044
Event: time 1531986636.347245, type 1 (EV_KEY), code 331 (BTN_STYLUS), value 0
Event: time 1531986636.347245, type 1 (EV_KEY), code 320 (BTN_TOOL_PEN), value 0
Event: time 1531986636.347245, -------------- SYN_REPORT ------------
Event: time 1531986636.387245, type 1 (EV_KEY), code 320 (BTN_TOOL_PEN), value 1

thinkedgatefoxe, 2018-07-19:
[event3.log](https://github.com/DIGImend/digimend-kernel-drivers/files/2208910/event3.log)
log here.

spbnick, 2018-07-19:
Judging from the output of `evdev`, the DIGImend kernel driver is not handling your tablet. Could you please reconnect your tablet and post the output of `sudo dmesg`?

thinkedgatefoxe, 2018-07-19:
Are we mistaken in setting up?
This is the log
[dmesg.log](https://github.com/DIGImend/digimend-kernel-drivers/files/2208952/dmesg.log)

spbnick, 2018-07-19:
Your tablet has a different USB product ID from what we've seen as XP-Pen DECO 01. That's why it is not recognized by the driver. We can try to make it work. Could you please post a link to the product you purchased, and also a photo of the product label on the back, if you have it? Thank you.

thinkedgatefoxe, 2018-07-19:
thanks!
This was purchased by Amazon in Japan
https://www.amazon.co.jp/1531988094
Is this the product label?
[image]
and, what. Deco 01 in Japan is not deco?

spbnick, 2018-07-19:
Apparently, yours uses a different vendor ID. It might be just a newer model, not specific to Japan. Yes, this label, could you post all of it, with model name, if there's any? Thank you.

Otherwise, let me try to tweak the driver to support your tablet.

spbnick, 2018-07-19:
OK, please download a test version from https://github.com/spbnick/digimend-kernel-drivers/archive/xp_pen_deco01_v2.zip and follow [Installing from source with DKMS](https://github.com/spbnick/digimend-kernel-drivers/tree/xp_pen_deco01_v2#installing-from-source-with-dkms) to install it.

thinkedgatefoxe, 2018-07-19:
sorry, Serial number and barcode only
Product name can not be found
we bought the 3day before so maybe it's a new model?

thinkedgatefoxe, 2018-07-19:
we will try. Thank you!

spbnick, 2018-07-19:
Yes, it can be just a newer model. Don't forget to uninstall the old driver!

thinkedgatefoxe, 2018-07-19:
we are sorry. We tried, but it will not move
Uninstallation, rebooting, re-installing. reboot. does not change the behavior
here new log
[xinput2.log](https://github.com/DIGImend/digimend-kernel-drivers/files/2209303/xinput2.log)
[evtest2.log](https://github.com/DIGImend/digimend-kernel-drivers/files/2209316/evtest2.log)
[evtest2-long.log](https://github.com/DIGImend/digimend-kernel-drivers/files/2209317/evtest2-long.log)
[dmesg2.log](https://github.com/DIGImend/digimend-kernel-drivers/files/2209301/dmesg2.log)

spbnick, 2018-07-19:
Please post the output of `sudo modinfo hid-uclogic`, and also `sudo cat /sys/module/hid_uclogic/version` after connecting the tablet.

thinkedgatefoxe, 2018-07-19:
ok, thanks.
>  sudo modinfo hid-uclogic
9
[uclogic.log](https://github.com/DIGImend/digimend-kernel-drivers/files/2209441/uclogic.log)

spbnick, 2018-07-19:
OK, it might be that rebinding doesn't work. Could you please post the output of `sudo journalctl | grep hid-rebind`?

thinkedgatefoxe, 2018-07-19:
ok, here.
[journalctl.log](https://github.com/DIGImend/digimend-kernel-drivers/files/2209572/journalctl.log)

thinkedgatefoxe, 2018-07-19:
Well, although the tablet is not working yet, we will try reinstalling again tomorrow
Thank you very much

spbnick, 2018-07-19:
It seems that rebinding works. Let's see what the driver actually does. Please execute `echo "module hid_uclogic +pfl" | sudo tee /sys/kernel/debug/dynamic_debug/control`, reconnect the tablet, run `sudo dmesg|tail -n100` and post its output. Thank you.

thinkedgatefoxe, 2018-07-19:
Okay, we tried
echo 
> module hid_uclogic + pfl

Log reconnected tablet
[dmsg2.log](https://github.com/DIGImend/digimend-kernel-drivers/files/2211674/dmsg2.log)
This procedure is
With tablet connected, echo> Reconnect tablet> dmesg,
Is this ok?

spbnick, 2018-07-20:
Hmmm, actually, I don't see the effect of rebinding in the log. Let me think...

thinkedgatefoxe, 2018-07-20:
Well, we set up KDEneon.
In conclusion, the deco01-new did not move.
we used the same setup, same procedure, same [for sh-xsetwacom].

we hope this helps something.
Each log in KDEneon will be.
[xinput-neon.log](https://github.com/DIGImend/digimend-kernel-drivers/files/2215091/xinput-neon.log)
[evtest-event9-neon.log](https://github.com/DIGImend/digimend-kernel-drivers/files/2215095/evtest-event9-neon.log)
[evtest-event10-neon.log](https://github.com/DIGImend/digimend-kernel-drivers/files/2215096/evtest-event10-neon.log)
[dmseg-neon.log](https://github.com/DIGImend/digimend-kernel-drivers/files/2215097/dmseg-neon.log)
[hid-uclogic-neon.log](https://github.com/DIGImend/digimend-kernel-drivers/files/2215098/hid-uclogic-neon.log)
[hid-uclogic-ver-neon.log](https://github.com/DIGImend/digimend-kernel-drivers/files/2215099/hid-uclogic-ver-neon.log)

thinkedgatefoxe, 2018-07-20:
and Thank you for dealing with that and so far.
We still can not use our deco01-new in ubuntu, but I hope someday we hope can use it.

Thanks to degiment we could use huion pro 1060-2048.
We were able to realize aspect ratio and button set with sh by xsetwacom in pro 1060.
It might be useful for someone if we can submit it somewhere, but we do not know its location.
But our pro 1060 was broken in normal use in just 3 months. haha.

spbnick, 2018-07-21:
Apparently something goes wrong with rebinding or with tablet initialization. I added some debug code to the driver. Please uninstall the driver, download https://github.com/spbnick/digimend-kernel-drivers/archive/xp_pen_deco01_v2.zip again, install it, reboot, reconnect the tablet, and send the output of `sudo dmesg | tail -n100` and `sudo journalctl -n1000`.

thinkedgatefoxe, 2018-07-24:
ok, sorry delay. we made our new system.
we get to work it.

thinkedgatefoxe, 2018-07-24:
Here is the submission. Created by a new our system
[dmesg-us.log](https://github.com/DIGImend/digimend-kernel-drivers/files/2222103/dmesg-us.log)
[journalctl-us.log](https://github.com/DIGImend/digimend-kernel-drivers/files/2222102/journalctl-us.log)
A part of ..sh moved as we tried!
But still the tablet's top button, right click is not working.

spbnick, 2018-07-24:
@thinkedgatefoxe, did you install the updated version of the driver from https://github.com/spbnick/digimend-kernel-drivers/archive/xp_pen_deco01_v2.zip? If you did, could you please reconnect the tablet, and send the output of `sudo dmesg | tail -n100` and `sudo journalctl -n1000`?

I don't see a reconnection in `dmesg-us.log` and `journalctl-us.log` you attached.

thinkedgatefoxe, 2018-07-25:
The result does not move but we are giving up. maybe.
Yes, we use https://github.com/spbnick/digimend-kernel-drivers/archive/xp_pen_deco01_v2.zip.
35db66201fed7ff9f3652e41364ce1b1 = MD5
1, digiment installation with git
https://github.com/DIGImend/digimend-kernel-drivers
sudo make dkms_install
2, we tried the installation we examined
sudo dkms build digimend / 9
sudo dkms install digimend / 9
Both are rebooted and cleaned and rebooted many times and confirmed

we will also attach conf for this time
1, conf's in git
2, what we created
we have both tried
It was 2 which moved mainly, it was the installation method that we created and examined
The top button of the pen still did not move
However, deco's new product 02 has only one pen button
As the aspect ratio and the mapping are moving with wacomsh. so, we think .. enough?

thinkedgatefoxe, 2018-07-25:
we think you will be tired of the correspondence so far.
But we are very grateful.
Most of all, since Ubuntu can use tablets.
There is the advantage that we do not have to use windows.
Thanks to digimend.
we think we may be able to contribute conf or wacomsh publishing or something else. sometime.

thinkedgatefoxe, 2018-07-25:
[dmesg-us2.log](
https://github.com/DIGImend/digimend-kernel-drivers/files/2227464/dmesg-us2.log)
[journalctl-us2.log](https://github.com/DIGImend/digimend-kernel-drivers/files/2227468/journalctl-us2.log)

[xinput-us2.log](https://github.com/DIGImend/digimend-kernel-drivers/files/2227470/xinput-us2.log)
[evtest-event2-us2.log](https://github.com/DIGImend/digimend-kernel-drivers/files/2227473/evtest-event2-us2.log)
[evtest-event3-us2.log](https://github.com/DIGImend/digimend-kernel-drivers/files/2227476/evtest-event3-us2.log)
[hid-uclogic-us2.log](https://github.com/DIGImend/digimend-kernel-drivers/files/2227477/hid-uclogic-us2.log)
[hid-uclogic-ver-us2.log](https://github.com/DIGImend/digimend-kernel-drivers/files/2227478/hid-uclogic-ver-us2.log)
[echo-us2.log](https://github.com/DIGImend/digimend-kernel-drivers/files/2227479/echo-us2.log)

[xinput-list-us2.log](https://github.com/DIGImend/digimend-kernel-drivers/files/2227480/xinput-list-us2.log)
[50-tabletconf.log](https://github.com/DIGImend/digimend-kernel-drivers/files/2227485/50-tabletconf.log)

spbnick, 2018-08-12:
@thinkedgatefoxe, I've returned from my vacation today and might be able to resume work on your tablet. Sorry for leaving without notice. Do you still need help?

thinkedgatefoxe, 2018-08-12:
@spbnick 
You do not have to apologize.
It is wonderful that there is a vacation! It was nice to hear that.
we are think, driver development seems to be a hard thing it.

And thank you!
If there is still help, we will be saved.
What information is required?
Since we will rebuild the system a bit, the submission may be delayed.

Z-Map, 2018-09-06:
Hi,
I just got a XP-PEN Deco 01 with the same vendor and product ID and, like thinkedgatefoxe, the right click doesn't work with the kernel driver (linux kernel 4.18.5) and the pen stopped working when I tried to install DIGImend from git with DKMS.
Here is the dmesg : 
`[18900.091615] new full-speed USB device number 12 using ehci-pci
[18900.189973] usb 1-1.4.1: New USB device found, idVendor=28bd, idProduct=0902, bcdDevice= 0.00
[18900.189979] usb 1-1.4.1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[18900.189983] usb 1-1.4.1: Product: DECO 01
[18900.189986] usb 1-1.4.1: Manufacturer: UGTABLET
[18900.192168] uclogic 0003:28BD:0902.0012: probing
[18900.192649] input: UGTABLET DECO 01 Mouse as /devices/pci0000:00/0000:00:1a.0/usb1/1-1/1-1.4/1-1.4.1/1-1.4.1:1.0/0003:28BD:0902.0012/input/input42
[18900.193068] input: UGTABLET DECO 01 Mouse as /devices/pci0000:00/0000:00:1a.0/usb1/1-1/1-1.4/1-1.4.1/1-1.4.1:1.0/0003:28BD:0902.0012/input/input43
[18900.193471] input: UGTABLET DECO 01 Keyboard as /devices/pci0000:00/0000:00:1a.0/usb1/1-1/1-1.4/1-1.4.1/1-1.4.1:1.0/0003:28BD:0902.0012/input/input44
[18900.248610] uclogic 0003:28BD:0902.0012: input,hidraw5: USB HID v1.00 Mouse [UGTABLET DECO 01] on usb-0000:00:1a.0-1.4.1/input0
[18900.250834] uclogic 0003:28BD:0902.0013: probing
[18900.251622] input: UGTABLET DECO 01 Pen as /devices/pci0000:00/0000:00:1a.0/usb1/1-1/1-1.4/1-1.4.1/1-1.4.1:1.1/0003:28BD:0902.0013/input/input45
[18900.252074] input: UGTABLET DECO 01 Pad as /devices/pci0000:00/0000:00:1a.0/usb1/1-1/1-1.4/1-1.4.1/1-1.4.1:1.1/0003:28BD:0902.0013/input/input46
[18900.252558] uclogic 0003:28BD:0902.0013: input,hidraw6: USB HID v1.00 Keypad [UGTABLET DECO 01] on usb-0000:00:1a.0-1.4.1/input1
[18900.253718] uclogic 0003:28BD:0902.0014: probing
[18900.254197] uclogic 0003:28BD:0902.0014: hiddev0,hidraw7: USB HID v1.00 Device [UGTABLET DECO 01] on usb-0000:00:1a.0-1.4.1/input2`

I can use it without the second button but if I can provide info to help or test a release I'll gladly do.
I use Arch, kernel 4.18.5, with gnome as DE.

enyachoke, 2018-09-08:
I also  got a similar tablet 28bd:0902  after installing the digimend driver the tablet does not work at all.

enyachoke, 2018-09-08:
Use the driver share above here is the result of  sudo dmesg | tail -n100
```
[ 1219.371356] input: UGTABLET DECO 01 Pen as /devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.1/0003:28BD:0902.0014/input/input55
[ 1219.428078] input: UGTABLET DECO 01 Mouse as /devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.0/0003:28BD:0902.0015/input/input53
[ 1219.428805] input: UGTABLET DECO 01 Pad as /devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.1/0003:28BD:0902.0014/input/input56
[ 1219.429286] input: UGTABLET DECO 01 Keyboard as /devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.0/0003:28BD:0902.0015/input/input54
[ 1219.429624] uclogic 0003:28BD:0902.0014: input,hidraw1: USB HID v1.00 Keypad [UGTABLET DECO 01] on usb-0000:00:14.0-1/input1
[ 1219.429917] uclogic 0003:28BD:0902.0015: input,hidraw2: USB HID v1.00 Mouse [UGTABLET DECO 01] on usb-0000:00:14.0-1/input0
[ 1301.928570] hid-generic 0003:28BD:0902.0016: hiddev0,hidraw0: USB HID v1.00 Device [UGTABLET DECO 01] on usb-0000:00:14.0-1/input2
[ 1301.931681] input: UGTABLET DECO 01 as /devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.1/0003:28BD:0902.0017/input/input57
[ 1301.945316] uclogic 0003:28BD:0902.0016: hiddev0,hidraw0: USB HID v1.00 Device [UGTABLET DECO 01] on usb-0000:00:14.0-1/input2
[ 1301.994817] hid-generic 0003:28BD:0902.0017: input,hiddev1,hidraw1: USB HID v1.00 Device [UGTABLET DECO 01] on usb-0000:00:14.0-1/input1
[ 1301.996270] input: UGTABLET DECO 01 as /devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.0/0003:28BD:0902.0018/input/input58
[ 1302.042371] input: UGTABLET DECO 01 Pen as /devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.1/0003:28BD:0902.0017/input/input59
[ 1302.100438] hid-generic 0003:28BD:0902.0018: input,hidraw1: USB HID v1.00 Mouse [UGTABLET DECO 01] on usb-0000:00:14.0-1/input0
[ 1302.101100] input: UGTABLET DECO 01 Pad as /devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.1/0003:28BD:0902.0017/input/input60
[ 1302.189220] uclogic 0003:28BD:0902.0017: input,hidraw1: USB HID v1.00 Keypad [UGTABLET DECO 01] on usb-0000:00:14.0-1/input1
[ 1302.197225] input: UGTABLET DECO 01 Mouse as /devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.0/0003:28BD:0902.0018/input/input61
[ 1302.198689] input: UGTABLET DECO 01 Mouse as /devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.0/0003:28BD:0902.0018/input/input62
[ 1302.199559] input: UGTABLET DECO 01 Keyboard as /devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.0/0003:28BD:0902.0018/input/input63
[ 1302.259487] uclogic 0003:28BD:0902.0018: input,hidraw2: USB HID v1.00 Mouse [UGTABLET DECO 01] on usb-0000:00:14.0-1/input0
[ 1410.646885] hid-generic 0003:28BD:0902.0019: hiddev0,hidraw0: USB HID v1.00 Device [UGTABLET DECO 01] on usb-0000:00:14.0-1/input2
[ 1410.652646] input: UGTABLET DECO 01 as /devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.1/0003:28BD:0902.001A/input/input64
[ 1410.693386] uclogic 0003:28BD:0902.0019: hiddev0,hidraw0: USB HID v1.00 Device [UGTABLET DECO 01] on usb-0000:00:14.0-1/input2
[ 1410.709299] hid-generic 0003:28BD:0902.001A: input,hiddev1,hidraw1: USB HID v1.00 Device [UGTABLET DECO 01] on usb-0000:00:14.0-1/input1
[ 1410.710648] input: UGTABLET DECO 01 as /devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.0/0003:28BD:0902.001B/input/input65
[ 1410.777533] input: UGTABLET DECO 01 Pen as /devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.1/0003:28BD:0902.001A/input/input66
[ 1410.831448] hid-generic 0003:28BD:0902.001B: input,hidraw1: USB HID v1.00 Mouse [UGTABLET DECO 01] on usb-0000:00:14.0-1/input0
[ 1410.831791] input: UGTABLET DECO 01 Pad as /devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.1/0003:28BD:0902.001A/input/input67
[ 1410.832901] uclogic 0003:28BD:0902.001A: input,hidraw2: USB HID v1.00 Keypad [UGTABLET DECO 01] on usb-0000:00:14.0-1/input1
[ 1410.967092] input: UGTABLET DECO 01 Mouse as /devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.0/0003:28BD:0902.001B/input/input68
[ 1410.967396] input: UGTABLET DECO 01 Mouse as /devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.0/0003:28BD:0902.001B/input/input69
[ 1410.967650] input: UGTABLET DECO 01 Keyboard as /devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.0/0003:28BD:0902.001B/input/input70
[ 1411.025262] uclogic 0003:28BD:0902.001B: input,hidraw1: USB HID v1.00 Mouse [UGTABLET DECO 01] on usb-0000:00:14.0-1/input0
[ 1437.324096] hid-generic 0003:28BD:0902.001C: hiddev0,hidraw0: USB HID v1.00 Device [UGTABLET DECO 01] on usb-0000:00:14.0-1/input2
[ 1437.327675] input: UGTABLET DECO 01 as /devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.1/0003:28BD:0902.001D/input/input71
[ 1437.385291] hid-generic 0003:28BD:0902.001D: input,hiddev1,hidraw1: USB HID v1.00 Device [UGTABLET DECO 01] on usb-0000:00:14.0-1/input1
[ 1437.387936] input: UGTABLET DECO 01 as /devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.0/0003:28BD:0902.001E/input/input72
[ 1437.407817] uclogic 0003:28BD:0902.001C: hiddev0,hidraw0: USB HID v1.00 Device [UGTABLET DECO 01] on usb-0000:00:14.0-1/input2
[ 1437.445160] hid-generic 0003:28BD:0902.001E: input,hidraw2: USB HID v1.00 Mouse [UGTABLET DECO 01] on usb-0000:00:14.0-1/input0
[ 1437.502574] input: UGTABLET DECO 01 Pen as /devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.1/0003:28BD:0902.001D/input/input73
[ 1437.586881] input: UGTABLET DECO 01 Mouse as /devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.0/0003:28BD:0902.001E/input/input75
[ 1437.641749] input: UGTABLET DECO 01 Pad as /devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.1/0003:28BD:0902.001D/input/input74
[... cut ...]
[ 2441.822171] input: UGTABLET DECO 01 Pad as /devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.1/0003:28BD:0902.0029/input/input96
[ 2441.823090] uclogic 0003:28BD:0902.0029: input,hidraw0: USB HID v1.00 Keypad [UGTABLET DECO 01] on usb-0000:00:14.0-1/input1
[ 2441.843614] input: UGTABLET DECO 01 Keyboard as /devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.0/0003:28BD:0902.0028/input/input94
[ 2441.844796] uclogic 0003:28BD:0902.0028: input,hidraw1: USB HID v1.00 Mouse [UGTABLET DECO 01] on usb-0000:00:14.0-1/input0
```
sudo journalctl -n1000 

https://pastebin.com/WYAbXxmT

thinkedgatefoxe, 2018-09-15:
https://www.xp-pen.com/download/index/cid/36.html
XPPEN seems to be issuing a linux driver.
In V1.0 we tried earlier we could not set all very well.
but It seems that it is now V 1.2.3.
@Z-Map, @enyachoke, Did you use this official driver in your environment?
If It may possibly move?
We have not tried it yet. in ubuntu KDE plasma. not neon.

We thought it was a problem with Distro but it seems not to be the only problem with KDE.
We wish to know where to adjust the source.
Even now right click on new Deco 01 is not functioning.

enyachoke, 2018-09-15:
@thinkedgatefoxe  Unfortunately I switched to Mac and stopped trying to make the driver work.

Z-Map, 2018-09-16:
@thinkedgatefoxe Yes it works, but it's a precompiled binary and need to be executed in root which is not safe at all.

And the last kernel update on my computer put the kernel driver in the same state as the DIGImend from source : the pen stop working and is listed under "keyboard" in xinput.

@spbnick I can investigate what the official driver do if you tell me what to do.