DIGImend/digimend-kernel-drivers #180: Still not working 1060 Plus on Debian 9 with v8 driver

ialebedev, 2018-07-23:
Installed the latest driver v8 on my debian 9 64bit system.
I used debian package for installation.

Still nothing.

The mouse doesn't moving, but the lights on the tablet working when pressing the buttons on the tablet.

ThePreviousOne, 2018-07-24:
did you restart after? did you check the support page?

ialebedev, 2018-07-25:
Yes restarted.

I've also checked the support page.
The lights on the tablet blinking when I press the buttons and when I touch the pad with the pen, but the mouse doesn't move.

spbnick, 2018-08-12:
@ialebedev, please post the output of `sudo dmesg | tail -n100` after you connect the tablet, and also the output of `sudo modinfo hid-uclogic` and `sudo cat /sys/module/hid_uclogic/version`.

ialebedev, 2018-08-16:
Here is the output

`$ sudo dmesg | tail -n100
[    3.775667] EDAC sbridge: Seeking for: PCI ID 8086:3cf5
[    3.775895] EDAC MC0: Giving out device to module sbridge_edac.c controller Sandy Bridge Socket#0: DEV 0000:7f:0e.0 (INTERRUPT)
[    3.776082] EDAC MC1: Giving out device to module sbridge_edac.c controller Sandy Bridge Socket#1: DEV 0000:ff:0e.0 (INTERRUPT)
[    3.776083] EDAC sbridge:  Ver: 1.1.1 
[    3.778674] intel_rapl: Found RAPL domain package
[    3.778677] intel_rapl: Found RAPL domain core
[    3.778683] intel_rapl: Found RAPL domain dram
[    3.778704] intel_rapl: Found RAPL domain package
[    3.778705] intel_rapl: Found RAPL domain core
[    3.778709] intel_rapl: Found RAPL domain dram
[    3.788317] media: Linux media interface: v0.10
[    3.792779] Linux video capture interface: v2.00
[    3.809468] usbcore: registered new interface driver snd-usb-audio
[    3.809512] uvcvideo: Found UVC 1.00 device Webcam C170 (046d:082b)
[    3.813709] uvcvideo 2-1.2.4:1.0: Entity type for entity Processing 2 was not initialized!
[    3.813712] uvcvideo 2-1.2.4:1.0: Entity type for entity Extension 6 was not initialized!
[    3.813714] uvcvideo 2-1.2.4:1.0: Entity type for entity Camera 1 was not initialized!
[    3.813783] input: Webcam C170 as /devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.2/2-1.2.4/2-1.2.4:1.0/input/input12
[    3.813842] usbcore: registered new interface driver uvcvideo
[    3.813843] USB Video Class driver (1.1.1)
[    3.856339] IPv6: ADDRCONF(NETDEV_UP): enp6s0f0: link is not ready
[    3.910962] usb 1-1.3.3: New USB device found, idVendor=046b, idProduct=ff40
[    3.910966] usb 1-1.3.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[    3.910981] usb 1-1.3.3: Product: Virtual Floppy Device
[    3.910983] usb 1-1.3.3: Manufacturer: American Megatrends Inc.
[    3.910985] usb 1-1.3.3: SerialNumber: AAAABBBBCCCC2
[    3.978557] ipmi_si IPI0001:00: Found new BMC (man_id: 0x001043, prod_id: 0x0c63, dev_id: 0x20)
[    3.978578] ipmi_si IPI0001:00: IPMI kcs interface initialized
[    4.042045] usb 1-1.3.4: new high-speed USB device number 7 using ehci-pci
[    4.098011] FS-Cache: Loaded
[    4.108653] FS-Cache: Netfs 'nfs' registered for caching
[    4.110933] Key type dns_resolver registered
[    4.119341] NFS: Registering the id_resolver key type
[    4.119344] Key type id_resolver registered
[    4.119345] Key type id_legacy registered                                                                           
[    4.218930] usb 1-1.3.4: New USB device found, idVendor=046b, idProduct=ff31                                        
[    4.218933] usb 1-1.3.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3                                   
[    4.218934] usb 1-1.3.4: Product: Virtual HardDisk Device
[    4.218936] usb 1-1.3.4: Manufacturer: American Megatrends Inc.
[    4.218937] usb 1-1.3.4: SerialNumber: AAAABBBBCCCC3
[    4.228945] usb-storage 1-1.3.2:1.0: USB Mass Storage device detected
[    4.229031] scsi host7: usb-storage 1-1.3.2:1.0
[    4.229147] usb-storage 1-1.3.3:1.0: USB Mass Storage device detected
[    4.229187] usb-storage 1-1.3.3:1.0: Quirks match for vid 046b pid ff40: 200
[    4.229206] scsi host8: usb-storage 1-1.3.3:1.0
[    4.229275] usb-storage 1-1.3.4:1.0: USB Mass Storage device detected
[    4.229329] scsi host9: usb-storage 1-1.3.4:1.0
[    4.229406] usbcore: registered new interface driver usb-storage
[    4.230773] usbcore: registered new interface driver uas
[    4.495441] input: HDA NVidia HDMI/DP,pcm=3 as /devices/pci0000:00/0000:00:02.0/0000:03:00.1/sound/card0/input13
[    4.495579] input: HDA NVidia HDMI/DP,pcm=7 as /devices/pci0000:00/0000:00:02.0/0000:03:00.1/sound/card0/input14
[    4.495699] input: HDA NVidia HDMI/DP,pcm=8 as /devices/pci0000:00/0000:00:02.0/0000:03:00.1/sound/card0/input15
[    4.495815] input: HDA NVidia HDMI/DP,pcm=9 as /devices/pci0000:00/0000:00:02.0/0000:03:00.1/sound/card0/input16
[    5.251717] scsi 8:0:0:0: Direct-Access     AMI      Virtual Floppy0  1.00 PQ: 0 ANSI: 0 CCS
[    5.251873] scsi 9:0:0:0: Direct-Access     AMI      Virtual HDISK0   1.00 PQ: 0 ANSI: 0 CCS
[    5.252045] scsi 7:0:0:0: CD-ROM            AMI      Virtual CDROM0   1.00 PQ: 0 ANSI: 0 CCS
[    5.252994] sd 8:0:0:0: Attached scsi generic sg1 type 0
[    5.253302] sd 9:0:0:0: Attached scsi generic sg2 type 0
[    5.253539] scsi 7:0:0:0: Attached scsi generic sg3 type 5
[    5.254456] sd 8:0:0:0: [sdb] Attached SCSI removable disk
[    5.254789] sd 9:0:0:0: [sdc] Attached SCSI removable disk
[    5.280161] sr 7:0:0:0: [sr0] scsi-1 drive
[    5.280163] cdrom: Uniform CD-ROM driver Revision: 3.20
[    5.280343] sr 7:0:0:0: Attached scsi CD-ROM sr0
[    7.470395] igb 0000:06:00.0 enp6s0f0: igb: enp6s0f0 NIC Link is Up 1000 Mbps Full Duplex, Flow Control: RX/TX
[    7.470625] IPv6: ADDRCONF(NETDEV_CHANGE): enp6s0f0: link becomes ready
[    9.624335] random: crng init done
[... 37 lines cut ...]
license:        GPL
author:         Nikolai Kondrashov
author:         Martin Rusko
alias:          hid:b0003g*v00002179p00000053
alias:          hid:b0003g*v00005543p00003031
alias:          hid:b0003g*v00005543p00000045
alias:          hid:b0003g*v00005543p00000081
alias:          hid:b0003g*v00005543p0000004D
alias:          hid:b0003g*v00005543p0000006E
alias:          hid:b0003g*v0000256Cp0000006E
alias:          hid:b0003g*v00005543p00000781
alias:          hid:b0003g*v00005543p00000522
alias:          hid:b0003g*v00005543p00000064
alias:          hid:b0003g*v00005543p00000005
alias:          hid:b0003g*v00005543p00000004
alias:          hid:b0003g*v00005543p00000003
alias:          hid:b0003g*v00005543p00000042
depends:        hid,usbcore
retpoline:      Y
intree:         Y
vermagic:       4.9.0-7-amd64 SMP mod_unload modversions 
`
`$ sudo cat /sys/module/hid_uclogic/version
cat: /sys/module/hid_uclogic/version: Нет такого файла или каталога
`

spbnick, 2018-08-16:
The driver doesn't seem to have installed properly. Please install again, on the terminal, and post the output here. Thank you.

ialebedev, 2018-08-16:
Reinstalled. Nothing changed.
What log do you need?
 
$ sudo dpkg -i ./digimend-dkms_8_all.deb 
(Чтение базы данных … на данный момент установлено 164408 файлов и каталогов.)
Подготовка к распаковке ./digimend-dkms_8_all.deb …

-------- Uninstall Beginning --------
Module:  digimend
Version: 8
Kernel:  4.9.0-7-amd64 (x86_64)
-------------------------------------

Status: Before uninstall, this module version was ACTIVE on this kernel.

hid-kye.ko:
 - Uninstallation
   - Deleting from: /lib/modules/4.9.0-7-amd64/updates/dkms/
 - Original module
   - No original module was found for this module on this kernel.
   - Use the dkms install command to reinstall any previous module version.

hid-uclogic.ko:
 - Uninstallation
   - Deleting from: /lib/modules/4.9.0-7-amd64/
rmdir: не удалось удалить '': Нет такого файла или каталога
 - Original module
   - No original module was found for this module on this kernel.
   - Use the dkms install command to reinstall any previous module version.
                                                                                                                       
                                                                                                                       
hid-polostar.ko:                                                                                                       
 - Uninstallation
   - Deleting from: /lib/modules/4.9.0-7-amd64/updates/dkms/
 - Original module
   - No original module was found for this module on this kernel.
   - Use the dkms install command to reinstall any previous module version.

hid-viewsonic.ko:
 - Uninstallation
   - Deleting from: /lib/modules/4.9.0-7-amd64/updates/dkms/
 - Original module
   - No original module was found for this module on this kernel.
   - Use the dkms install command to reinstall any previous module version.

depmod...

DKMS: uninstall completed.

------------------------------
Deleting module version: 8
completely from the DKMS tree.
------------------------------
Done.
Распаковывается digimend-dkms (8) на замену (8) …
Настраивается пакет digimend-dkms (8) …
Loading new digimend-8 DKMS files...
Building for 4.9.0-7-amd64
Building initial module for 4.9.0-7-amd64
Done.

hid-kye:
Running module version sanity check.
 - Original module
 - Installation
   - Installing to /lib/modules/4.9.0-7-amd64/updates/dkms/

hid-uclogic.ko:
Running module version sanity check.
Error! Module version 8 for hid-uclogic.ko
is not newer than what is already found in kernel 4.9.0-7-amd64 (8).
You may override by specifying --force.

hid-polostar.ko:
Running module version sanity check.
 - Original module
 - Installation
   - Installing to /lib/modules/4.9.0-7-amd64/updates/dkms/

hid-viewsonic.ko:
Running module version sanity check.
 - Original module
 - Installation
   - Installing to /lib/modules/4.9.0-7-amd64/updates/dkms/

depmod...

DKMS: install completed.
$ sudo apt install -f
Чтение списков пакетов… Готово
Построение дерева зависимостей       
Чтение информации о состоянии… Готово
обновлено 0, установлено 0 новых пакетов, для удаления отмечено 0 пакетов, и 0 пакетов не обновлено.
$

ialebedev, 2018-08-16:
Now it's working. Thanks a lot!

spbnick, 2018-08-16:
Cool. Don't forget to leave a comment in the [Debian bug](https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=890644) :)

closed 2018-08-16 by spbnick