DIGImend/digimend-kernel-drivers #181: Add support for Huion H950P tilt angle reporting

spbnick, 2018-07-24:
Apparently the Huion H950P firmware can be updated to report tilt angles. That needs Windows and an [update package](https://github.com/DIGImend/digimend-kernel-drivers/files/2224111/H950P.tilt.zip).

Attempt to implement support for tilt reporting.

NuDogma, 2018-08-08:
I have this model but I haven't found the firmware updater on the Huion website, just the drivers for Win/Mac and a pdf user manual.

Is it safe to update it? Has somebody tried?

ThePreviousOne, 2018-08-08:
I hate that I am even thinking this but, If you use windows, it should go fine assuming you follow the steps given. Assuming there is actually a firmware updater somewhere as it will undoubtedly be windows only

spbnick, 2018-08-09:
@NuDogma, I don't know if the update is on the official website anywhere. One person on Twitter said the one attached to this issue worked for him, though: https://twitter.com/ndee85/status/1021823707621285888

NuDogma, 2018-08-09:
I've just updated the firmware and I can confirm that it works and detects tilt.

spbnick, 2018-10-07:
I was too hasty putting "financed" label on this. Huion is not offering their support for this work so far. Taking the label off for now.

J5in, 2018-10-23:
Yes, it works. I've tried it on mine and now my tablet supports tilt. Also yes, it is for Windows. It's an exe file so mac might not support it.

sunil-ghorpade, 2019-03-27:
does tilt support work on ubuntu?. say on krita?. I tested my tablet h950p on windows and it works.

spbnick, 2019-05-05:
Latest development version of the drivers supports Huion tilt reporting.

closed 2019-05-05 by spbnick