DIGImend/digimend-kernel-drivers #182: please Add support for 10moons 10*6 inch graphic table

boubouh, 2018-07-25:
please Add support for 10moons 10*6 inch graphic table... It looks like Huion 1060 Plus Tablet which is already supported.

spbnick, 2018-08-12:
We cannot add support for it unless somebody provides us with the tablet, or with the data about the tablet. If you have it, please provide the output of `lsusb` with the tablet connected, for the start.

boubouh, 2018-08-29:
hi spbnich
thank you for reply ... 
`lsusb` output : 
```
Bus 001 Device 002: ID 13d3:5130 IMC Networks Integrated Webcam
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 002: ID 08f2:6811 Gotop Information Inc. 
Bus 002 Device 003: ID 15d9:0a4f Trust International B.V. Optical Mouse
Bus 002 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub

```
acutely the tablet works fine ... it needs just to configure the buttons and the pressure sensitivity

spbnick, 2018-08-30:
@boubouh, which of those lines would be your tablet?

boubouh, 2018-08-30:
This one :

'Bus 002 Device 002: ID 08f2:6811 Gotop Information Inc. '

spbnick, 2018-09-01:
I don't remember ever seeing this VID:PID pair. Could you please try running uclogic-probe on it and posting results?

boubouh, 2018-09-01:
Hi @spbnick .
this is the result of  `sudo uclogic-probe 2 2` :

```
M 53 00 5A 00 20 00 50 00 49 00 4E 00 47 00 2D 00 49 00 54 00 20 00 49 00 4E 00 43 00 2E 00 20 00
P 5B 00 54 00 35 00 30 00 31 00 5D 00 20 00 44 00 72 00 69 00 76 00 65 00 72 00 20 00 49 00 6E 00 73 00 69 00 64 00 65 00 20 00 54 00 61 00 62 00 6C 00 65 00 74 00

```

boubouh, 2018-09-01:
`sudo uclogic-probe 2 2 | uclogic-decode`
output:
```
Manufacturer: SZ PING-IT INC. 
       Product: [T501] Driver Inside Tablet
```

spbnick, 2018-09-03:
Thank you @boubouh, your tablet seems to be incompatible with UC-Logic tablets. Please proceed to the [regular diagnostics procedure](http://digimend.github.io/support/howto/trbl/diagnostics/).

spbnick, 2018-09-08:
Thank you, @boubouh. I took a look at the diagnostics you posted and it's definitely not something I've seen before. The tablet reports coordinates and pressure, but, as is, their resolution is very far from advertised. I could also not see proximity reports in the dumps, which makes driver implementation more difficult, plus I didn't find any reports for buttons on the side of the pen (not sure if you pressed them or not).

In theory, it would be possible to implement a driver on the tablet as is, but the input quality would be low, and I don't think we would be able to do much with the buttons on the frame. This would be not very useful, so the work to support this tablet would go to the end of my queue (which is not getting shorter).

One thing you could do is to find a Windows install, capture USB traffic for tablet initialization, and the diagnostics routine you did earlier. If you're up to that, write, and I'll give you a few tips on how to do that. We might be able to find a way to make it work then.

I might buy this tablet myself out of curiosity and try to make it work, but I cannot promise that, and it will take a while to deliver anyway. You can of course send me one yourself, alternatively.

Ticoeddy, 2018-12-01:
> We cannot add support for it unless somebody provides us with the tablet, or with the data about the tablet. If you have it, please provide the output of `lsusb` with the tablet connected, for the start.

Good morning, Gentlemen. I am a newcomer to the site and this is my first comment in this group. 
I have an unpack 10moons 6x10 in graphic tablet I have not been able to use, cause I intend to use with BricsCad application running in a corporate laptop that will not allow a (unsigned) generic driver to run, which is the one I've got from manufacturer. 
Any clues? Or Do we have any use for my tablet?

mosheliv, 2018-12-04:
+1 for me...
Just saying that this is the second most popular tablet on ali express, so probably userbase is large.

crxssrazr93, 2019-01-02:
Any updates? Planning get this or the Huion H430P....

alex-s-v, 2019-01-06:
Hello @spbnick. I captured USB traffic in Windows with native drivers installed. But I used the different tablet, the 10moons 6x4 inches one if it's inappropriate I will open a new issue. Files descriptors.txt and hid_report_descriptors.txt are obtained in Linux and .txt is a real format, but others captured using Wireshark in Windows and have .pcapng as a native format, I just renamed it to upload (GitHub swearing on me for some reason for .pcapng format).
[descriptors.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/2730746/descriptors.txt)
[frame_buttons.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/2730747/frame_buttons.txt)
[hid_report_descriptors.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/2730748/hid_report_descriptors.txt)
[pen_buttons.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/2730749/pen_buttons.txt)
[pen_coords.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/2730750/pen_coords.txt)
[pen_pressure.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/2730751/pen_pressure.txt)

DegradedMind, 2019-02-06:
A new version of the tablet came out - T503. Old version - T501. You can no longer buy T501.
The manufacturer claims that the new version (T503) has increased resolution - 5080 LPI: http://eng.10moons.com/info5529.html
Old version (T501) was claimed to have 2540 LPI. 
The official webpage (http://eng.10moons.com/info5529.html) was updated recently. Before the  update the page claimed that T503 had 2540 LPI, not 5080 LPI. The seller said that the official page hasn't been updated since the new version came out, so it said 2540 LPI, because the page served to be for old model (T501). 
Is it possible to find out the real resolution with the help of `uclogic-tools`? And is it possible to write a Linux driver for this tablet? I don't code and I am not an extensive Linux user. Don't have a clue how to write code, compile, make drivers and such.

``uclogic-probe * * | uclogic-decode` output is this: 
`    |           Manufacturer: SZ PING-IT INC. 
    |                Product: [T503] Driver Inside Tablet`

Tablet diagnostics:
[descriptors.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/2836458/descriptors.txt)
[hid_report_descriptors.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/2836459/hid_report_descriptors.txt)
[hid_report_descriptors.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/2836468/hid_report_descriptors.txt)
[pen_coordinates_up.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/2836461/pen_coordinates_up.txt)
[pen_coordinates_down.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/2836460/pen_coordinates_down.txt)
[pen_coordinates_left.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/2836463/pen_coordinates_left.txt)
[pen_coordinates_right.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/2836466/pen_coordinates_right.txt)
[pen_tilt.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/2836470/pen_tilt.txt)
[pen_pressure.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/2836472/pen_pressure.txt)
[pen_button1.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/2836497/pen_button1.txt) - 
`03 02 00 00 00` - Press
`03 00 00 00 00` - Release
[pen_button2.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/2836498/pen_button2.txt) - Can't find what it dumps.
[tablet_buttons.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/2836680/tablet_buttons.txt)

`002:030:002:STREAM
 00 00 00 00 00 00 00 00`
2. Pen tip touched tablet surface (00 C1 00 00 00 00 00 00)
3. Y position (00 00 9C 00 00 00 00 00)
4. Y position in segments (00 00 00 06 00 00 00 00) - Increases from up to down: 00 - 09 then 0A -> 0B -> ... -> 0F
5. X position (00 00 00 00 0A 00 00 00)
6. X position in segments (00 00 00 00 00 08 00 00) - Increases from right to left: 00 - 09 then 0A -> 0B -> ... -> 0F
7. Pen tilt (00 00 00 00 00 00 23 00)
8. Pen pressure (00 00 00 00 00 00 00 01) - From 01 to 07

Even without the proprietary driver installed, the table works, but not entirely: 
Issues: 
- X and Y movements are interpreted incorrectly, switched, inversed - by moving up, the cursor goes left; by moving down, the cursor goes right; by moving right, the cursor goes up. by moving left, the cursor goes down:
Up = Left
Down = Right
Right = Up
Left = Down

- The cursor periodically bounces: 
https://lh4.googleusercontent.com/DVj-xquEis7GH-rfaXdoJa0Um1se-NokuBaA3fB1OSwUZF6h-1PRg-6aRVLC5KqfqU8lY662VNsX7pl_L_ax=w1600-h726-rw
https://lh4.googleusercontent.com/DVj-xquEis7GH-rfaXdoJa0Um1se-NokuBaA3fB1OSwUZF6h-1PRg-6aRVLC5KqfqU8lY662VNsX7pl_L_ax=w1600-h726-rw
https://lh5.googleusercontent.com/8Uqwgej8j7LFDM-hOrc6LHawA9m_hWZII-JnPyS_reYUr2NThNtx3t2_wQTh1iblEDnA1dlJCOrkJaZosy21=w1045-h726-rw

Pen tip clicking, double-clicking and pressure works.

rurubell, 2019-03-06:
Sorry for my English!

I bought this tablet just today (or, correct, I get it from post office today). I read this topic and watch streams on YouTube. But after all, I bought it because I extremely poor and this thing extremely cheap. I think I will use it with virtualbox or native windows system.

But... It works! I try it with GIMP, and pressure sensitivity is work just like on windows (I try it too). And coordinates, left is left, right is right, and it fit screen size. And no bounces. Yes, buttons doesn't works.

I run it on my Debian 9 (stretch) with 4.9.0-7 kernel. I doesn't install drivers from this project.

Maybe I get something wrong, because this is my first tablet and I doesn't use any before. But I think this one works well, and this thing works on windows same way (with installed driver).

Maybe they use another chip for new devices? I can disassemble it and look on chip, and run diagnostic procedure, if you want. lsusb say same as OP-post: **ID 08f2:6811 Gotop Information Inc.**

DegradedMind, 2019-03-11:
@spbnick, you probably got lucky with the chip - I installed Debian 9.8, kernel version 4.9.0-8 and my issue is still present. I bought this tablet a month and a half ago. Well... shit.

crxssrazr93, 2019-03-11:
> Sorry for my English!
>
> I bought this tablet just today (or, correct, I get it from post office today). I read this topic and watch streams on YouTube. But after all, I bought it because I extremely [...]
@rurubell 

Could you please run
`sudo uclogic-probe 2 2 | uclogic-decode`

And follow instructions [here](http://digimend.github.io/support/howto/trbl/diagnostics/)?

rurubell, 2019-03-11:
```
$ sudo uclogic-probe 4 2 | uclogic-decode
    |           Manufacturer: SZ PING-IT INC. 
    |                Product: [T501] Driver Inside Tablet
```

[descriptors.txt](https://github.com/rurubell/10moons_diag/blob/master/descriptors.txt)
[frame_buttons.txt](https://github.com/rurubell/10moons_diag/blob/master/frame_buttons.txt)
[hid_report_descriptors.txt](https://github.com/rurubell/10moons_diag/blob/master/hid_report_descriptors.txt)
[pen_buttons.txt](https://github.com/rurubell/10moons_diag/blob/master/pen_buttons.txt)
[pen_coords.txt](https://github.com/rurubell/10moons_diag/blob/master/pen_coords.txt)
[pen_pressure.txt](https://github.com/rurubell/10moons_diag/blob/master/pen_pressure.txt)
[probe.txt](https://github.com/rurubell/10moons_diag/blob/master/probe.txt)

DegradedMind, 2019-03-11:
@rurubell, where and when did you buy this tablet? T501? I thought they were out of stock or, maybe, this is a T503 with a T501 chip. I wonder if the tablet of @boubouh is the same as yours.

rurubell, 2019-03-11:
https://www.aliexpress.com/item/10moons-10-6-Inch-Professional-Graphic-Tablet-8192-Levels-Digital-Drawing-Tablet-No-need-charge-Pen/32851786287.html?spm=2114.13010708.0.0.2fc04c4dDJuJ4T

I ordered it on February 19th

DegradedMind, 2019-03-12:
@rurubell, Oh, this is a 1060Plus! I thought this whole conversation was about T501 - a tablet that was selling before and looked identical with the recent T503, but it's about 1060Plus - a 10*6 inch tablet that uses a T501 chip, probably. My T503 has a different chip and has coordination issues on Linux unlike yours.

crxssrazr93, 2019-03-13:
It is the same, just rebranded as a 1060Plus for the newer hardware+product refresh

DegradedMind, 2019-03-15:
@spbnick, so what do you think? Is it possible to make these tablet work on Linux?

spbnick, 2019-03-15:
@DegradedMind, in theory, if they work on Windows, they could be made to work on Linux. It's only a question of time. Of which I have very little, as my current donations are still very low. However, two days ago I received two 10moons tablets from a Patreon patron: T503 and G10, and it will be much easier for me to support them, having them on hand. First I'll work on tilt reporting support for Huion tablets, though, as they're paying for it.

Blanzs, 2019-04-06:
Hi! 
I have a 10moons T503 graphic tablet (ordered in January, received 2 months ago). I tried it with Android 8 (Autodesk Sketchbook) and it worked fine, however, I realized today that it does not work with my Xubuntu OS laptop at all.

I accidentally resolved this problem: a few weeks ago I downloaded Wine to my computer for another reason, did not know how life-saver it is :)

If you press the upper left button on your graphic tablet (maybe you need to hold it down a bit, I only started to explore this tablet now, I guess it is also the switch on-off button by default), it will open a folder with an exe file in it: tablet_setup4.exe. Now, if you right click on it and open it with Wine, it will run it as you would have Windows OS. After the installation, type wine to your Xubuntu "start menu", and click on "run wine", and after that, Krita (and Xubuntu OS) works with the graphic tablet.

I hope this helps someone to be able to use this tablet on Linux as well.

DegradedMind, 2019-04-06:
@Blanzs, so basically you are saying that you have installed the the tablet drivers dedicated for windows on linux through wine? This doesn't seem to make sense because the driver is installed in the wine environment and, accordingly, it cannot affect anything outside of it.
If your tablet works fine on Android then that means it worked on Xubuntu even before any manipulations you've done. The tablet didn't worked initially because it took some time for the system to set it and start registering its output. 
Your very strange result is caused by the fact that the output is intended for android smartphones and the screen size they have.
What does `uclogic-probe * * | uclogic-decode` print?

vido89, 2019-04-13:
Im planing buying this tablet so is there any info about  keys which does not work under linux. Im looking on [this](https://www.aliexpress.com/item/10moons-10-6-Inch-Professional-Graphic-Tablet-8192-Levels-Digital-Drawing-Tablet-No-need-charge-Pen/32851786287.html?spm=2114.search0604.3.1.2b494908wgMwCv&s=p&ws_ab_test=searchweb0_0,searchweb201602_7_10065_10068_319_10059_10884_317_10887_10696_321_322_10084_453_10083_454_10103_10618_10307_537_536,searchweb201603_56,ppcSwitch_0&algo_expid=77ee476a-9968-46cc-b79b-e8dad989d4d7-0&algo_pvid=77ee476a-9968-46cc-b79b-e8dad989d4d7) tablet

firstuanl, 2019-04-21:
i use SZ PING-IT INC.  Tablet  plus，it work on ubunut 16.04 and gimp，but has a bug，sometime mouse click appears on all screen。not work on 17.04，17.10

spbnick, 2019-05-05:
I've spent some time looking at the 10moons 10x6 (T501 or G10) as well as 10moons T503, and found them to be of very low quality. They don't seem to be worth the effort of writing a driver. See https://twitter.com/spbnick/status/1119962571451379722

I might try to see how they work with the latest Windows driver downloaded from the manufacturer's website, maybe it's better, but I wouldn't bet on it.

DegradedMind, 2019-05-05:
@spbnick, yikes! People are being scammed! Both tablets got high ratings on AliExpress. 
People should be warned.

DegradedMind, 2019-05-05:
@spbnick, yikes! People are being scammed! Both tablets got high ratings on AliExpress. 
People should be warned. We are sorry that you had to spend money on these.

spbnick, 2019-05-05:
The tablets got contributed by Jean-Philippe, my supporter on Patreon, so didn't spend anything on them except my time and effort.

alex-s-v, 2019-05-06:
Hello, @spbnick. I'm just wondering if it is possible to install windows drivers using Wine and connect tablet through it, it seems possible on an Oracle virtual box, maybe Wine can do something like this. A quick search on the internet does not give me any reasonable answer. Maybe you can give some suggestions or it is not possible at all. Thank your support!

spbnick, 2019-05-06:
Hi @alex-s-v, I highly doubt that's possible.

polynamaude, 2019-07-07:
@alex-s-v Maybe you should learn some basic of computer system like OS isolation, virtualisation, what's a driver, what's an emulator... You'll save your self a truck-load of wasted time.

I'll make it dumb-ass simple : If you do install any USB driver inside VirtualBox then it will be limited to the virtual machine you installed on, not your whole system. Now as you talk about Wine, that's something different that allow you to run some Windows software over Linux. It doesn't deal with anything related to drivers. And now please... don't talk about NDisWrapper !

spbnick, 2019-07-08:
Hi @polynamaude, nice of you to help out @alex-s-v, but please be polite and try to be understanding. We were all newbies once, not everybody has to know everything about computers, and then @alex-s-v is just trying to solve his problem the way he knows at this moment.

polynamaude, 2019-07-08:
@spbnick Hi,
Sometime it's needed to get a little shake to remember us what we don't like (and that Microsoft made us lazy of doing so). Same as when we we're in school, that is, do some reading and get basic understanding. There's the difference between **playing around** and **being productive**. As you say, you don't have to know everything, I don't have to learn the driving regulation, unless I want to drive a car. If I want to dive deep inside the inner bits of my computer then yes I'll have to do some learning.

What's more useful ? That I tell you it's highly doubtful that you can install Windows drivers on Linux or that I tell you clearly that you're really going the wrong way ?

Maybe there's a reason why the acronym RTFM was also created some time ago...Don't be more clean than vacuum of space.

spbnick, 2019-07-08:
I'm sure these things could be explained without implying anybody is a dumb-ass. Nobody is hurt by @alex-s-v exploring which things are possible and which are not on his own computer. That's how people learn. So there's nothing here requiring crude responses. Just have some patience.

Believe me, I know it's not easy sometimes, I have to explain stuff here to people from various backgrounds, and, BTW, elsewhere people have to explain stuff to me. If the question shows lack of understanding, it's more productive to explain things, that to push people away. And if my resources are at an end, and irritation with ignorance is overpowering, then I just don't answer and let others help.

alex-s-v, 2019-07-08:
Thank you, @polynamaude, for criticizing and thank you, @spbnick, for defending.
Key point is that I wasted only @spbnick time, not mine, for which I'm very sorry.
And as a chemist, I can say that you don't need to learn thermodynamics to calculate how fast a solution is cooling, you can just ask a scientist for an equation.

cl3d, 2019-10-15:
I bought the model T503. Did anyone get a drive that works on linux?

rurubell, 2019-11-21:
Sorry for my English!

Can I ask one stupid question? I say about my 10moons tablet here, before. It works well, but buttons doesn't works. Can I use **usbhid-dump** utility to fix this problem? I'm not so smart, but I think. I can write some script that will parse it output and emulate keyboard shortcuts. But one problem, when I use this utility, tablet stops to work.

alex-s-v, 2019-12-31:
Hello. I wrote a little driver for 10moons T503 tablet. Idk if it is bad or not but it works fine for me. You can check the driver [here](https://github.com/alex-s-v/10moons-driver). It would be great if you @spbnick can check it out with your experience and give some suggestions or something like that.

spbnick, 2020-01-06:
Hi @alex-s-v, this looks to be a neat little driver and might be a better approach for this tablet. However, it seems to be very simple for the kind of nonsense reports the tablets I have produce, so I'm not sure if everything works right, but if it works for you, that's fine :)

Looking at how uinput has progressed recently and relative simplicity of Python I'm actually thinking whether it's the time to move DIGImend to userspace Python to simplify contributions and speed up development.

alex-s-v, 2020-01-06:
If you or somebody else interested I could take the time and write a little script from the code parts I used for determining raw bytes meaning. I don't know if it will be useful or not, but if it helped me maybe it will help others.

spbnick, 2020-01-14:
@alex-s-v, not sure what you mean exactly, but just documentation on the report format would certainly be useful.

jgobi, 2020-02-04:
@alex-s-v I tested your driver and it didn't worked in my tablet, so I searched a lot and made it work for me! I then forked your repo and edited the source.

In my version of the driver I removed all the button feature cause it wasn't working and I didn't need it, so I didn't searched for too much how to make it work..

Also the X and Y were _flipped_ in your driver for my tablet. I don't know if it applies just to my tablet or for all of them.. This way I didn't made a pull request.

thanks for the comment here, as it motivated me to try to solve these problems (since the driver was almost working)

Anyway, thank you for your comment here as it has motivated me to try to  , I am very proud of my research as it's the first time I try to make things like this and 

Here is my repo if anyone needs the driver: [https://github.com/jgobi/10moons-driver-t503](https://github.com/jgobi/10moons-driver-t503)

jgobi, 2020-02-04:
@alex-s-v I tested your driver and it didn't worked in my tablet, so I searched a lot and made it work for me! I then forked your repo and edited the source.

In my version of the driver I removed all the button feature cause it wasn't working and I didn't need it, so I didn't searched for too much how to make it work..

Also the X and Y were mirrored in your driver for my tablet. I don't know if it applies just to my tablet or for all of them.. This way I didn't made a pull request.

Anyway, thank you for your comment here as it has motivated me to try to solve the problems I found (since the driver was almost working). I am very proud of my research and job as it's the first time I try to make things like this. =)

Here is my repo if anyone needs the driver: [https://github.com/jgobi/10moons-driver-t503](https://github.com/jgobi/10moons-driver-t503).

polynamaude, 2020-02-04:
@jgobi Hi !
I have the same 10moons tablet that I bought on AliExpress.
I worked with some playing around using the built-in driver on Linux as a USB input device. I just had to configure the buttons.
Anyway, I forked your repository and will give it a try. Maybe it will offer me more precision in the movements of the pen.

Thanks for your work

jgobi, 2020-02-04:
@polynamaude Although I said I didn't need the buttons, I continued to work on them. Then I realized them didn't work because:
 - same as the pen: missing event
 - it appears that only BTN prefixed events are listened by the kernel (the examples were all prefixed by KEY)

I then added the buttons feature again and fixed the event bug (which in part were caused by my last change).

That said, the bottom button of the pen still don't work and crashes the driver when pressed.

Please check it out. :)

alex-s-v, 2020-02-04:
Hello, @jgobi, I'm happy that it helped you, but I don't get what you meant by X and Y flipped. If you mean that moving pen along X-axis on the tablet moves the cursor along Y-axis on the screen than it is quite interesting because this is exactly the problem I had without the driver. And regarding the bottom pen button, when pressed it reports 5 bytes instead of 7 and it breaks the evdev library because it waits for the byte array size multiple of 7. Solving this issue is beyond my knowledge. I will certainly try out your driver, I wonder if it will work for me either. Best regards.

jgobi, 2020-02-04:
Hi @alex-s-v when I said the X and Y were mirrored, I meant that if I move the pen to the left, it moves the cursor to the right and vice-versa. And when I move the pen up, the cursor moves down and vice-versa. 

About the pen button, thank you for the information, but it's also beyond my knowledge how to fix this problem.

Best regards!

rubenphi, 2020-05-04:
I had given up trying to install my tablet so I didn't try again. Some time later I updated my operating system to xubuntu 20.04 lts I never tried to install the tablet again and now it mysteriously detects all the available area. Before, it only detected the area intended for android, now it works correctly. the only problem is that the buttons don't work but I can solve that with the keyboard.

rubenphi, 2020-05-05:
I restarted my pc and the target area android was the only one available. I loaded windows, ran the tablet software and again the whole area was activated. Apparently the software activates some kind of switch within the tablet that enables the entire available area, that would explain why @Blanzs  managed to get the tablet to work using wine.

bgenc, 2020-12-11:
@rubenphi Did you by any chance solve this problem? It looks like you recovered the whole area for a day and then lost it again. I have the same issue with my vinsa 1060plus, which is basically the same tablet. In Linux, only android area of the tablet is recognized, but in windows through the software, whole tablet area can be activated. Alas, when you go back to Linux, it is Android only again. So is there a way to activate the whole tablet surface in Linux?

rubenphi, 2020-12-17:
@bgenc  in my case the following worked: First you must start Windows, make sure that the whole area is working. then you restart the pc in linux, but you must make sure that the tablet does not lose power through usb, that is, the tablet should not turn off at any time, therefore the power supply cannot be stopped through.

bgenc, 2020-12-17:
@rubenphi I see. That won't easily work for me though. I don't have a dual boot device. I have separate windows and linux machines. I will try with a powered usb hub to see if it works. There must be a very easy trick to switch between these two modes. Maybe a simple message to be sent to the tablet. Wish I had known it.

rubenphi, 2020-12-19:
@bgenc  Indeed, all this is a very curious fact, because it can be seen that the tablet can work with the entire area in linux, however, it is necessary that the order be sent so that the tablet activates the entire area, it is as if it existed some kind of "switch" inside the tablet. It is a pity that the company does not provide support, because the fact that the entire area of ​​the tablet works in Linux indicates that it is possible to develop a driver.

luoiliembac, 2021-04-05:
Hi everyone!
I bought a VINSA 1060Plus tablet and found that it is basically the same as 10moons as described here.
I found something interesting.

In Windows 10 (installed in Virtualbox of Manjaro KDE host, kernel 5.9)
My tablet cannot work in its full space (10 x 6 inch) when plugged in a 3.0 usb port, I must change to a 2.0 usb port, then Windows is recognized full working area.

So, while waiting for respected @spbnick to spare his time to help us, I think the workaround is to install Windows 10 in the Virtualbox.

And then, work hard to buy a Wacom, maybe?

Good luck to all!
By the way, in case someone can take a look at the driver of this tablet (Win10 and Macbook), I uploaded them here: (https://drive.google.com/drive/folders/1J3PA0PFU6XMnZPEDhXlLdTK5n_GABBtL?usp=sharing).

luoiliembac, 2021-04-05:
Good news!

Finally, my tablet works in full area.

I followed the instruction here: [https://github.com/DIGImend/10moons-tools](url)
But, please remember the correct command is: `autoreconf -i -f && ./configure && make`

The first time I tried, it did not work with 3.0 usb port, then **changed to 2.0 usb** => full area recognized!

Some button on the left hand did not work, however, but I don't care too much about that.
The most important thing is now my tablet can draw normally in full 10 x 6 inch area.

Thank you very much @spbnick!

dhelly, 2021-11-06:
> Good news!
> 
> Finally, my tablet works in full area.
> 
> I followed the instruction here: https://github.com/DIGImend/10moons-tools But, please remember the correct command is: `autoreconf -i -f && ./configure && make`
> 
> The first time I tried, it did not work with 3.0 usb port, then **changed to 2.0 usb** => full area recognized (after system reboot)!
> 
> Some button on the left hand did not work, however, but I don't care too much about that. The most important thing is now my tablet can draw normally in full 10 x 6 inch area.
> 
> Thank you very much @spbnick!

Work to me too....

[5 more replies]