DIGImend/digimend-kernel-drivers #184: Need help with my Huion New 1060Plus on Ubuntu Gnome 16.04

jcelmeta14, 2018-08-04:
Hello everybody,

I am trying to get my Huion New 1060Plus to work. I have followed the installation procedure and tried locating possible problems by using [this guide](https://digimend.github.io/support/howto/trbl/locating_failure/). Out of all those steps I encountered two problems:
1. The files `events` and `rdesc` at the kernel debug interfaces seem to be empty (0 bytes).
2. When I run `xinput list`, the pen and pad do not appear (I tested all other devices and I do not think they are related to the tablet).
Everything else seems to work correctly, at least according to the locating failures guide.

Kind regards,
Jurgen

ThePreviousOne, 2018-08-04:
Duplicate of #180

jcelmeta14, 2018-08-04:
Good evening,

I would not say that this a duplicate of #180 because `sudo usbhid-dump -es -m 256c:006e` and `sudo evtest` actually show data that makes sense. I mean, I can see that the output changes as I move the pen around, press the pad or the keys. Also, my pen and pad do not appear on `xinput list`.

Thank you for your help!
Kind regards,
Jurgen

ThePreviousOne, 2018-08-05:
what is your Kernel version? `uname -a`
there seems to be (or at least I believe so) a problem with 4.15.x though 4.14 / 4.16 and even 4.17 work fine

also is your tablet plugged in through USB 2 or 3 USB3 #40

jcelmeta14, 2018-08-06:
Hello again!
My kernel version is 4.15.0-29-generic, it looks like it may be related to 4.15.x. What should I do?
Thank you for your support!
Kind regards,
Jurgen

jcelmeta14, 2018-08-07:
Good evening,
I was doing some experiments on an old laptop. I installed Linux Mint 19. Kernel version 4.15.0-20-generic and the tablet is working. So this is kinda weird :smile: 
Kind regards,
Jurgen

ThePreviousOne, 2018-08-07:
I assume you tried on USB2. Are you running mint on both systems? the only thing I can come up with is that is something specific to that kernel or its configuration

ThePreviousOne, 2018-08-07:
from running Debian I know that 4.15.0-x is not a real version but rather a package version and your kernel is more like 4.15.4 or something.
I find it funny and a bit troubling that kernel.org doesn't even maintain 4.15 anymore, was it that bad?

spbnick, 2018-08-08:
Hi Jurgen, do you still need to solve this problem?

jcelmeta14, 2018-08-08:
Hello again,
Unfortunately, I still cannot use the tablet on my laptop that is running Ubuntu 16.04. I suppose that I will upgrade to 18.04 and try again. Or maybe I can try to downgrade/upgrade the kernel. If it were for me, I would just switch to Linux Mint, however, the problem is that I have some work-related stuff in this laptop, so I was hoping for some quick solution.I will figure out something eventually.
Thank you for your support and great work! 
KInd regards, Jurgen

ThePreviousOne, 2018-08-08:
@jcelmeta14 I did a quick search and found out 18.04.1 so it should be stable now, also its running 4.17 kernel which i know from testing is significantly better than 4.15, having used both

spbnick, 2018-08-09:
> The files events and rdesc at the kernel debug interfaces seem to be empty (0 bytes).

Those files are supposed to be empty, as they're not regular "files". Just run `cat` on them to see the contents.

Could you post the output of `sudo dmesg | tail -n100` after reconnecting the tablet?

closed 2025-08-27 by jcelmeta14