DIGImend/digimend-kernel-drivers #187: Installing v8 on Manjaro:  "No rule to make target `modules'" error

moshpirit, 2018-08-11:
Hi, I'm trying to install the latest version (8th) so I can use my Huion 1060 Plus but I'm getting this error.

**Steps to reproduce**

1. Go to [Releases](https://github.com/DIGImend/digimend-kernel-drivers/releases/tag/v8)
2. Download "digimend-kernel-drivers-8.tar.gz" and extract its content
3. $ `cd ~/Downloads/digimend-kernel-drivers-8 modules` _(in my case ~/Descargas/digimend-kernel-drivers-8 modules, since I'm Spanish)_
4. $ `make`

**What I get**

```
make -C /lib/modules/4.14.60-1-MANJARO/build SUBDIRS=/home/alberto/Descargas/digimend-kernel-drivers-8 modules
make[1]: Entering directory '/usr/lib/modules/4.14.60-1-MANJARO/build'
make[1]: *** No rule to make target `modules'.  Stop.
make[1]: Leaving directory '/usr/lib/modules/4.14.60-1-MANJARO/build'
make: *** [Makefile:19: modules] Error 2
```

danghai, 2018-08-11:
@moshpirit I think you probably do not have kernel header.

moshpirit, 2018-08-12:
You're right! Thank you so much!! 

I'm trying to configure it but I'm not sure how, someone told me to use [this configuration](https://wiki.manjaro.org/Wacom_Tablet_And_Pen#Slow_drag_in_File_Manager) and I can use the tablet and the pen, but not most of the buttons of the tablet. I [asked on the official Manjaro forum](https://forum.manjaro.org/t/huion-1060plus-graphic-tablet-just-uses-3-buttons/50283/9) but I still don't know how to configure it. I couldn't install kcm-wacomtablet-git because of [a building error](https://pastebin.com/raw/hqTKtxr7), though. I know you already help me enough, but the problem still remains and since you are an expert in this topic, I though you also might helped me with this too (I don't have any configuration guide for the tablet so I'm thinking in sharing my experience after all of this is done to help others)

spbnick, 2018-08-12:
@moshpirit, take a look at the [README.md](https://github.com/DIGImend/digimend-kernel-drivers/blob/master/README.md) for an example of button configuration. At the moment the Wacom configuration GUI tool won't support your tablet, though.