DIGImend/digimend-kernel-drivers #188: XP-PEN Deco 02

ronnietucker, 2018-08-12:
Attached are the required files for the **XP-PEN Deco 02**.

The current kernel modules (Deco 01?) get the Deco 02 pressure sensitivity working with xinput. Buttons and wheel don't work with the current kernel modules.

[frame-buttons.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/2280966/frame-buttons.txt)
[frame-dials.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/2280967/frame-dials.txt)
[pen_buttons.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/2280968/pen_buttons.txt)
[pen_pressure.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/2280969/pen_pressure.txt)
[pen_coords.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/2280970/pen_coords.txt)
[descriptors.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/2280971/descriptors.txt)
[hid_report_descriptors.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/2280972/hid_report_descriptors.txt)

arunkumar9102, 2019-03-22:
How can we install and configure Deco 02 on ubuntu 18.04?
Following above steps will help me configure keys, wheel to any keys I want?

arunkumar9102, 2019-03-22:
I have basic knowledge in python programming, if I can get little guidance on how to add support, I will be happy to work on this, as I have XP-PEN Deco 02.

arunkumar9102, 2019-03-23:
Please provide support for XP-PEN Deco 02

https://www.xp-pen.com/product/88.html
https://xp-pen.co.in/product/deco02/
https://www.amazon.com/XP-PEN-Deco-02-Graphics-Battery-Free/dp/B07DCS9D9Q

I selectively checked and copied some info in "Selective Hardware info of Deco 02.txt". Use it if needed or discard if unnecessary. Thanks in advance.

[descriptors.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/2998821/descriptors.txt)
[hid_report_descriptors.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/2998822/hid_report_descriptors.txt)
[probe.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/2998823/probe.txt)
[frame_wheel_srolling.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/2998824/frame_wheel_srolling.txt)
[pen_coords.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/2998825/pen_coords.txt)
[pen_pressure.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/2998828/pen_pressure.txt)
[pen_pressure_evtest.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/2998829/pen_pressure_evtest.txt)
[eraser_pressure.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/2998830/eraser_pressure.txt)
[eraser_pressure_evtest.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/2998831/eraser_pressure_evtest.txt)
[pen_buttons.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/2998832/pen_buttons.txt)
[frame_dials.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/2998834/frame_dials.txt)
[frame_buttons.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/2998835/frame_buttons.txt)
[decode.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/2998837/decode.txt)
[Selective Hardware info of Deco 02.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/2998838/Selective.Hardware.info.of.Deco.02.txt)

EssGeeEich, 2020-11-12:
[descriptors.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/5533567/descriptors.txt)
[hid_report_descriptors.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/5533569/hid_report_descriptors.txt)
[probe.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/5533570/probe.txt)
[pen_coords_top.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/5533571/pen_coords_top.txt)
[pen_coords_right.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/5533572/pen_coords_right.txt)
[pen_coords_bottom.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/5533573/pen_coords_bottom.txt)
[pen_coords_left.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/5533574/pen_coords_left.txt)
[pen_tilt.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/5533576/pen_tilt.txt)
[pen_pressure.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/5533577/pen_pressure.txt)
[pen_buttons.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/5533578/pen_buttons.txt)
[frame_dials.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/5533579/frame_dials.txt)
[frame_buttons.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/5533580/frame_buttons.txt)
[pen_pressure_eraser.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/5533581/pen_pressure_eraser.txt)

I did the diagnostics before realizing somebody else already did all of them.
I'm still gonna upload them just in case.
Any other way to contribute?