DIGImend/digimend-kernel-drivers #192: Gaomon S56K on Debian 4.9 and Xorg, drivers, break existing support for the tablet via Xorg

Veek, 2018-09-11:
Hello,
The S56K was working in Xorg and applications/WindowManager/OpenBoard could display/recognize a moving pointer and clicks with default kernel support. **However the pen-buttons didn't do anything** and xinput list did not show the tablet. Therefore **I installed the digimend-kernel .deb file** and ran the sed oneliner on the dkms file.

Now the new driver loads and xinput list shows the tablet:
↳ Tablet Monitor Pad                        id=11   [slave  pointer  (2)]
↳ Tablet Monitor Pen                        id=12   [slave  keyboard (3)]

and dmesg shows the tablet being plugged in
[  222.762610] input: Tablet Monitor Pad as /devices/pci0000:00/0000:00:1d.1/usb5/5-1/5-1:1.0/0003:256C:006E.000A/input/input30
[  222.762831] uclogic 0003:256C:006E.000A: input,hidraw1: USB HID v1.11 Keypad [Tablet Monitor] on usb-0000:00:1d.1-1/input0
[  222.770651] input: Tablet Monitor Mouse as /devices/pci0000:00/0000:00:1d.1/usb5/5-1/5-1:1.1/0003:256C:006E.000B/input/input31

but X**org seems to have some problem with it**
[    38.460] (**) Tablet Monitor Pen: always reports core events
[    38.460] (**) Option "Device" "/dev/input/event13"
[    38.460] (**) Option "_source" "server/udev"
[    38.461] (II) input device 'Tablet Monitor Pen', /dev/input/event13 is tagged by udev as: Tablet
[    38.549] (II) Tablet Monitor Pen: tablet unknown to libwacom
[    38.549] (II) input device 'Tablet Monitor Pen', /dev/input/event13 is a tablet
[    38.549] (**) Option "config_info" "udev:/sys/devices/pci0000:00/0000:00:1d.1/usb5/5-1/5-1:1.0/0003:256C:006E.0006/input/input21/event13"
[    38.549] (II) XINPUT: Adding extended input device "Tablet Monitor Pen" (type: TABLET, id 11)
[    38.550] (II) input device 'Tablet Monitor Pen', /dev/input/event13 is tagged by udev as: Tablet
[    38.563] (II) **Tablet Monitor Pen: tablet unknown to libwacom**
[    38.563] (II) input device 'Tablet Monitor Pen', /dev/input/event13 is a tablet
[    38.564] (II) config/udev: Adding input device Tablet Monitor Pen (/dev/input/mouse2)
[    38.564] (II) **No input driver specified, ignoring this device**.
[    38.564] (II) This **device may have been added with another device file.**
[    38.565] (II) config/udev: Adding input device Tablet Monitor Pad (/dev/input/event14)
[    38.565] (**) Tablet Monitor Pad: Applying InputClass "libinput tablet catchall"
[    38.565] (II) Using input driver 'libinput' for 'Tablet Monitor Pad'
[    38.566] (II) systemd-logind: got fd for /dev/input/event14 13:78 fd 29 paused 0
[    38.566] (**) Tablet Monitor Pad: always reports core events
[    38.566] (**) Option "Device" "/dev/input/event14"
[    38.566] (**) Option "_source" "server/udev"
[    38.567] (II) input device 'Tablet Monitor Pad', /dev/input/event14 is tagged by udev as: Tablet TabletPad

Veek, 2018-09-16:
1. yep 'sed' was run correctly
2.  xserver-xorg-input-wacom is installed
I did xsetwacom list devices --> nothing
grep -i wacom  /home/veek/.local/share/xorg/Xorg.0.log --> nothing
dmesg | grep -i wacom --> nothing
lsusb --> nothing Wacom
ls /dev/input/wacom* --> nothing
dmesg lists it as a Nuvoton Tablet
https://wiki.archlinux.org/index.php/wacom_tablet#Installation (I'm on Debian btw)
I tried insmod wacom.ko + plug out/plug in the tablet --> nothing, it's not used

I also tried changing the order of the config files - there's a /usr/share/X11/xorg.conf.d/40-libinput.conf
that acts as a catchall so I moved it out and tried using the /usr/share/X11/xorg.conf.d/70-wacom.conf
but the tablet didn't work with just the 70-wacom.conf.

I'm going to try that stuff about static and udev config of X with  the tablet

spbnick, 2018-09-16:
@Veek, you don't need to use the kernel Wacom driver, it won't work anyway. You need to use the X.org Wacom driver (xfserver-xorg-input-wacom), and set it up as described in [our README.md](https://github.com/DIGImend/digimend-kernel-drivers#configuration). Look into /var/log/Xorg.0.log to see if the driver binds to all of the tablet's /dev/input/event* devices.

Veek, 2018-09-16:
yep did that and Xorg's seeing it kind off: 'Tablet Monitor Pad: tablet unknown to libwacom'
evtest
/dev/input/event13:     Tablet Monitor Pen
/dev/input/event14:     Tablet Monitor Pad
/dev/input/event15:     Tablet Monitor Mouse
/dev/input/event16:     Tablet Monitor Keyboard
/dev/input/event17:     Tablet Monitor Consumer Control
/dev/input/event18:     Tablet Monitor System Control

but Xorg says
[  1371.956] (II) config/udev: Adding input device Tablet Monitor Pen (/dev/input/event13)
[  1371.956] (**) Tablet Monitor Pen: Applying InputClass "Tablet"
[  1371.956] (**) Tablet Monitor Pen: **Applying InputClass "libinput tablet catchall"**
[  1371.956] (II) Using input driver 'libinput' for 'Tablet Monitor Pen'
[  1371.957] (II) systemd-logind: got fd for /dev/input/event13 13:77 fd 26 paused 0
[  1371.957] (**) Tablet Monitor Pen: always reports core events
[  1371.957] (**) Option "Device" "/dev/input/event13"
[  1371.957] (**) Option "_source" "server/udev"
[  1371.958] (II) input device 'Tablet Monitor Pen', /dev/input/event13 is tagged by udev as: Tablet
[  1372.046] (II) Tablet Monitor Pen: tablet unknown to libwacom
[  1372.046] (II) input device 'Tablet Monitor Pen', /dev/input/event13 is a tablet
[  1372.046] (**) Option "config_info" "udev:/sys/devices/pci0000:00/0000:00:1d.0/usb4/4-1/4-1:1.0/0003:256C:006E.0006/input/input28/event13"
[  1372.046] (II) XINPUT: Adding extended input device "Tablet Monitor Pen" (type: TABLET, id 10)
[  1372.047] (II) input device 'Tablet Monitor Pen', /dev/input/event13 is tagged by udev as: Tablet
[  1372.060] (II) Tablet Monitor Pen: tablet unknown to libwacom
[  1372.060] (II) input device 'Tablet Monitor Pen', /dev/input/event13 is a table

Veek, 2018-09-16:
Xorg.0.conf has
[  3031.087] (II) XINPUT: Adding extended input device "Tablet Monitor Pen stylus" (type: STYLUS, id 10)

> xsetwacom set 10 MapToOutput 1366x768+0+0 --> does nothing and Gimp lists the stylus but moving the pen on the pad does nothing..

xsetwacom get "Tablet Monitor Pen stylus"  all
```
Option "Area" "0 0 25196 18896"
Button: Actions are not supported by xorg.conf. Try shell format (-s) instead.
Button: Actions are not supported by xorg.conf. Try shell format (-s) instead.
Button: Actions are not supported by xorg.conf. Try shell format (-s) instead.
Option "DebugLevel" "0"
Option "CommonDBG" "0"
Option "Suppress" "2"
Option "RawSample" "4"
Option "PressCurve" "0 0 100 100"
Option "Mode" "Absolute"
Option "TPCButton" "off"
Option "Touch" "off"
Property 'Wacom Hardware Touch Switch' does not exist on device.
Option "Gesture" "off"
Option "ZoomDistance" "0"
Option "ScrollDistance" "0"
Option "TapTime" "250"
Property 'Wacom Proximity Threshold' does not exist on device.
Option "Rotate" "none"
Property 'Wacom Wheel Buttons' does not exist on device.
Property 'Wacom Wheel Buttons' does not exist on device.
Property 'Wacom Wheel Buttons' does not exist on device.
Property 'Wacom Wheel Buttons' does not exist on device.
Property 'Wacom Wheel Buttons' does not exist on device.
Property 'Wacom Wheel Buttons' does not exist on device.
Property 'Wacom Strip Buttons' does not exist on device.
Property 'Wacom Strip Buttons' does not exist on device.
Property 'Wacom Strip Buttons' does not exist on device.
Property 'Wacom Strip Buttons' does not exist on device.
Option "Threshold" "27"
Option "Serial" "0"
Option "PressureRecalibration" "on"

```

spbnick, 2018-09-16:
Your xorg.conf sections are too specific. Use `MatchDevicePath "/dev/input/event*"` instead. You don't seem to need `NoMatchProduct` lines either. The `tablet unknown to libwacom` message is irrelevant. It only concerns GUI configuration tool, AFAIK. You need to get the following lines in your Xorg.0.log:

    Using input driver 'wacom' for 'Tablet Monitor Pen'
    Using input driver 'wacom' for 'Tablet Monitor Pad'

Afterwards check that you have them in `xsetwacom list` output, and test that you get proper input with `xinput`.

spbnick, 2018-09-16:
OK, Xorg.0.log, xinput list, and xsetwacom list look OK. However, the usbhid-dump output doesn't look OK at all. There should be no report ID 0x0A. Could you please post the output of `usbhid-dump -es -m 256c:006e`, where you do a few strokes with the pen? Thank you.

Veek, 2018-09-16:
[descriptors.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/2386532/descriptors.txt)
[hid_report_descriptors.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/2386533/hid_report_descriptors.txt)
[pen_buttons.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/2386534/pen_buttons.txt)
[pen_coords.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/2386535/pen_coords.txt)
[pen_pressure.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/2386536/pen_pressure.txt)

Veek, 2018-09-19:
how long would it take to get an update.. right now.. i have an unusable pad so i'm thinking of uninstalling..

spbnick, 2018-09-19:
I might try to implement the driver changes on the weekend, but the whole next week I'll be traveling and unable to help you.

Veek, 2018-09-19:
okay ciao, have fun

alxvll, 2019-07-15:
> I might try to implement the driver changes on the weekend, but the whole next week I'll be traveling and unable to help you.

There are some shifts for s56K?

Veek, 2019-07-17:
it seems to be working great - nothing is the matter - the buttons work fine
(sorry I thought this bug had been closed)

alxvll, 2019-07-17:
ABOUT! I love it.
Please tell me the correct link to the driver configuration for S56K.
Thank you!

Veek, 2019-07-17:
I did nothing special - standard Debian install + Xorg - I did not
need to install the digimend drivers