DIGImend/digimend-kernel-drivers #193: Huion K46

lucidprogrammer, 2018-09-14:
Any plans for supporting Huion K46.

Or are there any suggestions to make it work with the current digimed codebase? Any help is appreciated.

spbnick, 2018-09-14:
No plans so far. It all depends on how much information about it we have, or even patches to support it. For an introduction on how to make a tablet work try watching my talk at FrOSCon 2018: https://youtu.be/Qi73_QFSlpo?t=9m50s then look at the source code and see how Huion tablets are supported. Or just post the output `uclogic-probe` for your tablet, and collect diagnostics as described in http://digimend.github.io/support/howto/trbl/diagnostics/ without reconnecting it afterwards.

lucidprogrammer, 2018-09-14:
[pen_buttons.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/2383538/pen_buttons.txt)
[pen_pressure.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/2383539/pen_pressure.txt)
[pen_tilt.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/2383540/pen_tilt.txt)
[pen_coords.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/2383541/pen_coords.txt)
[frame_wheel_srolling.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/2383542/frame_wheel_srolling.txt)
[hid_report_descriptors.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/2383543/hid_report_descriptors.txt)
[descriptors.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/2383544/descriptors.txt)
[uclogic-probe.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/2383569/uclogic-probe.txt)

@spbnick see data as above. will greatly appreciate.