DIGImend/digimend-kernel-drivers #196: HUION h950p

arimassaa, 2018-09-17:
Hi, 
I recently bought a pen tablet, the huion h950p, and I run on my computer Linux Mint 19.
At first it didn't work at all, so I installed the drivers from this page that the Huion Support sent me:
https://github.com/DIGImend/digimend-kernel-drivers/releases/v8

Now it works, but it's still not recognized by the graphic tablet application and I can not make the configuration or use its keys.
I'm not an expert so I'm not really confortable in programming, I hope you can help me.
Thank you.

realmadpuppy, 2019-02-13:
I may be able to help a little, with the buttons at least, I have the same tablet and with help from the dev and a few people here I was able to get my H950P working. just an FYI if you install the V9 drivers after reboot you will need to install the kernel headers and reboot again. there is some issue with the package that it doesn't automatically install them during a kernel upgrade. 

As for the buttons from top to bottom are numbered "1,2,3,8,9,10,11,12", 4,5,6 and 7 are not used. 
If you enter in bash `xsetwacom list` and this is displayed:
[image]
You should be able to assign your buttons to what you want.
to do that you have to enter this in bash `xsetwacom set "HID 256c:006e Pad pad" button 9 key Ctrl Z`
then just change the button number to the button on the tablet you want to change and change the key or key combination to what you want the button to do. 
for example if I want button 1 on my H950P to be "undo" in krita it would be the exact command above. If I want "redo" in krita to be on button 2 this is what I would enter:
`xsetwacom set "HID 256c:006e Pad pad" button 2 key Ctrl Space Z` 
I hope this helps, I am not a very good teacher. :P

Quercetin, 2020-11-02:
Hello!!! 

I decided to connect to a laptop on debian 9.8 LXDE, tablet Huion H950P
https://blog.aboni.dev/2020-03-30-huion-h950p-linux-working/

The digimend-dkms_10_all.deb driver took from
https://github.com/DIGImend/digimend-kernel-drivers/releases

Installed the driver like this
https://github.com/DIGImend/digimend-kernel-drivers#installing

Everything went well.
Then I added xsetwacom with dependencies in the package, and configurators for wacom.

The tablet didn't work. And it was not determined.
Rebooted, launched the xsetwacom device detection, a tablet, an eraser and a pen were found.

But the tablet required adjustment and the line thickness from the degree of pressure and the reaction to the angle of inclination of the pen did not work. The buttons also did not work, all 8.

On Windows 10 on a PC, after installing the drivers, everything works fine.

I wanted to use the kcm-wacomtablet configurator for KDE, but it is not in the repositories. I use LXDE.

To put a third party or compile?
Am I wright, that I have to configure my huion h950p as wacom like in this instruction:
https://github.com/DIGImend/digimend-kernel-drivers#configuration

Help set up please. :)