DIGImend/digimend-kernel-drivers #197: No X Server after configuration

foobuntu, 2018-09-19:
Preamble:  
Xubuntu 18.04 LTS

Issue:
I followed the installation from the README.md up to where it says put  
```
Section "InputClass"
    Identifier "Tablet"
    Driver "wacom"
    MatchDevicePath "/dev/input/event*"
    MatchUSBID "<VID>:<PID>"
EndSection
```  
in `/etc/X11/xorg.conf.d/50-tablet.conf`.   
First "problem" there was no `xorg.conf.d` directory. Is that correct?  
I created the directory and then the `50-tablet.conf` inside the directory, with the appropriate VID:PID of course, restarted the machine and there is no xserver anymore, just a black command line I can log-in to. I then remove the created `xorg.conf.d` directory, restarted the machine and the xserver is running again.
What am I missing? What can I do to get it working correctly?

ThePreviousOne, 2018-09-23:
X now uses udev to setup, adding xorg.conf.d files overrides this. causing your nonexistent defaults to be used... Is my guess.

https://gist.github.com/ThePreviousOne/2d81e1c6e951cf541f2061330fc93493
here are mine as an example.

foobuntu, 2018-09-23:
So, I created the `50-tablet.conf` in `/etc/udev/rules.d/`, I restarted my machine. This time I have xserver running properly but when I do `xsetwacom list` like it's said in the README.md to check if my device is handled by the Wacom driver, I get nothing. No output, no error, just nothing. So, I'm still kind of lost with this.  
The tablet and pen is working though, I just can't configure it...and the tablet is also recognized as a usb storage. It's a Huion 1060Plus 8192, btw.

spbnick, 2018-10-08:
@foobuntu, could you please attach your `50-tablet.conf` and X.org log to a comment?

foobuntu, 2018-10-09:
Sure.
This is my entire `/etc/udev/rules.d/50-tablet.conf`:
```
Section "InputClass"
	Identifier "Tablet"
	Driver "wacom"
	MatchDevicePath "/dev/input/event"
	MatchUSBID "256:006e"
EndSection
```
I'm not so familiar with Xorg logs, so I'm not quite sure if I'm doing it right but since `/var/log/Xorg.1.log` doesn't seem to log much I only post the content of `/var/log/Xorg.0.log`.
This is what `/var/log/Xorg.0.log` logs when I plug my tablet in:
```
[  7009.029] (II) config/udev: Adding input device HID 256c:006e Pen (/dev/input/mouse1)
[  7009.029] (II) No input driver specified, ignoring this device.
[  7009.029] (II) This device may have been added with another device file.
[  7009.080] (II) config/udev: Adding input device HID 256c:006e Pad (/dev/input/js0)
[  7009.080] (II) No input driver specified, ignoring this device.
[  7009.080] (II) This device may have been added with another device file.
[  7009.110] (II) config/udev: Adding input device HID 256c:006e Pad (/dev/input/event16)
[  7009.110] (**) HID 256c:006e Pad: Applying InputClass "libinput tablet catchall"
[  7009.110] (II) Using input driver 'libinput' for 'HID 256c:006e Pad'
[  7009.110] (**) HID 256c:006e Pad: always reports core events
[  7009.110] (**) Option "Device" "/dev/input/event16"
[  7009.110] (**) Option "_source" "server/udev"
[  7009.162] (II) event16 - HID 256c:006e Pad: is tagged by udev as: Tablet TabletPad
[  7009.169] (II) event16 - HID 256c:006e Pad: tablet 'HID 256c:006e Pad' unknown to libwacom
[  7009.174] (II) event16 - HID 256c:006e Pad: device is a tablet pad
[  7009.174] (II) event16 - HID 256c:006e Pad: device removed
[  7009.204] (**) Option "config_info" "udev:/sys/devices/pci0000:00/0000:00:01.3/0000:03:00.0/usb1/1-3/1-3.1/1-3.1:1.0/0003:256C:006E.0008/input/input22/event16"
[  7009.204] (II) XINPUT: Adding extended input device "HID 256c:006e Pad" (type: PAD, id 12)
[  7009.204] (**) Option "AccelerationScheme" "none"
[  7009.204] (**) HID 256c:006e Pad: (accel) selected scheme none/0
[  7009.204] (**) HID 256c:006e Pad: (accel) acceleration factor: 2.000
[  7009.204] (**) HID 256c:006e Pad: (accel) acceleration threshold: 4
[  7009.256] (II) event16 - HID 256c:006e Pad: is tagged by udev as: Tablet TabletPad
[  7009.263] (II) event16 - HID 256c:006e Pad: tablet 'HID 256c:006e Pad' unknown to libwacom
[  7009.269] (II) event16 - HID 256c:006e Pad: device is a tablet pad
[  7009.270] (II) config/udev: Adding input device HID 256c:006e Pen (/dev/input/event15)
[  7009.270] (**) HID 256c:006e Pen: Applying InputClass "libinput tablet catchall"
[  7009.270] (II) Using input driver 'libinput' for 'HID 256c:006e Pen'
[  7009.270] (**) HID 256c:006e Pen: always reports core events
[  7009.270] (**) Option "Device" "/dev/input/event15"
[  7009.270] (**) Option "_source" "server/udev"
[  7009.270] (II) event15 - HID 256c:006e Pen: is tagged by udev as: Tablet
[  7009.275] (II) event15 - HID 256c:006e Pen: tablet 'HID 256c:006e Pen' unknown to libwacom
[  7009.275] (II) event15 - HID 256c:006e Pen: device is a tablet
[  7009.275] (II) event15 - HID 256c:006e Pen: device removed
[  7009.307] (**) Option "config_info" "udev:/sys/devices/pci0000:00/0000:00:01.3/0000:03:00.0/usb1/1-3/1-3.1/1-3.1:1.0/0003:256C:006E.0008/input/input21/event15"
[  7009.307] (II) XINPUT: Adding extended input device "HID 256c:006e Pen" (type: TABLET, id 13)
[  7009.307] (II) event15 - HID 256c:006e Pen: is tagged by udev as: Tablet
[  7009.316] (II) event15 - HID 256c:006e Pen: tablet 'HID 256c:006e Pen' unknown to libwacom
[  7009.316] (II) event15 - HID 256c:006e Pen: device is a tablet
```
If there's anything else I can do to help debugging let me know, I appreciate.

spbnick, 2018-10-14:
@foobuntu, you seem to be missing a `c` in your `MatchUSBID` option value. I.e. you have:

    	MatchUSBID "256:006e"

but should have:

    	MatchUSBID "256c:006e"

You're also missing an asterisk in your `MatchDevicePath` option value. You have:

    	MatchDevicePath "/dev/input/event"

but should have:

    	MatchDevicePath "/dev/input/event*"

foobuntu, 2018-10-15:
Oh man....thanks so much, I was really blind for this. I don't know how many times I've been looking into this 50-tablet.conf, checked and rechecked it so many times, I didn't see it.
Summary:
I corrected the typos so now I got 
```
Section "InputClass"
	Identifier "Tablet"
	Driver "wacom"
	MatchDevicePath "/dev/input/event*****"
	MatchUSBID "256**c**:006e"
EndSection
```
stored at `/etc/X11/xorg.conf.d/50-tablet.conf`.
And with that I now have xsetwacom tool and X server working.
I really appreciate. Your work is amazing.

closed 2018-10-15 by foobuntu